openwrt/openwrt #14464: ipq40xx as dumb AP, kernel 6.1 results in "received packet on [network device] with own address as source address" on router

plouds, 2024-01-23:
### Describe the bug

hAP ac2 is used as a dumb AP and works as intended on current stable 23.05. Using snapshot results in "received packet on [device*] with own address as source address (addr:[router's MAC], vlan:[any in operation**])". There can be as many as 10 shown in logread in a single second, preceded by "et_ratelimit: [#] callbacks suppressed".
This happens when I upgrade and retain config. If I start afresh, the issue appears as soon as I change IP to match the home network. The WAN port of the dumb AP is added to br-lan to enable PoE.
* individual port reported, which is included in a bridge
** two VLANs used to allow for a guest wifi, when the latter is enabled, it gets reported too

### OpenWrt version

snapshot

### OpenWrt release

r24885-7ad6e8c312

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

MikroTik hAP ac2

### Image kind

Official downloaded image

### Steps to reproduce

Convert MikroTik hAP ac2 to a dumb AP while using a snapshot with kernel 6.1. Power the dumb AP over ethernet via its WAN port included in br-lan. (I did not test powering the AP by a brick, sorry.). Connect a device by an ethernet cable.

### Actual behaviour

As soon as a device is connected via an ethernet cable, the router begins reporting packets with own address coming at quite a high rate.

### Expected behaviour

Dumb AP should behave as a dumb AP with its ethernet ports serving as a switch. Works on kernel 5.15.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-23:
Can yku share /etc/config/nefwork and wireless?
Also for better diagnostics enable
```
sysctl net.ipv4.conf.defaultmlog_martians=1 net.ipv4.conf.all.log_martians=1
```
You get reports of l2 loop, this logs l3 loop.
both means you have a cable loop somewhere else.

plouds, 2024-01-24:
`
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config device
	option name 'lan1'
	option macaddr '48:8f:5a:2b:3b:21'

config device
	option name 'lan2'
	option macaddr '48:8f:5a:2b:3b:21'

config device
	option name 'lan3'
	option macaddr '48:8f:5a:2b:3b:21'

config device
	option name 'lan4'
	option macaddr '48:8f:5a:2b:3b:21'

config interface 'lan'
	option device 'br-lan.1'
	option proto 'static'
	option ipaddr '10.0.10.140'
	option netmask '255.255.255.0'
	option gateway '10.0.10.2'
	list dns '10.0.10.2'

config device
	option name 'wan'
	option macaddr '48:8f:5a:2b:3b:20'

config interface 'wan'
	option proto 'none'
	option auto '0'

config interface 'wan6'
	option proto 'none'
	option auto '0'

config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '3'
	list ports 'wan:t'

config interface 'guest'
	option proto 'none'
	option device 'br-lan.3'
	option auto '0'
	option type 'bridge'
	list dns '10.0.10.2'`

`
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/a000000.wifi'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option txpower '11'
	option country 'CZ'
	option cell_density '0'
	option beacon_int '400'
	option log_level '4'
	option disabled '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'guest'
	option mode 'ap'
	option ssid 'Internet'
	option encryption 'none'
	option isolate '1'
	option dtim_period '1'
	option disabled '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/a800000.wifi'
	option channel '153'
	option band '5g'
	option htmode 'VHT80'
	option txpower '11'
	option country 'CZ'
	option cell_density '0'
	option beacon_int '400'
	option log_level '4'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'kainga'
	option encryption 'psk2'
	option dtim_period '1'
	option disassoc_low_ack '0'
	option key '[kupihipa]'
	option ieee80211r '1'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option ieee80211k '1'
	option time_advertisement '2'
	option time_zone 'CET-1CEST,M3.5.0,M10.5.0/3'
	option wnm_sleep_mode '1'
	option wnm_sleep_mode_no_keys '1'
	option bss_transition '1'
	option proxy_arp '1'`

plouds, 2024-01-24:
Must have taken a wrong turn trying to format the above correctly, advice most welcome.
On which device do I enable the "martians" param, the router or the offending AP?
Thank you very much.

brada4, 2024-01-24:
Three backticks on separate line before and after code section helps.

brada4, 2024-01-24:
The question is if you get martians logged in dmesg?
Does disabling proxy_arp help.

plouds, 2024-01-24:
> ```
> sysctl net.ipv4.conf.defaultmlog_martians=1 net.ipv4.conf.all.log_martians=1
> ```

I suppose I need to enter this line in the CLI, on the router would be my guess, but can do on both, of course. No notion of martians in the router's dmesg now. AP back to 5.15 but a spare one, same device, available. Will flash with 6.1, configure the same (different IP) and connect to the network to test. Including disabling proxy_arp. Later tonight hopefully. Your guidance much appreciated, thank you

brada4, 2024-01-24:
If proxy_arp is at fault either disable it while waiting wait for fix in next releases or apply nftables workaround:
https://github.com/openwrt/openwrt/issues/14309#issuecomment-1879206309

plouds, 2024-01-25:
I have configured the spare hAP ac2. As long as it's on 23.05.2, i.e. kernel 5.15, all seems fine. Flash with snapshot (keep config this time round) results in complaints of packets with own address once a device is connected by cable to the dumb AP. Reports of own-address packets by the router appear to coincide with ARP requests by the router to the just-connected device (tecra laptop) which never seem to be answered.

logread router
```
25. 01. 2024 19:57:04 warning kernel[]: [690793.818967] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 19:57:05 warning kernel[]: [690794.870944] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 19:57:06 warning kernel[]: [690795.888970] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 19:57:07 warning kernel[]: [690796.928993] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
```
tcpdump AP
```
19:57:04.843941 ARP, Request who-has tecra.lan tell gate, length 46
19:57:05.895874 ARP, Request who-has tecra.lan tell gate, length 46
19:57:06.913946 ARP, Request who-has tecra.lan tell gate, length 46
19:57:07.953936 ARP, Request who-has tecra.lan tell gate, length 46
```
tcpdump router
```
19:57:04.846797 ARP, Request who-has tecra.lan tell gate.lan, length 28
19:57:04.847003 ARP, Request who-has tecra.lan tell gate.lan, length 46
19:57:05.898743 ARP, Request who-has tecra.lan tell gate.lan, length 28
19:57:05.898917 ARP, Request who-has tecra.lan tell gate.lan, length 46
19:57:06.916792 ARP, Request who-has tecra.lan tell gate.lan, length 28
19:57:06.917004 ARP, Request who-has tecra.lan tell gate.lan, length 46
19:57:07.956795 ARP, Request who-has tecra.lan tell gate.lan, length 28
19:57:07.957032 ARP, Request who-has tecra.lan tell gate.lan, length 46
```

Running the 
```
sysctl net.ipv4.conf.defaultmlog_martians=1 net.ipv4.conf.all.log_martians=1
```
command told me that the first (defaultmlog) was "an unknown key". What should dmesg show once it's enabled?

brada4, 2024-01-25:
Disable arp proxy, it runs in br-lan even it originates from hostapd.

brada4, 2024-01-25:
`sysctl net.ipv4.conf.default.log_martians=1 net.ipv4.conf.all.log_martians=1

plouds, 2024-01-25:
> `sysctl net.ipv4.conf.default.log_martians=1 net.ipv4.conf.all.log_martians=1

No errors this time. Will report, tomorrow I suppose.

plouds, 2024-01-25:
In the meantime, with arp disabled and AP rebooted, result pretty much the same.

router's syslog
```
25. 01. 2024 21:34:25 warning kernel[]: [696635.290688] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:34:27 warning kernel[]: [696636.380468] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:34:28 warning kernel[]: [696637.420540] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
```
tcpdump router
```
21:34:25.917149 ARP, Request who-has tecra.lan tell gate.lan, length 28
21:34:25.917339 ARP, Request who-has tecra.lan tell gate.lan, length 46
21:34:27.006792 ARP, Request who-has tecra.lan tell gate.lan, length 28
21:34:27.007044 ARP, Request who-has tecra.lan tell gate.lan, length 46
21:34:28.046792 ARP, Request who-has tecra.lan tell gate.lan, length 28
21:34:28.047045 ARP, Request who-has tecra.lan tell gate.lan, length 46
```
tcpdump AP
```
21:34:25.914408 ARP, Request who-has tecra.lan tell gate, length 46
21:34:27.004060 ARP, Request who-has tecra.lan tell gate, length 46
21:34:28.044065 ARP, Request who-has tecra.lan tell gate, length 46
```

plouds, 2024-01-25:
Before I disconnected the AP I got this on the router:
```
25. 01. 2024 21:46:23 warning kernel[]: [697352.939983] IPv4: martian source 10.0.10.21 from 10.0.10.2, on dev br-lan.1
25. 01. 2024 21:46:23 warning kernel[]: [697352.947040] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
25. 01. 2024 21:46:24 warning kernel[]: [697353.948699] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:46:24 warning kernel[]: [697353.958926] IPv4: martian source 10.0.10.21 from 10.0.10.2, on dev br-lan.1
25. 01. 2024 21:46:24 warning kernel[]: [697353.965982] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
25. 01. 2024 21:46:25 warning kernel[]: [697354.978742] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:46:25 warning kernel[]: [697354.988966] IPv4: martian source 10.0.10.21 from 10.0.10.2, on dev br-lan.1
25. 01. 2024 21:46:25 warning kernel[]: [697354.996022] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
25. 01. 2024 21:46:27 warning kernel[]: [697356.450952] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:46:27 warning kernel[]: [697356.461176] IPv4: martian source 10.0.10.21 from 10.0.10.2, on dev br-lan.1
25. 01. 2024 21:46:27 warning kernel[]: [697356.468231] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
25. 01. 2024 21:46:28 warning kernel[]: [697357.468902] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:46:28 warning kernel[]: [697357.479131] IPv4: martian source 10.0.10.21 from 10.0.10.2, on dev br-lan.1
25. 01. 2024 21:46:28 warning kernel[]: [697357.486186] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
25. 01. 2024 21:46:29 warning kernel[]: [697358.508955] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:1)
25. 01. 2024 21:46:29 warning kernel[]: [697358.519179] IPv4: martian source 10.0.10.21 from 10.0.10.2, on dev br-lan.1
25. 01. 2024 21:46:29 warning kernel[]: [697358.526235] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06

```
where 10.0.10.21 is another device, a tablet connected via wifi to another dumb AP and 10.0.10.2 is the router. What does it tell you?

brada4, 2024-01-25:
That you use VLAN ID 1 and something reflects your own packets back.

brada4, 2024-01-26:
1- avoid using VLAN ID 1
2- disable proxy_arp

plouds, 2024-01-26:
1- VLAN ID 1 is the default one, and all devices concerned have been using it; only VLAN ID 5 in operation on the router but not the AP at the time of testing
2- proxy_arp was disabled for the last attempt; no issue with waiting for a fix - if indeed it is not my configuration that's at fault

The AP will remain a spare device until the weather is warm enough in the northern hemisphere and it's needed outside the house, and even then it can run a version with kernel 5.15.

brada4, 2024-01-26:
Any chance changing VLAN ID away from 1 to link untagged ports? 
And stop all arp proxies on the network.

plouds, 2024-01-26:
Can do both, will report back. Just to make sure I understand correctly: I need to change the default/untagged VLAN to an ID other than 1 (or the IDs of the other two VLANs configured), right? Thank you

brada4, 2024-01-26:
Yes, then you can run tcpdump and try to capture adverse packet.
Now its like "default" and new vlan 1 mixed that may reflect packet on its own.

If you see packet from own mac coming in, it is quite likely other arp_proxy (made in last dozen years) is reflecting it back. Indication alone is not a problem.

plouds, 2024-01-26:
No dice. Removed arp_proxy from all APs, removed VLANs from router's bridge but still get the same
```
26. 01. 2024 23:08:32 warning kernel[]: [788688.343631] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
26. 01. 2024 23:08:32 warning kernel[]: [788688.353858] IPv4: martian source 10.0.10.25 from 10.0.10.2, on dev br-lan
26. 01. 2024 23:08:32 warning kernel[]: [788688.360739] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
26. 01. 2024 23:08:36 warning kernel[]: [788691.860642] br-lan: received packet on lan0 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
26. 01. 2024 23:08:36 warning kernel[]: [788691.870869] IPv4: martian source 10.0.10.16 from 10.0.10.2, on dev br-lan
26. 01. 2024 23:08:36 warning kernel[]: [788691.877768] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
```
where 10.0.10.25 is a laptop connected via the 6.1 kernel AP, and 10.0.10.16 a laptop connected over wifi to a 5.15 AP (same as the tablet on a previous occasion mentioned earlier).

brada4, 2024-01-26:
the reflected packet (own address ...:8e) comes from 5.15 AP arp proxy.
Is there something else cached in br-lan(-s) like non-standard fib lifetime?

plouds, 2024-01-27:
No idea what a non-standard fib lifetime could be, but not sure if it increases the chances of my not tinkering with it.
Still I can't get my head around it: I first encountered the issue maybe a month ago, perhaps even a bit longer, when I noticed that the ipq40xx target was switched to 6.1 in snapshots. I flashed the snapshot of the day and everything seemed to play ball until an Android TV box was turned on. It is connected by a cable to one of the hAP ac2 dumb APs in the house. Then it all went belly up. There was no problem as long as only wifi clients were connected to either of the two APs in operation. I also believe that arp_proxy was not enabled back then. I tried that later, in an unrelated bout of tinkering.
Now only the test AP is on 6.1 and problems start only after I connect a laptop to it via cable, see syslog below. The first four lines are the test AP being connected directly to the router, no switches between. The rest begins less than a second after the laptop is plugged into the AP. The last line is when I unplugged the laptop again. No more complaints afterwards.
```
27. 01. 2024 09:02:51 notice netifd[]: Network device 'lan1' link is up
27. 01. 2024 09:02:51 notice kernel[]: [824349.462488] lan1: 1000 Mbps Full duplex, port 2, queue 2
27. 01. 2024 09:02:51 info kernel[]: [824349.467939] br-lan: port 2(lan1) entered blocking state
27. 01. 2024 09:02:51 info kernel[]: [824349.473277] br-lan: port 2(lan1) entered forwarding state
27. 01. 2024 09:13:15 warning kernel[]: [824973.692780] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.703022] IPv4: martian source 10.0.10.25 from 10.0.10.2, on dev br-lan
27. 01. 2024 09:13:15 warning kernel[]: [824973.709920] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
27. 01. 2024 09:13:15 warning kernel[]: [824973.718050] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.728325] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.738543] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.749263] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.759527] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.769745] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.780763] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:15 warning kernel[]: [824973.791053] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:17 warning dnsmasq[1]: ignoring query from non-local network 169.254.180.71 (logged only once)
27. 01. 2024 09:13:20 warning kernel[]: [824978.699414] net_ratelimit: 141 callbacks suppressed
27. 01. 2024 09:13:20 warning kernel[]: [824978.699431] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.707327] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.714646] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.725056] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.735018] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.755464] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.759691] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.765673] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.782467] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:20 warning kernel[]: [824978.786011] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:32 warning kernel[]: [824990.704112] net_ratelimit: 2143 callbacks suppressed
27. 01. 2024 09:13:32 warning kernel[]: [824990.704129] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:32 warning kernel[]: [824990.721029] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
27. 01. 2024 09:13:32 warning kernel[]: [824990.731277] IPv4: martian source 10.0.10.8 from 10.0.10.2, on dev br-lan
27. 01. 2024 09:13:32 warning kernel[]: [824990.738082] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
27. 01. 2024 09:13:37 warning kernel[]: [824995.832604] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
```

brada4, 2024-01-27:
The arp_proxy reflects arp requests with own source address back to system that sent it (received .... messages)
You can prevent this by disabling arp_proxy on AP-s bridged to the network.
fib (the table of mac-interface mappings) in any switch stays for 5-15-60s thus you get few messages when eg client roams between AP-s (the martian messages)

plouds, 2024-01-28:
What recourse is there if arp_proxy IS disabled on all APs in the LAN and the issue persists?

plouds, 2024-01-28:
Did a few more tests HW-wise: took the WAN port out of the equation both in terms of powering the device and as a member of br-lan. No difference, regardless of the WAN port being in or out of br-lan, or powering the AP by PoE or brick.
Final round of tests done with wifi disabled – only by means of having no SSID active, though. Still no difference – as soon as I plug in the laptop’s ethernet cable, the router begins reporting
```
28. 01. 2024 11:36:49 warning kernel[]: [919994.727674] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
28. 01. 2024 11:36:50 warning kernel[]: [919995.701448] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
28. 01. 2024 11:36:50 warning kernel[]: [919995.711677] IPv4: martian source 10.0.10.25 from 10.0.10.2, on dev br-lan
28. 01. 2024 11:36:50 warning kernel[]: [919995.718573] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
28. 01. 2024 11:36:51 warning kernel[]: [919996.762331] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
28. 01. 2024 11:36:51 warning kernel[]: [919996.772553] IPv4: martian source 10.0.10.25 from 10.0.10.2, on dev br-lan
28. 01. 2024 11:36:51 warning kernel[]: [919996.779436] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
28. 01. 2024 11:36:53 warning kernel[]: [919997.802451] br-lan: received packet on lan1 with own address as source address (addr:74:ac:b9:13:16:8e, vlan:0)
28. 01. 2024 11:36:53 warning kernel[]: [919997.812677] IPv4: martian source 10.0.10.25 from 10.0.10.2, on dev br-lan
28. 01. 2024 11:36:53 warning kernel[]: [919997.819559] ll header: 00000000: ff ff ff ff ff ff 74 ac b9 13 16 8e 08 06
```

brada4, 2024-01-28:
Do you brdge wifi and eth via your laptop?

plouds, 2024-01-28:
Nope. Wifi disabled on laptop, so it's connected by cable or not at all.
Oddly enough, when I first realised I had an issue, it was the TV box that refused to play, the TV app would not launch. The OS itself was usable and when I switched to wifi – in settings only, with the cable still connected – I was able to launch and use the TV app. So both connections were available, and with wifi in use, the network behaved itself.
Again, it’s only on kernel 6.1. With the APs on stable, i.e. 5.15, no problem with the same configuration.

brada4, 2024-01-28:
You need snapshots from last 2 weeks to fix packet reflection.
6.1 indicates problem, not causes. 
Connect your laptop to uplink to 6.1 router and start it up with wireshark active.
We looking for packets that come back as if from its own MAC.
YES- > its not 6.1
NO -> 6.1 it is

raenye, 2024-01-28:
```
config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan:u*'
```

Is that standard? (for lanX)
I thought one needs to mention `:t` explicitly.

plouds, 2024-01-28:
> You need snapshots from last 2 weeks to fix packet reflection.

The snapshot used is from 23 Jan

plouds, 2024-01-28:
Thank you for looking into it.
> I thought one needs to mention `:t` explicitly.

I was banking on the belief that a port can carry up to one untagged VLAN. WAN port untagged for home LAN, tagged for guest VLAN (3). Worked in v21 (swconfig if memory serves) and v23. (DSA).
```
config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '3'
	list ports 'wan:t'
```
As for the issue at hand, it exists on my device even with changed config and no VLANs.

raenye, 2024-01-28:
But are lan1-4 tagged or untagged for vlan 1?

brada4, 2024-01-28:
vlan1 is default, like make completely untagged bridge and it turns up in (ip-)bridge as VLAN1

raenye, 2024-01-28:
OK, so let me ask this way: why `wan` needs PVID 1 for egress? why not simply untagged?

plouds, 2024-01-29:
> But are lan1-4 tagged or untagged for vlan 1?

They are untagged and the above is what LuCI produces for me. Unlike wan, lan 1-4 are not marked primary since they are absent from VLAN 3 - which may be just a nonstandard shortcut of mine and one I've been getting away with until now.
I did try to mark the four as untagged and primary, without the desired effect of resolving the issue. I did not try rebooting the router though after the change.

plouds, 2024-01-29:
Two more tests: marking 1-4 as untagged + primary has no effect pre or post reboot. Marking them manually as :u (no asterisk) eliminates reports of martian packets but does not affect the own-address packets.

raenye, 2024-01-29:
And removing primary from wan?

plouds, 2024-01-29:
No difference :(

brada4, 2024-02-18:
Archer C7 V2 -> enable firewall offload and WAN (eth0.2) starts receiving martians like crazy, seems ethernet addresses swapped in card buffer of supposedly outgoing packet. Nothing on the wire via swconfig mirror port.
No more access to device, disabled offload and left it behind.

mrkiko, 2024-03-12:
Maybe unrelated, but still my setup looked like this, assuming I remember correctly:

FRITZ!Bx 7530:
	- to one port was connected a Netgear R7800
	- to another a GL-MT2500

this generated the error of packets received with own sources.
I used no VLANs at all, really plain config since 7530 was doing PPPOE over it's VDSL chip, R7800 acting as AP, GL-MT2500A was doing basically nothing.
Connecting the GL-MT2500 to the R7800 fixed this all.

siwind, 2024-04-07:
bug这么久了，问题还是没有解决吗~~

raenye, 2024-04-07:
> bug这么久了，问题还是没有解决吗~~

> The bug has been around for so long and the problem is still not solved~~

FTFY.

siwind, 2024-06-02:
The problem is still there:
```
Sun Jun  2 15:10:29 2024 kern.warn kernel: [  183.632178] br-lan: received packet on lan3 with own address as source address (addr:e4:ea:83:af:98:8f, vlan:0)
Sun Jun  2 15:10:29 2024 kern.warn kernel: [  183.652566] br-lan: received packet on lan3 with own address as source address (addr:e4:ea:83:af:98:8f, vlan:0)
```

RolandoMagico, 2024-06-13:
I have the same problem with main branch builds for a Linksys MR6350 (https://github.com/openwrt/openwrt/pull/14098) and the current snapshot builds for FritzBox 4040. Is there any further information required to fix the issue?

RolandoMagico, 2025-02-09:
Hello @siwind, @plouds,
can you try OpenWrt 24.10.0? I updated my FB4040 to this version and until now, the error is not present anymore.
BR
Roland

plouds, 2025-02-09:
I don't remember which 24.10.0-rc I tried first on the MikroTik hAP ac2 but the issue never reared its head thoughout, stable release included. So all good as far as I can see.

closed 2025-02-09 by plouds