openwrt/openwrt #14473: vlan traffic doesn't traverse bridge to wifi interface on wl-wn573hx1

timeport0, 2024-01-24:
### Describe the bug

When attempting to create an ssid bridged to a vlan-tagged lan interface, Openwrt does not forward the traffic received to the wifi interface.

Traffic from the wifi interface is forwarded and tagged correctly on the lan port, but the received tagged traffic is not forwarded to the wifi interface.

### OpenWrt version

r24806-735efbfb7c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

WAVLINK WL-WN573HX1

### Image kind

Self-built image

### Steps to reproduce

/etc/config/network
`config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd34:a801:269f::/48'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan'

config interface 'lan'
	option device 'br-lan'
	option proto 'dhcp'

config interface 'iot'
	option proto 'static'
	option device 'br-iot'
	option ipaddr '192.168.3.2'
	option netmask '255.255.255.0'

config device
	option type '8021q'
	option ifname 'lan'
	option vid '1027'
	option name 'lan.1027'

config device
	option type 'bridge'
	option name 'br-iot'
	list ports 'lan.1027'`

/etc/config/wireless
`config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HE20'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option network '1027 iot'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'`

configure upstream device to tag vlan 1027 on port facing the AP, configure  192.168.3.1/24 on the vlan interface
join "OpenWrt" network with a wifi device.
Ping wifi device from the AP(works)
ping AP from the wifi device(works)
ping router IP (192.168.3.1) from the AP(works)
ping AP(192.168.3.2) from the router(works)
Run a packet sniffer on the router vlan 1027 interface and ping the router from the client device. Observe arp requests from the AP and arp responses from the router. 
Run a packet sniffer on the wifi interface of the client device. Observe arp responses from router are never received.

### Actual behaviour

[image]

### Expected behaviour

Traffic from the router should be forwarded to the client device

### Additional info

forum thread 
https://forum.openwrt.org/t/vlans-on-wavlink-wl-wn573hx1/184928/18?u=timeport

Note eoip/tunneling packages were installed after observing the behavior. I can tunnel around it with EOIP/gretap.

Also this same config (except using lan instead of eth0) works fine on a wax220

### Diffconfig

```text
tmp/.config-package.in:138609:error: recursive dependency detected!
tmp/.config-package.in:138609:	symbol PACKAGE_prometheus-node-exporter-ucode depends on PACKAGE_ucode
tmp/.config-package.in:39406:	symbol PACKAGE_ucode is selected by PACKAGE_uspotfilter
tmp/.config-package.in:124963:	symbol PACKAGE_uspotfilter is selected by PACKAGE_uspot
tmp/.config-package.in:94662:	symbol PACKAGE_uspot depends on PACKAGE_uhttpd
tmp/.config-package.in:117179:	symbol PACKAGE_uhttpd is selected by PACKAGE_prometheus-node-exporter-ucode
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:138609:error: recursive dependency detected!
tmp/.config-package.in:138609:	symbol PACKAGE_prometheus-node-exporter-ucode depends on PACKAGE_ucode
tmp/.config-package.in:39406:	symbol PACKAGE_ucode is selected by PACKAGE_uspotfilter
tmp/.config-package.in:124963:	symbol PACKAGE_uspotfilter is selected by PACKAGE_uspot
tmp/.config-package.in:94662:	symbol PACKAGE_uspot depends on PACKAGE_uhttpd
tmp/.config-package.in:117179:	symbol PACKAGE_uhttpd is selected by PACKAGE_prometheus-node-exporter-ucode
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_wavlink_wl-wn573hx1=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_eoip=y
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_gre=y
CONFIG_PACKAGE_kmod-gre=y
CONFIG_PACKAGE_kmod-gre6=y
CONFIG_PACKAGE_kmod-ip6-tunnel=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel6=y
CONFIG_PACKAGE_kmod-mmc=y
CONFIG_PACKAGE_kmod-mt76=y
CONFIG_PACKAGE_kmod-mt76-sdio=y
CONFIG_PACKAGE_kmod-mt76-usb=y
CONFIG_PACKAGE_kmod-mt7603=y
CONFIG_PACKAGE_kmod-mt7615-common=y
CONFIG_PACKAGE_kmod-mt7615-firmware=y
CONFIG_PACKAGE_kmod-mt7615e=y
CONFIG_PACKAGE_kmod-mt76x02-common=y
CONFIG_PACKAGE_kmod-mt76x2=y
CONFIG_PACKAGE_kmod-mt76x2-common=y
CONFIG_PACKAGE_kmod-mt7921-common=y
CONFIG_PACKAGE_kmod-mt7921-firmware=y
CONFIG_PACKAGE_kmod-mt7921e=y
CONFIG_PACKAGE_kmod-mt7921s=y
CONFIG_PACKAGE_kmod-mt7921u=y
CONFIG_PACKAGE_kmod-mt7922-firmware=y
CONFIG_PACKAGE_kmod-mt7925-common=y
CONFIG_PACKAGE_kmod-mt7925e=y
CONFIG_PACKAGE_kmod-mt7925u=y
CONFIG_PACKAGE_kmod-mt792x-common=y
CONFIG_PACKAGE_kmod-mt792x-usb=y
CONFIG_PACKAGE_kmod-mt7996e=y
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libevdev=y
CONFIG_PACKAGE_libkmod=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libpci=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libudev-zero=y
CONFIG_PACKAGE_libusb-1.0=y
CONFIG_PACKAGE_lrzsz=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-eoip=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-gre=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_mikrotik-btest=y
CONFIG_PACKAGE_mt7601u-firmware=y
CONFIG_PACKAGE_mt7622bt-firmware=y
CONFIG_PACKAGE_mt7921bt-firmware=y
CONFIG_PACKAGE_mt7922bt-firmware=y
CONFIG_PACKAGE_nano-full=y
CONFIG_PACKAGE_pciids=y
CONFIG_PACKAGE_pciutils=y
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_zlib=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-24:
```
 option network '1027 iot'
```
This is never defined. AP provides wifi but is not connected anywhere.

timeport0, 2024-01-24:
Hi -- thanks for the note, that line was added by luci. As part of the troubleshooting process, we changed that to just 
`
option network 'iot'
`
with no change in behavior. I'm not sure why luci created the line like that.

timeport0, 2024-01-24:
we also tried the bridge-vlan configuration(as suggested by forum user psherman) with somewhat identical results. Except packets for vlan1 were coming through tagged instead of untagged as configured:

[image]

`
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd34:a801:269f::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan'

config interface 'lan'
        option device 'br-lan.1'
        option proto 'dhcp'

config interface 'iot'
        option proto 'none'
        option device 'br-lan.1027

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'lan:u*'

config bridge-vlan
        option device 'br-lan'
        option vlan '1027'
        list ports 'lan:t'

`

brada4, 2024-01-24:
Try brctl show and/or bridge vlan show.
vlan id 1 is kind of default in dsa switch, anything else will work better especially during reconfigurations.

brada4, 2024-01-24:
Example (no vlans enabled in luci)
```
bridge vlan show
port              vlan-id  
lan1              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan4              1 PVID Egress Untagged
br-lan            1 PVID Egress Untagged
phy1-ap0          1 PVID Egress Untagged
phy0-ap0          1 PVID Egress Untagged
```

timeport0, 2024-01-24:
I appreciate the help -- give me a bit to go back to that config. I had changed it to use gretap to (successfully) tunnel around the problem but gretap performance isn't ideal.

brada4, 2024-01-24:
Just a tip - add another AP just for testing with workaround in place.....

timeport0, 2024-01-24:
What package provides the "bridge" command? It's apparently not installed here. 
```
root@OpenWrt:/etc/config# bridge vlan show
-ash: bridge: not found
```
```
root@OpenWrt:/etc/config# brctl show

bridge name	bridge id		STP enabled	interfaces
br-lan		7fff.803f5d72a315	no		lan
br-iot		7fff.803f5d72a315	no		lan.1027
							phy0-ap0
```

brada4, 2024-01-24:
So lan.1027 is bridged transparently with phy0-ap0,i.e i see nothing wrong with your setup.

Alternative option would be to add vlan 1027 to br-lan and use br-lan.1027 as the network for wifi.

bridge command is in package ip-bridge. It is like more modern covering more corner cases.

timeport0, 2024-01-24:
yes -- the same config is working on another model (wax220)

This is a newly supported hardware(only in snapshot) -- the MTD device is not even supported in snapshot, I have separate patch for that.

The supporting commit says there is a switch chip in this model even though it only has 1 lan port.

My guess is something is not happening quite right here with that.

The AP can ping the client 
the client can ping the AP

the AP can ping the router (on vlan 1027)
and the router can ping the AP (on vlan 1027)

The router receives packets from the client on the correct vlan, 
the AP receives the packets from the router on the correct vlan. 
But the AP does not forward those packets to the client on wifi

output from bridge vlan show after installing ip-bridge
```
root@OpenWrt:~# bridge vlan show
port              vlan-id  
lan               1 PVID Egress Untagged
br-lan            1 PVID Egress Untagged
br-iot            1 PVID Egress Untagged
lan.1027          1 PVID Egress Untagged
phy0-ap0          1 PVID Egress Untagged
root@OpenWrt:~# 
```

brada4, 2024-01-24:
The switch is dsa device connected via (g)mii and besides gigabit link controlled using special gmii registers or other kind of sideband IO.

Now it is more clear, add vlan to the bridge.
[image]

brada4, 2024-01-24:
@jow- did netifd rewrite port configuration? Initial configuration was legit too as per [DOC](https://openwrt.org/docs/guide-user/network/vlan/switch_configuration#assigning_vlan_ids_using_dsa_on_devices_with_one_physical_port)

PDS-GH, 2024-02-11:
This looks awfully similar to the issue I am describing here:
https://forum.openwrt.org/t/openwrt-23-05-0-first-stable-release/174239/363

brada4, 2024-02-11:
vlan id 1 is used internally by DSA making it unusable outside.