openwrt/openwrt #14477: qualcommax/ipq807x/ax3600: Bootlop after adding led configuration

McGiverGim, 2024-01-25:
### Describe the bug

I have a led configuration for my mesh interface:
```
config led
        option trigger 'netdev'
        list mode 'link'
        option name ''
        option dev 'm-11s-0'
        option sysfs 'blue:aiot'
```
The "m-11-s" is a mesh 802.11s (the strange name comes for this bug https://github.com/openwrt/openwrt/issues/14301)
After updating to latest snapshot, the router entered in bootloop. After entering failsafe and removing this configuration, the router starting to boot again without problem.
If I add the config again, the router starts the bootloop again.

### OpenWrt version

r24894-d82930c7c7

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

I'm not too sure if it happens with other led configuration or only for this.

### Actual behaviour

The router enter in bootloop.

### Expected behaviour

The router boot normally.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

McGiverGim, 2024-01-25:
Looking at the code, I see this commit recently related to LED and ipq807x: https://github.com/openwrt/openwrt/commit/fb45194305e6bf9b58f610c1528769320879c749 I don't know if it's related

McGiverGim, 2024-01-25:
I've done more tests, if it helps:
- If I enable the LED after the router has boot and is online, the led lights up and works, but if then I reboot the router, the bootloop starts.
- If I select another led device, different than the `option dev 'm-11s-0'` the bootloop disappears. The router boots without problem.
- If I select a different led than the `option sysfs 'blue:aiot'` but I use the `option dev 'm-11s-0'` here we have the bootloop again.

So the problem is with the `option dev 'm-11s-0'` and only when booting. If the router is up and I configure it the light works and all is ok until the next reboot.

McGiverGim, 2024-01-25:
Another thing, I don't know if related: The 5G radio, has two networks: one is the MESH that is producing the problem, and the other one is a standard network, where the clients connect. When I reboot the router, this standard network appears as enabled, but the icon in LuCi shows as deactivated, and it is really down.
If I disable and enable it again, with the LuCi button, the network starts to work again without problem.

McGiverGim, 2024-01-27:
Today I've updated to latest master, some differences:
- The problem with the wifi is gone.
- The bootloop remains, but after some loops it boots correctly. I suppose some kind of race condition.

McGiverGim, 2024-01-27:
It seems both problems are related:
- If I disable the LED config, it boots at the first try and the wifi is down.
- If I enable de LED config, it enters the bootloop, after some tries it boots, and the wifi is up.

@Ansuel @robimarko both of you have Xiaomi AX3600 and know how they work. Some idea of what can be happening here?

McGiverGim, 2024-01-27:
If it helps, here is the log after issuing a `wifi` command:
```
Sat Jan 27 09:53:18 2024 daemon.info hostapd: wlan5g: STA ac:67:84:56:2c:af IEEE 802.11: authenticated
Sat Jan 27 09:53:18 2024 daemon.info hostapd: wlan5g: STA ac:67:84:56:2c:af IEEE 802.11: associated (aid 2)
Sat Jan 27 09:53:18 2024 daemon.notice hostapd: wlan5g: AP-STA-CONNECTED ac:67:84:56:2c:af auth_alg=open
Sat Jan 27 09:53:18 2024 daemon.info hostapd: wlan5g: STA ac:67:84:56:2c:af RADIUS: starting accounting session 2AF5066D0DAE98CB
Sat Jan 27 09:53:18 2024 daemon.info hostapd: wlan5g: STA ac:67:84:56:2c:af WPA: pairwise key handshake completed (RSN)
Sat Jan 27 09:53:18 2024 daemon.notice hostapd: wlan5g: EAPOL-4WAY-HS-COMPLETED ac:67:84:56:2c:af
Sat Jan 27 09:53:37 2024 daemon.notice hostapd: Set new config for phy phy2:
Sat Jan 27 09:53:37 2024 daemon.notice hostapd: Remove interface 'phy2'
Sat Jan 27 09:53:37 2024 daemon.notice hostapd: wlan24g: interface state ENABLED->DISABLED
Sat Jan 27 09:53:37 2024 daemon.notice hostapd: wlan24g: AP-DISABLED
Sat Jan 27 09:53:37 2024 daemon.notice hostapd: wlan24g: CTRL-EVENT-TERMINATING
Sat Jan 27 09:53:37 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Jan 27 09:53:37 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan24g disabled_11b_rates=0
Sat Jan 27 09:53:37 2024 kern.info kernel: [ 1967.544641] device wlan24g left promiscuous mode
Sat Jan 27 09:53:37 2024 kern.info kernel: [ 1967.544799] br-lan: port 5(wlan24g) entered disabled state
Sat Jan 27 09:53:37 2024 daemon.notice netifd: Network device 'wlan24g' link is down
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: Set new config for phy phy1:
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: Remove interface 'phy1'
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: wlan5g: interface state ENABLED->DISABLED
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: wlan5g: AP-STA-DISCONNECTED ac:67:84:56:2c:af
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: wlan5g: AP-STA-DISCONNECTED ee:c3:e5:2d:d6:d4
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: wlan5g: AP-DISABLED
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: wlan5g: CTRL-EVENT-TERMINATING
Sat Jan 27 09:53:38 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Jan 27 09:53:38 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan5g disabled_11b_rates=0
Sat Jan 27 09:53:38 2024 kern.info kernel: [ 1967.974487] device wlan5g left promiscuous mode
Sat Jan 27 09:53:38 2024 kern.info kernel: [ 1967.974653] br-lan: port 6(wlan5g) entered disabled state
Sat Jan 27 09:53:38 2024 daemon.notice netifd: Network device 'wlan5g' link is down
Sat Jan 27 09:53:38 2024 daemon.notice wpa_supplicant[1677]: Set new config for phy phy2
Sat Jan 27 09:53:38 2024 daemon.notice wpa_supplicant[1677]: Set new config for phy phy1
Sat Jan 27 09:53:38 2024 daemon.notice wpa_supplicant[1677]: Remove interface mesh
Sat Jan 27 09:53:38 2024 daemon.notice wpa_supplicant[1677]: mesh: MESH-GROUP-REMOVED mesh
Sat Jan 27 09:53:38 2024 daemon.notice wpa_supplicant[1677]: mesh: leaving mesh
Sat Jan 27 09:53:38 2024 daemon.notice netifd: Wireless device 'radio2' is now down
Sat Jan 27 09:53:38 2024 daemon.notice netifd: radio2 (10796): WARNING: Variable 'data' does not exist or is not an array/object
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: mesh: interface state ENABLED->DISABLED
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: mesh: AP-DISABLED
Sat Jan 27 09:53:39 2024 daemon.notice netifd: Network device 'mesh' link is down
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: mesh: CTRL-EVENT-DSCP-POLICY clear_all
Sat Jan 27 09:53:39 2024 kern.info kernel: [ 1968.755663] br-lan: port 7(mesh) entered disabled state
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: mesh: CTRL-EVENT-DSCP-POLICY clear_all
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: nl80211: deinit ifname=mesh disabled_11b_rates=0
Sat Jan 27 09:53:39 2024 daemon.err wpa_supplicant[1677]: rmdir[ctrl_interface=/var/run/wpa_supplicant]: Permission denied
Sat Jan 27 09:53:39 2024 kern.info kernel: [ 1968.795087] br-lan: port 7(mesh) entered blocking state
Sat Jan 27 09:53:39 2024 kern.info kernel: [ 1968.795130] br-lan: port 7(mesh) entered forwarding state
Sat Jan 27 09:53:39 2024 kern.info kernel: [ 1968.884892] br-lan: port 7(mesh) entered disabled state
Sat Jan 27 09:53:39 2024 kern.info kernel: [ 1968.924695] device mesh left promiscuous mode
Sat Jan 27 09:53:39 2024 kern.info kernel: [ 1968.924734] br-lan: port 7(mesh) entered disabled state
Sat Jan 27 09:53:39 2024 daemon.notice hostapd: Set new config for phy phy2:
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: Set new config for phy phy2
Sat Jan 27 09:53:39 2024 daemon.notice netifd: Wireless device 'radio1' is now down
Sat Jan 27 09:53:39 2024 daemon.notice netifd: radio1 (10834): WARNING: Variable 'data' does not exist or is not an array/object
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: Set new config for phy phy2
Sat Jan 27 09:53:39 2024 daemon.notice hostapd: Set new config for phy phy1:
Sat Jan 27 09:53:39 2024 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Sat Jan 27 09:53:39 2024 daemon.notice hostapd: Restart interface for phy phy2
Sat Jan 27 09:53:39 2024 daemon.notice wpa_supplicant[1677]: Set new config for phy phy1
[... cut ...]
Sat Jan 27 09:53:45 2024 daemon.notice wpa_supplicant[1677]: mesh: mesh plink with 2e:d1:27:4c:52:a1 established
Sat Jan 27 09:53:45 2024 daemon.notice wpa_supplicant[1677]: mesh: MESH-PEER-CONNECTED 2e:d1:27:4c:52:a1
Sat Jan 27 09:53:45 2024 kern.info kernel: [ 1975.424458] device br-lan left promiscuous mode
```
This is on an AP, but I face the same problem with the Router (both are AX3600, using mesh). 
I don't understand too much about OpenWrt, but this line seems problematic:
```
Sat Jan 27 09:53:40 2024 daemon.err hostapd: Could not read interface wlan5g flags: No such device
```

McGiverGim, 2024-01-27:
I've observed this line too:
```
Sat Jan 27 09:53:40 2024 kern.info kernel: [ 1969.981285] ath11k c000000.wifi mesh: renamed from wlan5g
```
This implies that the `wlan5g` is renamed to `mesh`? If yes, then the `wlan5g` does not exist more and thus the:
```
Sat Jan 27 09:53:40 2024 daemon.err hostapd: Could not read interface wlan5g flags: No such device
```
that produces all the problem?

McGiverGim, 2024-01-27:
I did some more tests:
- If I remove the mesh wireless network, the wlan5g starts without problem.
- If I change the mesh wireless by another wlan5g (so I have two wlan5g) both networks start without problem.
- If I let the mesh and the wlan5g, the problem happens again.

This are mi entries, to see if someone could reproduce it:
```
config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'
        option country 'ES'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'PIMINET'
        option ifname 'wlan5g'
        option key '*****'
        option ieee80211r '1'
        option ft_over_ds '0'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option wnm_sleep_mode '1'
        option bss_transition '1'
        option network 'lan'
        option encryption 'psk2'
        option ft_psk_generate_local '1'
        option multicast_to_unicast_all '1'

config wifi-iface 'wifinet2'
        option device 'radio1'
        option mode 'mesh'
        option mesh_fwding '1'
        option mesh_rssi_threshold '0'
        option network 'lan'
        option encryption 'sae'
        option mesh_id 'PIMIMESH'
        option key '*****'
        option ifname 'mesh'
```

McGiverGim, 2024-01-27:
One latest test: if I disable the mesh, the router boots normally, as commented before. If then I try to enable it, it breaks the wlan5g. If I do the inverse it works (enable the wlan5g after the mesh).

McGiverGim, 2024-01-31:
Finally I've been able to compile by myself OpenWrt. I'm tryting to bisect this, I have confirmed that commit that added kernel 6.1.69 works without problem, and latest master fails. I will try to find the culprit. Some idea of what could be the commit?

McGiverGim, 2024-01-31:
Finally I've found the culprit.

This commit works: https://github.com/openwrt/openwrt/commit/c0c3234e17207a9287a08757fc1752490144a1cd

This commit fails: https://github.com/openwrt/openwrt/commit/ced3fbcda18e200315cb077f7e2b362ae52ee065

So it seems the problem has been the update of `ubox`. I don't know what is `ubox` and how it can break this. There is some newer version with fixes? Better to do a rollback? Or maybe let it as is with this bug? @Ansuel you added this commit, some idea of what is wrong?

Ansuel, 2024-01-31:
@McGiverGim no chance you have output of the log right?

McGiverGim, 2024-01-31:
I don't have serial connection, but I suppose it's not needed. If you see my latest messages it happens simply using a wifi command. Maybe some way to make this command show more logs?
There is a way to get the log you want without serial?
I can give you the files tomorrow (I'm out of the computer now) but I think the only needed is two wireless in the 5GHz radio: one for wifi and the other for mesh.

Ansuel, 2024-01-31:
that log don't show a bootloop... i would expect a crash or a panic for a bootloop

McGiverGim, 2024-01-31:
The bootloop was the final result, what happens is that the init of the mesh network crashes the 5GHz wifi network, it seems the rename of the interface is renaming the wrong one, the bootloop was produced because the LED was associated to the mesh, and with this crash the LED crashes too and reboots.

McGiverGim, 2024-02-01:
@Ansuel I've been doing more tests, and it seems is some kind of race condition. I explain:
- I updated to latest master, reverted the ubox commit, and all seems to work as expected.
- I tried to find the ubox commit that produced the problem
- Commit https://github.com/openwrt/ubox/commit/c006dccecb6f1cf694c9fb9efc6727cbd2ab9ae4 seems to produce the error always (or almost always)
- Commit https://github.com/openwrt/ubox/commit/ba3908351232d4166cfc89c23c5c629e4092169d produce the error in a random manner. Each time I issue a `wifi` command, sometimes the 5GHz wifi is down and others is up
- Commit https://github.com/openwrt/ubox/commit/4ffc29e4041ca1b5d6b0f007859394c23fba034e it seems to be ok but I think I've seen it crash once (not too sure)
- Commit https://github.com/openwrt/ubox/commit/8c95fc7039cb7db79f0f484cf5a29648e3118057 seems to be ok, but maybe I've not issued the `wifi` command enough times to make it crash

After a `wifi` command, if the 5GHz wifi has crashed, I can see in the log the message, the full log is similar to the once I appended before:
```
kern.info kernel: [ 1969.981285] ath11k c000000.wifi mesh: renamed from wlan5g
```
that is what I think is starting the problem. 

To try to reproduce this, I think this is simply enough:
wireless file:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/20000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '1'
        option cell_density '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'
        option country 'ES'

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi+1'
        option band '2g'
        option cell_density '0'
        option htmode 'HE40'
        option channel '6'
        option txpower '20'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'PIMINET'
        option ifname 'wlan5g'
        option key '*****'
        option network 'lan'
        option encryption 'psk2'

config wifi-iface 'wifinet2'
        option device 'radio1'
        option mode 'mesh'
        option mesh_fwding '1'
        option mesh_rssi_threshold '0'
        option network 'lan'
        option encryption 'sae'
        option mesh_id 'PIMIMESH'
        option key '*****'
        option ifname 'mesh'

config wifi-iface 'wifinet3'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'PIMINET_N'
        option ifname 'wlan24g'
        option key '*****'
        option encryption 'psk2'
```
As you see, a 2.4G (wlan24g) network, a 5G (wlan5g) network and a MESH (mesh) network.
If you issue the `wifi` command, and after that a `iwinfo` you will see the 5G network down. If you add this to the recipe in the `system` file:
```
config led
        option trigger 'netdev'
        list mode 'link'
        option name 'MESH'
        option dev 'mesh'
        option sysfs 'blue:aiot'
```
is when the bootloop start.

Ansuel, 2024-02-03:
Manage to repro...
```
[  149.393036] br-lan: port 4(wlan24g) entered disabled state
[  149.397645] device wlan24g entered promiscuous mode
[  149.403025] br-lan: port 4(wlan24g) entered blocking state
[  149.407642] br-lan: port 4(wlan24g) entered forwarding state
[  149.413775] br-lan: port 4(wlan24g) entered disabled state
[  149.547927] br-lan: port 5(wlan5g) entered blocking state
[  149.547975] br-lan: port 5(wlan5g) entered disabled state
[  149.552663] device wlan5g entered promiscuous mode
[  149.558881] device wlan5g left promiscuous mode
[  149.562566] br-lan: port 5(wlan5g) entered disabled state
[  149.656738] ath11k c000000.wifi mesh: renamed from wlan5g
[  149.709795] Unable to handle kernel access to user memory outside uaccess routines at virtual address 0000000000000038
[  149.709848] Mem abort info:
[  149.719383]   ESR = 0x0000000096000005
[  149.722108]   EC = 0x25: DABT (current EL), IL = 32 bits
[  149.725911]   SET = 0, FnV = 0
[  149.731384]   EA = 0, S1PTW = 0
[  149.734228]   FSC = 0x05: level 1 translation fault
[  149.737270] Data abort info:
[  149.742390]   ISV = 0, ISS = 0x00000005
[  149.745254]   CM = 0, WnR = 0
[  149.748812] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004f3d2000
[  149.752083] [0000000000000038] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  149.758374] Internal error: Oops: 0000000096000005 [#1] SMP
[  149.766873] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath11k_pci ath11k_ahb ath11k pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 tmp103 slhc rfcomm regmap_i2c qrtr_smd qrtr_mhi qrtr qmi_helpers nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mhi mdio_netlink libcrc32c hidp hci_uart compat btusb btrtl btmtk btintel bnep bluetooth hid evdev input_core ecdh_generic ecc sha512_generic seqiv jitterentropy_rng drbg michael_mic kpp hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug ext4 mbcache jbd2 aquantia hwmon crc_ccitt crc32c_generic
[  149.829361] CPU: 0 PID: 1629 Comm: wpa_supplicant Tainted: G        W          6.1.74 #0
[  149.851599] Hardware name: Xiaomi AX3600 (DT)
[  149.859753] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  149.864010] pc : get_device_state+0x1c/0x70
[  149.870776] lr : netdev_trig_notify+0xbc/0x294
[  149.874942] sp : ffffffc00bd43400
[  149.879455] x29: ffffffc00bd43400 x28: 0000000000000003 x27: ffffffc009441308
[  149.882847] x26: ffffff800fd24000 x25: ffffffc009441200 x24: ffffffc009494940
[  149.889964] x23: ffffff8004830748 x22: ffffff800fd24000 x21: ffffff8007777500
[  149.897084] x20: ffffff8007777578 x19: ffffff8007777500 x18: ffffffc00f27bcb0
[  149.904202] x17: 0000000000000000 x16: ffffffc008857988 x15: ffffff800f0be20c
[  149.911321] x14: ffffffffffffffff x13: 0000000000000020 x12: 0101010101010101
[  149.918438] x11: 7f7f7f7f7f7f7f7f x10: fffffffff7a535a0 x9 : 0000000000000020
[  149.925555] x8 : 0101010101010101 x7 : 0000000080008080 x6 : 0000000080678080
[  149.932674] x5 : 0000670000000000 x4 : 0000000000000000 x3 : 0000000000000000
[  149.939792] x2 : ffffff8003b68c00 x1 : ffffffc00bd43428 x0 : 0000000000000000
[  149.946911] Call trace:
[  149.954019]  get_device_state+0x1c/0x70
[  149.956278]  netdev_trig_notify+0xbc/0x294
[  149.960098]  raw_notifier_call_chain+0x54/0x74
[  149.964265]  call_netdevice_notifiers_info+0x54/0xa0
[  149.968694]  dev_change_name+0x1d8/0x370
[  149.973813]  do_setlink+0x720/0xedc
[  149.977718]  rtnl_setlink+0xe0/0x180
[  149.980931]  rtnetlink_rcv_msg+0x24c/0x354
[  149.984751]  netlink_rcv_skb+0x5c/0x12c
[  149.988656]  rtnetlink_rcv+0x18/0x24
[  149.992389]  netlink_unicast+0x1e8/0x2d4
[  149.996209]  netlink_sendmsg+0x1a0/0x3d0
[  150.000115]  ____sys_sendmsg+0x1c8/0x270
[  150.004022]  ___sys_sendmsg+0x7c/0xc0
[  150.007926]  __sys_sendmsg+0x48/0xb0
[  150.011486]  __arm64_sys_sendmsg+0x24/0x30
[  150.015133]  invoke_syscall.constprop.0+0x5c/0x104
[  150.019041]  do_el0_svc+0x58/0x17c
[  150.023811]  el0_svc+0x18/0x54
[  150.027197]  el0t_64_sync_handler+0xf4/0x120
[  150.030238]  el0t_64_sync+0x174/0x178
[  150.034669] Code: f9000bf3 aa0003f3 9100a3e1 f9404c00 (f9401c02)
[  150.038230] ---[ end trace 0000000000000000 ]---
[  150.044303] Kernel panic - not syncing: Oops: Fatal exception
[  150.048990] SMP: stopping secondary CPUs
[  150.054634] Kernel Offset: disabled
[  150.058620] CPU features: 0x00000,00000000,0000400b
[  150.061836] Memory Limit: none
[  150.066696] Rebooting in 3 seconds..
```

Very interesting bug! Even more strange that ubox is the culprit... (still have to confirm this)

McGiverGim, 2024-02-03:
Hi! Thanks for looking at it.
I don't think ubox is the culprit exactly, I think this bug was there since a lot of time ago but the changes in ubox make it more evident, but maybe is in another part.
But this is only a supposition seeing the behavior in my tests.

robimarko, 2024-02-03:
Now, this is a "fun" bug.
How is ubox causing a NULL pointer in `netdev_trig_notify` which is in kernel space?

Ansuel, 2024-02-03:
As soon as ba3908351232d4166cfc89c23c5c629e4092169d is used the thing crash on boot
4ffc29e4041ca1b5d6b0f007859394c23fba034e is fine.

robimarko, 2024-02-03:
And https://github.com/openwrt/ubox/commit/2c5887cb46883a28d69071c4349c3dabbbe3972c doesnt fix it?

McGiverGim, 2024-02-03:
I'm not too sure of that. If you issue the `wifi` command you will see that it crashes, but not always. Sometimes yes, sometimes no.

Ansuel, 2024-02-03:
@McGiverGim i tested the command 10 times and it didn't crash...

robimarko, 2024-02-03:
That is why I find it rather weird that ubox can cause a NULL pointer dereference in `netdev_trig_notify` in any way

Ansuel, 2024-02-03:
@robimarko that fix doesn't fix our bug... the netdev_trig_notify might be a false positive but worth to try to backport some leds fixes... let me check... there were some for interface rename so maybe... (it's still strange that ubox cause it... maybe it's a time thing and additional code in ubox made it slow (or fast???) enough to trigger it)

robimarko, 2024-02-03:
Well, it might be somehow accidentally making a race even worse thus uncovering the issue cause this seems like race to me and possible TOCTU as well

McGiverGim, 2024-02-03:
I'm not too sure because my knowledge on this is reduced, but it seems you are centered on the notify for the led, that produces the bootloop, but I think the important is why the wlan5g crashes. Knowing this I think will be easier to understand why the LED produces the bootloop.

McGiverGim, 2024-02-03:
> @McGiverGim i tested the command 10 times and it didn't crash...

If it helps, when I was in bootloop with master, if I let it some time, after some loops the router starts. So I can be sure that with latest ubox, after some tries and under certain conditions, sometimes the wifi does not crash and the bootloop ends.

Ansuel, 2024-02-03:
Interesting I found the problem and I'm trying to understand what was the logic behind it...

Ansuel, 2024-02-03:
@robimarko if you are interested and have a clue... netdev_trig_notify
 
case NETDEV_CHANGENAME

we call get_device_state before dev_put and dev_hold but i'm trying to understand why we did that...

robimarko, 2024-02-03:
If you dont figure it out before, I can have a look tommorow, its sleep time currently

Ansuel, 2024-02-03:
@robimarko nha i found the bug... an interesting one! And yes it seems for real a timing thing so ubox wasn't the culprit at all

Ansuel, 2024-02-03:
It all comes down to a refactor and get_device_state using the previous net_dev on interface rename. Will submit a patch upstream and in the meantime will just push a downstream patch here to handle and close this issue!

McGiverGim, 2024-02-03:
Thanks to both of you for giving OpenWrt to my device!

closed 2024-02-04 by openwrt-bot

McGiverGim, 2024-02-06:
@Ansuel with the patch, the wlan5g must work always? Or it only fixes the bootloop? I've compiled latest master and the problem remains. I'm not an expert building OpenWrt, so I don't know if I need to make something "special" for this to work.

McGiverGim, 2024-02-07:
I've opened a new issue to try to fix the wireless network, because this issue is closed.

commit 508f2dbf (2024-02-04): generic: 6.1: add patch fixing bugs with LED netdev trigger

Backport one patch merged upstream that prevent a deadlock for LED
netdev trigger and add a pending patch that fix kernel panic on
interface rename trigger notification with invalid dev.

Fixes: #14477