openwrt/openwrt #14490: dnsmasq: unreliable interface ignore configuration

yegorich, 2024-01-26:
### Describe the bug

My `/etc/config/dhcp` has the following section:

```
config dhcp wan
    option interface wan
    option ignore 1
```

When the system starts `/var/etc/dnsmasq.conf.*` doesn't have `no-dhcp-interface` entry. Only when I manually invoke `/etc/init.d/dnsmasq restart`, this entry appears and I don't get the dnsmasq warnings concerning the missing addess range.

### OpenWrt version

r24436-0f283ab4c9

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

custom hardware

### Device

custom device

### Image kind

Self-built image

### Steps to reproduce

Start the system and check the generated  `/var/etc/dnsmasq.conf.*`.

### Actual behaviour

No `no-dhcp-interface` entry.

### Expected behaviour

The `no-dhcp-interface` entry is part of the generated  `/var/etc/dnsmasq.conf.*` file.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-27:
dhcp server is never enabled on wan (eth1) on default x86_64. Look in your local customisations very carefully, especially repeat your problem on any VM image.

jow-, 2024-01-27:
This is an actual bug. If the wan ifname cannot be resolved on boot (e.g. slow proto or interface with auto 0) the ignore setting will be skipped. Normally an ifup hotplug reload handler will take care of reconfiguring dnsmasq if the iface comes up but since wan is ignored, no reload trigger will be installed for it...

Solution is likely to also install an interface trigger for ignored interfaces in case their ifname cannot be resolved

yegorich, 2024-01-29:
@jow- thanks for the hint. This hack is working for me:

```
diff --git a/package/network/config/netifd/files/etc/hotplug.d/iface/00-netstate b/package/network/config/netifd/files/etc/hotplug.d/iface/00-netstate
index 71ccb01915..4150c900ae 100644
--- a/package/network/config/netifd/files/etc/hotplug.d/iface/00-netstate
+++ b/package/network/config/netifd/files/etc/hotplug.d/iface/00-netstate
@@ -1,6 +1,13 @@
 [ ifup = "$ACTION" ] && {
+       config_load dhcp
+       config_get_bool ignore "$INTERFACE" ignore 0
+
        uci_toggle_state network "$INTERFACE" up 1
        [ -n "$DEVICE" ] && {
                uci_toggle_state network "$INTERFACE" ifname "$DEVICE"
        }
+
+       [ "$ignore" == 1 ] && {
+               service dnsmasq restart
+       }
 }
```

jow-, 2024-01-29:
@yegorich - thanks for confirming, can you try this fix?

```diff
diff --git a/package/network/services/dnsmasq/files/dnsmasq.init b/package/network/services/dnsmasq/files/dnsmasq.init
index 510d328cd2..a742fa2a43 100755
--- a/package/network/services/dnsmasq/files/dnsmasq.init
+++ b/package/network/services/dnsmasq/files/dnsmasq.init
@@ -1244,12 +1244,13 @@ dnsmasq_stop()
 
 add_interface_trigger()
 {
-       local interface ignore
+       local interface ifname ignore
 
        config_get interface "$1" interface
        config_get_bool ignore "$1" ignore 0
+       network_get_device ifname "$interface"
 
-       [ -n "$interface" ] && [ $ignore -eq 0 ] && procd_add_interface_trigger "interface.*" "$interface" /etc/init.d/dnsmasq reload
+       [ -n "$interface" ] && [ $ignore -eq 0 -o -z "$ifname" ] && procd_add_interface_trigger "interface.*" "$interface" /etc/init.d/dnsmasq reload
 }
 
 service_triggers()
```

yegorich, 2024-01-29:
I confirm that this fix works.

yegorich, 2024-02-01:
@jow- are you going to create a patch, or should I make it?

yegorich, 2024-03-26:
Issue closed via https://github.com/openwrt/openwrt/commit/62acd9a2f9f18f30997d23b479abaea4940a4273.

closed 2024-03-26 by yegorich