openwrt/openwrt #14494: r8000 - Packages for block-mount found, but incompatible with the architectures configured

Mile-Lile, 2024-01-27:
### Describe the bug

root@R8000:~# opkg install block-mount
Unknown package 'block-mount'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for block-mount
 * pkg_hash_check_unresolved: cannot find dependency libblobmsg-json20240126 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package block-mount.

### OpenWrt version

r24913-7441c2bef6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Netgear R8000 (BCM4709)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-27:
Run opkg update again soon Probably mirror is not synced up with latest snapshot.

Mile-Lile, 2024-01-28:
@brada4 
Thak you for reply. I did what you suggested (tried a couple of times) but I got same message again...
Even if I try to build with luci-app-attendedsysupgrade I get similar message:
[image]
is there I something I could do to provide more info to you?

brada4, 2024-01-28:
Mirrors are still not in sync. You can check directly here for ubox20240126 
https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9/base/
Enough of crossposting random screenshots. It will resolve by itself with time.

closed 2024-01-29 by Mile-Lile