openwrt/openwrt #14502: QCA8081 PHY LED can't shut down

yyyzz123, 2024-01-29:
### Describe the bug

QCA8081 get upstream driver and add led control, it work, but it can't shut down
It work in previously versions https://github.com/Ansuel/openwrt/commit/57afcfb9213f29a838c28f79e8fec78c542e6bb5

### OpenWrt version

r24936-60ffcfdabc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-01-29:
What does "cant shut down" mean?
LED is always on or?

yyyzz123, 2024-01-29:
> What does "cant shut down" mean? LED is always on or?

Yes, it always on, can't turn off in LED Configuration or uci command, the previously versions can.

Ansuel, 2024-02-03:
Fixed by 7e0a7bf21da68d4f8044c43012d82c1ea10a46b2

closed 2024-02-03 by Ansuel

motolav, 2024-02-04:
@Ansuel The LEDs are broken on the Dynalink still, plugged in it's solid orange only and unplugged it's solid green and orange.

yyyzz123, 2024-02-04:
> Fixed by [7e0a7bf](https://github.com/openwrt/openwrt/commit/7e0a7bf21da68d4f8044c43012d82c1ea10a46b2)

YES, I test on my router, Thanks

robimarko, 2024-02-04:
> @Ansuel The LEDs are broken on the Dynalink still, plugged in it's solid orange only and unplugged it's solid green and orange.

Dynalink QCA8081 LED config needs to be tweaked as it was never correct in OpenWrt, vendor configured it after boot with:
```
## QCA8081 setting
			## Askey, Harry Chen, 2020/02/25, modify LED behavior for QCA-8081 (WAN)
			ssdk_sh debug phy set 0x1c 0x40078078 0x8600
```

Ansuel, 2024-02-04:
@motolav can you give me the content of your system file? Also can you confirm the existence of the leds and if they correctly turn off and on?

motolav, 2024-02-04:
There's no phy leds in `/sys/class/leds` just `blue:system  red:system`
There's no blinking but that was the same before the LED/upstream phy patches
And yea its a 1gb connection

Ansuel, 2024-02-04:
> There's no phy leds in `/sys/class/leds` just `blue:system red:system`

this is not good... and something is wrong here... can you post here a dmesg from a just booted system?

motolav, 2024-02-04:
```
root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.75 () (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 unknown) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Feb  3 22:12:50 2024
[    0.000000] Machine model: Dynalink DL-WRX36
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
[    0.000000]   node   0: [mem 0x0000000051100000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 17 pages/cpu s31080 r8192 d30360 u69632
[    0.000000] pcpu-alloc: s31080 r8192 d30360 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 887600K/1048576K available (7808K kernel code, 858K rwdata, 2288K rodata, 1280K init, 273K bss, 160976K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000106] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000120] pid_max: default: 32768 minimum: 301
[    0.000602] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000616] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002437] cblist_init_generic: Setting adjustable number of callback queues.
[    0.002445] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002640] rcu: Hierarchical SRCU implementation.
[    0.002644] rcu:     Max phase no-delay instances is 1000.
[    0.003427] smp: Bringing up secondary CPUs ...
[    0.004054] Detected VIPT I-cache on CPU1
[    0.004190] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.004839] Detected VIPT I-cache on CPU2
[    0.004941] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.005547] Detected VIPT I-cache on CPU3
[    0.005641] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.005708] smp: Brought up 1 node, 4 CPUs
[    0.005716] SMP: Total of 4 processors activated.
[    0.005721] CPU features: detected: 32-bit EL0 Support
[    0.005725] CPU features: detected: CRC32 instructions
[    0.005788] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.005794] CPU: All CPU(s) started at EL1
[    0.005797] alternatives: applying system-wide alternatives
[    0.016155] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016180] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.016394] pinctrl core: initialized pinctrl subsystem
[    0.017544] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.018090] DMA: preallocated 128 KiB GFP_KERNEL pool for atomic allocations
[... 174 lines cut ...]
[    8.139424] GACT probability on
[    8.139917] Mirror/redirect action on
[    8.144366] u32 classifier
[    8.145186]     input device check on
[    8.147790]     Actions configured
[    8.157877] Loading modules backported from Linux version v6.5-0-g2dde18cd1d8f
[    8.157914] Backport generated by backports.git v5.15.92-1-56-g5fb2ccb6b9e8
[    8.169306] NET: Registered PF_QIPCRTR protocol family
[    8.178465] xt_time: kernel timezone is -0000
[    8.210968] PPP generic driver version 2.4.2
[    8.211830] NET: Registered PF_PPPOX protocol family
[    8.215707] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    8.216134] urngd: v1.0.2 started.
[... cut ...]
[    9.253004] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[    9.257844] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.297019] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[    9.304525] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[    9.304572] ath11k c000000.wifi: fw_version 0x290104a5 fw_build_timestamp 2023-08-02 20:32 fw_build_id WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1
[    9.686122] ath11k c000000.wifi: htt event 48 not handled
```

Ansuel, 2024-02-04:
@robimarko mhh nothing in the log... the thing is not getting probed??? Do you see any problem in the dts?

motolav, 2024-02-05:
@Ansuel It's probably related to what robimarko said above with how Askey configured the board differently from what openwrt uses

Ansuel, 2024-02-05:
@motolav you don't even have the leds in sysfs and that doesn't make any sense...

robimarko, 2024-02-05:
PR https://github.com/openwrt/openwrt/pull/14556 should fix this

DenseSpace, 2024-10-21:
Can confirm as well that PR #14556  allows me to control the green LED on the Dynalink but it's mislabeled as the "yellow" one and I can't get control of the actual yellow one. Both Lights still solid when WAN is unplugged.

commit 2d077913 (2025-01-18): qualcommax: dl-wrx36: fix 2.5G port LED-s

Currently, 2.5G port LED-s on Dynalink are incorrectly configured and thus
they will light up all of the time.

So, lets fix this by:
1. Current green LED is actually yellow, change the color
2. Fix its polarity as its actually active-low
3. The yellow LED that was registered as being connected to LED_1 pin on
the PHY is not actually connected at all, so remove it.
4. The actual green LED is connected to LED_2 on the PHY so add it.

Fixes: 75ad5c24142a ("qualcommax: switch to qca8081 upstream PHY driver")
Fixes: #14502
Link: https://github.com/openwrt [... cut]
PR #17656 (2025-01-18): qualcommax: dl-wrx36: fix 2.5G port LED-s
Currently, 2.5G port LED-s on Dynalink are incorrectly configured and thus they will light up all of the time.

So, lets fix this by:
1. Current green LED is actually yellow, change the color
2. Fix its polarity as its actually active-low
3. The yellow LED that was registered as being connected to LED_1 pin on the PHY is not actually connected at all, so remove it.
4. The actual green LED is connected to LED_2 on the PHY so add it.

Fixes: 75ad5c24142a ("qualcommax: switch to qca8081 upstream PHY driver")
Fixes: #14502
commit a1c78167 (2025-01-18): qualcommax: dl-wrx36: fix 2.5G port LED-s

Currently, 2.5G port LED-s on Dynalink are incorrectly configured and thus
they will light up all of the time.

So, lets fix this by:
1. Current green LED is actually yellow, change the color
2. Fix its polarity as its actually active-low
3. The yellow LED that was registered as being connected to LED_1 pin on
the PHY is not actually connected at all, so remove it.
4. The actual green LED is connected to LED_2 on the PHY so add it.

Fixes: 75ad5c24142a ("qualcommax: switch to qca8081 upstream PHY driver")
Fixes: #14502
Link: https://github.com/openwrt [... cut]