openwrt/openwrt #14503: Sporadic overlayfs failure

yegorich, 2024-01-29:
### Describe the bug

From time to time, the overlay seems to stop functioning:

- `/overlay/upper`, `/etc` are empty. `/usr` is not empty compared to when one issues the `firstboot` command
- after reboot the overlay is workinig again, but the previuos data is lost

```
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/ubi0_1 on /overlay type ubifs (rw,noatime,assert=read-only,ubi=0,vol=1)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
/dev/mmcblk0p1 on /sdcard type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                21.3M     21.3M         0 100% /rom
tmpfs                   117.9M    152.0K    117.7M   0% /tmp
/dev/ubi0_1             167.2M    228.0K    162.3M   0% /overlay
overlayfs:/overlay      167.2M    228.0K    162.3M   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk0p1            3.7G      3.9M      3.7G   0% /sdcard
```

My bootparams: `ubi.mtd=5 ubi.block=0,0 root=/dev/ubiblock0_0 rootfstype=squashfs rootwait console=ttyO0,115200n8`

Kernel: Linux OpenWrt 6.1.62 #0 SMP Tue Nov 21 06:36:08 2023 armv7l GNU/Linux

I cannot see any kernel warnings/errors compared to the normal boot.

### OpenWrt version

r24436-0f283ab4c9

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

TI OMAP3/4/AM33xx

### Device

custom device

### Image kind

Self-built image

### Steps to reproduce

Boot the system and check `/overlay/upper`.

### Actual behaviour

`/overlay/upper` is empty.

### Expected behaviour

`/overlay/upper` is not empty.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-30:
It is you to debug your dts file obviously.

yegorich, 2024-01-30:
@brada4 do you mean checking [NAND timings](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm/boot/dts/ti/omap/am335x-baltos.dtsi?h=v6.7.2#n185)?

yegorich, 2024-03-26:
I found the root of the issue. It was my script that excidently deleted the `/overlay/upper` folder. Soory for the noise.

closed 2024-03-26 by yegorich