openwrt/openwrt #14504: Line / speed detection on NETGEAR WAX220 is faulty

dnnscoder, 2024-01-29:
### Describe the bug

When connecting the NETGEAR WAX220 to a 1Gbps port the duplex detection fails and ethtool states: `Speed: 1000Mb/s Duplex: Unknown! (255)` when connected to a 2.5Gbps port its works correctly and ethtool states Speed: 2500Mb/s Duplex: Full. This result in a-sync TX en RX when connected to a 1Gbps port `369 Mbits/sec receiver, 940 Mbits/sec sender` 

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Netgear WAX220

### Image kind

Official downloaded image

### Steps to reproduce

1. Connect WAP to a 1Gbps 
2. View the duplex setting with ethtool
3. Run a iperf3 test with the flag --bidir 
4. View results. 
5. Repeat step 1 till 4 when connected to a 2.5Gbps port.

### Actual behaviour

Speed seems to be async and TX and RX differ a lot see results below:
```
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Unknown! (255)
	Port: Twisted Pair
	PHYAD: 6
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Current message level: 0x000000ff (255)
			       drv probe link timer ifdown ifup rx_err tx_err
	Link detected: yes

Connecting to host 192.168.10.66, port 5201
[  5] local 192.168.10.9 port 36152 connected to 192.168.10.66 port 5201
[  7] local 192.168.10.9 port 36162 connected to 192.168.10.66 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec  45.4 MBytes   380 Mbits/sec  209   82.0 KBytes
[  7][RX-C]   0.00-1.00   sec   110 MBytes   926 Mbits/sec
[  5][TX-C]   1.00-2.00   sec  44.0 MBytes   369 Mbits/sec  267   70.7 KBytes
[  7][RX-C]   1.00-2.00   sec   112 MBytes   937 Mbits/sec
[  5][TX-C]   2.00-3.00   sec  43.8 MBytes   367 Mbits/sec  249   55.1 KBytes
[  7][RX-C]   2.00-3.00   sec   112 MBytes   937 Mbits/sec
[  5][TX-C]   3.00-4.00   sec  44.6 MBytes   374 Mbits/sec  170    109 KBytes
[  7][RX-C]   3.00-4.00   sec   112 MBytes   937 Mbits/sec
[  5][TX-C]   4.00-5.00   sec  44.2 MBytes   371 Mbits/sec  219   77.8 KBytes
[  7][RX-C]   4.00-5.00   sec   112 MBytes   938 Mbits/sec
[  5][TX-C]   5.00-6.00   sec  45.1 MBytes   379 Mbits/sec  137   66.5 KBytes
[  7][RX-C]   5.00-6.00   sec   112 MBytes   937 Mbits/sec
[  5][TX-C]   6.00-7.00   sec  42.9 MBytes   360 Mbits/sec  147    139 KBytes
[  7][RX-C]   6.00-7.00   sec   112 MBytes   937 Mbits/sec
[  5][TX-C]   7.00-8.00   sec  45.6 MBytes   383 Mbits/sec  230   65.0 KBytes
[  7][RX-C]   7.00-8.00   sec   112 MBytes   937 Mbits/sec
[  5][TX-C]   8.00-9.00   sec  43.4 MBytes   364 Mbits/sec  223   63.6 KBytes
[  7][RX-C]   8.00-9.00   sec   112 MBytes   938 Mbits/sec
[  5][TX-C]   9.00-10.00  sec  42.0 MBytes   352 Mbits/sec  214   62.2 KBytes
[  7][RX-C]   9.00-10.00  sec   112 MBytes   937 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec   441 MBytes   370 Mbits/sec  2065             sender
[  5][TX-C]   0.00-10.01  sec   440 MBytes   369 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  1.09 GBytes   940 Mbits/sec                  sender
[  7][RX-C]   0.00-10.01  sec  1.09 GBytes   935 Mbits/sec                  receiver 
```

### Expected behaviour

Correct detection of the duplex setting and full utilization of the line speed as with the use of a 2.5Gpbs port see below

```
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  100baseT/Half 100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 6
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Current message level: 0x000000ff (255)
			       drv probe link timer ifdown ifup rx_err tx_err
	Link detected: yes

Connecting to host 192.168.10.66, port 5201
[  5] local 192.168.10.9 port 36150 connected to 192.168.10.66 port 5201
[  7] local 192.168.10.9 port 36158 connected to 192.168.10.66 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec   114 MBytes   960 Mbits/sec    0    539 KBytes
[  7][RX-C]   0.00-1.00   sec   111 MBytes   933 Mbits/sec
[  5][TX-C]   1.00-2.00   sec   112 MBytes   936 Mbits/sec    0    539 KBytes
[  7][RX-C]   1.00-2.00   sec   112 MBytes   935 Mbits/sec
[  5][TX-C]   2.00-3.00   sec   112 MBytes   944 Mbits/sec    0    609 KBytes
[  7][RX-C]   2.00-3.00   sec   110 MBytes   926 Mbits/sec
[  5][TX-C]   3.00-4.00   sec   111 MBytes   934 Mbits/sec    0    609 KBytes
[  7][RX-C]   3.00-4.00   sec   112 MBytes   935 Mbits/sec
[  5][TX-C]   4.00-5.00   sec   112 MBytes   944 Mbits/sec    0    609 KBytes
[  7][RX-C]   4.00-5.00   sec   112 MBytes   936 Mbits/sec
[  5][TX-C]   5.00-6.00   sec   111 MBytes   934 Mbits/sec    0    609 KBytes
[  7][RX-C]   5.00-6.00   sec   112 MBytes   935 Mbits/sec
[  5][TX-C]   6.00-7.00   sec   112 MBytes   943 Mbits/sec    0    609 KBytes
[  7][RX-C]   6.00-7.00   sec   112 MBytes   935 Mbits/sec
[  5][TX-C]   7.00-8.00   sec   111 MBytes   933 Mbits/sec    0    609 KBytes
[  7][RX-C]   7.00-8.00   sec   112 MBytes   935 Mbits/sec
[  5][TX-C]   8.00-9.00   sec   112 MBytes   943 Mbits/sec    0    609 KBytes
[  7][RX-C]   8.00-9.00   sec   112 MBytes   935 Mbits/sec
[  5][TX-C]   9.00-10.00  sec   112 MBytes   944 Mbits/sec    0    609 KBytes
[  7][RX-C]   9.00-10.00  sec   112 MBytes   936 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  1.10 GBytes   941 Mbits/sec    0             sender
[  5][TX-C]   0.00-10.02  sec  1.09 GBytes   938 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  1.09 GBytes   936 Mbits/sec                  sender
[  7][RX-C]   0.00-10.02  sec  1.09 GBytes   933 Mbits/sec                  receiver
```

### Additional info

The Iperf test are done with the same hardware, cables and settings. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-30:
I think it must be `hsgmii` phy here instead
https://github.com/openwrt/openwrt/blob/0b498c3c3f1ebc6ffb5055ca07129eb21d8aef9c/target/linux/mediatek/dts/mt7986b-netgear-wax220.dts#L97
@dangowrt  should know better.

dnnscoder, 2024-01-30:
@brada4 Thanks lets wait and see if @dangowrt  has time to check this. Is there anything I can do to help? can i patch this easily or test it?

brada4, 2024-01-30:
If you have serial connected you can rebuild image changing only one line in dts file.  ethtool will show multiple speeds supported and you need to check if it connects.

dnnscoder, 2024-01-30:
Unfortunately not, and I'm scrolling through this ( https://openwrt.org/docs/techref/hardware/port.serial ) and it doesn't look like something I can easily do.

brada4, 2024-01-30:
That is the limitation with 1 port devices.  If you applied  broke wan config on others you could still connect via lan and flash original back.

dnnscoder, 2024-01-30:
Yeah I understand, but can't i just flash it back to a working version via tftp?

brada4, 2024-01-30:
First step in tftp procedure is to connect the serial console. No doubt after that you can.

dnnscoder, 2024-01-30:
mmm I'm having doubts now. I thought the WAX220 has a tftp server in the bootloader where i can just connect to and flash the device, but I'm not sure, but i can find out. However then the issue still remains on how do I get a patched firmware? or how do i build a new firmware, I do not have a dev environment setup for this. 

On a side note, thank you for your time i really appreciate it.

brada4, 2024-01-30:
It is described in wiki, long way at 1st attempt.

Djfe, 2024-02-06:
I also noticed that connecting to 100mbit/s ports isn't working for this device. (on release 23 there is no link up on the eth port)
I need to conduct further tests still and report actual output from master here (like ethtool)

dnnscoder, 2024-02-06:
I haven't tested it with a 100Mbit/s port, so i can not confirm this. But It would make sense and more or less confirms that is only works correctly if connected to a 2.5Gbits/s port.

dnnscoder, 2024-02-25:
@Ansuel I see you have also committed to this https://github.com/openwrt/openwrt/blob/0b498c3c3f1ebc6ffb5055ca07129eb21d8aef9c/target/linux/mediatek/dts/mt7986b-netgear-wax220.dts do you maybe have time to take a look at it?

dnnscoder, 2024-02-28:
@brada4 I've got my development env setup and able to build a new firmware. However it's not completely clear to me what i need to change? 

shoud it be:

`phy-mode = "2500base-x";  -> hsgmii-mode = "2500base-x"; ` or
`phy-mode = "2500base-x";  -> phy-mode = "2500base-x;"`

dnnscoder, 2024-02-28:
Tried both, did not work. No connection at all.

brada4, 2024-02-28:
The extra modes need to be added to phy driver, as for now driver just reports back what phy says possible.

brada4, 2024-02-28:
PHY (the controller around connector) does not report all supported speeds properly, the software side needs to add them.

dangowrt, 2024-02-28:
@dnnscoder Please share a complete bootlog so we can see which exact PHY is found there. I know it's RealTek 2.5G, but there have been a few variants of the PHY and this devices comes with a particularly early and buggy one.

Other devices (such as TL-XDR6068 I got here on my desk) work fine with those RTL8221B PHYs in all speeds, but those are also later revisions of the PHY chip.

> PHY (the controller around connector) does not report all supported speeds properly, the software side needs to add them.

Please explain what you mean by that. How would "properly" report it without software involvement?!

> phy-mode = "hsgmii"

There is no such a thing defined anywhere in the kernel sources, and there is no standard definition of HSGMII. In this case here, HSGMII is a RealTel-specific thing to connect 2.5G PHYs and allow them to operate at lower speeds by repeating the symbol -- which isn't as trivial as for actual SGMII supporting 10M/100M/1000M and hence having to repeat a symbol 10x or 100x. Now with 2500M rate this isn't so easy, because you can't repeat a symbol 2.5x, obviously. And, of course, in-band-status (sometimes also called "SGMII Auto-Negotiation") has to signal the new speed somehow. So sure, if you connect a RealTek PHY to a RealTek switch SoC (as done on many boards supported by the `realtek` target) we define HSGMII mode and use it, because it's actually compatible.

However, in heterogeneous systems we have to do what we do the same as also with MaxLinear GPY211: Use SGMII for 10M/100M/1000M and use 2500Base-X for 2500M, as those are all well-defined and well-supported interface modes. As this is kinda complex, some vendors choose always operate the link between MAC and PHY with 2500Base-X and use flow-control (as in: pause frames) to have the PHY adapt MAC speed to the speed of the external link partner. Obviously this is a waste of energy (operating SerDes clock much faster than needed) and also doesn't work as well as having the MAC follow the PHY speed.

In any case, the reporting of the speed for the user always happens using a driver and is always specific for the PHY, simply because most 2.5G PHYs don't go as far as behaving like 10G PHYs without support for 5G and 10G (which would make them Generic Clause-45 PHYs), but rather use vendor-defined methods to deal with 2.5G and usually all the rest is like 1G PHY.
(Aquantia being the exception here, their 2.5G PHYs *are* practically reduced 10G PHYs; but then there are other reasons to again have a chip-specific PHY driver even for those...)

dnnscoder, 2024-02-29:
Thanks for your reply, I'll try to get the bootlog. I don't have anything setup for serial access and i need to read up on how to do that. For my understanding there is no other way of getting the bootlog then via a serial connection which requires me to open the case and connect the pins? 

Is there anything else that you would need to research this?

dangowrt, 2024-02-29:
Grabing the output of `dmesg` right after boot will also do the trick.

dnnscoder, 2024-02-29:
Cool, I can do that. See below:

```
root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.137 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Nov 14 13:38:11 2023
[    0.000000] Machine model: Netgear WAX220
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 17 pages/cpu s30040 r8192 d31400 u69632
[    0.000000] pcpu-alloc: s30040 r8192 d31400 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line:
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1012400K/1048576K available (8384K kernel code, 912K rwdata, 1472K rodata, 448K init, 303K bss, 36176K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: 0xffffffc008422e34
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000108] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000116] pid_max: default: 32768 minimum: 301
[    0.000322] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000332] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001358] rcu: Hierarchical SRCU implementation.
[    0.001622] smp: Bringing up secondary CPUs ...
[    0.001882] Detected VIPT I-cache on CPU1
[    0.001905] GICv3: CPU1: found redistributor 1 region 0:0x000000000c0a0000
[    0.001932] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.002195] Detected VIPT I-cache on CPU2
[    0.002206] GICv3: CPU2: found redistributor 2 region 0:0x000000000c0c0000
[    0.002217] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.002455] Detected VIPT I-cache on CPU3
[    0.002465] GICv3: CPU3: found redistributor 3 region 0:0x000000000c0e0000
[    0.002474] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.002499] smp: Brought up 1 node, 4 CPUs
[    0.002513] SMP: Total of 4 processors activated.
[    0.002517] CPU features: detected: 32-bit EL0 Support
[    0.002520] CPU features: detected: CRC32 instructions
[... 133 lines cut ...]
[    8.598611] Backport generated by backports.git v5.15.92-1-44-gd6ea70fafd36
[    8.718411] urngd: v1.0.2 started.
[    9.010329] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20221012174648a
[    9.010329]
[    9.047022] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Unknown - flow control off
[    9.055565] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    9.353584] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20221012174725
[    9.503924] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20221012174937
[   12.532918] PPP generic driver version 2.4.2
[   12.537858] NET: Registered PF_PPPOX protocol family
[   12.545041] kmodloader: done loading kernel modules from /etc/modules.d/*
[   12.756367] mtdblock: MTD device 'u-boot-env' is NAND, please consider using UBI block devices instead.
[... cut ...]
[   16.188868] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/2500base-x link mode
[   16.237119] br-lan: port 1(eth0) entered blocking state
[   16.242348] br-lan: port 1(eth0) entered disabled state
[   16.247751] device eth0 entered promiscuous mode
[   20.407003] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Unknown - flow control off
[   20.415564] br-lan: port 1(eth0) entered blocking state
[   20.420785] br-lan: port 1(eth0) entered forwarding state
[   20.426454] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
root@OpenWrt:~#
```

dnnscoder, 2024-03-07:
@dangowrt Any progress / idea what is can be, is there anything i can help you with?

fakemanhk, 2024-07-11:
I observed the same problem on snapshot as well

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.35",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Netgear WAX220",
        "board_name": "netgear,wax220",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r26843-4cc69a6542",
                "target": "mediatek/filogic",
                "description": "OpenWrt SNAPSHOT r26843-4cc69a6542"
        }
}
```

Below is the **_dmesg_** output:

```
root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.35 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26843-4cc69a6542) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Wed Jul  3 11:11:02 2024
[    0.000000] Machine model: Netgear WAX220
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
[    0.000000] OF: reserved mem: 0x000000004fc00000..0x000000004fcfffff (1024 KiB) nomap non-reusable wmcpu-reserved@4fc00000
[    0.000000] OF: reserved mem: 0x000000004fd00000..0x000000004fd3ffff (256 KiB) nomap non-reusable wo-emi@4fd00000
[    0.000000] OF: reserved mem: 0x000000004fd40000..0x000000004fd7ffff (256 KiB) nomap non-reusable wo-emi@4fd40000
[    0.000000] OF: reserved mem: 0x000000004fd80000..0x000000004ffbffff (2304 KiB) nomap non-reusable wo-data@4fd80000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s35048 r8192 d30488 u73728
[    0.000000] pcpu-alloc: s35048 r8192 d30488 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007fb06000-0x000000007fc06000] (1MB)
[    0.000000] Memory: 1009988K/1048576K available (8768K kernel code, 900K rwdata, 2564K rodata, 384K init, 304K bss, 38588K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[... 187 lines cut ...]
[   10.324509] br-lan: port 1(eth0) entered disabled state
[   10.329813] mtk_soc_eth 15100000.ethernet eth0: entered allmulticast mode
[   10.336738] mtk_soc_eth 15100000.ethernet eth0: entered promiscuous mode
[   13.448996] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Unknown - flow control off
[   13.457549] br-lan: port 1(eth0) entered blocking state
[   13.462770] br-lan: port 1(eth0) entered forwarding state
[   69.620590] br-lan: port 1(eth0) entered disabled state
[   69.626706] mtk_soc_eth 15100000.ethernet eth0: left allmulticast mode
[   69.633302] mtk_soc_eth 15100000.ethernet eth0: left promiscuous mode
[   69.639794] br-lan: port 1(eth0) entered disabled state
[   69.688816] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[   70.227379] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:06] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[   70.241207] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/2500base-x link mode
[   70.288014] br-lan: port 1(eth0) entered blocking state
[   70.293231] br-lan: port 1(eth0) entered disabled state
[   70.298478] mtk_soc_eth 15100000.ethernet eth0: entered allmulticast mode
[   70.305366] mtk_soc_eth 15100000.ethernet eth0: entered promiscuous mode
[   70.312907] br-lan: port 1(eth0) entered blocking state
[   70.318139] br-lan: port 1(eth0) entered forwarding state
[   70.638237] br-lan: port 1(eth0) entered disabled state
[   73.436511] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Unknown - flow control off
[   73.445062] br-lan: port 1(eth0) entered blocking state
[   73.450283] br-lan: port 1(eth0) entered forwarding state
root@OpenWrt:~# 
```

fakemanhk, 2024-07-11:
I don't have the UART to connect my WAX220, however I think the WAX220 is unable to talk to 100Mbps peer because when I tried to connect it to a travel router with only 100Mbps LAN port (Buffalo WCR-1166DS w/OpenWrt 23.05.2) the LAN port keeps flapping and the WAX220 is unable to get any IP from it at all (observed in syslog in WCR-1166DS).

mvhuizen, 2024-07-20:
Another bootlog connected to an GS108PP.

> [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.150 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23809-234f1a2efa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Mar 22 22:09:42 2024
[    0.000000] Machine model: Netgear WAX220
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33112 r8192 d32424 u73728
[    0.000000] pcpu-alloc: s33112 r8192 d32424 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1012448K/1048576K available (8384K kernel code, 906K rwdata, 1472K rodata, 448K init, 300K bss, 36128K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: 0xffffffc008010090
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000001] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000106] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000115] pid_max: default: 32768 minimum: 301
[    0.000318] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000327] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001350] rcu: Hierarchical SRCU implementation.
[    0.001617] smp: Bringing up secondary CPUs ...
[    0.001877] Detected VIPT I-cache on CPU1
[    0.001899] GICv3: CPU1: found redistributor 1 region 0:0x000000000c0a0000
[    0.001926] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.002190] Detected VIPT I-cache on CPU2
[    0.002203] GICv3: CPU2: found redistributor 2 region 0:0x000000000c0c0000
[    0.002215] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.002450] Detected VIPT I-cache on CPU3
[    0.002459] GICv3: CPU3: found redistributor 3 region 0:0x000000000c0e0000
[    0.002468] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.002494] smp: Brought up 1 node, 4 CPUs
[    0.002507] SMP: Total of 4 processors activated.
[    0.002511] CPU features: detected: 32-bit EL0 Support
[    0.002514] CPU features: detected: CRC32 instructions
[    0.002540] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.002593] CPU: All CPU(s) started at EL2
[... 150 lines cut ...]
[   15.282489] br-lan: port 1(eth0) entered disabled state
[   15.287877] device eth0 entered promiscuous mode
[   15.457108] mtdblock: MTD device 'u-boot-env' is NAND, please consider using UBI block devices instead.
[   16.966243] br-lan: port 2(phy0-ap0) entered blocking state
[   16.971868] br-lan: port 2(phy0-ap0) entered disabled state
[   16.977631] device phy0-ap0 entered promiscuous mode
[   16.982644] br-lan: port 2(phy0-ap0) entered blocking state
[   16.988208] br-lan: port 2(phy0-ap0) entered forwarding state
[   16.994000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   17.131134] br-lan: port 2(phy0-ap0) entered disabled state
[   17.136867] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   17.143570] br-lan: port 2(phy0-ap0) entered blocking state
[... cut ...]
[   17.669179] br-lan: port 3(phy1-ap0) entered forwarding state
[   18.166869] br-lan: port 3(phy1-ap0) entered disabled state
[   18.390036] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[   18.396748] br-lan: port 3(phy1-ap0) entered blocking state
[   18.402304] br-lan: port 3(phy1-ap0) entered forwarding state
[   18.417160] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Unknown - flow control off
[   18.425708] br-lan: port 1(eth0) entered blocking state
[   18.430930] br-lan: port 1(eth0) entered forwarding state

andreagdipaolo, 2024-08-15:
Could anyone try to fix this in the snapshot build?

Djfe, 2024-08-16:
I'm lending the wax220 of a friend to a dev soonish. That might help get this fixed eventually. I'll update here if I know more.

closed 2024-10-09 by openwrt-bot

Djfe, 2024-10-19:
I just verified on the Netgear WAX220, that
10mbit/s half and full duplex
100mbit/s half and full duplex
and 1000mbit/s full duplex
are working correctly now (on snapshots).
I will verify 2.5gbit/s eventually once I get access to such a device.

@dangowrt do you think it's possible to backport the pending patches to 5.15 (openwrt-23.05)?

Djfe, 2024-10-19:
I just verified 2.5gbit/s as well.
I connected a Netgear WAX220 to a Zyxel NWA50AX Pro (both running a fairly recent snapshot version).
It works and shows the correct link speed in ethtool.

commit 69906789 (2024-10-09): generic: net: phy: realtek: various improvements

Follow the advise of Russell King allows to greatly improve the driver
for RealTek's 1G and 2.5G Ethernet PHYs. The results are full/half
duplex as well as Gbit master/slave property being read from PHY
Specific Status Register (PHYSR), and fixes regarding link-partner
advertisement.

Fixes: #14504