openwrt/openwrt #14505: dnsmasq is crashing after the last snapshots

VictorxD70, 2024-01-29:
### Describe the bug

dnsmasq is not working after the latest snapshots on my device.

This appears in the log:

`Sun Jan 28 23:16:03 2024 user.err : jail: failed to clone/fork: Invalid argument
Sun Jan 28 23:16:08 2024 user.err : jail: failed to clone/fork: Invalid argument
Sun Jan 28 23:16:08 2024 daemon.info procd: Instance sysntpd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Sun Jan 28 23:16:58 2024 user.err : jail: failed to clone/fork: Invalid argument
Sun Jan 28 23:16:58 2024 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 9 crashes, 0 seconds since last crash`

### OpenWrt version

r24936-60ffcfdabc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link Archer C20 v4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

VictorxD70, 2024-01-31:
I managed to solve it by removing the `procd-ujail` package that was causing the crashes.

closed 2024-01-31 by VictorxD70

kizill, 2024-04-07:
I've got the same problem with `user.err : jail: failed to clone/fork: Invalid argument` on 
```
OpenWrt SNAPSHOT r25845-7f85104ebb / LuCI Master git-24.089.54606~0ecb5ed
```
Solved by removing `procd-ujail`, @VictorxD70 thank you :)