openwrt/openwrt #14513: qualcommax/ipq807x/ax9000: No Upstream network

Soromeister, 2024-01-30:
### Describe the bug

There is no Upstream link for the network, basically, both `wan` and `wan6` interfaces are not getting an IP.

This was working for build `r24909-65f599223d`, but somehow broke after this build.

### OpenWrt version

r24939-0b498c3c3f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Official downloaded image

### Steps to reproduce

Simply upgrading to a version higher than `r24909-65f599223d` will have no upstream IP.

### Actual behaviour

No upstream IP (neither `wan` nor `wan6` interfaces get IP).

### Expected behaviour

Upstream IP should be assigned (Both `wan` and `wan6` interfaces should get IP address).

### Additional info

[config.txt](https://github.com/openwrt/openwrt/files/14097532/config.txt)
[kernel.log](https://github.com/openwrt/openwrt/files/14097536/kernel.log)
[system.log](https://github.com/openwrt/openwrt/files/14097538/system.log)

### Diffconfig

```text
-ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yyyzz123, 2024-01-30:
There has a fix about it.
https://github.com/openwrt/openwrt/pull/14500

Ansuel, 2024-01-30:
Fix merged.

closed 2024-01-30 by Ansuel

Soromeister, 2024-02-13:
Confirmed it's working. Thanks for fixing.