openwrt/openwrt #14523: MX4200v1"  (EU) |  5Ghz unusable due to wrong BDF in OpenWRT

Razerwire, 2024-01-31:
### Describe the bug

**Describe the bug**

Radios do come  up after installing OpenWRT on the MX4200v1   and configuring the necessary. 

However the 5Ghz (phy0-ap0)  will not send any data as in "_RX: 0B_" and it cannot be connected to via "phy0-ap0", the 5Ghz WiFi. 

**Cause of the issue**

The issue at play here, is the same as is mentioned in the [OpenWRT-Wiki page](url) for the "Buffalo WXR-5950AX12 / WXR-6000AX12x".  The page also mentions [workaround](https://openwrt.org/toh/buffalo/wxr-5950ax12#potential_issueslimitations)  than can be adapted to fix the issues with the MX4200v1. 

(there is a lot of conversation surrounding this issue on the [OpenWRT Forum) 
](https://forum.openwrt.org/t/openwrt-support-for-linksys-mx4200/86477)

**Workaround / temporary-solution:** 

For the  MX4200v1, one can use the following  workaround that overrides the package and use the router's
factory calibration data stored in MTD as a hack BDF:

```
opkg remove ipq-wifi-linksys_mx4200

mv /lib/firmware/ath11k/IPQ8074/hw2.0/board-2.bin /lib/firmware/ath11k/IPQ8074/hw2.0/_IGNORE_board-2.bin

. /lib/functions/caldata.sh

FIRMWARE=ath11k/IPQ8074/hw2.0/board.bin caldata_extract "0:art" 0x1000 0x20000

reboot

```

### OpenWrt version

r24939-0b498c3c3f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX4200v1

### Image kind

Official downloaded image

### Additional info

The official OpenWRT images for MX4200v1 are currently being built with a hard-coded WiFi
Board Data File (BDF) firmware package preinstalled, that does not work well with MX4200v1
It is known to break 5GHz radio RX signal on MX4200v1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Razerwire, 2024-01-31:
> I keep seeing this constantly but no explanation why. You are extracting the caldata and claiming that its the BDF instead, can you provide any evidence this is the right thing to do?

The former commands:

-   Removes the package that supplies the wrong Board-Data-File.
-   Renames the faulty Board Data File
-   Creates a working (accurate) Board Data File from MTD (containing accurate calibration data) 
-   Commands OpenWrt to use the newly created Board-Data-File
-   Reboots the device.

It solves the issue. 

What is it that you need? a forensic report?

robimarko, 2024-01-31:
I just want an explanation of where this idea originates from:
Creates a working (accurate) Board Data File from MTD (containing accurate calibration data)

Because this is not correct, caldata is not equal to BDF.
You can probably "fix" the issue you are seeing by using the QCA generic BDF as well, but this should be rather investigated whether the BDF pulled from the stock FW is the correct one.

Razerwire, 2024-01-31:
> I just want an explanation of where this idea originates from: Creates a working (accurate) Board Data File from MTD (containing accurate calibration data)
>
> Because this is not correct, caldata is [...]

The cause for this issue is know and I  have reported the issue here,  and I have  provided a interim solution (workaround) for the issue above.  

Should you wish to investigate anything please do, i look forward to your findings. 
Perhaps we can all  benefit from your investigation and (once you have completed your investigation)  the subsequent report.

OpenWRT is a community effort, after all.

robimarko, 2024-01-31:
You are not getting the point, this is not a valid workaround as you are feeding caldata as the BDF and those are not the same.

Razerwire, 2024-01-31:
> You are not getting the point, this is not a valid workaround as you are feeding caldata as the BDF and those are not the same.

The issue is clear and concise as is the workaround (which once-more) fixes the problem. 
Should you still not understand the issue described here and/or fail to grasp the workaround offered, please do re-read the former. 

I'm going to leave it at that, thanks.

testuser7, 2024-01-31:
Issue should be fixed with this two PR:
- https://github.com/openwrt/firmware_qca-wireless/pull/27
- https://github.com/openwrt/openwrt/pull/14512

Razerwire, 2024-01-31:
> Issue should be fixed with this two PR:
> 
>     * [ipq8074: remove regdomain from MX4200 BDF firmware_qca-wireless#27](https://github.com/openwrt/firmware_qca-wireless/pull/27)
> 
>     * [ipq807x: use ath11k_patch_mac and ath11k_remove_regdomain for MX4200 #14512](https://github.com/openwrt/openwrt/pull/14512)
> 
> 
> @Razerwire You can help if you share your `pre-caldata` file with @Ansuel.

Thanks will do!

Razerwire, 2024-02-02:
> Issue should be fixed with this two PR:
>
>     * [ipq8074: remove regdomain from MX4200 BDF firmware_qca-wireless#27](https://github.com/openwrt/firmware_qca-wireless/pull/27)
>
>     * [ipq807x: use [...]

For administrative purposes:  caldata from the MX4200v1-EU was shared here:
 [https://github.com/openwrt/openwrt/pull/14512#issuecomment-1924421757](url)

testuser7, 2024-05-28:
This issue should be closed now.