openwrt/openwrt #14538: TL-WR841N v13 (SoC MT7628AN) Internet access lost every several days

wenqingj, 2024-02-02:
### Describe the bug

The router hardware is TL-WR841N v13 (SoC MT7628AN), running OpenWRT 23.05.02.
The internet connection of the router's clients is lost randomly in several days. Wifi links (and ethernet links) seem not lost, but the router just won't forward the requests out.

A simple "solution" (workaround) I found is to restart the WAN interface (LuCI menu Network -> Interfaces, find the row for wan, and press the restart button). This resolves the issue every time.

The logs (LuCI Status -> System Log) shows nothing interesting (same for Kernel log tab). I understand that this makes it hard to investigate. Please advise how I can get you more information (like via certain command line operations). Thank you.

### OpenWrt version

23.05.02

### OpenWrt release

23.05.02

### OpenWrt target/subtarget

ramips/mt76x8/tplink_tl-wr841n-v13-squashfs-sysupgrade.bin

### Device

TPLink TL-WR841N v13

### Image kind

Official downloaded image

### Actual behaviour

Internet access is lost after random number of days (mostly 2 ~ 5 days), while there is no apparent reason (like power outage).  

### Expected behaviour

Internet access should not be lost.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-02:
First slowly fix release tags in your post.
First effort would be to run network diagnostics in luci or plainly ping and arping wan n€twork gateway{s}.
Also try to recall at whivh release problems started.

wenqingj, 2024-02-02:
Thanks, brada4. I've updated the tags/release etc. and hope they're correct now.
The network diagnostics in LuCI returns normal result at the moment. I'll run it again when the reported issue occurs again and report back.
The issue appears in release 23.05.2, but I'm not certain if it started in 23.05.0 - because very soon the new release (23.05.2) showed up and I jumped to the new one. Before 23.05.0, my router was not usable because of bug #10074.
Several days ago I tried the snapshot (OpenWrt SNAPSHOT r24939-0b498c3c3f / LuCI Master git-24.025.10336-0617d30), manually installed LuCI, and the issue still exists.

brada4, 2024-02-02:
Once device gets stuck also check 
`logread | tee -a /tmp/copylog.txt`

Download the file and try to look through it on your computer. Your router has rather small flash, lets try to get around without installing extra.

wenqingj, 2024-02-02:
@brada4 Got it and will do. Thanks for the advice!

wenqingj, 2024-02-07:
FYI the issue has not occurred yet. I'm still watching...

brada4, 2024-02-07:
Very good, we scared it together ;-) If it repeats and you dont manage to capture logs you can try forwarding syslog to some 24x7 fileserver if you have one at home.

wenqingj, 2024-02-08:
It came last night (always like that - found it in the first use of a morning). 
3 tests in the LuCI diagnostics:
  ping: bad address 'openwrt.org'
  traceroute: bad address 'openwrt.org'
  ;; connection timed out; no servers could be reached
[KernelLog.txt](https://github.com/openwrt/openwrt/files/14210482/KernelLog.txt)
[SystemLog.txt](https://github.com/openwrt/openwrt/files/14210484/SystemLog.txt)
Captured system log and kernel log attached.

wenqingj, 2024-02-08:
Note that it's a OpenWRT snapshot, not a formal release (the issue is observed on both).

brada4, 2024-02-08:
Try to set fixed link speed in "port 2":
swconfig dev switch0 port 2 get link
swconfig dev switch0 port 2 set link "speed 100 duplex full"
Or add "autoneg off" even to link specification.
Is it "wan" port? What is connected there, or more accurately can you change device to another port.

> snapshot
Your effort to investigate and bisect issue is appreciated.

wenqingj, 2024-02-08:
Yes it's a "wan" port. There's a "wan6" port but I've disabled it (I don't need IPv6). The wan port is connected to a cable modem via an CAT6 Ethernet cable. Physically there is only one port on the router to connect to the modem so I cannot switch to another port (the other 4 ports are for clients of the internal network).

root@TL-WR841N:~# swconfig dev switch0 port 0 get link
port:0 link:up speed:100baseT full-duplex 
root@TL-WR841N:~# swconfig dev switch0 port 1 get link
port:1 link:up speed:100baseT full-duplex 
root@TL-WR841N:~# swconfig dev switch0 port 2 get link
port:2 link:down
root@TL-WR841N:~# swconfig dev switch0 port 3 get link
port:3 link:down
root@TL-WR841N:~# swconfig dev switch0 port 4 get link
port:4 link:down
root@TL-WR841N:~# swconfig dev switch0 port 5 get link
port:5 link:down
root@TL-WR841N:~# swconfig dev switch0 port 6 get link
port:6 link:up speed:1000baseT full-duplex 
root@TL-WR841N:~# 

The output above was done while the router is working well (after I've restart wan interface the router is working well). I'll check the ports again next time when the issue occurs.

I guess ports #0 to #2 include wan and wan6 (not sure what's the 3rd one), ports #3 to #6 are the 4 wired ports for clients (and I'm using one of them). Hence the output looks normal. Please correct me if you disagree...

brada4, 2024-02-08:
Use triple backticks around code blocks.

wenqingj, 2024-02-08:
Sorry it should be like the following, after I check the switch config in LuCI.
port 6 is CPU (1000 mbps).
port 1 to 4 for clients (and 1 is being used, the other 3 are not used at the moment), each 100 mbps.
not sure of port 0 and port 5 - I guess port 0 is wan as it shows a 100 mbps connection. No idea what's port 5.

brada4, 2024-02-08:
Port2 is glitching even no device is connected to it as observable in dmesg. If there is physical damage like shorted pins it may help to plainly take it out of VLANs and leave unconnected hoping that it gets no power and does not glitch. If it was a logical programming problem it would be various ports entering up down loop randomly  killing switch.

Port 5 is either not connected anywhere or it is disabled second ser-des + gmii of SoC, which could make a direct gigabit for WAN, but does not matter in your device.

wenqingj, 2024-02-08:
Sorry my bad - port 2 is actually connected to a device that's turned on only occasionally. I checked the pins at port 2 (router side) and they look pretty clean and normal (not touching each other). The pins at the device side also look normal but I have just disconnected the device to the port so as to kick this factor out. Will keep observing.

brada4, 2024-02-08:
Plug cable  in different port and see how it goes. If it crashes on new port try to set fixed speed, if old port crashes remove it from switch (off in all column)

wenqingj, 2024-02-08:
@brada4 Thanks. I've just changed port 2 from untagged to off (hope I read you right). Will wait and see what happens...

brada4, 2024-02-08:
Yes, you got it correctly, and plug suspect device in other switch port next time, it could be some wirdness with autonegotiation too.

wenqingj, 2024-02-11:
The issue occurred again last night. The following is captured:

[KernelLog.txt](https://github.com/openwrt/openwrt/files/14232743/KernelLog.txt)
[SystemLog.txt](https://github.com/openwrt/openwrt/files/14232744/SystemLog.txt)
```
BusyBox v1.36.1 (2024-01-29 21:23:01 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r24939-0b498c3c3f
 -----------------------------------------------------
root@TL-WR841N:~# swconfig dev switch0 port 0 get link
port:0 link:up speed:100baseT full-duplex 
root@TL-WR841N:~# swconfig dev switch0 port 1 get link
port:1 link:up speed:100baseT full-duplex 
root@TL-WR841N:~# swconfig dev switch0 port 2 get link
port:2 link:down
root@TL-WR841N:~# swconfig dev switch0 port 3 get link
port:3 link:down
root@TL-WR841N:~# swconfig dev switch0 port 4 get link
port:4 link:down
root@TL-WR841N:~# swconfig dev switch0 port 5 get link
port:5 link:down
root@TL-WR841N:~# swconfig dev switch0 port 6 get link
port:6 link:up speed:1000baseT full-duplex 
root@TL-WR841N:~# 
```
After that I restarted the wan interface to resolve the issue.

brada4, 2024-02-11:
```
Sun Feb 11 06:41:45 2024 daemon.notice netifd: wan (2880): udhcpc: sending renew to server 10.72.2.3
```
It never got a response.

```
# udhcpc -h
...
        -x OPT:VAL      Include option OPT in sent packets (cumulative)
                        Examples of string, numeric, and hex byte opts:
                        -x hostname:bbox - option 12
                        -x lease:3600 - option 51 (lease time)
                        -x 0x3d:0100BEEFC0FFEE - option 61 (client id)
                        -x 14:'"dumpfile"' - option 14 (shell-quoted)
...
```
Your provider forgets your lease over time, based on at least 12h lease time from your log quite likely if local DNS device has limited memory etc.
Ass something like shorter lease time in /etc/config/network next to proto 'dhcp' in wan section:
```
         option sendopts "lease:3600"
```
Among other options - disable renew (not obvious from parameters how to), send (more) unique string as hostname etc.
`lease:3600 hostname:hal9000` and so on....
You can restart interface from luci or learn ubus to restart one interface to apply changes.

wenqingj, 2024-02-13:
Sorry replying late, as I am still waiting for a time (that has no connected clients) to apply the suggested change and reboot.
In the mean time, I observed the following:
```
Tue Feb 13 09:13:51 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:25:06 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:30:43 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:33:32 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:34:56 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:35:38 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:35:59 2024 daemon.notice netifd: wan (6173): udhcpc: sending renew to server 10.72.2.3
Tue Feb 13 09:36:02 2024 daemon.notice netifd: wan (6173): udhcpc: broadcasting renew
Tue Feb 13 09:36:02 2024 daemon.notice netifd: wan (6173): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 90015
```
Wondering why sending to server won't get a renew but broadcasting achieved it (although the renew is from the same sever). The new least time (90015) is also kind of odd.
By the way, why DNS is a factor? The renew request is directly on a specific IP, and my browser requests (at client side) use browser-side DNS settings (i.e. 8.8.8.8, 1.1.1.1) which are also direct IPs. Not sure if I missed something.

brada4, 2024-02-13:
Add broadcast option, since WAN flops all the time I see no reason of NOT restarting interface alone in 5 minutes.
https://openwrt.org/docs/guide-user/network/ipv4/configuration#protocol_dhcp

brada4, 2024-02-13:
Explanations:
Lease time is ~25h, likely original was 48h and the reminder is not extended server side unless it falls under half.
DHCP server is not directly reachable , good as per DOCSIS, contrary to RFC
DoH canary configuration is not involved in DHCP client. Probably ask in forums if you want to change something about it.

wenqingj, 2024-02-13:
@brada4 Thanks for the tech details! I'm not sure I fully understand but I just checked (and saved) the following checkbox in LuCI:
network -> interfaces -> wan edit -> "advanced settings" tab
a checkbox called "Use broadcast flag" (required for certain ISPs, e.g. Charter with DOCSIS 3)

I guess this does not demand a reboot. Will monitor and see if it resolves the IP renew issue.

wenqingj, 2024-02-14:
Seems that the checkbox does not have an effect. This is what I see after that:
```
Wed Feb 14 10:25:33 2024 daemon.notice netifd: wan (8962): udhcpc: sending renew to server 10.72.2.3
Wed Feb 14 10:30:57 2024 daemon.notice netifd: wan (8962): udhcpc: sending renew to server 10.72.2.3
Wed Feb 14 10:33:39 2024 daemon.notice netifd: wan (8962): udhcpc: sending renew to server 10.72.2.3
Wed Feb 14 10:35:00 2024 daemon.notice netifd: wan (8962): udhcpc: sending renew to server 10.72.2.3
Wed Feb 14 10:35:40 2024 daemon.notice netifd: wan (8962): udhcpc: sending renew to server 10.72.2.3
Wed Feb 14 10:36:00 2024 daemon.notice netifd: wan (8962): udhcpc: sending renew to server 10.72.2.3
Wed Feb 14 10:36:03 2024 daemon.notice netifd: wan (8962): udhcpc: broadcasting renew
Wed Feb 14 10:36:05 2024 daemon.notice netifd: wan (8962): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 172800
Wed Feb 14 10:36:06 2024 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
```
I will reboot the router when a convenient time comes and see if that will make a difference...

wenqingj, 2024-02-15:
@brada4 Thanks for the tip! I've just restarted the wan interface. Will keep monitoring.
```
Wed Feb 14 20:28:51 2024 daemon.notice netifd: Interface 'wan' is now down
Wed Feb 14 20:28:51 2024 daemon.warn dnsmasq[2262]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Wed Feb 14 20:28:51 2024 daemon.notice netifd: Interface 'wan' is setting up now
Wed Feb 14 20:28:51 2024 daemon.notice netifd: wan (11108): udhcpc: started, v1.36.1
Wed Feb 14 20:28:51 2024 daemon.notice netifd: wan (11108): udhcpc: broadcasting discover
Wed Feb 14 20:28:51 2024 daemon.notice netifd: wan (11108): udhcpc: broadcasting select for 108.162.136.122, server 10.72.2.3
Wed Feb 14 20:28:51 2024 daemon.notice netifd: wan (11108): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 137234
Wed Feb 14 20:28:51 2024 daemon.notice netifd: Interface 'wan' is now up
```

wenqingj, 2024-02-15:
The issue occurred again. Attached is the system log (seems no error inside)...
[SystemLog.txt](https://github.com/openwrt/openwrt/files/14296845/SystemLog.txt)

brada4, 2024-02-15:
No sign whatsoever of disconnect.
Try short lease time. Maybe it keeps you connected. https://github.com/openwrt/openwrt/issues/14538#issuecomment-1937787767
as side effect you will trigger renew more often.

wenqingj, 2024-02-15:
@brada4 Sure. The file is updated with the last line below added:
```
config interface 'wan'                    
        option device 'eth0.2'          
        option proto 'dhcp'   
        option sendopts 'lease:12000'
```
After that the wan interface is restarted. Back to monitoring...

wenqingj, 2024-02-15:
I saw the least time was still more than a day after that wan interface restart mentioned in the previous comment. Hence I restart it again, and captured the following from system log:
```
Thu Feb 15 09:24:46 2024 daemon.notice netifd: wan (12048): udhcpc: received SIGTERM
Thu Feb 15 09:24:46 2024 daemon.notice netifd: wan (12048): udhcpc: unicasting a release of 108.162.136.122 to 10.72.2.3
Thu Feb 15 09:24:46 2024 daemon.notice netifd: wan (12048): udhcpc: sending release
Thu Feb 15 09:24:46 2024 daemon.notice netifd: wan (12048): udhcpc: entering released state
Thu Feb 15 09:24:46 2024 daemon.notice netifd: wan (12048): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Thu Feb 15 09:24:46 2024 daemon.notice netifd: Interface 'wan' is now down
Thu Feb 15 09:24:46 2024 daemon.notice netifd: Interface 'wan' is setting up now
Thu Feb 15 09:24:46 2024 daemon.warn dnsmasq[2262]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Thu Feb 15 09:24:47 2024 daemon.notice netifd: wan (12164): udhcpc: started, v1.36.1
Thu Feb 15 09:24:47 2024 daemon.notice netifd: wan (12164): udhcpc: broadcasting discover
Thu Feb 15 09:24:47 2024 daemon.notice netifd: wan (12164): udhcpc: broadcasting select for 108.162.136.122, server 10.72.2.3
Thu Feb 15 09:24:47 2024 daemon.notice netifd: wan (12164): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 90678
Thu Feb 15 09:24:47 2024 daemon.notice netifd: Interface 'wan' is now up
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: reading /tmp/resolv.conf.d/resolv.conf.auto
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using nameserver 206.248.154.22#53
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using nameserver 206.248.154.170#53
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for test
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for onion
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for localhost
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for local
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for invalid
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for bind
Thu Feb 15 09:24:47 2024 daemon.info dnsmasq[2262]: using only locally-known addresses for lan
Thu Feb 15 09:24:48 2024 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
```
Not sure why the actual lease time (90648) is not the custom value of 12000. And not sure what the "command failed" message means?

wenqingj, 2024-02-20:
Again the checkbox does not have any effect:
```
Tue Feb 20 08:49:24 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Tue Feb 20 08:54:45 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Tue Feb 20 08:57:27 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Tue Feb 20 08:58:47 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Tue Feb 20 08:59:27 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Tue Feb 20 08:59:47 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Tue Feb 20 08:59:50 2024 daemon.notice netifd: wan (2867): udhcpc: broadcasting renew
Tue Feb 20 08:59:50 2024 daemon.notice netifd: wan (2867): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 98421
Tue Feb 20 08:59:51 2024 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
```
The lease time above (98421 seconds) is about 27 hours 20 minutes. However the LuCI status page (checked within a couple of minutes after getting the lease above) shows expiration in about 9 hours. That's quite puzzling.
Will continue to monitor.

brada4, 2024-02-20:
Broadcast flag is in packet content asking response via broadcast. Probably this is request for enhancement to allow configuring renew via broadcast.

wenqingj, 2024-02-20:
(to continue with the story from my previous comment) Now 9 hours was near. I was curious and watched what would happen. I bet you can never guess out what I observed -

The system log did not have anything that relates to sending out a new lease renew request (either before or after the 9 hour threshold), and the LuCI status page, after the 9 hour time ran out (I watched it running from 1 minute remaining down to naught), it became 1 day 3 hours 20 minutes-ish remaining - exactly the time that got leased 9 hours ago! And (as I have just said) in the system log there is no any renew request.

Kind of spooky. Perhaps I missed something?

wenqingj, 2024-02-21:
When the remaining time from LuCI dropped to around 9 hours (please read my previous comment to help figure out why this time is important), I observed that the router renewed the lease successfully:
```
Wed Feb 21 12:07:26 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Wed Feb 21 12:13:50 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Wed Feb 21 12:17:02 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Wed Feb 21 12:18:38 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Wed Feb 21 12:19:26 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Wed Feb 21 12:19:50 2024 daemon.notice netifd: wan (2867): udhcpc: sending renew to server 10.72.2.3
Wed Feb 21 12:19:53 2024 daemon.notice netifd: wan (2867): udhcpc: broadcasting renew
Wed Feb 21 12:19:53 2024 daemon.notice netifd: wan (2867): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 164397
Wed Feb 21 12:19:54 2024 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
'''
and the LuCI status page again shows a time (about 18 hours 20 minutes) that is not compatible with that of the log (164397 seconds, which is more than 45 hours).
If the renewals from the log keeps working like that for several more iterations, I'll consider the issue resolved (resolved by checking the checkbox "use broadcast flag" for the wan interface), although there is this lingering issue of mismatching remaining time between LuCI status page and system log...

brada4, 2024-02-21:
You can try projecting next check to after advertised server lease time expiration. There is nothing that fails in a stable way, thus blaming maintenance at your provider.
`logread -e udhcpc`
or with pprocps-pkill trigger renew:
`pkill -usr1 udhcpc`
Id speculate requests get denser towards lease expiry until server extends it?

brada4, 2024-02-29:
Set lease to 7200..3600..1800, it may not work, anything minute to week should be fine. Maybe it serves as keeo-alive.

I am pretty sure dhcp was not responded in the log when it went down. No explanation in logs if it was chicken or egg first - internet line or dhcp.

Another approach would be ifup wan every night via (luci cron) schedule to force dhcp refresh.

googele sent me here, you can get dome indpiration:
https://github.com/mchsk/openwrt-lte-keep-alive

wenqingj, 2024-03-02:
@brada4 Thank you. I am going to try it out, now that I know that the broadcast flag value does not make any difference.
This is the log just captured, with broadcast flag off (the default). It is same as the the log captured when the flag was on:
```
Sat Mar  2 06:53:39 2024 daemon.notice netifd: wan (17231): udhcpc: sending renew to server 10.72.2.3
Sat Mar  2 06:59:14 2024 daemon.notice netifd: wan (17231): udhcpc: sending renew to server 10.72.2.3
Sat Mar  2 07:02:02 2024 daemon.notice netifd: wan (17231): udhcpc: sending renew to server 10.72.2.3
Sat Mar  2 07:03:26 2024 daemon.notice netifd: wan (17231): udhcpc: sending renew to server 10.72.2.3
Sat Mar  2 07:04:08 2024 daemon.notice netifd: wan (17231): udhcpc: sending renew to server 10.72.2.3
Sat Mar  2 07:04:29 2024 daemon.notice netifd: wan (17231): udhcpc: sending renew to server 10.72.2.3
Sat Mar  2 07:04:32 2024 daemon.notice netifd: wan (17231): udhcpc: broadcasting renew
Sat Mar  2 07:04:32 2024 daemon.notice netifd: wan (17231): udhcpc: lease of 108.162.136.122 obtained from 10.72.2.3, lease time 91001
Sat Mar  2 07:04:33 2024 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
```
Will try out the lease time and script...

wenqingj, 2024-03-23:
I think I cannot use a script to automatically restart the wan interface because any connections (like file downloads) at the time of the restarts will be broken. But I can manually restart the wan interface whenever I am sure nobody is using the Internet.
I tried another router with 802.11ax and OpenWrt 23.05.2, and it's pretty stable. It has the same weird IP renewal behavior (more and more frequent dhcp requests, eventually a broadcast resulting in a new lease), but it does not suffer from the connection issues of TL-WR841N.
Today OpenWrt 23.05.3 shows up. I just put it to TL-WR841N (802.11n) and will see if that makes a difference...

brada4, 2024-03-23:
Wait for dust to settle on 23.5.3

wenqingj, 2024-04-05:
It's been 12 days since release 23.05.3 was put on the router and it has not encountered any issue yet. The IP renewal from log seems as before (as shown in the following sample). So if the issue is really gone in the new release I've no idea how it was fixed...
```
Fri Apr  5 00:20:31 2024 daemon.notice netifd: wan (4493): udhcpc: sending renew to server 10.72.2.3
Fri Apr  5 00:20:58 2024 daemon.notice netifd: wan (4493): udhcpc: sending renew to server 10.72.2.3
Fri Apr  5 00:21:01 2024 daemon.notice netifd: wan (4493): udhcpc: broadcasting renew
Fri Apr  5 00:21:01 2024 daemon.notice netifd: wan (4493): udhcpc: lease of 108.162.136.108 obtained from 10.72.2.3, lease time 151732
Fri Apr  5 00:21:01 2024 user.notice firewall: Reloading firewall due to ifupdate of wan (eth0.2)
```

brada4, 2024-04-05:
(certainly would be nice to know the patch that fixed issue to approach same code piece when problem occurs again)

wenqingj, 2024-04-16:
The issue is no more for me. I considered it fixed in release 23.05.3 (instead of my sheer luck in the past 20+ days).
By the way, during those days I encountered loss of wifi twice, and resolved it by restarting the wireless interface. That should be a separate issue (if not just a glitch) since the problem reported in this ticket relates only to the wan (wired) interface.

brada4, 2024-04-16:
Esamine logread, add extra logging for hostapd, should not happen under normal conditions. Only thing related - if you run all wired at max speed and wireless are actually used at 1w 30dBm , then some old USB HDD, it may stress out the power adapter.

wenqingj, 2024-04-20:
@brada4 Is that addressing another ticket? My wifi router does not have any USB port, hence no USB HDD connected.
Anyway, thank you for all the help in trying to resolve this. Since the issue reported by this ticket (wan connection lost) is no more after release 23.05.3. I'm gonna close it now ...

closed 2024-04-20 by wenqingj