openwrt/openwrt #14541: ath10k: 5 GHz radio not detected.

castillofrancodamian, 2024-02-02:
### Describe the bug

After upgrading to a newer snapshot, the 5 GHz radio is not detected. I think [this commit](https://github.com/openwrt/openwrt/pull/14416) generates the problem.

### OpenWrt version

r24969-44cd90c49a

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C60 v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2024-02-02:
Hi! Sorry for.the trouble. Can you install openwrt 23.05.2 and show me the full kernel log? From the wiki kernel log, it seems that this model never work correctly. https://openwrt.org/toh/tp-link/archer_c60_v2

castillofrancodamian, 2024-02-03:
@DragonBluep That information is totally outdated. The 5 GHz band has always worked for me.

```
[    0.000000] Linux version 5.15.137 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Tue Nov 14 13:38:11 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C60 v1
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55908K/65536K available (6079K kernel code, 597K rwdata, 780K rodata, 1240K init, 217K bss, 9628K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000001] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008290] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074814] pid_max: default: 32768 minimum: 301
[    0.080510] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.088242] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.102629] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.113047] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.120390] pinctrl core: initialized pinctrl subsystem
[    0.127617] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.134390] thermal_sys: Registered thermal governor 'step_wise'
[    0.146026] clocksource: Switched to clocksource MIPS
[    0.159003] NET: Registered PF_INET protocol family
[    0.164361] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.172914] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.181793] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.189984] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.198081] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.205527] TCP: Hash tables configured (established 1024 bind 1024)
[    0.212383] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.219343] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.227075] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.233088] PCI: CLS 0 bytes, default 32
[    0.240559] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.252052] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.258275] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.269900] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.281330] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.288627] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.297824] printk: console [ttyS0] disabled
[    0.302395] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.311551] printk: console [ttyS0] enabled
[    0.320620] printk: bootconsole [early0] disabled
[    0.348449] spi-nor spi0.0: gd25q64 (8192 Kbytes)
[    0.353414] 5 fixed-partitions partitions found on MTD device spi0.0
[    0.360067] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.366951] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.374360] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.381289] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.388352] Creating 5 MTD partitions on "spi0.0":
[... 149 lines cut ...]
[   45.378328] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_invitados: link becomes ready
[   45.528310] br-lan: port 7(5g_wds) entered blocking state
[   45.534026] br-lan: port 7(5g_wds) entered disabled state
[   45.560348] device 5g_wds entered promiscuous mode
[   45.565722] br-lan: port 7(5g_wds) entered blocking state
[   45.571384] br-lan: port 7(5g_wds) entered forwarding state
[   45.736341] br-lan: port 7(5g_wds) entered disabled state
[   46.816136] br-lan: port 5(5g_jitter) entered disabled state
[   46.822489] br-lan: port 6(5g_invitados) entered disabled state
[... cut ...]
[   49.011075] 5g_wds: associated
[   49.147400] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_wds: link becomes ready
[   49.154479] br-lan: port 7(5g_wds) entered blocking state
[   49.160138] br-lan: port 7(5g_wds) entered forwarding state
[   49.190954] br-lan: port 5(5g_jitter) entered blocking state
[   49.196898] br-lan: port 5(5g_jitter) entered forwarding state
[   49.441182] br-lan: port 6(5g_invitados) entered blocking state
[   49.447396] br-lan: port 6(5g_invitados) entered forwarding state
```

DragonBluep, 2024-02-03:
And please upload art partition backup here so that I can do some test. thanks.

DragonBluep, 2024-02-03:
By the way, does CT version driver works for you?

Ansuel, 2024-02-03:
@DragonBluep problem is probably that in board-2 there isn't any ID 20...

`bus=pci,bmi-chip-id=0,bmi-board-id=20`

This is why I had some fear for those case that linked the calibration to the board file... It was to handle these funny case... But I wonder how many of these are around... Ideally we can sort all by just passing the calibration nvmem but I think we need some patching to ath10k.

castillofrancodamian, 2024-02-03:
> And please upload art partition backup here so that I can do some test. thanks.

> By the way, does CT version driver works for you?

On snapshots with ath10k or with ath10k-ct? Before or after trying ath10-ct?

castillofrancodamian, 2024-02-03:
> By the way, does CT version driver works for you?

<details><summary>Kernel Log</summary>
<p>

```
[    0.000000] Linux version 5.15.148 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24991-3595931934) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Fri Feb 2 16:32:50 2024
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C60 v1
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55332K/65536K available (6120K kernel code, 592K rwdata, 1336K rodata, 1224K init, 216K bss, 10204K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000001] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008291] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074815] pid_max: default: 32768 minimum: 301
[    0.080505] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.088241] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.102383] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.112803] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.120156] pinctrl core: initialized pinctrl subsystem
[    0.127380] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.134183] thermal_sys: Registered thermal governor 'step_wise'
[    0.150518] clocksource: Switched to clocksource MIPS
[    0.163558] NET: Registered PF_INET protocol family
[    0.168891] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.177416] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.186401] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.194583] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.202678] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.210117] TCP: Hash tables configured (established 1024 bind 1024)
[    0.216971] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.223934] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.231696] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.237726] PCI: CLS 0 bytes, default 32
[    0.245174] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.256705] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.262928] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.274570] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.285978] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.293251] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.302486] printk: console [ttyS0] disabled
[    0.307061] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.316210] printk: console [ttyS0] enabled
[    0.325282] printk: bootconsole [early0] disabled
[    0.352965] spi-nor spi0.0: gd25q64 (8192 Kbytes)
[    0.357934] 5 fixed-partitions partitions found on MTD device spi0.0
[    0.364583] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.371467] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[... 292 lines cut ...]
ubox - 2024-01-24-2c5887cb-1
ubus - 2023-11-28-f84eb599-1
ubusd - 2023-11-28-f84eb599-1
uci - 2023-08-10-5781664d-1
ucode - 2023-11-30-6e89b89e-1
ucode-mod-fs - 2023-11-30-6e89b89e-1
ucode-mod-html - 1
ucode-mod-lua - 1
ucode-mod-math - 2023-11-30-6e89b89e-1
ucode-mod-nl80211 - 2023-11-30-6e89b89e-1
ucode-mod-rtnl - 2023-11-30-6e89b89e-1
ucode-mod-ubus - 2023-11-30-6e89b89e-1
ucode-mod-uci - 2023-11-30-6e89b89e-1
ucode-mod-uloop - 2023-11-30-6e89b89e-1
uhttpd - 2023-06-25-34a8a74d-2
uhttpd-mod-ubus - 2023-06-25-34a8a74d-2
urandom-seed - 3
urngd - 2023-11-01-44365eb1-1
usign - 2020-05-23-f1f65026-1
wireless-regdb - 2023.09.01-1
wpad-mbedtls - 2023-09-08-e5ccbfc6-6
```

</p>
</details>

castillofrancodamian, 2024-02-03:
> And please upload art partition backup here so that I can do some test. thanks.

[archer-c60.mtd7.bin.tar.gz](https://github.com/openwrt/openwrt/files/14149368/archer-c60.mtd7.bin.tar.gz)

DragonBluep, 2024-02-03:
> @DragonBluep problem is probably that in board-2 there isn't any ID 20...
> 
> `bus=pci,bmi-chip-id=0,bmi-board-id=20`

Oh I see it. Both CT version and none-CT version board doesn't have this id.

> This is why I had some fear for those case that linked the calibration to the board file... It was to handle these funny case... But I wonder how many of these are around... Ideally we can sort all by just passing the calibration nvmem but I think we need some patching to ath10k.

@Ansuel I think a simple patch checking `ret` value and reading nvmem pre-calibration data after this line should be enough. Would you want to cook such a patch? If not, maybe I can have a try tomorrow.
https://github.com/greearb/ath10k-ct/blob/fadd0768cbd22248a60efbb219ccefc9d86cd78c/ath10k-6.4/core.c#L2001

DragonBluep, 2024-02-11:
This patch should fix the CT version driver. Place it in `package/kernel/ath10k-ct/patches`.

Anyway, I think add missing board id to board-2.bin is a better fix. One week has passed, and we still only received one failed feedback. ath10k use `firmware` struct to store the board data file. It's designed to read data from filesystem. Though we can manually load it from NVMEM cell, that make me uncomfortable.
```
--- a/ath10k-6.4/core.c
+++ b/ath10k-6.4/core.c
@@ -1000,6 +1000,35 @@ static const struct firmware *ath10k_fetch_fw_file(struct ath10k *ar,
 	return fw;
 }
 
+static struct firmware *ath10k_fetch_fw_nvmem(struct ath10k *ar,
+					      const char *cell_name)
+{
+	struct firmware *fw;
+	struct nvmem_cell *cell;
+	void *buf;
+	size_t len;
+
+	cell = devm_nvmem_cell_get(ar->dev, cell_name);
+	if (IS_ERR(cell))
+		return ERR_PTR(-ENOENT);
+
+	buf = nvmem_cell_read(cell, &len);
+	nvmem_cell_put(cell);
+	if (IS_ERR(buf))
+		return buf;
+
+	fw = kzalloc(sizeof(*fw), GFP_KERNEL);
+	if (!fw) {
+		kfree(buf);
+		return fw;
+	}
+
+	fw->data = (u8 *)buf;
+	fw->size = len;
+
+	return fw;
+}
+
 static int ath10k_push_board_ext_data(struct ath10k *ar, const void *data,
 				      size_t data_len)
 {
@@ -1625,6 +1654,15 @@ static int ath10k_core_fetch_board_data_api_1(struct ath10k *ar, int bd_ie_type)
 			ar->normal_mode_fw.board = fw;
 		}
 
+		/* workaround: load board data from NVMEM calibration data */
+		if (IS_ERR(ar->normal_mode_fw.board)) {
+			ath10k_err(ar, "failed to find board id from %s\n",
+				   ar->hw_params.fw.board);
+
+			fw = ath10k_fetch_fw_nvmem(ar, "pre-calibration");
+			ar->normal_mode_fw.board = fw;
+		}
+
 		if (IS_ERR(ar->normal_mode_fw.board))
 			return PTR_ERR(ar->normal_mode_fw.board);
 

```

Ansuel, 2024-02-11:
@DragonBluep main problem is that we have way too much ath79 target and some device are used as dumb ap and use only stable build... Adding stuff to board-2 is a no go as we use upstream board-2 and we can't provide a custom one...

I really think we need this patch, I would add a big warning tho that we are using pre-calibration.

DragonBluep, 2024-02-12:
@Ansuel Hi! I opened a new PR in https://github.com/openwrt/openwrt/pull/14611 to submit this patch.

pesa1234, 2024-03-20:
I'm on Archer C6 v2 and I also get this error. How can I help? @DragonBluep

mrkiko, 2024-03-22:
Would be the R6100 affected by this?

pesa1234, 2024-03-22:
> Would be the R6100 affected by this?

Sorry, but I don't know, please try to install latest snapshot and check if wifi 5ghz it works.

pesa1234, 2024-06-19:
Just to confirm that the problem is still present

castillofrancodamian, 2024-06-19:
> Just to confirm that the problem is still present

It hasn't been resolved yet. I guess they want to let 8/64 devices die and this is one of them.

Ansuel, 2024-06-19:
@castillofrancodamian ???? no conspiracy theory.... just very busy and it's hard to track everything... @DragonBluep we really need to find a solution to this...

castillofrancodamian, 2024-06-19:
@Ansuel I know you are very busy and loaded, but the removal of 8/64 devices is something that is going to happen.

Ansuel, 2024-06-19:
Sure but I have to make it clear that we aren't doing anything like that with these funny procedure....

If we want to drop a family of devices we just make the statement and the clear commit... We don't cause intentional regression. This problem for example was caused by a rework and us noticing the bad state of some firmware blob. And me not giving enough priority to handle this ANNND... it's stalled...

DragonBluep, 2024-06-21:
Please test this file (extract from `Archer C60(EU)_V1_160811.zip`).
[board-2.zip](https://github.com/user-attachments/files/15930043/board-2.zip)
1. Uncompress it to get `board-2.bin` and replace `/lib/firmware/ath10k/ath10k/QCA9888/hw2.0/board-2.bin` with it.
2. Reboot router.

castillofrancodamian, 2024-06-21:
@DragonBluep I have downloaded that file and saved it in `/lib/firmware/ath10k/QCA9888/hw2.0/board-2.bin` replacing the original.

<details><summary>logread with original board-2.bin</summary>
<p>

```
root@archer-c60:~# logread
Fri Jun 21 15:31:03 2024 kern.notice kernel: [    0.000000] Linux version 6.6.34 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26733-2b6772c82c) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Thu Jun 20 21:23:30 2024
Fri Jun 21 15:31:03 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Fri Jun 21 15:31:03 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
Fri Jun 21 15:31:03 2024 kern.info kernel: [    0.000000] MIPS: machine is TP-Link Archer C60 v1
Fri Jun 21 15:31:03 2024 kern.info kernel: [    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Zone ranges:
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Early memory node ranges
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Fri Jun 21 15:31:04 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Fri Jun 21 15:31:04 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Fri Jun 21 15:31:04 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] Memory: 54500K/65536K available (6712K kernel code, 596K rwdata, 1560K rodata, 1236K init, 215K bss, 11036K reserved, 0K cma-reserved)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] CPU clock: 775.000 MHz
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 388MHz, resolution 2ns, wraps every 5541893118ns
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.008303] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.074807] pid_max: default: 32768 minimum: 301
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.089177] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.096918] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.113454] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.126053] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.136474] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.146102] pinctrl core: initialized pinctrl subsystem
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.154350] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.161151] thermal_sys: Registered thermal governor 'step_wise'
Fri Jun 21 15:31:04 2024 kern.warn kernel: [    0.163833] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.190685] clocksource: Switched to clocksource MIPS
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.206202] NET: Registered PF_INET protocol family
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.211622] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.220664] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jun 21 15:31:04 2024 kern.info kernel: [    0.229645] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[... cut ...]
Fri Jun 21 15:46:48 2024 daemon.notice netifd: Interface 'wwan' is now up
Fri Jun 21 15:46:49 2024 user.notice firewall: Reloading firewall due to ifup of wwan (5g_wwan)
Fri Jun 21 15:46:49 2024 daemon.warn odhcpd[1801]: No default route present, overriding ra_lifetime to 0!
Fri Jun 21 15:49:18 2024 daemon.notice netifd: wwan (3577): udhcpc: sending renew to server 10.192.0.1
Fri Jun 21 15:49:19 2024 daemon.notice netifd: wwan (3577): udhcpc: lease of 10.196.193.198 obtained from 10.192.0.1, lease time 300
```

</p>
</details>

DragonBluep, 2024-06-21:
Please try this one: [board-2.zip](https://github.com/user-attachments/files/15934474/board-2.zip)

castillofrancodamian, 2024-06-22:
@DragonBluep 

<details><summary>logread with second downloaded board-2.bin</summary>
<p>

```
root@archer-c60:~# logread
Sat Jun 22 00:14:54 2024 kern.notice kernel: [    0.000000] Linux version 6.6.34 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26733-2b6772c82c) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Thu Jun 20 21:23:30 2024
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] MIPS: machine is TP-Link Archer C60 v1
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Zone ranges:
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Early memory node ranges
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Sat Jun 22 00:14:54 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sat Jun 22 00:14:54 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Sat Jun 22 00:14:54 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] Memory: 54500K/65536K available (6712K kernel code, 596K rwdata, 1560K rodata, 1236K init, 215K bss, 11036K reserved, 0K cma-reserved)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] CPU clock: 775.000 MHz
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 388MHz, resolution 2ns, wraps every 5541893118ns
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.008302] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.074807] pid_max: default: 32768 minimum: 301
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.089160] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.096902] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.113441] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.126046] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.136466] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.146099] pinctrl core: initialized pinctrl subsystem
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.154348] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.161149] thermal_sys: Registered thermal governor 'step_wise'
Sat Jun 22 00:14:54 2024 kern.warn kernel: [    0.163829] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.190681] clocksource: Switched to clocksource MIPS
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.206207] NET: Registered PF_INET protocol family
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.211631] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.220665] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.229646] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
Sat Jun 22 00:14:54 2024 kern.info kernel: [    0.237832] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... cut ...]
Sat Jun 22 00:24:04 2024 daemon.notice netifd: wwan (3769): udhcpc: broadcasting discover
Sat Jun 22 00:24:07 2024 daemon.notice netifd: wwan (3769): udhcpc: broadcasting discover
Sat Jun 22 00:24:09 2024 authpriv.info dropbear[3906]: Child connection from 192.168.70.135:39092
Sat Jun 22 00:24:10 2024 daemon.notice netifd: wwan (3769): udhcpc: broadcasting discover
Sat Jun 22 00:24:10 2024 authpriv.notice dropbear[3906]: Password auth succeeded for 'root' from 192.168.70.135:39092
```

</p>
</details> 

It seems to be working. What would I have to send in that PR?

DragonBluep, 2024-06-22:
You can rename `board-2.bin` to something like `board-bmi-board-id-20.qca9888` and then push it.
ref: https://github.com/openwrt/firmware_qca-wireless/commit/f9ff251c463bb7989d4daa88648e152a4447c3b5 https://github.com/openwrt/openwrt/commit/f9e6218b82b3d2827e3d781ebc0c4eea59bf8bb6

kofec, 2024-11-26:
I see the same issue:
on OpenWrt 24.10-SNAPSHOT, r28040-141ccfd548:
```
[    0.000000] Linux version 6.6.63 (MyBuild@OnServer) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28040-141ccfd548) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Mon Nov 25 15:42:03 2024
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C6 v2 (EU/RU/JP)
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
...............................
[   19.569811] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
[   19.637222] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   19.644032] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   23.199731] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   23.209313] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   23.223852] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 59e741e7
[   23.603853] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
[   24.130344] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   24.140118] ath10k_pci 0000:00:00.0: failed to fetch board file: -12
[   24.166452] ath10k_pci 0000:00:00.0: could not probe fw (-12)
```
On OpenWrt 23.05-SNAPSHOT, r24113-fa0a7e00e8:
```
[    0.000000] Linux version 5.15.167 (MyBuild@OnServer) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24113-fa0a7e00e8) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Sat Sep 28 16:33:21 2024
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C6 v2 (EU/RU/JP)
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
.........................
[   22.431624] ath10k_pci 0000:00:00.0: Loading BDF type 0
[   22.441813] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
[   22.664615] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id 0:20 crc32 e0ac4ebe
[   24.502406] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   24.510209] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[   24.540332] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   24.551712] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12644 sram: 29508'
[   24.786204] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1

```

hauke, 2024-12-08:
There are multiple complains about this problem in the OpenWrt 24.10 threads. @Ansuel  or @robimarko  or @DragonBluep  How should we handle this?

hauke, 2024-12-08:
Should we revert the calibration data extract back to the old userspace process for some devices?

Ansuel, 2024-12-08:
@hauke yes that is an option but it just postponing the problem... The problem is that those board file are missing... Or better say the ID are missing... I need to check how this can correctly be fixed.

robimarko, 2024-12-08:
Basically, we need to dig out the BDF for them from the stock FW from my point of view, faking the BDF by passing caldata instead of it is not correct.

mrkiko, 2024-12-12:
Hi!!

I sent some patches in the ML for all the three C60 Revisions - I only tested V3.
It would be nice if someone could test the others and give some feedback on how it works, hoping it does :) .

A test procedure could look like:
1 - Clone https://github.com/openwrt/firmware_qca-wireless.
2 - Pick up the patches from the mailing list or patchwork:
V1: https://patchwork.ozlabs.org/project/openwrt/patch/20241209214157.94055-1-mrkiko.rs@gmail.com/
V2: https://patchwork.ozlabs.org/project/openwrt/patch/20241209214157.94055-2-mrkiko.rs@gmail.com/
V3: https://patchwork.ozlabs.org/project/openwrt/patch/20241209214157.94055-3-mrkiko.rs@gmail.com/
.
3 - Create a local branch:
$ git branch c60test
$ git checkout c60test
.
4 - Apply the patches via "git am".
5 - Upload the relevant file replacing board-2.bin.
6 - Reload ath10k driver modules.

I don't know if these patches will be accepted as they are and some work is missing anyway: but maybe getting some feedback on performance and so on can help.
Also, some decisions may need to be taken - so this is for testing purposes only.

mrkiko, 2024-12-12:
Oh - and do note these BDFs are extracted from EU firmware.

dwardor, 2024-12-28:
I'm seeing the same issue when upgrading a Linksys WHW03 V2 to self built 24.10.0-rc4 (using ct-full-htt)

```
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-full-htt=y
CONFIG_PACKAGE_ath10k-firmware-qca9888-ct-full-htt=y
``` 

At first boot I had no wifi and dmesg showed
ath10k/QCA9888/hw2.0/board-2.bin & ath10k/QCA4019/hw1.0/board-2.bin
could not be loaded

```
> dmesg | grep ath10k
[   14.686125] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
[   14.687219] ath10k_pci 0000:01:00.0: enabling device (0140 -> 0142)
[   14.693833] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   15.512408] ath10k_pci 0000:01:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   15.512483] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   15.524943] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9888-fH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 9a038867
[   15.843519] ath10k_pci 0000:01:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=23,variant=linksys-whw03v2 from ath10k/QCA9888/hw2.0/board-2.bin
[   16.040671] ath10k_pci 0000:01:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   16.040754] ath10k_pci 0000:01:00.0: failed to fetch board file: -12
[   16.049231] ath10k_pci 0000:01:00.0: could not probe fw (-12)
[   16.833249] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   16.833324] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   16.845212] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 cc518b9b
[   16.900152] ath10k_ahb a000000.wifi: failed to fetch board data for bus=ahb,bmi-chip-id=0,bmi-board-id=20,variant=linksys-whw03v2 from ath10k/QCA4019/hw1.0/board-2.bin
[   17.097864] ath10k_ahb a000000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   17.097947] ath10k_ahb a000000.wifi: failed to fetch board file: -12
[   17.106376] ath10k_ahb a000000.wifi: could not probe fw (-12)
[   17.621042] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   17.621170] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   17.633110] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 cc518b9b
[   17.687495] ath10k_ahb a800000.wifi: failed to fetch board data for bus=ahb,bmi-chip-id=0,bmi-board-id=21,variant=linksys-whw03v2 from ath10k/QCA4019/hw1.0/board-2.bin
[   17.885173] ath10k_ahb a800000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   17.885259] ath10k_ahb a800000.wifi: failed to fetch board file: -12
[   17.893671] ath10k_ahb a800000.wifi: could not probe fw (-12)
```

The files are actually there but are much smaller than on a 23.05.5 install.
I copied over the files from 23.05.5 and rebooted to working wifi.

Leo-PL, 2025-01-15:
@Ansuel, @robimarko I've noticed that ath10k devs are slow to review submitted BDFs recently. Files I've submitted haven't seen any update for 3 months.
Kind-of offtopic, but in this case - would adding them to `ipq-wifi` package temporarily be a good idea, as we do for newer devices?
I'm facing similar issue with ZTE MF286A/C boards, in which lack of custom BDF severely reduces 5GHz performance. I've prepared fix for that, but haven't yet submitted it.

grosjo, 2025-01-26:
With git snapshot, even with teh board-2.bin regenerated, it seems there is something wrong
https://github.com/openwrt/firmware_qca-wireless/pull/56

rpodric, 2025-02-06:
Has anyone confirmed with Archer C7v2, which is ath10k?

mfullerca, 2025-02-06:
> Has anyone confirmed with Archer C7v2, which is ath10k?

I have one, but for unrelated reasons related to my switch/VLAN config being incompatible, I haven't had time to upgrade.  I do, however, have 2 Ubiquiti UniFi AC Lites, which I believe are ath10k (Qualcomm Atheros QCA9880).  They work.  IDK about performance -- I'm seeing 416 Mbps on iPerf from a phone (vs 1.42 Gbps on an MT6000 for reference on my phone and LAN limits), which seems ballpark for hardware that old.

Leo-PL, 2025-02-06:
> Has anyone confirmed with Archer C7v2, which is ath10k?

This issue is related only to ath10k wave2 devices. C7v2 is wave1.

khs-kks, 2025-02-06:
Has anyone tested with TP-Link Archer C6 v2 (EU/RU/JP)?

Architecture: Qualcomm Atheros QCA956X ver 1 rev 0
Target Platform: 	ath79/generic

Salomon-MH, 2025-02-06:
This issue affects OpenWrt 24.10.0 on Archer C6 v2 (EU/RU/JP) devices.
OpenWrt 23.05.5 was/is working fine for these devices.

Neither custom builds which I normally use, nor the public build works.
Neither ath10k nor ath10k-ct show any 5GHz adapters.

[image]

[image]

[image]

[image]

fundef1, 2025-02-06:
just adding the my Zyxel NBG6716 (ath79/nand QCA9880) also doesn't load with the release build openwrt24.10.0

```
# dmesg | grep ath10
[   17.949969] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x3c.
[   18.007694] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   18.014241] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   20.808285] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   20.817699] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   20.828019] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fW-022-ecad3248 api 2 features wmi-10.x,has-wmi-mgmt-tx,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,txrate-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 3e4cf97f
[   21.377333] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   23.317003] ath10k_pci 0000:00:00.0: failed to receive control response completion, polling..
[   24.357007] ath10k_pci 0000:00:00.0: ctl_resp never came in (-145)
[   24.363297] ath10k_pci 0000:00:00.0: failed to connect to HTC: -145
[   24.512443] Modules linked in: ath10k_pci(O+) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat(O) ledtrig_usbport sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) phy_ar7200_usb phy_ar7100_usb usbcore nls_base usb_common crc32c_generic
[   24.584431] Workqueue: ath10k_wq 0x82d20480 [ath10k_core@b6da697f+0x56000]
[   24.696029] [<82d236b8>] 0x82d236b8 [ath10k_core@b6da697f+0x56000]
[   24.702473] [<82d2e7e4>] 0x82d2e7e4 [ath10k_core@b6da697f+0x56000]
[   24.708887] [<82d1e8e0>] 0x82d1e8e0 [ath10k_core@b6da697f+0x56000]
[   24.718952] [<82d20b6c>] 0x82d20b6c [ath10k_core@b6da697f+0x56000]
[   24.782516] Modules linked in: ath10k_pci(O+) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat(O) ledtrig_usbport sha512_generic seqiv sha3_generic jitterentropy_rng drbg hmac geniv rng cmac fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) phy_ar7200_usb phy_ar7100_usb usbcore nls_base usb_common crc32c_generic
[   24.854493] Workqueue: ath10k_wq 0x82d20480 [ath10k_core@b6da697f+0x56000]
[   24.966065] [<82d236c4>] 0x82d236c4 [ath10k_core@b6da697f+0x56000]
[   24.972509] [<82d2e7e4>] 0x82d2e7e4 [ath10k_core@b6da697f+0x56000]
[   24.978923] [<82d1e8e0>] 0x82d1e8e0 [ath10k_core@b6da697f+0x56000]
[   24.988988] [<82d20b6c>] 0x82d20b6c [ath10k_core@b6da697f+0x56000]
[   25.040911] ath10k_pci 0000:00:00.0: could not init core (-145)
[   25.047238] ath10k_pci 0000:00:00.0: could not probe fw (-145)
```

rmaeder, 2025-02-06:
I don't know what to think of this; releasing 24.10.0 even though it is known to break these devices (Archer C6(EU) V2.0).
Not only did it break the 5GHz network, but it tried to create a bunch of WLAN networks with SSID "OpenWRT", and no encryption. Reverted to 23.05.5, and things are back to normal.

MrObvious, 2025-02-06:
Glad I saw this because I think it would affect my Archer C7v2.

rpodric, 2025-02-06:
@MrObvious It shouldn't per the post earlier today:

https://github.com/openwrt/openwrt/issues/14541#issuecomment-2639237291

MrObvious, 2025-02-07:
Ah missed that thanks.

rmaeder, 2025-02-07:
> [@rmaeder](https://github.com/rmaeder) how did you revert to release 23.05.5?

I always use imagebuilder, so I built the old image again, copied it to the device (scp), and did (via ssh)
`sysupgrade -v openwrt-23.05.5-ath79-generic-tplink_archer-c6-v2-squashfs-sysupgrade.bin`
then I restored the backup I did before the upgrade attempt.

zorxd, 2025-02-07:
Is the Linksys EA8500 affected by this issue? ath10k wave 2 from what I understand, Qualcomm Atheros QCA9990. So are they all affected?

esbeeb, 2025-02-08:
I have a TP-Link Archer C60 v3, and the 5GHz radio is **not** detected (but yes, the 2.4GHz radio is detected).

scrutinizer80, 2025-02-08:
This should have been a showstopper.
Releasing something with this kind of regressive bug is rather unprofessional.

OW87, 2025-02-08:
If it could help, I know my device may not have the same WiFi hardware, but I got "generic unknown" after upgrading (configuration not preserved) my Xiaomi AX320 to 24.10 yesterday. 
What I did to resolve this issue was by removing the entire WiFi networks on the 5Ghz from this device and rebooting. I don't remember if the problem was resolved after creating a new WiFi network or if it is already good after booting.

khs-kks, 2025-02-08:
> This should have been a showstopper. Releasing something with this kind of regressive bug is rather unprofessional.

Hey, but "Support for over 100 new devices was added in addition to the device support by OpenWrt 23.05." per https://openwrt.org/releases/24.10/notes-24.10.0

/s

g6094199, 2025-02-08:
i see a non working 5GHz on a MikroTik cAP ac on 24.10

Leo-PL, 2025-02-08:
There is a workaround, thanks to the people, who bothered to extract missing board data from their devices and submit it upstream:

1. Get the board data for your device from here: https://github.com/openwrt/firmware_qca-wireless - look through pull requests as well. For example, Archer C6 v2 EU: https://github.com/openwrt/firmware_qca-wireless/pull/52/files
2. Locate which board file is missing: 
```
dmesg | grep 'failed to fetch board data'
```
3. Put this file under /lib/firmware/ath10k, for C6v2 EU it is `/lib/firmware/ath10k/QCA9888/hw2.0/board-2.bin`, or similar location for your device - taken from previous point.
4. Reboot your device and check if the 5GHz is up.
5. Add this path to `/etc/sysupgrade.conf`, to get it preserved over upgrades:
```
echo /lib/firmware/QCA9888/hw2.0/board-2.bin >> /etc/sysupgrade.conf
```
5. ...
6. Profit.

ashr123, 2025-02-08:
@Leo-PL regarding https://github.com/openwrt/firmware_qca-wireless/pull/52/files
that means I need to change it's name from `board-tplink_archer-c6-v2.qca9888` to `board-2.bin`?

and than do

```shell
scp ~/Downloads/board-tplink_archer-c6-v2.qca9888 root@openwrt.local:/lib/firmware/ath10k/QCA9888/hw2.0/board-2.bin
```
?

mrkiko, 2025-02-08:
> scp ~/Downloads/board-tplink_archer-c6-v2.qca9888 ***@***.***:/lib/firmware/ath10k/QCA9888/hw2.0/board-2.bin

I sent the files for C60 V{1, 2, 3}; but over ML, not PR.

Leo-PL, 2025-02-08:
@ashr123 Yes, exactly.

ashr123, 2025-02-08:
@Leo-PL I've needed to install `openssh-sftp-server` for `scp` to work

in addition I've executed `scp` as I've asked and verified the file via

```bash
ll /lib/firmware/ath10k/QCA9888/hw2.0
```

and got

> ```bash
> drwxr-xr-x    1 root     root             0 Feb  4 01:09 ./
> drwxr-xr-x    1 root     root             0 Feb  4 01:09 ../
> -rw-r--r--    1 root     root         12176 Feb  8 16:04 board-2.bin
> lrwxrwxrwx    1 root     root            30 Feb  4 01:09 board.bin -> ../../cal-pci-0000:01:00.0.bin
> -rw-r--r--    1 root     root        630784 Feb  4 01:09 firmware-5.bin
> ```

in the GUI I've checked what've changed and I still saw
[image]

g6094199, 2025-02-08:
> [@Leo-PL](https://github.com/Leo-PL) I've needed to install `openssh-sftp-server` (140KB) for `scp` to work

no you dont need to. the internal ssh server works. you just MUST use legacy protocoll setting in scp to get it working.

ashr123, 2025-02-08:
> > [@Leo-PL](https://github.com/Leo-PL) I've needed to install `openssh-sftp-server` (140KB) for `scp` to work
> 
> no you dont need to. the internal ssh server works. you just MUST use legacy protocoll -O setting in scp to get it working.

already installed and `scp`ed it LOL but it seems that it didn't help

Leo-PL, 2025-02-08:
@ashr123 could you `dmesg | grep ath10k`? Maybe ath10k doesn't pick up the file because it only has board data annotated with a variant, that isn't set in the device tree. Then, this won't work indeed :(

[145 more replies]

closed 2026-09-10 by openwrt-bot