openwrt/openwrt #14542: FRITZ!Repeater 3000 throughput performance

meiser79, 2024-02-02:
### Describe the bug

Hi,

I try to use a FRITZ!Repeater 3000 (FR3000) as a switch between my OpenWrt router and a FRITZ!Repeater 1200 (FR1200). All devices are running on 23.05 with DSA config.
It's like router lan --> lan1 FR3000 lan2 --> FR1200 lan interface

If I connect the FR1200 directly to my router, my wifi client (macOS) gets throughput of approx. 340 Mbps (iperf3, IPv4, 4 TCP flows) connected to SSID 'LAN'.
If I connect the FR1200 to the FR3000, my wifi client (macOS) gets throughput of approx. 190 Mbps (iperf3, IPv4, 4 TCP flows) connected to SSID 'LAN'. It's the same for connecting the wifi client to FR3000 or FR1200.
But on the FR3000 or FR1200 directly (logged in via SSH and running iperf3), I still see the 340 Mbps.

As the counters do not increase on eth0 of FR3000, I assume that the packets are switched in hardware and do not go via the CPU. Changing settings like "software offload" or "packet steering" have no effect at all, I also installed irqbalance without any possible effect.

BTW, the FR3000 performance issue was also there with 22.03 (with swconfig configuration).

More info including a topology diagram and configs can be found in the [forum](https://forum.openwrt.org/t/fritz-repeater-3000-worse-throughput-performance/181626)

I doubt that it's a configuration issue, so maybe someone of the devs can have a look at the kernel part or give some troubleshooting advice.

Thanks a lot for your help,
meiser

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Repeater 3000

### Image kind

Official downloaded image

### Expected behaviour

throughput is the same (approx. 340Mbps) on client side, with the FR3000 inbetween.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-03:
Is same defect observable with guest network where vlan is not remapped at ports?
If you connect laptop wired via 2-port AP - is it slow?
Also you can change firewall forward to drop, the dumb ap does not receive packets for other IPs. Should have zero effect though.
`ethtool -S` from physical ports may help.

meiser79, 2024-02-03:
Yes, it's the same on the guest network.
No, the speed is fine when connected to the 2-port AP (FR3000). Same as if I use iperf on the FR1200 which is connected to the FR3000.

Please find below the ethtool output from lan1, lan2, eth0:
```
root@repeater3000:~# ethtool -S lan1
NIC statistics:
     tx_packets: 2563
     tx_bytes: 255659
     rx_packets: 547
     rx_bytes: 136620
     RxBroad: 9
     RxPause: 0
     RxMulti: 511
     RxFcsErr: 2050
     RxAlignErr: 0
     RxRunt: 0
     RxFragment: 0
     Rx64Byte: 978
     Rx128Byte: 153325
     Rx256Byte: 2761
     Rx512Byte: 541
     Rx1024Byte: 530
     Rx1518Byte: 10890
     RxMaxByte: 0
     RxTooLong: 0
     RxGoodByte: 25203030
     RxBadByte: 3088591
     RxOverFlow: 0
     Filtered: 22
     TxBroad: 1828
     TxPause: 0
     TxMulti: 735
     TxUnderRun: 0
     Tx64Byte: 2048
     Tx128Byte: 7369
     Tx256Byte: 3075
     Tx512Byte: 798
     Tx1024Byte: 556
     Tx1518Byte: 782063
     TxMaxByte: 0
     TxOverSize: 0
     TxByte: 1188903410
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 166455
     TXunicast: 793346
root@repeater3000:~# ethtool -S lan2
NIC statistics:
     tx_packets: 60997
     tx_bytes: 4291929
     rx_packets: 190924
     rx_bytes: 281188670
     RxBroad: 1845
     RxPause: 0
     RxMulti: 643
     RxFcsErr: 0
     RxAlignErr: 0
     RxRunt: 0
     RxFragment: 0
     Rx64Byte: 2090
     Rx128Byte: 8196
     Rx256Byte: 3162
     Rx512Byte: 846
     Rx1024Byte: 632
     Rx1518Byte: 973489
     RxMaxByte: 0
     RxTooLong: 0
     RxGoodByte: 1479432705
     RxBadByte: 0
     RxOverFlow: 0
     Filtered: 24
     TxBroad: 16
     TxPause: 0
     TxMulti: 586
     TxUnderRun: 0
     Tx64Byte: 1272
     Tx128Byte: 212968
     Tx256Byte: 3055
     Tx512Byte: 731
     Tx1024Byte: 567
     Tx1518Byte: 12955
     TxMaxByte: 0
     TxOverSize: 0
     TxByte: 35756909
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 985927
     TXunicast: 230946
root@repeater3000:~# ethtool -S eth0
NIC statistics:
     tx_q0_pkt: 2227
     tx_q1_pkt: 0
     tx_q2_pkt: 0
     tx_q3_pkt: 0
     tx_q4_pkt: 45209
     tx_q5_pkt: 0
     tx_q6_pkt: 0
     tx_q7_pkt: 0
     tx_q8_pkt: 15002
     tx_q9_pkt: 0
     tx_q10_pkt: 0
     tx_q11_pkt: 0
     tx_q12_pkt: 1372
     tx_q13_pkt: 0
     tx_q14_pkt: 0
     tx_q15_pkt: 0
     tx_q0_byte: 217458
     tx_q1_byte: 0
     tx_q2_byte: 0
     tx_q3_byte: 0
     tx_q4_byte: 3022777
     tx_q5_byte: 0
     tx_q6_byte: 0
     tx_q7_byte: 0
     tx_q8_byte: 1116570
     tx_q9_byte: 0
     tx_q10_byte: 0
     tx_q11_byte: 0
     tx_q12_byte: 236360
     tx_q13_byte: 0
     tx_q14_byte: 0
     tx_q15_byte: 0
     rx_q0_pkt: 48420
     rx_q1_pkt: 0
     rx_q2_pkt: 48602
     rx_q3_pkt: 0
     rx_q4_pkt: 93241
     rx_q5_pkt: 0
     rx_q6_pkt: 1421
     rx_q7_pkt: 0
     rx_q0_byte: 70626362
     rx_q1_byte: 0
     rx_q2_byte: 71998727
     rx_q3_byte: 0
     rx_q4_byte: 140254426
     rx_q5_byte: 0
     rx_q6_byte: 1165064
     rx_q7_byte: 0
     tx_desc_error: 0
     p00_RxBroad: 1930
     p00_RxPause: 0
     p00_RxMulti: 1448
     p00_RxFcsErr: 0
     p00_RxAlignErr: 0
     p00_RxRunt: 0
     p00_RxFragment: 0
     p00_Rx64Byte: 0
     p00_Rx128Byte: 62541
     p00_Rx256Byte: 642
     p00_Rx512Byte: 430
     p00_Rx1024Byte: 179
     p00_Rx1518Byte: 1
     p00_RxMaxByte: 17
     p00_RxTooLong: 0
     p00_RxGoodByte: 4976025
     p00_RxBadByte: 0
     p00_RxOverFlow: 0
     p00_Filtered: 23
     p00_TxBroad: 1914
     p00_TxPause: 0
     p00_TxMulti: 1258
     p00_TxUnderRun: 0
     p00_Tx64Byte: 1632
     p00_Tx128Byte: 1805
     p00_Tx256Byte: 417
     p00_Tx512Byte: 286
     p00_Tx1024Byte: 215
     p00_Tx1518Byte: 187329
     p00_TxMaxByte: 0
     p00_TxOverSize: 0
     p00_TxByte: 284812363
     p00_TxCollision: 0
     p00_TxAbortCol: 0
     p00_TxMultiCol: 0
     p00_TxSingleCol: 0
     p00_TxExcDefer: 0
     p00_TxDefer: 0
     p00_TxLateCol: 0
     p00_RXUnicast: 60432
     p00_TXunicast: 188512

```

brada4, 2024-02-03:
fcs-errors means eth1 has bad cable connected. 
chech ethtool without parameters on both sides of the link to confirm they negotiates same mode/speed.

meiser79, 2024-02-03:
The RxFCsErr counter does not increase, don't know why it's at 2050. The link speed is 1G everywhere.

brada4, 2024-02-03:
Please check negotiation reports. You can change negotiated features if there is any mismatch.
Obviously should be identical speed/duplex/eee/pause on both ends of cable.

Another is `tc -s qdisc` - should be noqueue on switch ports, may be by default fq_codel on supposed WAN ports.

meiser79, 2024-02-03:
It's noqueue on lan1 and lan2, and fq_codel on eth0. The ethtool values show some results.
As my internet connection is not that stable today, I have to wait till this is better again. Then, I can check again.

brada4, 2024-02-03:
`tc qdisc replace dev eth0 root noqueue`

brada4, 2024-02-03:
You can still manipulate ethernet negotiation and correct if anything is loose.

meiser79, 2024-02-14:
I checked again, ethtool is fine, changing tc to "noqueue" has no impact.

What else could it be? Any kernel PHY issue?

brada4, 2024-02-14:
Just that rxbadbyte counter, it is very small still adding up retransmits and not allowing full speed. If you could replace cable like carrying devices next to eachother and connect using brand new wire to see if error is just cable.

meiser79, 2024-04-30:
Not reproducable anymore

closed 2024-04-30 by meiser79