openwrt/openwrt #14545: AVM Fritz!Box 7430 bad performance

Snowiiii, 2024-02-03:
### Describe the bug

Hello. 2 Days ago i switched my 7430 from FritzOS to OpenWRT.
I expected same or better Internet Performance but this was not the case i got much worse performance.

Currently i pay for an 50Mibit/s download and 10Mbit/s upload speed at Vodafone. And thats what i also got with ethernet on FritzOS, but now the Results are just terrible 
[image]
I looked at the Internet for an solution, some people said that SQM should help but it hasn't.

These are my current wan settings:
[image]
As default the device was just dsl0, This broke everything then with dsl0.7 i had an Internet connection.

So people said the it maybe an Code issue and not an Config issue so im here.

@Djfe 

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM FRITZ!Box 7430

### Image kind

Official downloaded image

### Steps to reproduce

See that you have an bad internet connection

### Actual behaviour

I have an bad Internet connection and an high ping
(about 15Mbit/s download and ~1Mbit/s upload)

### Expected behaviour

To have an good Internet connection like with FritzOS
(50Mbit/s download and 10Mbit/s upload)

### Additional info

The results are from an ethernet connection, but WIFI is slow too

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-03:
Try this https://github.com/openwrt/openwrt/issues/13379#issuecomment-1911833638

In principle duplicate of (and full of maybe-workarounds) https://github.com/openwrt/openwrt/issues/10224

Snowiiii, 2024-02-03:
> Try this [#13379 (comment)](https://github.com/openwrt/openwrt/issues/13379#issuecomment-1911833638)
> 
> In principle duplicate of (and full of maybe-workarounds) #10224

gives an error
[image]

brada4, 2024-02-03:
You need to switch route to ip address of pppoe server from pppoo-wan to dsl0.7
Either way the problem in 10224 is still not understood well enough to fix.

Snowiiii, 2024-02-03:
> Either way the problem in 10224 is still not understood well enough to fix.

This need to change

brada4, 2024-02-03:
Any success with swapping interface of route to ppp server? 
i.e
```
ip -4 ro sh
<try to identify one individual route attached to pppoe-wan>
ip ro del "the full text"
ip ro add "swap pppoe-wan to dsl0.7"
service firewall restart ; sleep 61 ; service firewall restart
```

brada4, 2024-02-03:
Also `ping -I pppoo-wan ppoe-server-ip` vs `-I dsl0.7` may show some latency weirdness.

Snowiiii, 2024-02-03:
i dont understand

brada4, 2024-02-03:
There is a specific route to your provider's PPPoE server.
It is (mistakenly|uncommonly) assigned via pppoe tunnel, not via physical interface. 
```
# ip -4 ro sh
default via 192.0.2.1 dev pppoe-wan  src 192.0.2.2
10.11.12.13  dev pppoe-wan  scope link # !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```

You need to replace device in that specific route. Then reload firewall twice with pause to assure any offload states (can be) offloaded again.

Snowiiii, 2024-02-03:
> There is a specific route to your provider's PPPoE server. It is (mistakenly|uncommonly) assigned via pppoe tunnel, not via physical interface.
>
> ```
> # ip -4 ro sh
> default via 192.0.2.1 dev [...]

looks like this
[image]
what now?

brada4, 2024-02-03:
The second route needs to be replaced:
```
ip route del "84.59.211.1 dev pppoe-wan scope link src 2.204.253"
ip route add "84.59.211.1 dev dsl0.7 scope link"
```

Snowiiii, 2024-02-03:
replaced with what ?, now it says: RTNETLINK anseers: No such device

brada4, 2024-02-03:
so check DSL device name in 'ip addr'

Snowiiii, 2024-02-03:
``` 
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1508 qdisc fq_codel state UNKNOWN qlen 1000
    link/ether 96:14:f8:bb:16:d2 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9414:f8ff:febb:16d2/64 scope link 
       valid_lft forever preferred_lft forever
3: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 7c:ff:4d:49:c2:c0 brd ff:ff:ff:ff:ff:ff
4: lan4@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 7c:ff:4d:49:c2:c0 brd ff:ff:ff:ff:ff:ff
5: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 7c:ff:4d:49:c2:c0 brd ff:ff:ff:ff:ff:ff
6: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether 7c:ff:4d:49:c2:c0 brd ff:ff:ff:ff:ff:ff
10: dsl0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 7c:ff:4d:49:c2:c3 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7eff:4dff:fe49:c2c3/64 scope link 
       valid_lft forever preferred_lft forever
29: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 7c:ff:4d:49:c2:c0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd75:95dc:4e62::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::7eff:4dff:fe49:c2c0/64 scope link 
       valid_lft forever preferred_lft forever
30: dsl0.7@dsl0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 7c:ff:4d:49:c2:c3 brd ff:ff:ff:ff:ff:ff
31: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN qlen 3
    link/ppp 
    inet 2.204.169.253 peer 84.59.211.1/32 scope global pppoe-wan
       valid_lft forever preferred_lft forever
32: phy0-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 00:02:03:04:05:06 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::202:3ff:fe04:506/64 scope link 
       valid_lft forever preferred_lft forever
```

brada4, 2024-02-03:
Maybe without tickmarks indeed.

Snowiiii, 2024-02-03:
I will test this tomorrow, my mom is a mad that the Internet stops working haha

Snowiiii, 2024-02-03:
> Maybe without tickmarks indeed.
> 
> ```
> ip route del 84.59.211.1 dev pppoe-wan scope link src 2.204.253
> ip route add 84.59.211.1 dev dsl0.7 scope link src 2.204.253
> ```
> 
> EDIT: since no IP on the interface it needs SRC IP specified in route.

what exactly this does ?, or why it should help

brada4, 2024-02-03:
It changes route to pppoe concentrator to physical interrface in a manner similar to debian pppoeconf

Snowiiii, 2024-02-04:
Now both commands are failing:

[image]

Snowiiii, 2024-02-04:
> The second route needs to be replaced:
> 
> ```
> ip route del "84.59.211.1 dev pppoe-wan scope link src 2.204.253"
> ip route add "84.59.211.1 dev dsl0.7 scope link"
> ```

so the first command worked here, the secound not

brada4, 2024-02-04:
You can restore first route.
Can work without src ip, hopefully takes one from pppoewan

Snowiiii, 2024-02-04:
> You can restore first route. Can work without src ip, hopefully takes one from pppoewan

i still don't understand, You must consider that i have openwrt since a few days and not much fimiliar with Networking at all

Snowiiii, 2024-02-04:
I saw that for some reason dsl0.7 is disabled while dsl0 is enabled, Does it make a diffrence or is it right ?
[image]

brada4, 2024-02-04:
It is likely brought UP along pppoe interface.
but still the route spec looks invalid.

Snowiiii, 2024-02-04:
> It is likely brought UP along pppoe interface. but still the route spec looks invalid.

i enabled dsl0.7 instead but nothing changed
[image]

Snowiiii, 2024-02-04:
Hey i think i found the issue, I looked at https://forum.openwrt.org/t/fritz-7412-slow-internet-connection/108306/9
and they said its an CPU issue and i guess it is
[image]

brada4, 2024-02-04:
`timeout 5 ubus monitor | tee /tmp/ubus.log`
Try to figure out what is being done in tight loop
Serms dhcp requests are sent to pppoe-wan at max speed.
since you havre ip from ppp it is safe to kill dhcp client and see if cpu usage drops(or in general disable dhcp on ppp wan connection, it gets ip from ppp)

brada4, 2024-02-04:
timeout == coreutils-timeout
just to kill outputting process even if shell is slowbto follow

Snowiiii, 2024-02-04:
`timeout: not found`

> `timeout 5 ubus monitor | tee /tmp/ubus.log` Try to figure out what is being done in tight loop Serms dhcp requests are sent to pppoe-wan at max speed. since you havre ip from ppp it is safe to kill dhcp client and see if cpu usage drops(or in general disable dhcp on ppp wan connection, it gets ip from ppp) PS impressed by your no-knowledge of network.

Snowiiii, 2024-02-04:
> `timeout 5 ubus monitor | tee /tmp/ubus.log` Try to figure out what is being done in tight loop Serms dhcp requests are sent to pppoe-wan at max speed. since you havre ip from ppp it is safe to kill [...]

gives:
```
root@Baumhaus:~# timeout 5 ubus monitor | tee /tmp/ubus.log
-> 48300086 #00000003         status: {"status":0}
Terminated
``

danpawlik, 2024-02-04:
Just curious: did you try to make factory reset and restore config from a backup if there would be still an issue? I know that's not a fix or solution...

Snowiiii, 2024-02-04:
> Just curious: did you try to make factory reset and restore config from a backup if there would be still an issue? I know that's not a fix or solution...

Why i should ?

brada4, 2024-02-04:
Run `ps ww| grep dhcp`
You get something like
`12345 root      1234 S    udhcpc -p /var/run/udhcpc-wan.pid -s /lib/netifd/dhcp.script -f -t 0 -i wan -x hostname:OpenWRT -C -R -O`
note the `-i wan` part - the interface DHCP client is querying.

Now kill the process
`kill -9 12345`

Measure "speed"
(dhcp client may come back)

`/etc/config//network` has section regarding dhcp client.
```
config interface 'wan'
        option device 'wan'
        option proto 'dhcp'
```
change to
`option proto 'none'`
to disable DHCP client. You dont need DHCP client as IP is acquired from pppd.

Snowiiii, 2024-02-04:
> Run `ps ww| grep dhcp`
> You get something like
> `12345 root 1234 S udhcpc -p /var/run/udhcpc-wan.pid -s /lib/netifd/dhcp.script -f -t 0 -i wan -x hostname:OpenWRT -C -R -O`
> note the `-i wan` part - the interface DHCP client is querying.

i dont have an udhcpc process

[image]

brada4, 2024-02-04:
I see it in htop picture against `-I pppoe-wan`
Try to find any chance it starts by `/dhcp` in /etc/config/network.... And change to none.

Snowiiii, 2024-02-04:
in `/etc/config/network.` my wan6 is the dhcpv6 client, I have also wan thats is pppoe
[image]

brada4, 2024-02-04:
Weird high-cpu dhcpv4 client is in your htop
If you do not get ipv6 address you can disable (dont start at boot) the wan6.
Does not seem either is at fault in UBUS CPU usage.

Try to trace ubus for longer. Either it is infinite loop in itself or something punching same useless request at it less often than every 5s.
ubus is supposed to be low overhead ipc.

Snowiiii, 2024-02-04:
I had tried around dhcpv4 and dhcpv6 but it had no difference. But ubus is not the Problem i looked at htop again an ubus have basicly no CPU usage

brada4, 2024-02-04:
Strange you cannot divert route. No point in qos if bw limit is local to router.

danpawlik, 2024-02-04:
> > Just curious: did you try to make factory reset and restore config from a backup if there would be still an issue? I know that's not a fix or solution...
> 
> Why i should ? And i just made an fresh install 3 days ago, I don't see an point for an reset

In that case there is no reason to reset to defaults.

Snowiiii, 2024-02-04:
> Strange you cannot divert route. No point in qos if bw limit is local to router.

how what should i do now ?

brada4, 2024-02-04:
Try to add arbitrary route via dev dsl0.7 like
`ip route add 0.1.2.3 dev dsl0.7 src 0.2.3.4`

Snowiiii, 2024-02-04:
> Try to add arbitrary route via dev dsl0.7 like `ip route add 0.1.2.3 dev dsl0.7 src 0.2.3.4`

root@Baumhaus:~# ip route add 0.1.2.3 dev dsl0.7 src 0.2.3.4
ip: RTNETLINK answers: Invalid argument

brada4, 2024-02-04:
See this says vlan 132....
https://forum.vodafone.de/t5/Archiv-Internet-Ger%C3%A4te/Eigene-Fritzbox-7560-nutzen-PPPoE-Pakete/m-p/1708471

Snowiiii, 2024-02-04:
> See this says vlan 132.... https://forum.vodafone.de/t5/Archiv-Internet-Ger%C3%A4te/Eigene-Fritzbox-7560-nutzen-PPPoE-Pakete/m-p/1708471

Dosn't work, no connection at all. The Post is a bit older and back in the day this was a thing the Every Provider (In Germany) would have their own VLAN id, But today the default is ID 7, And should work for every provider at least in germany

Snowiiii, 2024-02-04:
So slowly i more and more coming to the conclussion that its not an Config error, Its something to do with the Code, Btw PfSense has somewhat the same issue: https://www.youtube.com/watch?v=jpdQ9cstvxU, There are many people saying thats in upstream issue and the 100% CPU usage is very sus.

Snowiiii, 2024-02-04:
btw something weird too is that when uploading the CPU seems fine but i still have an bottleneck and only getting ~1Mbit/s

brada4, 2024-02-04:
Another (not very bright) idea:
`speedtest-netperf.sh -H netperf-eu.bufferbloat.net`

yes, low cpu seems like delay accumulating in mystery place hard to locate because it does not eat up cpu.

you may try htop show kernel threads in case you dont see any usage by user processes.

Snowiiii, 2024-02-04:
Htop with some more details while running speed test:
[image]

Snowiiii, 2024-02-04:
idk why but in the last time CPU is not maxing out anymore but is still very high, I changed nothing

brada4, 2024-02-04:
magenta bar is softirq on 1st cpu. nstat and tc -s qdisc may shlw packet drop points. It says cpu too weak for sqm, likely even needing firewall offloads

Snowiiii, 2024-02-04:
> idk why but since a while ago CPU is not maxing out anymore but is still very high, I changed nothing

nvm i guess haha, 
I installed `net-perf`, thats what i got:
[image]

brada4, 2024-02-04:
I mean the exact script speedtest-netperf against public server.
You run client on 1 cpu server on other, they get 50MB/s around over localhost. 100x better than slow internet.

Snowiiii, 2024-02-04:
So i currently running netperf over my router, Should i run it over my PC ?, These are the results:

```
root@Baumhaus:~# netperf -H netperf-eu.bufferbloat.net
MIGRATED TCP STREAM TEST from 0.0.0.0 (0.0) port 0 AF_INET to demo.tohojo.dk (193) port 0 AF_INET : demo
Recv   Send    Send                          
Socket Socket  Message  Elapsed              
Size   Size    Size     Time     Throughput  
bytes  bytes   bytes    secs.    10^6bits/sec  

131072  16384  16384    11.13       1.45   
```
I got 2 ssh sessions one for htop and the other for netperf

brada4, 2024-02-04:
Full speedtest-netperf.sh script from synonymous package. I.e if it gets radically faster than what waveform sees over nat.

brada4, 2024-02-04:
(if the speeds/latencies are same just flash back to fritzos and let your mum enjoy internets)

brada4, 2024-02-05:
Please try speedtest-netperf script,  if it yelds no improvement over current lan speeds issue to subscribe while hetting back vendor fw is https://github.com/openwrt/openwrt/issues/10224
As you see the$e many tweaks been t$ied but non is unive$sally good and thus corrrect

brada4, 2024-02-05:
To shake slow code paths you may create br-wan containing either dsl0 or dsl0.7 
Another is to use alternative vlan encap method akin docker-typical vxlan https://enterprise-support.nvidia.com/s/article/Configuring-VXLAN-Encap-Decap-Offload-Using-tc
The rest is pretty much same as most common gige-gige NAT setup.

Snowiiii, 2024-02-05:
I think we need an OpenWRT coder here, Someone who can Help to Profile this stuff, Code Profiling and especially Network Profiling is a not funny Job, But im also tired of just trying random stuff, Maybe we should ping someone ?

Snowiiii, 2024-02-05:
Okay fine, Lets investigate further when you think, so i saw in htop that 'ksoftirqd' is eating up most of the CPU usage, a quick google search showed that its an common problem
e.g. https://portal.nutanix.com/page/documents/kbs/details?targetId=kA07V000000LUR3SAO

Snowiiii, 2024-02-05:
Btw consider that not until the Issue is fixed i can't really play Multiplayer Games anymore or Watch Youtube. So i try my best to get this fixed, Rolling back to FritzOS is not an option too me

[60 more replies]