openwrt/openwrt #14546: kernel: status of 6.1 version

hauke, 2024-02-03:
### Describe the bug

Status of Linux kernel 6.1 migration in master:

 - [ ] airoha
 - [ ] apm821xx
   - testing
 - [ ] archs38
 - [x] armsr
 - [ ] at91
 - [ ] ath25
 - [ ] ath79
   - testing
 - [x] bcm27xx
 - [ ] bcm47xx
 - [ ] bcm4908
   - testing
 - [ ] bcm53xx
   - testing
 - [ ] bcm63xx
 - [x] bmips
 - [x] gemini
 - [ ] generic
 - [ ] imx
   - testing
 - [x] ipq40xx
 - [x] ipq806x
 - [x] ixp4xx
 - [ ] kirkwood
   - testing
 - [ ] lantiq
 - [ ] layerscape
   - testing
 - [x] malta
 - [ ] mediatek
   - testing
 - [ ] mpc85xx
 - [ ] mvebu
   - testing
 - [ ] mxs
   - testing
 - [x] octeon
 - [ ] octeontx
 - [ ] omap
 - [ ] oxnas
 - [ ] pistachio
 - [ ] qoriq
 - [x] qualcommax
 - [ ] ramips
 - [ ] realtek
 - [x] rockchip
 - [x] sifiveu
 - [x] sunxi
 - [ ] tegra
 - [ ] uml
   - testing
 - [x] x86
 - [ ] zynq

### OpenWrt version

24.x

### OpenWrt release

24.x

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xback, 2024-02-03:
@hauke 
I have a large update for target **Tegra** pending [here](https://git.openwrt.org/?p=openwrt/staging/xback.git;a=shortlog;h=refs/heads/nvidia_jetson).
- bump to 6.1
- split to subtargets
- adds nvidia jetson board support

It already received a lot if testing offlist through nvidia channels.

I'll finish this up asap.

svanheule, 2024-02-03:
@hauke
Testing 6.1 for realtek is mostly ready:
https://git.openwrt.org/?p=openwrt/staging/svanheule.git;a=shortlog;h=refs/heads/realtek/testing-6.1

namiltd, 2024-02-04:
In reference to http://lists.openwrt.org/pipermail/openwrt-devel/2024-February/042204.html maybe it would be worth creating a similar list for the 6.6 kernel?

Noltari, 2024-02-05:
Speaking of bcm63xx I don't want to maintain it any further because it's such a PITA considering that there's no full device tree support, so I propose to drop it.
I know that a lot of bcm63xx devices aren't supported on bmips yet, but all of them could be added again in the future if users are still interested on them...

remittor, 2024-02-07:
Patch [arm-add-cmdline-override](https://github.com/openwrt/openwrt/blob/main/target/linux/ipq806x/patches-6.1/900-arm-add-cmdline-override.patch) is missing on many platforms!
Status OK: ipq806x, mediatek/filogik, mpc85xx

dhewg, 2024-02-08:
omap @ https://github.com/openwrt/openwrt/pull/14582

namiltd, 2024-02-10:
PR https://github.com/openwrt/openwrt/pull/12738 is merged.

dhewg, 2024-02-14:
ramips got merged

ptpt52, 2024-02-16:
with kernel 6.1 WED reboot/crash on mt7981
```
insmod mt7915e wed_enable=Y
[   74.734159] mt798x-wmac 18000000.wifi: attaching wed device 0 version 2
[   74.787920] platform 15010000.wed: MTK WED WO Firmware Version: DEV_000000, Build Time: 20221208202138
[   74.797280] platform 15010000.wed: MTK WED WO Chip ID 00 Region 3
(hang...and...reboot)
```

raenye, 2024-02-16:
What is needed to make 6.1 the current version on `apm821xx`? I've been running 6.1 fine on my WD my book live since September or so.

castillofrancodamian, 2024-02-17:
> * [x]  ramips
>   
>   * testing: [ramips: add 6.1 support #12738](https://github.com/openwrt/openwrt/pull/12738)

I think the check is incorrect since kernel 6.1 has not been enabled by default.

jeffguillaume, 2024-02-17:
[mediatek has finished the move to 6.1](https://github.com/openwrt/openwrt/pull/14140), apparently

romanovj, 2024-02-18:
threaded backlog not working on 6.1 kernel, isn't it? 

echo 1 > /proc/sys/net/core/backlog_threaded

CHKDSK88, 2024-02-19:
mpc85xx: https://github.com/openwrt/openwrt/pull/14678

Neustradamus, 2024-02-23:
@hauke: Thanks to have listenned to me! :)
- https://github.com/openwrt/openwrt/issues/14262

It is important to use latest Kernel LTS: 6.6.x.

And please like LibreOffice, plan RC versions with the real official month release date.
LibreOffice 24.2 has been released the last day of January: 2024-01-31:
- https://blog.documentfoundation.org/blog/2024/01/31/libreoffice-24-2/

Mailing list message:
- https://lists.openwrt.org/pipermail/openwrt-devel/2024-February/042204.html

edrikk, 2024-02-24:
Not sure how / if it’s tracked, but although MediaTek is checked, the [AX6S](https://github.com/openwrt/openwrt/commit/dadad6bb731ff1025125c619da7ccc196ad6ea01) and [rfb1](https://github.com/openwrt/openwrt/commit/6e2962d4c5487d60faf5554c4bfe1d5deaa3b4ad) builds are being skipped and require attention.

ttc0419, 2024-02-29:
> > Not sure how / if it’s tracked, but although MediaTek is checked, the [AX6S](https://github.com/openwrt/openwrt/commit/dadad6bb731ff1025125c619da7ccc196ad6ea01) and [rfb1](https://github.com/openwrt/openwrt/commit/6e2962d4c5487d60faf5554c4bfe1d5deaa3b4ad) builds are being skipped and require attention.
> 
> [#14140 (comment)](https://github.com/openwrt/openwrt/pull/14140#issuecomment-1949634229)

Is that mean AX6S is no longer supported in future releases?

namiltd, 2024-03-01:
I am asking for the same topic for kernel v6.6.

CHKDSK88, 2024-03-06:
Layerscape is switched: https://github.com/openwrt/openwrt/commit/73f99b6849f3d0c40b1da6a5b35b69578b72d7d1
Kirkwood have PR for switching: https://github.com/openwrt/openwrt/pull/14803

khanjui, 2024-03-16:
Why SNAPSHOTS aren't building for ramips/mt7621 for more than 4 days (after switching to 6.1(?))

csharper2005, 2024-03-16:
> Why SNAPSHOTS aren't building for ramips other than mt7620 for more than 4 days (after switching to 6.1(?))
> 
> https://buildbot.openwrt.org/images/#/builders/61/builds/113

Some devices exceed their limits:
```
	Line 23718:     WARNING: Image file /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ubnt_edgerouter-x-kernel.bin is too big: 3175504 > 3145728
	Line 23729: if [ -e /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin -a "$(stat -c%s /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin)" -lt "3145728" ]; then echo '21001:7' > /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar.compat; tar -cf /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar --transform='s/^.*/compat/' /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar.compat; tar -rf /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar --transform='s/^.*/vmlinux.tmp/' /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin; /builder/shared-workdir/build/staging_dir/host/bin/mkhash md5 /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin > /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar.md5; tar -rf /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar --transform='s/^.*/vmlinux.tmp.md5/' /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-initramfs-factory.tar.md5; echo "dummy" > /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/o ...
	Line 23732: WARNING: initramfs kernel image too big, cannot generate factory image (actual 5840174; max 3145728)
	Line 23806:     WARNING: Image file /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ubnt_edgerouter-x-sfp-kernel.bin is too big: 3175658 > 3145728
	Line 23826: if [ -e /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-kernel.bin -a "$(stat -c%s /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-kernel.bin)" -lt "3145728" ]; then echo '21001:7' > /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar.compat; tar -cf /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar --transform='s/^.*/compat/' /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar.compat; tar -rf /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar --transform='s/^.*/vmlinux.tmp/' /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-kernel.bin; /builder/shared-workdir/build/staging_dir/host/bin/mkhash md5 /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-kernel.bin > /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar.md5; tar -rf /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar --transform='s/^.*/vmlinux.tmp.md5/' /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp-initramfs-factory.tar.md5; echo "dummy" > /builder/shared-workdir/build/build_dir/targe ...
	Line 23828: WARNING: initramfs kernel image too big, cannot generate factory image (actual 5840327; max 3145728)
	Line 34122: file-system partition too big (more than 2858911 bytes): Success
	Line 34123: file-system partition too big (more than 2818048 bytes): Success
	Line 34130:     WARNING: Image file /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-tplink_re350-v1-squashfs-sysupgrade.bin is too big:  > 6160384

```

namiltd, 2024-03-16:
Take a look at this: https://github.com/openwrt/openwrt/pull/14713
Use kernel_bump.sh to avoid losing file history.

castillofrancodamian, 2024-03-17:
@csharper2005 And what is happening that they don't fix it?

csharper2005, 2024-03-17:
@castillofrancodamian anyone can figure it out, fix it and open a pull request. Therefore, we can ask ourselves the same question: why haven't we fixed it yet?

castillofrancodamian, 2024-03-17:
@csharper2005 If anyone can fix it, why isn't it at this point?

csharper2005, 2024-03-17:
> @csharper2005 If anyone can fix it, why isn't it at this point?

I think this is obvious, but let's look at the possible reasons:
1. Lack of resources (motivation, knowledge, time, fast (or even any) build environment);
2. No one opened an issue and collected diagnostic data here https://github.com/openwrt/openwrt/issues;
3. No one indicated that this was important or posted on IRC;
4. There are more important tasks to do (according to priorities);
5. Passive behavior (it's easier to wait for someone else to fix it);
6. Passive-aggressive behavior (why should I fix everything);
7. etc

wigyori, 2024-03-18:
The pistachio target doesn't seem to have too much traction, but I'll see if I can get together a 6.1 kernel for it.

schuettecarsten, 2024-04-05:
Maybe this pinned issue should be updated to a new "kernel: status of 6.6 version"?

namiltd, 2024-04-11:
Take look at: https://github.com/openwrt/openwrt/commit/076f945dfbc8afa6ced7c5f67d95ab27383dd1e4
It appears that kernel 6.6 has been selected as the default for the next OpenWRT release.

namiltd, 2024-04-17:
> Maybe this pinned issue should be updated to a new "kernel: status of 6.6 version"?

https://github.com/openwrt/openwrt/issues/15192

ascendbeing, 2024-04-18:
I flashed an ea3500 to the new 6.1/apk styled packages around mar 31. I was going to add a kmod for a USB wifi dongle. But the kernel from mar 31 isn't still on the kmods dir. There are newer 6.1.82s but not the same one.
So I tried to sysupgrade today, and to my shock, I got the message, again, that I would have to do the fw_setenv commands again with a factory image. It claimed that I was still on a 2.0 image, not a 3.0 image.
Is that normal? Or was there another image upgrade after mar 31? For Kirkwood and specifically for arm xscale based ea3500/ea4500/e4200v2.
I didn't want to make an issue because the bot is sensitive and I don't have the router up rn to grab the specific log but you probably know what I'm talking about.
It may be worth mentioning that this router is dual partition so maybe it's the other partition that's not the new format? Maybe it would work if I advanced reboot to the other partition, then sysupgrade how I was trying to from that partition?
I think I answered my own question with that last paragraph, but I'll leave this up for anyone who might run into the same thing. Or if I'm missing something. Most likely, you have to do the process twice: 1x for 1 partition, then do the same thing while in your new boot environment, to have them both on the updated format. After that it should be smooth sailing, right?
My only concern is if I'm missing something here.

sch-m, 2024-04-22:
I've add a new PR for lantiq 6.1 support: #15233
When this is merged, I will have a look at the 6.6 support.

ascendbeing, 2024-05-09:
BTW, i didn't end up having answered my own question: it appears to want the factory firmware process to be done every single time you upgrade after you've upgraded from 5.15 or older to 6.1.
that is to say, 6.1 on Kirkwood , at least for ea3500, and probably its cousins, ea4500+e4200v2, have a major issue on 6.1.
if you're on 6.1, or migrating to 6.1, you can't do a straight sysup. it will whinge that you need to edit certain boot parameters and use factory fw.
it does end up working, but even after you've done so, and you're on the new one, it wants you to do it every single time for the rest of time.
is the warning being overly cautious? that is to say, is it triggering when it no longer needs to?
is there a compatibility issue with the advanced reboot software? i have that installed, and I've seen it does weird stuff like mounting the other partition, when you visit the advanced reboot page.
that's all i can think of as a largely non-coder. if i knew that's what upgrading to 6.1 would've gotten me to, I would've left things at 5.15 on the ea3500, cause it didn't claim every single time that the alleged format vs what it alleges the format i currently have are incompatible and require running and double-checking some fw_setenv cmds, before upgrading, and even after having done so like 3x so far, it always requires me to do that.

also, i still dont know why running an ath9k_htc USB AP on the ea3500 renders 1/2 of the built-in wifi devices useless. openwrt acts like they all work. i can get 2.4sta to work, and 2.4ap with the ath9k_htc to work. but 5ap, no matter what i have the settings set to, different tx powers, diff ssid, etc, no device is ever able to find it or connect. do i need to set a different mac? i never changed them

ascendbeing, 2024-05-31:
after removing the ath9k_htc i still couldn't use 5ghz. I factoried to 23.05 and 5ghz AP works as expected. when i was using snapshots on ea3500, i was always restoring my config after the flashes. maybe if i was using a fresh config,; 1. 5ghz would've worked 2. i wouldn't be getting the nags to do -F -n

i may explore this further but on the older fw things work more or less as expected although i believe i see more frequently ```tx rings drained ```

Neustradamus, 2024-09-28:
Dear @OpenWrt, @hauke: Can you speak with slh to reopen the forum post, I can not do an answer on it that I have published some hours after:
- https://forum.openwrt.org/t/openwrt-version-naming-the-first-official-stable-version-must-have-the-good-year-month/211523

I am not always on computer.

Some people have not understand, the year.month is not currently the first final stable version, but a previous year.month when it was unstable.

Thanks in advance.

closed 2024-10-04 by Noltari