openwrt/openwrt #14548: Kernel Panic when wifi station disconnects ZyXEL EX5700 (Telenor)

ttyridal, 2024-02-04:
### Describe the bug

Kernel Panic when wifi station disconnects ZyXEL EX5700 (Telenor)

### OpenWrt version

r23963-ef8e2f66f5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

ZyXEL EX5700 (Telenor)

### Image kind

Self-built image

### Steps to reproduce

1) connect station (in this case a galaxy s22) to accesspoint (openwrt)
2) disconnect station

### Actual behaviour

Mon Sep 18 10:53:14 2023 daemon.notice hostapd: phy2-ap0: AP-STA-DISCONNECTED 02:xx:xx:xx:xx:xx
Mon Sep 18 10:53:15 2023 daemon.info hostapd: phy2-ap0: STA 02:xx:xx:xx:xx:xx IEEE 802.11: disassociated
[   98.108342] Unable to handle kernel read from unreadable memory at virtual address 0000000000000024
[   98.117383] Mem abort info:
[   98.120174]   ESR = 0x0000000096000005
Mon Sep 18 10:53[   98.123906]   EC = 0x25: DABT (current EL), IL = 32 bits
:15 2023 kern.al[   98.130597]   SET = 0, FnV = 0
ert kernel: [   [   98.135019]   EA = 0, S1PTW = 0
98.108342] Unabl[   98.139534]   FSC = 0x05: level 1 translation fault
e to handle kern[   98.145775] Data abort info:
el read from unr[   98.150028]   ISV = 0, ISS = 0x00000005
eadable memory a[   98.155229]   CM = 0, WnR = 0
t virtual addres[   98.159570] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000044f59000
s 00000000000000[   98.167371] [0000000000000024] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
24
Mon Sep 18 1[   98.177432] Internal error: Oops: 96000005 [#1] SMP
[   98.183668] Modules linked in: xt_connlimit pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_conncount xt_state xt_helper xt_conntrack xt_connmark xt_connbytes xt_CT pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_recent xt_policy xt_multiport xt_mark xt_mac xt_limit xt_esp xt_comment xt_TCPMSS xt_LOG slhc nfnetlink nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ah ip_tables crc_ccitt compat crypto_safexcel ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4
[   98.183811]  xfrm_user xfrm_ipcomp af_key xfrm_algo crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic seqiv md5 echainiv des_generic libdes chacha20poly1305 cbc authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ubootenv_nvram gpio_button_hotplug usbcore usb_common
[   98.298256] CPU: 0 PID: 1872 Comm: hostapd Not tainted 5.15.131 #0
[   98.304417] Hardware name: ZyXEL EX5700 (Telenor) (DT)
[   98.309536] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   98.316475] pc : 0xffffffc000b3aee8 [mt76@000000002046597e+0xf000]
[   98.322647] lr : 0xffffffc000b3aec8 [mt76@000000002046597e+0xf000]
[   98.328807] sp : ffffffc00aae36d0
[   98.332105] x29: ffffffc00aae36d0 x28: ffffff80034c0000 x27: ffffffc00aae3dd0
[   98.339220] x26: ffffff80000af880 x25: ffffffc008b56440 x24: 0000000000000001
[   98.346336] x23: ffffffc000b3b184 x22: 0000000000000000 x21: ffffff80042eace8
[   98.353451] x20: ffffffc00aae3718 x19: ffffff8003f82040 x18: 0000000000000000
[   98.360566] x17: 0000000000001bc0 x16: ffffffc008e35000 x15: 0000000000000de0
[   98.367682] x14: ffffff80042eae50 x13: ffffff80042eae50 x12: 0000000000000000
[   98.374798] x11: 0000000000000000 x10: ffffff80042eae58 x9 : ffffff80042eae50
[   98.381914] x8 : 0000000000000000 x7 : ffffff8003f86040 x6 : 0000000000000000
[   98.389029] x5 : 0000000000000031 x4 : 0000000000000000 x3 : 0000000000000001
[   98.396144] x2 : 0000000000000000 x1 : ffffff80034c0000 x0 : 0000000000000024
[   98.403260] Call trace:
[   98.405692]  0xffffffc000b3aee8 [mt76@000000002046597e+0xf000]
[   98.411507]  0xffffffc000b3b184 [mt76@000000002046597e+0xf000]
[   98.417322]  0xffffffc000b3b25c [mt76@000000002046597e+0xf000]
[   98.423135]  0xffffffc000b5b864 [mt7915e@00000000ea7a3ee8+0x21000]
[... 269 lines cut ...]
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"
CONFIG_USE_LLVM_BUILD=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/23.05.2"
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_WOLFSSL_HAS_NO_HW=y
# CONFIG_VERSION_CODE_FILENAMES is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-04:
not related to wifi but seems you use nft conncount togehter with xt conntrack. Can you check if  `nft list ruleset` shows no xt bridge rules?

ttyridal, 2024-02-06:
@brada4  Not exactly sure what you're asking. there is no trace of conncount or conntrack in nft list ruleset  . This is a near "out of the box" configuration. only renamed the wifi's and opened ports for ipsec

[ruleset.txt](https://github.com/openwrt/openwrt/files/14185575/ruleset.txt)

brada4, 2024-02-06:
Conntrack is called `ct`
what i was worried is accidental `xt` offload to old engine, seems nf-nat got pulled as dependency and no worries.

brada4, 2024-02-06:
Try (after boot)
```
echo N > /sys/module/mt7915e/parameters/wed_enable
```
other module parameters are at boot only. If it survives disconnets tell.

ttyridal, 2024-02-06:
@brada4  that didn't help, unfortunately.

last good commit is 13b24340d7f2e11173 by the way, so something happened with commit ef8e2f66f56
@nbd168

brada4, 2024-02-07:
@nbd168

hsand, 2024-04-12:
I'm experiencing the same behavior with this router, tested with the latest release and snapshot.

brada4, 2024-04-12:
You may try disabling mt76 wed via module parameter.

hsand, 2024-04-12:
> You may try disabling mt76 wed via module parameter.

The wed parameter seems to be missing as an option for mt76, but I tried the command for the mt7915e as mentioned above and no change in behavior. :(

```
root@OpenWrt:~# ls /sys/module/mt76
coresize   holders    initsize   initstate  notes      refcnt     sections   taint      uevent
```

brada4, 2024-04-12:
it is of the driver in ethtool -i phy0-ap0

hsand, 2024-04-12:
> it is of the driver in ethtool -i phy0-ap0

Not sure I completely understand, but here's the output of the command

 ```
 root@OpenWrt:~# ethtool -i phy0-ap0
driver: mt7915e
version: 5.15.150
firmware-version: DEV_000000-20230202143332
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
 ```

brada4, 2024-04-12:
`/sys/module/mt7915e/parameters`

lukasz1992, 2024-07-04:
https://github.com/lukasz1992/openwrt/commit/f0affbb50ed191c2d33ba48a2554893a98507346
https://github.com/lukasz1992/openwrt/blob/bc641e7998f445752a9368146f0e567bcdd1ed00/package/kernel/mac80211/patches/subsys/991-wifi-mac80211-do-not-pass-AP_VLAN-vif-pointer-to-dri.patch
https://github.com/lukasz1992/openwrt/blob/bc641e7998f445752a9368146f0e567bcdd1ed00/package/kernel/mac80211/patches/subsys/993-wifi-mac80211-only-call-drv_sta_rc_update-for-upload.patch

try after those patches applied

pcslide, 2024-07-05:
@lukasz1992 
Were those two patches already merged in the latest snapshot?

peikelias, 2024-07-11:
Same here running the EX5700 on
**OpenWrt version**
23.05.3 (r23809-234f1a2efa)
**OpenWrt release**
Release

Unable to try @lukasz1992's patch links as they seem to be applied/removed. Will be trying latest snapshot if possible this weekend.

I also take any suggestions for having the issue resolved

lukasz1992, 2024-07-12:
23.05-SNAPSHOT contains them, or wait for 23.05.4

pcslide, 2024-07-12:
@lukasz1992  @peikelias Unfortunately,  this issue still exists in the latest snapshot.

bmork, 2024-07-12:
I can confirm this bug.  It is related to the 5GHz phy attached to the mt7986 SoC, I believe.  I have no client I can use for 6GHz testing, but it doesn't seem to affect 2GHz at least.

I added this debug print (too lazy to calculate the suspected dereferenced fields myself) to mt76_wcid_cleanup():
```	
dev_warn(dev->dev, "phy=%p, offsetof(tx_lock)=%zu, offsetof(pktid)=%zu\n",
		phy, offsetof(struct mt76_phy, tx_lock), offsetof(struct mt76_wcid, pktid));
```
and got

```
[   89.220004] mt798x-wmac 18000000.wifi: phy=0000000000000000, offsetof(tx_lock)=36, offsetof(pktid)=360
```
so that "virtual address 0000000000000024" is clearly the &phy->tx_lock dereference added by
https://github.com/openwrt/mt76/commit/f1e1e67d97d1e9a8bb01b59ab20c45ebc985a958

Simple workaround is adding	
```
if (!phy)
		return;
```

right before the
```
spin_lock_bh(&phy->tx_lock);
```

But that might be hiding some other bug here?  I have no clue why the phy is NULL in this case.

On the PCI attached 2GHz radio I got
`[   62.116403] mt7915e 0000:01:00.0: phy=000000004f4a303d, offsetof(tx_lock)=36, offsetof(pktid)=360`

which explains why that didn't crash.

bmork, 2024-07-12:
OK, so I changed the debug printk to
```
	dev_warn(dev->dev, "phy=0x%p, offsetof(tx_lock)=0x%zx, wcid->phy_idx=%d\n",
		phy, offsetof(struct mt76_phy, tx_lock), wcid->phy_idx);
```

and got
```
[  138.737215] mt798x-wmac 18000000.wifi: phy=0x0000000000000000, offsetof(tx_lock)=0x24, wcid->phy_idx=1
```

So the problem seems to be that wcid->phy_idx.  It is used as an index into the `phys` table:
```
	struct mt76_phy *phy = dev->phys[wcid->phy_idx];
```

But there is only one phy here.  The index should be 0. Wonder where that wcid->phy_idx comes from?

(too bad I can't test 6GHz where it should be 1).

bmork, 2024-07-12:
Right.  I think I found the explanation:

```
static bool mt7915_band_config(struct mt7915_dev *dev)
{
	bool ret = true;
	u32 sku = 0;

	dev->phy.mt76->band_idx = 0;

	if (is_mt798x(&dev->mt76)) {
		sku = mt7915_check_adie(dev, true);

		/*
		 * for mt7986, dbdc support is determined by the number
		 * of adie chips and the main phy is bound to band1 when
		 * dbdc is disabled.
		 */
		if (sku == MT7975_ONE_ADIE || sku == MT7976_ONE_ADIE) {
			dev->phy.mt76->band_idx = 1;
			ret = false;
		}
	} else {
		ret = is_mt7915(&dev->mt76) ?
		      !!(mt76_rr(dev, MT_HW_BOUND) & BIT(5)) : true;
	}

	dev_warn(dev->mt76.dev, "%s: ret=%d, band_idx=%d, sku=0x%x\n",
		__FUNCTION__, !!ret, dev->phy.mt76->band_idx, sku);
	return ret;
}
```

I believe the comment tries to tell us that we SHOULD expect band_idx == 1 when DBDC is disable on mt7986. Which is the case for this router.  The sku is 0xa.  I.e MT7976_ONE_ADIE.  So we do end up with band_idx = 1.

But the phy is still allocated as dev->phys[0].

There's a bit of a mess wrt the indexes here.  I guess the right thing to do is to use dev->phys[1] and let dev->phys[0] be NULL?  Pretty sure that will break something else, though....

pcslide, 2024-07-12:
@bmork  Can we try  to set dev->phy.mt76->band_idx = 0 and see what happens?  Though we have ADIE in our router(ZyXEL ex5700/ Acer w6), our mt7986 chip only connect to one single RF chip.

bmork, 2024-07-13:
I believe this is the proper fix:
https://github.com/bmork/mt76/commit/7d7eea43c696a302e653a94a6817edc0ea21208d

Tested and "Works For Me"

I'll see if I can find some time to submit that against mainline linux repo, which I believe is where the mt76 developement is supposed to happen.  But vacation... don't know when.

rotanid, 2024-07-28:
@ttyridal @hsand should be fixed in OpenWrt master branch snapshot builds since 2 weeks, please test with a snapshot build

peikelias, 2024-07-28:
> @ttyridal @hsand should be fixed in OpenWrt master branch snapshot builds since 2 weeks, please test with a snapshot build

Tested and working perfectly, no more issues with disconnecting from the radio.

sqyexu, 2024-08-31:
Can you share your EX5700 mtd3_Factory.bin? 86680889@qq.com

sqyexu, 2024-08-31:
> > 应该在 OpenWrt master 分支快照构建中修复 2 周后，请使用快照构建进行测试
> 
> 经过测试并运行良好，不再有断开无线电连接的问题。

Can you share your EX5700 mtd3_Factory.bin? [86680889@qq.com](mailto:86680889@qq.com)

sqyexu, 2024-08-31:
> 应该在 OpenWrt master 分支快照构建中修复 2 周后，请使用快照构建进行测试

Can you share your EX5700 mtd3_Factory.bin? [86680889@qq.com](mailto:86680889@qq.com)

sqyexu, 2024-08-31:
> 23.05-SNAPSHOT 包含它们，或等待 23.05.4

Can you share your EX5700 mtd3_Factory.bin? [86680889@qq.com](mailto:86680889@qq.com)

sqyexu, 2024-08-31:
> 遗憾的是，此问题仍然存在于最新快照中。
Can you share your EX5700 mtd3_Factory.bin? [86680889@qq.com](mailto:86680889@qq.com)

sqyexu, 2024-08-31:
> > 您可以尝试通过 module 参数禁用 mt76 wed。
> 
> wed 参数似乎缺少作为 mt76 的一个选项，但我如上所述尝试了 mt7915e 的命令，行为没有变化。:(
> 
> ```
> root@OpenWrt:~# ls /sys/module/mt76
> coresize   holders    initsize   initstate  notes      refcnt     sections   taint      uevent
> ```
Can you share your EX5700 mtd3_Factory.bin? [86680889@qq.com](mailto:86680889@qq.com)

bmork, 2024-08-31:
> Can you share your EX5700 mtd3_Factory.bin? [](mailto:86680889@qq.com)

Well, I can share mine.  But it is known faulty, lacking some of the 5GHz eeprom data, cause I only have is an early device sample. It mostly works anyway, only with 6dBm limits on all 5GHz channels.

I have confirmed that the disitributed devices have valid 5GHz eeproms.
[mtd3.gz](https://github.com/user-attachments/files/16824163/mtd3.gz)

AFAICS, this isn't related to this bug at all?  The bug is present whether or not the eeprom is valid, as long as you can bring up 5Ghz and have clients on it.

sqyexu, 2024-09-01:
EX5700 wireless imperfect, no perfect eeprom. Thank you for sharing. Thank you for coming to China. Currently, this device is available for sale in the Chinese market

sqyexu, 2024-09-01:
> 我相信这是正确的解决方法：[bmork/mt76@7d7eea4](https://github.com/bmork/mt76/commit/7d7eea43c696a302e653a94a6817edc0ea21208d)
> 
> 经过测试，“Works For Me”
> 
> 现在提交： https://lore.kernel.org/linux-wireless/20240713130010.516037-1-bjorn@mork.no/T/#u
https://github.com/openwrt/openwrt/issues/14548#issuecomment-2322845616

sqyexu, 2024-09-01:
> > 应该在 OpenWrt master 分支快照构建中修复 2 周后，请使用快照构建进行测试
> 
> 经过测试并运行良好，不再有断开无线电连接的问题。
Can you share your active EX5700 5G EEPROM? My earlier version, this issue is currently very difficult to fix 860680889@qq.com

hsand, 2024-10-12:
Just tested snapshot r27722-e00b6ba0c8 and it seems to be fixed 🥳

sqyexu, 2024-10-12:
> 刚刚测试了快照 r27722-e00b6ba0c8，它似乎已修复 🥳

你的无线数据还好吗

hsand, 2024-10-12:
> > 刚刚测试了快照 r27722-e00b6ba0c8，它似乎已修复 🥳
> 
> 你的无线数据还好吗

I tested 2.4/5/6 GHz, as far as I can tell all bands are working as expected.

sqyexu, 2024-10-14:
> I tested 2.4/5/6 GHz, as far as I can tell all bands are working as expected.

Can you share MTD3 wireless data eeprom.bin 860680889@qq.com

closed 2024-10-14 by nbd168

Doanduy09, 2024-12-15:
> EX5700 mtd3_Factory.bin

can you guide me to install mdt3, i don't know where to start... thanks

Doanduy09, 2024-12-15:
> > Can you share your EX5700 mtd3_Factory.bin?
> 
> Well, I can share mine: [mtd3.gz](https://github.com/user-attachments/files/16824163/mtd3.gz) But it is known faulty, lacking some of the 5GHz eeprom data, cause I only have is an early device sample. It mostly works anyway, only with 6dBm limits on all 5GHz channels.
> 
> I have confirmed that the disitributed devices have valid 5GHz eeproms.
> 
> AFAICS, this isn't related to this bug at all? The bug is present whether or not the eeprom is valid, as long as you can bring up 5Ghz and have clients on it.

can you guide me to install mdt3, i don't know where to start... thanks