openwrt/openwrt #14550: 18.06.9 mips_24kc Packages SHA256sum mismatch

astro, 2024-02-04:
### Describe the bug

I need to use the imagebuilder for a 4 MB flash device. Using nix-openwrt-imagebuilder I rely on valid checksums but I got a mismatch. Could someone please rebuild these files?

### OpenWrt version

r8077-7cbbab7246

### OpenWrt release

18.06.9

### OpenWrt target/subtarget

ar71xx/tiny

### Device

TP-Link TL-WR841N/ND v11

### Image kind

Official downloaded image

### Steps to reproduce

```
wget https://downloads.openwrt.org/releases/18.06.9/packages/mips_24kc/base/Packages
wget https://downloads.openwrt.org/releases/18.06.9/packages/mips_24kc/base/ath10k-firmware-qca9887-ct_2018-05-12-952afa49-1_mips_24kc.ipk
grep -e ath10k-firmware-qca9887-ct\$ -A 10 Packages|grep SHA256
```
SHA256sum: 531c205c557a993567bb02cfe3f97f227637cac841f93032dfee542883de88c1
```
sha256sum ath10k-firmware-qca9887-ct_2018-05-12-952afa49-1_mips_24kc.ipk
```
31b7660e4b485867b01b0de2018757f1d73c27c1349391ea54ce638dba4598e1  ath10k-firmware-qca9887-ct_2018-05-12-952afa49-1_mips_24kc.ipk

### Actual behaviour

SHA256 sum mismatch

### Expected behaviour

SHA256 sums should match

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-02-16:
18.06 is EoL so therefore not supported anymore

astro, 2024-02-16:
But I just need valid checksum files

astro, 2024-02-16:
I would like to point out that this issue has not much do with the precise OpenWrt release. The problem is that SHA256sums on downloads.openwrt.org become invalid out of a sudden.

Please reopen or just restore the matching files?

robimarko, 2024-02-17:
Wait, so you are telling this happens with any release?

astro, 2024-02-17:
No, just the files of 18.06.9. If nobody regenerates these sha256sums anymore but they've become invalid, has downloads.openwrt.org been compromised?

robimarko, 2024-02-17:
They are not regenerated AFAIK, there was probably an issue while the release was made.

As I mentioned, 18.06 is EOL and fully unsupported.

astro, 2024-02-17:
> while the release was made.

Unlikely, I've been running 4 MiB flash devices for years on that release.

Ansuel, 2024-02-17:
```
Package: ath10k-firmware-qca9887-ct
Version: 2018-05-12-952afa49-1
Depends: libc
Provides: ath10k-firmware-qca9887
Section: firmware
Architecture: mips_24kc
Installed-Size: 191441
Filename: ath10k-firmware-qca9887-ct_2018-05-12-952afa49-1_mips_24kc.ipk
Size: 192462
SHA256sum: d25f66d0e5bbfe54c3bb1b478e2e8bd2603317502b0a10dc214a83615c09acf8
Description:  Alternative ath10k firmware for QCA9887 from Candela Technologies.
 Enables IBSS and other features.  See:
 http://www.candelatech.com/ath10k-10.1.php
 This firmware conflicts with the standard 9887 firmware, so select only
 one.
```
From 
https://emn178.github.io/online-tools/sha256_checksum.html
`d25f66d0e5bbfe54c3bb1b478e2e8bd2603317502b0a10dc214a83615c09acf8`

Not following this honestly...

astro, 2024-02-17:
lol, I get d25f66d0e5bbfe54c3bb1b478e2e8bd2603317502b0a10dc214a83615c09acf8 too now!

Ansuel, 2024-02-17:
probably file got corrupted in transit. That's the use of hash :D