openwrt/openwrt #14551: DHCP/DNS打不开

sf2322571, 2024-02-05:
### Describe the bug

DHCP/DNS选项卡打不开 提示上面错误
[image]

### OpenWrt version

22.03

### OpenWrt release

22.03

### OpenWrt target/subtarget

22.03

### Device

x86

### Image kind

Official downloaded image

### Steps to reproduce

无

### Actual behaviour

无

### Expected behaviour

无

### Additional info

无

### Diffconfig

```text
无
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sf2322571, 2024-02-05:
openwrt-22.03 版本是5.10.208

brada4, 2024-02-05:
22.03.6 has kernel 5.10.201, please tell the story how you got your image upgraded.

Snowiiii, 2024-02-05:
Looks like a duplicate of https://github.com/openwrt/openwrt/issues/14552

brada4, 2024-02-05:
luci git tag is NOT from openwrt. Either install openwrt or report to fork firmware project.

sf2322571, 2024-02-05:
> 22.03.6 的内核为 5.10.201，请讲述一下您的映像升级过程。

git pull后 在编译就成这样了

sf2322571, 2024-02-05:
> 请重复固件选择器版本的问题（安装中文翻译和主题就可以了） 每个新软件包后检查网络/DHCP菜单是否仍然存在。

[image]

sf2322571, 2024-02-05:
[image]
无法载入

brada4, 2024-02-05:
Just use download release. It has full luci as opposed to parts scrambled together with own build.
Also check immortalwrt fork in place of creating your own