openwrt/openwrt #14554: [rpi4] Kernel crash and reboot after Internet connection since 6.1.75 kernel version bump

schuettecarsten, 2024-02-05:
### Describe the bug

On my Raspberry PI 4, I get a kernel exception that causes a reboot about  two seconds after the internet connection was established using PPPoE. Provider is Deutsche Telekom. This error occurs since kernel 6.1.75 bump in commit https://github.com/openwrt/openwrt/commit/2c363a070e3a007a2f4f5e12491676cbe670fe5a, reverting this commit fixes the issue. I have no idea what happens, the only thing I can see is the crash and then the reboot.

```
Unable to handle kernel access to user memory outside uaccess routines at virtual address 000000000000007a
Mem abort info:
  ESR = 0x0000000096000005
  EC = 0x25: DABT (current EL), IL = 32 bits
  SET = 0, FnV = 0
  EA = 0, S1PTW = 0
  FSC = 0x05: level 1 translation fault
Data abort info:
  ISV = 0, ISS = 0x00000005
  CM = 0, WnR = 0
user pgtable: 4k pages, 39-bit VAs, pgdp=0000000047866000
[000000000000007a] pgd=0800000049eee003, p4d=0800000049eee003, pud=0800000049eee003, pmd=0000000000000000
```

If it is neccessary to compile a special kernel version that will output more details, I can do that but I need some instructions how to do this. I've also deactivated some services like docker, sqm, 

### OpenWrt version

r25004-b463737826

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2024-02-05:
@schuettecarsten are you sure it's the kernel bump or not some "near" commits?

schuettecarsten, 2024-02-05:
> @schuettecarsten are you sure it's the kernel bump or not some "near" commits?

Yes, I am quite sure because I am on `master` and reverted the single commit https://github.com/openwrt/openwrt/commit/2c363a070e3a007a2f4f5e12491676cbe670fe5a to make it work again (`git revert 2c363a070e3a007a2f4f5e12491676cbe670fe5a`), nothing else.

schuettecarsten, 2024-02-05:
@Ansuel The first time I've attached a serial console to my RPi :-). Here is the full exception:

```
[   86.839851] Unable to handle kernel access to user memory outside uaccess routines at virtual address 000000000000007a
[   86.850727] Mem abort info:
[   86.853530]   ESR = 0x0000000096000005
[   86.857295]   EC = 0x25: DABT (current EL), IL = 32 bits
[   86.862610]   SET = 0, FnV = 0
[   86.865672]   EA = 0, S1PTW = 0
[   86.868814]   FSC = 0x05: level 1 translation fault
[   86.873685] Data abort info:
[   86.876567]   ISV = 0, ISS = 0x00000005
[   86.880400]   CM = 0, WnR = 0
[   86.883360] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000052a18000
[   86.889808] [000000000000007a] pgd=0800000049a40003, p4d=0800000049a40003, pud=0800000049a40003, pmd=0000000000000000
[   86.900437] Internal error: Oops: 0000000096000005 [#1] SMP
[   86.906010] Modules linked in: brcmfmac_wcc pppoe ppp_async brcmfmac wireguard pppox ppp_generic nft_fib_inet nf_flow_table_inet lz4 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 cdc_ncm cdc_ether zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet usbhid ts_fsm ts_bm slhc sch_cake r8169 r8152 poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda
[   86.906174]  nf_log_syslog nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount lzo_rle lzo lz4hc_compress lz4_decompress lz4_compress libcurve25519_generic libchacha lan78xx iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hid_generic crc_ccitt compat cls_flower brcmutil br_netfilter asn1_decoder act_vlan fuse ntfs3 cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact hid evdev raspberrypi_hwmon gpio_fan hwmon i2c_dev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port
[   86.992553]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfs ifb ip6_udp_tunnel udp_tunnel oid_registry veth lockd sunrpc grace dns_resolver nls_utf8 zram zsmalloc bfq crypto_acompress nls_iso8859_1 nls_cp437 ehci_pci fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd vfat fat exfat btrfs xor zstd_decompress zstd_compress zstd_common zlib_inflate zlib_deflate xxhash xor_neon raid6_pq lzo_decompress lzo_compress libcrc32c realtek microchip
[   87.131621] CPU: 2 PID: 12208 Comm: igmpproxy Not tainted 6.1.75 #0
[   87.137884] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[   87.143709] pstate: 20400005 (nzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   87.150666] pc : ip_mr_forward+0x1c8/0x35c
[   87.154766] lr : ipmr_mfc_add+0x6ac/0x800
[   87.158770] sp : ffffffc011cabb40
[   87.162076] x29: ffffffc011cabb40 x28: ffffffc008ab6000 x27: ffffffc008c12b80
[   87.169209] x26: 0000000000000000 x25: 0000000000000002 x24: ffffff8045ccbd00
[   87.176342] x23: ffffff8045f1f000 x22: ffffff80478f3e40 x21: ffffff804002e000
[   87.183474] x20: 0000000000000001 x19: 0000000000000001 x18: 0000000000000000
[   87.190605] x17: 0000000000000000 x16: 0000000000000000 x15: 0000007fd264de28
[   87.197737] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[   87.204868] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
[   87.212000] x8 : ffffff80478f3ea8 x7 : ffffffffffffffff x6 : ffffff804002e068
[   87.219131] x5 : 0000000000000000 x4 : ffffff80478f3e40 x3 : 0000000000000001
[   87.226263] x2 : 0000000000000000 x1 : ffffff80457c5000 x0 : 0000000000000000
[   87.233395] Call trace:
[   87.235833]  ip_mr_forward+0x1c8/0x35c
[   87.239577]  ipmr_mfc_add+0x6ac/0x800
[   87.243234]  ip_mroute_setsockopt+0x420/0x510
[   87.247585]  do_ip_setsockopt+0x1d0/0xd6c
[   87.251590]  ip_setsockopt+0x34/0xa0
[   87.255158]  raw_setsockopt+0xd4/0x140
[   87.258902]  sock_common_setsockopt+0x1c/0x30
[   87.263254]  __sys_setsockopt+0x8c/0x170
[   87.267169]  __arm64_sys_setsockopt+0x28/0x40
[   87.271519]  invoke_syscall.constprop.0+0x5c/0x104
[   87.276305]  do_el0_svc+0x50/0x13c
[   87.279700]  el0_svc+0x18/0x54
[   87.282750]  el0t_64_sync_handler+0x114/0x120
[   87.287101]  el0t_64_sync+0x174/0x178
[   87.290759] Code: eb0102ff 54fff840 f9402f00 927ff800 (3941e800)
[   87.296845] ---[ end trace 0000000000000000 ]---
[   87.301455] Kernel panic - not syncing: Oops: Fatal exception
[   87.307194] SMP: stopping secondary CPUs
[   87.311110] Kernel Offset: disabled
[   87.314589] CPU features: 0x80000,00034080,0000400b
[   87.319460] Memory Limit: none
[   87.322507] Rebooting in 3 seconds..
```

Ansuel, 2024-02-05:
Looks like a regression... let me check what was merged with that kernel version...

schuettecarsten, 2024-02-05:
If I understand this correctly, the process causing this is igmpproxy?

Ansuel, 2024-02-05:
@schuettecarsten can you try revert this? https://github.com/torvalds/linux/commit/bb7403655b3c3eb245d0ee330047cd3e20b3c4af

Ansuel, 2024-02-05:
https://gist.github.com/Ansuel/0e903f2d288d1c0ce9b87cbb88be5a40 this patch put in generic hack

schuettecarsten, 2024-02-05:
> https://gist.github.com/Ansuel/0e903f2d288d1c0ce9b87cbb88be5a40 this patch put in generic hack

@Ansuel The patch fixes the issue.

Ansuel, 2024-02-05:
@schuettecarsten oh! wow I sniped it ahahahah let me check it better tho

Ansuel, 2024-02-05:
@schuettecarsten ok... good news!

remove that patch but add this patch

https://gist.github.com/Ansuel/07831372645fa3e252043c7e813b87c5

I assume should fix your problem.

schuettecarsten, 2024-02-05:
@Ansuel will try and let you know, might take a day or two. Just to let you know, there is a pull request pending to bump kernel to 6.1.76 which is also affected. Looks like the patch was submitted somewhere but did not yet find it's way into the kernel.

Ansuel, 2024-02-05:
@schuettecarsten please test so I can give the stable kernel guy more info... that patch should fix that but I need you to confirm that :D

schuettecarsten, 2024-02-05:
> @schuettecarsten please test so I can give the stable kernel guy more info... that patch should fix that but I need you to confirm that :D

The new patch does not apply correctly:

```
Applying /home/cschuette/openwrt/snapshot/openwrt/target/linux/generic/hack-6.1/0001-ipmr-fix-kernel-panic-when-forwarding-mcast-packets.patch using plaintext:
patching file include/net/ip.h
Hunk #1 succeeded at 752 (offset -15 lines).
patching file net/ipv4/ip_sockglue.c
Hunk #1 succeeded at 1406 with fuzz 1 (offset 43 lines).
Hunk #2 succeeded at 1441 (offset 43 lines).
patching file net/ipv4/ipmr.c
Hunk #1 succeeded at 1074 (offset 1 line).
patching file net/ipv4/raw.c
Hunk #1 FAILED at 292.
1 out of 1 hunk FAILED -- saving rejects to file net/ipv4/raw.c.rej
patching file net/ipv4/udp.c
Hunk #1 succeeded at 2196 (offset 27 lines).
Patch failed!  Please fix /home/cschuette/openwrt/snapshot/openwrt/target/linux/generic/hack-6.1/0001-ipmr-fix-kernel-panic-when-forwarding-mcast-packets.patch!
```

```
--- net/ipv4/raw.c
+++ net/ipv4/raw.c
@@ -292,7 +292,7 @@ static int raw_rcv_skb(struct sock *sk, struct sk_buff *skb)

        /* Charge it to the socket. */

-       ipv4_pktinfo_prepare(sk, skb);
+       ipv4_pktinfo_prepare(sk, skb, true);
        if (sock_queue_rcv_skb_reason(sk, skb, &reason) < 0) {
                kfree_skb_reason(skb, reason);
                return NET_RX_DROP;
```

Ansuel, 2024-02-05:
@schuettecarsten try now https://gist.github.com/Ansuel/07831372645fa3e252043c7e813b87c5

schuettecarsten, 2024-02-05:
> @schuettecarsten try now https://gist.github.com/Ansuel/07831372645fa3e252043c7e813b87c5

Patch applied, compiling...

closed 2024-02-05 by openwrt-bot

Ansuel, 2024-02-05:
@schuettecarsten well the thing is already backported... they just need to release .77

commit 5e395f0c (2024-02-05): generic: 6.1: backport patch fixing kernel panic with mcast packets

Backport patch fixing kernel panic with mcast packets, patch is already
scheduled to be backported to stable kernels and will be dropped once
new stable kernel version are released.

Fixes: #14554