openwrt/openwrt #14562: LLDP PDUs should not be forwarded in 802.1d but some devices are forwarding them

howels, 2024-02-06:
### Describe the bug

Noticed on several devices that I see LLDP entries for the device itself when plugged into another OpenWRT switch. This shouldn't happen as 802.1d compliant bridges should never forward frames sent to 01:80:c2:00:00:0e, the LLDP multicast address.
Everything is running the latest 23.05 release but I have noticed on my Ubiquiti EdgeRouter X and TP-Link C2600, that I can see multiple LLDP neighbours on a single local interface port. Not a bridge port, but eth0 for example. These neighbours are not locally connected but are connected to a neighbour of the device and seem to be forwarded from the neighbour.
Doesn't happen on all devices - an OpenWRT x86 machine with Intel E810 NIC seems to show LLDP neighbours correctly. So I guess it's an issue with the switch configuration on the affected devices?

Example - only Edgerouter-1 is locally attached to this device, the other entries are connected to edgerouter-1 and are not locally attached.

```
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
Interface:    eth1.1, via: LLDP
  Chassis:     
    ChassisID:    local edgerouter-x-1
    SysName:      edgerouter-x-1
  Port:        
    PortID:       mac 74:83:c2:08:d4:f6
    PortDescr:    eth2
    TTL:          120
-------------------------------------------------------------------------------
Interface:    eth1.1, via: LLDP
  Chassis:     
    ChassisID:    local edgerouter-x-2
    SysName:      edgerouter-x-2
  Port:        
    PortID:       mac 74:83:c2:4d:cb:ce
    PortDescr:    eth0
    TTL:          120
-------------------------------------------------------------------------------
Interface:    eth1.1, via: LLDP
  Chassis:     
    ChassisID:    mac d8:5e:d3:ad:90:57
    SysName:      bruce
  Port:        
    PortID:       mac d8:5e:d3:ad:90:57
    PortDescr:    enp7s0
    TTL:          120
-------------------------------------------------------------------------------
```

I found this setting which would be where the bridge normally allows LLDP forwarding, but it's set to 0x0 so no mask if allowed for forwarding.

```
root@C2600:~# cat /sys/class/net/br-lan/bridge/group_fwd_mask
0x0
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Ubiquiti EdgeRouter X

### Image kind

Official downloaded image

### Steps to reproduce

Enable LLDP on multiple devices, view neighbours on multiple devices.

### Actual behaviour

Non-locally connected neighbours are visible on local devices. Only local LLDP neighbours should be visible but neighbours of non-local LLDP devices are visible, including seeing the local device as a neighbour from itself.

### Expected behaviour

We should only see LLDP PDUs from systems locally connected to the device because these PDUs should not be forwarded by 802.1d compliant Linux bridges. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

howels, 2024-02-06:
> > The default value is 0, which means the bridge does not forward any link-local frames coming on this port.

Correct, that should happen but that is not the observed behaviour.

brada4, 2024-02-06:
Are the reserved standard packets unnecessarily forwarded by DSA switch (like between ports lan1@dsa lan2@dsa etc or they manage to cross CPU like linux brctl/bridge? (and nftables bridge or ebtables may be used to fix up)

howels, 2024-02-06:
> Are the reserved standard packets unnecessarily forwarded by DSA switch (like between ports lan1@dsa lan2@dsa etc or they manage to cross CPU like linux brctl/bridge? (and nftables bridge or ebtables may be used to fix up)

Forwarded by DSA switch I think.

brada4, 2024-02-06:
Try ebtables-like nft on interim bridge(s).
It is supposed to get those packets to CPU and not forward via DSA.
(either it works around the issue or extends scope of the defect to nftables too)
```
#!/usr/sbin/nft -f
table bridge link-local
flush table bridge link-local
table bridge link-local {
        chain drop-local {
                type filter hook forward priority filter; policy accept;
                ether daddr & ff:ff:ff:ff:ff:f0 == 01:80:c2:00:00:00 counter drop
        }
}
```

howels, 2024-02-07:
> Try ebtables-like nft on interim bridge(s). It is supposed to get those packets to CPU and not forward via DSA. (either it works around the issue or extends scope of the defect to nftables too)
>
> ``` [...]

Does not appear to change the visible LLDP neighbours.

KanjiMonster, 2024-02-07:
I guess for Edgerouter-X's MT7621 [this commit](https://github.com/torvalds/linux/commit/8332cf6fd7c7087dbc2067115b33979c9851bbc4) is missing and needs to be backported. @arinc9, this is from you. Can you take care of that? Looks like this fix was never backported to 5.15 (but is included in 6.1).

brada4, 2024-02-07:
It is 0-F not to be forwarded. https://en.wikipedia.org/wiki/Multicast_address#Ethernet , while not zapping PoE devices flow control and LACP will certainly make adverse effects.

arinc9, 2024-02-07:
@KanjiMonster @howels, I'll submit my remaining patches that fix handling of link-local frames to 5.15 stable releases. It should appear on OpenWrt 23.02 once the kernel version is bumped to a recent patch version of 5.15 which will include my patches.

howels, 2024-02-08:
> @KanjiMonster @howels, I'll submit my remaining patches that fix handling of link-local frames to 5.15 stable releases. It should appear on OpenWrt 23.05 once the kernel version is bumped to a recent patch version of 5.15 which will include my patches.

Any options to build these targets with 6.1 or other newer kernels whilst we wait for 5.15 patches to land upstream and be pulled into OpenWRT?

KanjiMonster, 2024-02-09:
> Any options to build these targets with 6.1 or other newer kernels whilst we wait for 5.15 patches to land upstream and be pulled into OpenWRT?

Be aware that the referenced fix only applies to the MT7621 found in the EdgeRouter X, for which no 6.1 exists yet in OpenWrt. For the TP-Link C2600 a different fix will be needed, as thre is a different switch chip, the AR8337, in there.

arinc9, 2024-03-24:
@KanjiMonster @howels my patches are applied to the 5.15 stable tree. It should appear on the next patch version of 5.15 and the next version of OpenWrt 23.05.

howels, 2024-03-25:
> @KanjiMonster @howels my patches are applied to the 5.15 stable tree. It should appear on the next patch version of 5.15 and the next version of OpenWrt 23.05.

Did that just miss the 23.05.3 release? I guess it did because my devices are still echoing LLDP PDUs across bridges after upgrading.

arinc9, 2024-03-25:
It must've. Next release then.

vincele, 2024-09-11:
@howels did you test 23.05.4 ?

howels, 2024-09-11:
> @howels did you test 23.05.4 ?

It's fixed in 23.05.4 for my devices

vincele, 2024-09-11:
great, maybe the issue can be closed then...

closed 2024-09-12 by howels