openwrt/openwrt #14566: OpenWRT Factory image bricks NWA50AX-PRO when installed on latest OEM firmware

linuxtim, 2024-02-06:
### Describe the bug

Unboxed new NWA50AX-PRO, upgraded to latest OEM firmware, installed OpenWRT 23.05.2 factory image.  Device bricked.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

ZyXEL NWA50AX Pro

### Image kind

Official downloaded image

### Steps to reproduce

- Unbox brand new NWA50AX-PRO.
- Boot OEM firmware (acquires IP address by DHCP).
- Open web interface.
- Upgrade OEM firmware to latest release `NWA50AX PRO_6.65(ACGE.1)C0.zip`:
- `df7316b9f568c5cded26655d36152302e6f750863ac5d8e8ec0ed78af825fb1d  /tmp/665ACGE1C0.bin`
- Open web interface.
- Install `openwrt-23.05.2-mediatek-filogic-zyxel_nwa50ax-pro-squashfs-factory.bin`

### Actual behaviour

Looks like it's maybe using OpenWRT kernel with Zyxel rootfs?

Serial boot log:

```
F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 2400 0041 [0000]
G0: 1190 0000
EC: 0000 0000 [1000]
T0: 0000 024B [010F]
Jump to BL

NOTICE:  BL2: v2.7(release):645p0c0-49-g10d224bb-dirty
NOTICE:  BL2: Built : 17:12:31, Oct 20 2022
NOTICE:  WDT: disabled
NOTICE:  EMI: Using DDR4 settings

dump toprgu registers data:
1001c000 | 00000000 0000ffe0 00000000 00000000
1001c010 | 00000fff 00000000 00f00000 00000000
1001c020 | 00000000 00000000 00000000 00000000
1001c030 | 003c0003 003c0003 00000000 00000000
1001c040 | 00000000 00000000 00000000 00000000
1001c050 | 00000000 00000000 00000000 00000000
1001c060 | 00000000 00000000 00000000 00000000
1001c070 | 00000000 00000000 00000000 00000000
1001c080 | 00000000 00000000 00000000 00000000

dump drm registers data:
[... 260 lines cut ...]
        0x459aa4        argc:6  args:ap vlan vlan-id 1 untag
can not get genl family id!
[wifimgr_genl_connect:81]ERROR: not find family:zyUMAC id:-12
[WiFiNetlinkConnect:594]ERROR: fail: -12, zyUMAC
can not get genl family id!
[wifimgr_genl_connect:81]ERROR: not find family:zyUMAC id:-12
[WiFiNetlinkConnect:594]ERROR: fail: -12, zyUMAC
can not get genl family id!
[wifimgr_genl_connect:81]ERROR: not find family:zyUMAC id:-12
[WiFiNetlinkConnect:594]ERROR: fail: -12, zyUMAC
[wifimgr_genl_connect:81]ERROR: not find family:zyUMAC id:-12
[WiFiNetlinkConnect:594]ERROR: fail: -12, zyUMAC
```

### Expected behaviour

OpenWRT first boot

### Additional info

Possibly the Zyxel firmware is running an A/B setup, and the wrong slot is either being selected, or was written by the Zyxel programmer during the factory image install.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kirdesde, 2024-02-06:
https://forum.openwrt.org/t/zyxel-nwa50ax-pro-bricked/186051/6

linuxtim, 2024-02-08:
OK, perhaps I should have said requires recovery via serial console.  For many users this is equivalent tho' because it's unreasonable to require users to have a UART adapter available (and know how to use it). I've added a note to the device wiki page regarding checking which slot it active prior to install.  Presumably the factory image could be modified so that it will refuse to install to the incorrect slot and/or reconfigure the firmware slot settings before installing.

ynezz, 2024-02-25:
> I've also added a note to the device wiki page yesterday regarding checking which slot it active prior to install.

Thanks!

> OK, perhaps I should have said requires recovery via serial console

Have you tried to reboot one more time? According to @blocktrron 's description in f0445746f6fd9 it looks like the device should recover from that:

"As the device uses a dual-image partition layout, OpenWrt can only installed on Slot A. This requires the current active image prior flashing the device to be on Slot B. In case this is not the case, OpenWrt will boot only one time, returning
to the ZyXEL firmware the second boot."

> Presumably the factory image could be modified so that it will refuse to install to the incorrect slot and/or reconfigure the firmware slot settings before installing.

Vendor/Zyxel would need to do this. OpenWrt can't do that, because OpenWrt is not running yet when you flash the factory image, it's vendor's code running.

I'm afraid, that you can't do much about that, you would basically need [some kind of installer](https://github.com/dangowrt/owrt-ubi-installer) and maybe even replacing the complete bootchain.

closed 2024-02-25 by ynezz