openwrt/openwrt #14573: hostapd freezes & restarts are unreliable when running as a service (but works when started manually) - procd?

yogo1212, 2024-02-07:
### Describe the bug

After boot, 2/3 or all 3 interfaces are configured on phy0. None on phy1 (race-y but pretty reliable).

Restarting hostapd later, when everything has settled, usually leads to all interfaces being configured.
The restart only works with some extra help.
See 'additional details'.

In either case, no wifi connections are possible.

I've encountered the issue in a close-to-upstream fork and reproduced it from the main branch, carrying over only the wifi config and some packages I like having on the router for use with my Nextcloud server (installed via feed).

Since then, I've noticed that stopping the wpad service, killing all remaining process, and launching hostapd manually (with all the shenanigans) leads to a working, unfrozen process:

```
$ /sbin/ujail -t 5 -n hostapd -U network -G network -C /etc/capabilities/wpad.json -c -- /usr/sbin/hostapd -s -g /var/run/hostapd/global
-> happy me
```

:eyes: 

Playing around some more, I've found another application that used to work as a service but doesn't anymore.
It uses wget to get my public IP (that works) and then sets a DNS record via a self-written C program that POSTs the address via TLS.
The C program has it's TLS socket closed (read event with length 0) before it can complete the POST - but only if running as a service.
When running from an SSH session, it works.

This is the service definition of that service:

```
#!/bin/sh /etc/rc.common

USE_PROCD=1

start_service() {
	procd_open_instance
	procd_set_param limits core="unlimited"
	procd_set_param command /usr/bin/nextcloud_dns
	procd_set_param nice 12
	procd_set_param respawn 0 1 0
	procd_set_param stdout 1
	procd_set_param stderr 1
	procd_close_instance
}
```

I'm confident that hostapd and it's packaging is not the cause of the problem.

:flamingo:

### OpenWrt version

r24979-097f0f035e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Cudy M1800

### Image kind

Self-built image

### Additional info

```
$ ubus call hostapd reload
Command failed: Request timed out
```

Log for `/etc/init.d/wpad restart`:

```
Wed Feb  7 10:05:07 2024 daemon.notice hostapd: ctrl_iface exists and seems to be in use - cannot override it
Wed Feb  7 10:05:07 2024 daemon.notice hostapd: Delete '/var/run/hostapd/global' manually if it is not used anymore

Wed Feb  7 10:06:27 2024 daemon.notice hostapd: ctrl_iface exists and seems to be in use - cannot override it
Wed Feb  7 10:06:27 2024 daemon.notice hostapd: Delete '/var/run/hostapd/global' manually if it is not used anymore
Wed Feb  7 10:06:28 2024 daemon.err hostapd: eloop: could not process SIGINT or SIGTERM in two seconds. Looks like there is a bug that ends up in a busy loop that prevents clean shutdown. Killing program forcefully.
```

```
 1389 network   8532 S    /usr/sbin/hostapd -s -g /var/run/hostapd/global
 8030 root      2912 S    {hostapd} /sbin/ujail -t 5 -n hostapd -U network -G network -C /etc/capabilities/wpad.json -c -- /usr/sbi
 8033 network   7992 S    /usr/sbin/hostapd -s -g /var/run/hostapd/global
```

Killing either hostapd or ujail doesn't restore functionality, both need to be killed in one go.

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_cudy_m1800=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_cudy_wr2100=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_BUSYBOX_CONFIG_MKFS_VFAT=y
CONFIG_BUSYBOX_CONFIG_NC_110_COMPAT=y
CONFIG_BUSYBOX_CONFIG_NC_SERVER=y
# CONFIG_BUSYBOX_CONFIG_SWAPOFF is not set
# CONFIG_BUSYBOX_CONFIG_SWAPON is not set
CONFIG_BUSYBOX_CONFIG_TIMEOUT=y
CONFIG_BUSYBOX_CONFIG_USLEEP=y
CONFIG_FASTD_ENABLE_CIPHER_NULL=y
CONFIG_FASTD_ENABLE_CIPHER_SALSA2012=y
CONFIG_FASTD_ENABLE_MAC_UHASH=y
CONFIG_FASTD_ENABLE_METHOD_COMPOSED_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_GENERIC_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_NULL=y
CONFIG_FASTD_ENABLE_METHOD_NULL_L2TP=y
CONFIG_FASTD_WITH_OFFLOAD_L2TP=y
CONFIG_FASTD_WITH_STATUS_SOCKET=y
CONFIG_IMAGEOPT=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_batctl-default=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bridge_mac_table=y
CONFIG_PACKAGE_fastd=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-input-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-l2tp=y
CONFIG_PACKAGE_kmod-l2tp-eth=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-mii=y
# CONFIG_PACKAGE_kmod-nft-fib is not set
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-utf8=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-hid=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-cdc-mbim=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-serial=y
CONFIG_PACKAGE_kmod-usb-serial-option=y
CONFIG_PACKAGE_kmod-usb-serial-qualcomm=y
CONFIG_PACKAGE_kmod-usb-serial-wwan=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_kmod-veth=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libevent2-core=y
CONFIG_PACKAGE_libevent2-extra=y
CONFIG_PACKAGE_libevent2-openssl=y
CONFIG_PACKAGE_libjson-c-parser=y
CONFIG_PACKAGE_libnames=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-legacy=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libringbuffer=y
CONFIG_PACKAGE_librt=y
# CONFIG_PACKAGE_libuclient is not set
CONFIG_PACKAGE_libuecc=y
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_uclient-fetch is not set
CONFIG_PACKAGE_ucode-mod-nl80211=y
CONFIG_PACKAGE_ucode-mod-rtnl=y
# CONFIG_PACKAGE_ucode-mod-uci is not set
CONFIG_PACKAGE_ucode-mod-uloop=y
CONFIG_PACKAGE_umbim=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_utftp=y
CONFIG_PACKAGE_uthash=y
CONFIG_PACKAGE_vpn-routes=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_zlib=y
CONFIG_PREINITOPT=y
CONFIG_TARGET_PREINIT_DISABLE_FAILSAFE=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
# CONFIG_KERNEL_WEXT_CORE is not set
# CONFIG_KERNEL_WEXT_PRIV is not set
# CONFIG_KERNEL_WEXT_PROC is not set
# CONFIG_KERNEL_WEXT_SPY is not set
# CONFIG_KERNEL_WIRELESS_EXT is not set
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yogo1212, 2024-02-07:
I've tried to determine hostapd's broken state using a coredump.

```
Program terminated with signal SIGABRT, Aborted.
#0  0x77e32aa4 in ?? ()
(gdb) bt
#0  0x77e32aa4 in ?? ()
```

Rebuilt wpad with `STRIP=/bin/true TARGET_OPTIMIZATION="-ggdb3 -O0"`, same thing.
This is what I usually do with my applications but, probably, that doesn't do it for hostapd.

yogo1212, 2024-02-07:
Put the same firmware on another M1800 and it had only the closing socket with my dyidns. hostapd works.
This is just one test, so I'm hesitant to call 'indeterministic behaviour of procd services' yet.

yogo1212, 2024-02-07:
Last known working: c35ff41ccc31f2ecaea7406830de04bd77940d48.
I guess it's a bisect then.

brada4, 2024-02-07:
Public ip detection is here
https://github.com/openwrt/openwrt/blob/c6fabd0bc3354099bcfc73aa4064139b23fb509d/package/base-files/files/lib/functions/network.sh#L209
It lacks netmask, probably causing somemaintainer to resort to wget.

yogo1212, 2024-02-09:
The first bad commit is 7863fb15fc6dfe1555d34b223e7ecc96d2915a75.

@nbd168 @jow- can you maybe spot a potential cause with those changes?

My first attempt at building libubox from source went horribly but I can try again tomorrow.

yogo1212, 2024-02-09:
I have an unrelated problem where all of my serial adapters don't communicate with other devices.
Two serial adapters can, however, and the loopback works.
My router is a brick since I installed [f7d156911](https://git.openwrt.org/?p=project/libubox.git;a=commit;h=f7d1569113110ea8df071d2ea64fd17aaf5b42c9)  (not saying there's a casual relation)

yogo1212, 2024-02-09:
update:
```
bad  82fa6480de7a uloop: add support for interval timers
     13d9b04fb09d uloop: add support for user defined signal handlers (current)
     f7d156911311 uloop: properly initialize signal handler mask (brick?)
good 8a5a4319a85c uloop: fix typo in signal handling rework
```

why would it be either of those?

yogo1212, 2024-02-09:
it's none of them.
depending on from which side i approach the commit in the openwrt repo, the results are different.
:disappointed: 

i'll have to bisect again with `make clean` :computer: :fire:

yogo1212, 2024-02-09:
Nope, it's https://github.com/openwrt/openwrt/commit/7863fb15fc6dfe1555d34b223e7ecc96d2915a75.

[image]

yogo1212, 2024-02-10:
i'll take the weekend off of this bug.

why this commit?
why are my ath79 and ipq40xx devices not affected?
only the M1800 :brain: :hammer:

yogo1212, 2024-02-12:
Why this commit?! How?

My ath79 and ipq40xx devices are not affected.
Same config. Just changed the devices.

Only the M1800, the WR2100 (both ramips, mt7621), and the M3000 (mediatek, filogic). :brain: :hammer:

yogo1212, 2024-02-14:
`strace`'ing the differences between the calls that work and those that don't (service vs. shell), this happens right at the beginning for the failing instance:

```
open("/lib/libsetlbf.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=65583, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 135168, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77ddb000
mmap2(0x77dfa000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0x77dfa000
close(3)                                = 0
```

Almost everything else is effectively the same (naturally, there are differences with read/write on the TLS socket and a lot of mmap/munmap, also because of openssl, i suppose).
There's one more mprotect - for the additional library?
Read/write length's are absolutely identical until the last read before the socket closure event from libevent2.

Closing socket:

```
read(12, "\326\340\245\205\25h\324\335\260hk\347\35\306\342-\263\21\34j\263\247\5 \355\316", 26) = 26
epoll_ctl(3, EPOLL_CTL_DEL, 12, 0x7fbfe990) = 0
writev(2, [{iov_base="socket closed", iov_len=13}, {iov_base=NULL, iov_len=0}], 2socket closed) = 13
writev(2, [{iov_base="", iov_len=0}, {iov_base="\n", iov_len=1}], 2) = 1
```

Working socket:

```
read(12, "\272H\\\325\27WD\1\353{\253\367\346\355\265\7rD\36\23\311R\365\345H\362\231\251", 28) = 28
munmap(0x7768b000, 28672)               = 0
ioctl(1, TIOCGWINSZ, 0x7f9626e8)        = -1 ENOTTY (Not a tty)
writev(1, [{iov_base="connected", iov_len=9}, {iov_base="\n", iov_len=1}], 2connected
) = 10
```

[image]

yogo1212, 2024-02-14:
With `libsetlbf` removed from `LD_PRELOAD`, it works again on all the commits following 7863fb15fc6dfe1555d34b223e7ecc96d2915a75 that I tested.

That still doesn't explain *how* it breaks the connection.

yogo1212, 2024-02-14:
@nbd168 @jow- Can you help me out with a pointer?
I'm willing to put in the time but I don't how to advance.

Since [these changes](https://github.com/openwrt/libubox/compare/75a3b870cace1171faf57bd55e5a9a2f1564f757..d4c3066e7c5efa5f395c21db77609516d386cbf8) in libubox ([this](https://github.com/openwrt/openwrt/commit/7863fb15fc6dfe1555d34b223e7ecc96d2915a75) package update), there are connection closures internal to openssl that don't happen when `libsetlbf` is removed `LD_PRELOAD` (Linux socket is still open. Output of `strace` diff'd above on the Github issue).

But only with rampis and mediatek/filogic, not with ipq40xx and ath79 (at least the devices I own/have access to).

yogo1212, 2024-02-19:
I've removed all output before error handling and result handling:

```
$ LD_PRELOAD=/lib/libsetlbf.so /tmp/test_wurst
error: socket closed
$ /tmp/test_wurst
200 OK
```

`SSL_get_error` returns `SSL_ERROR_ZERO_RETURN`. So, `SSL_RECEIVED_SHUTDOWN` has to have been set.

jow-, 2024-02-19:
Can you please share the code of `/tmp/test_wurst` ?

yogo1212, 2024-02-19:
Hi and thanks for tuning in :-)

Previously, I've been testing with HTTP and TLS.
This has proven to be difficult for reproduction and sharing publicly [1].
But there seems to be a more reliable way using MQTT.
The MQTT CONNECT handshake can be easily customized to look like this:

```hexdump
00000000  10 11 00 04 4d 51 54 54  04 02 00 1e 00 05 74 65   ....MQTT ......te
00000010  73 74 00                                           st.
```

The above is what goes into a bufferevent (abstraction in libevent that can be used with TLS).
This is what Wireshark later decodes from the TLS socket (if the application was started with `LD_PRELOAD=/lib/libsetlbf.so`):

```hexdump
00000000  10 11 00 04 4d 51 54 54  04 a2 00 1e 00 05 74 65   ....MQTT ......te
00000010  73 74 00                                           st.
```

Byte 9 has two flipped bits (User Name Flag and Will Retain). Since the user name is missing from the rest of the frame, the connection is dropped by the server.
And now that I'm using my own server, I have access to the logs and, indeed, `MQTT cannot parse a frame on connection`.
The server is closing the connection. It's the data that is different on the client side.

This is where the writes happen:
https://github.com/yogo1212/evmqtt/blob/master/src/mqtt.c#L217 

I'm planning to throw together a single-file C program that works without linking against my MQTT library.
But it'll have to wait - it's time to go to bed for me now.

Still no idea what is going on with hostapd, though (other than that it busy loops waiting for a udebugd socket to appear).
Still no idea what libsetlbf (procd) might have to do with it (`setbuf(stdout, NULL);` is all there is to it - or not?).

:shrug: 

[1]
The API relay I use for the DNS update closes the connection instantly upon being unhappy with the payload.
Also, I don't know any public endpoints speaking protobuf.

yogo1212, 2024-02-21:
Long story short:
The problem MQTT is a problem with uninitialized memory, compiler optimizations, and such - libsetlbf only pushes data into the heap region. That problem is entirely a bug with my code.
I thought the MQTT problem was related to hostapd freezing - which puts me back on tile 1 debugging that.

Here's the (non-openwrt) bug that I did figure out:
`connect_flags |= ((data->x ? 1 : 0) << y);` is optimized to `connect_flags |= (data->x << y)` (x being a `bool` which should only be 1 or 0). I made the extra effort to deal with implementations that don't do `bool` properly and this compiler throws it out because it does `bool` so properly it doesn't believe there could be anything but 0 or 1 in it.

```
lbu     v0,9(s1)
sll     v0,4
or      a2,v0
```

The byte underlying the bool was `3a` (== `0b00111010`), which became `0b10100000` rather than (`0b00100000`).

- shortcuts for assuming program state based on the results are always risky (my assumption about the data passed to libevent/openssl based on how other program invocations went was bogus)
- rather add a printf to check - even if you think it might be part of the problem
- these investigations shouldn't be rushed

yogo1212, 2024-03-11:
I've encountered the first ath79 device (Compex WPJ531) with the freeze. I copied the entire `/etc/` onto another board from the same batch and it doesn't freeze there. The respective behaviour persists through reboots.
This is bug is haunting.

```
Mon Mar 11 15:33:53 2024 daemon.notice netifd: radio0 (6971): Command failed: ubus call hostapd config_set { "phy": "phy0", "config": "", "prev_config": "/var/run/hostapd-phy0.conf" } (Request timed out)
```

yogo1212, 2024-03-31:
still present after a forward to 0c96d20bf9fba6d814efa88c6fb08a5df094103a 🫣

not all freezes are equal, it seems.
today, i was able to call methods of the interface that did get loaded:

```
[root@hostapd-test1 ~] $ ubus call hostapd.test-ssid2 rrm_nr_get_own
{
  "value": [
    "02:0c:43:26:46:28",
    "ssid2",
    "020c43264628ef4900005306070603010800"
  ]
}
[root@hostapd-test1 ~] $ ubus call hostapd config_set '{ "phy": "phy1", "config":"/var/run/hostapd-phy1.conf", "prev_config": "/var/run/hostapd-phy1.conf.prev"}'
Command failed: Request timed out
[root@hostapd-test1 ~] $ ubus call hostapd config_set '{ "phy": "phy0", "config":"/var/run/hostapd-phy0.conf", "prev_config": "/var/run/hostapd-phy0.conf.prev"}'
Command failed: Request timed out
[root@hostapd-test1 ~] $ ubus call hostapd.test-ssid2 rrm_nr_get_own
{
  "value": [
    "02:0c:43:26:46:28",
    "ssid2",
    "020c43264628ef4900005306070603010800"
  ]
}
```

No idea whether it has anything to do with it, but this happened when I ran `/etc/init.d/wpad restart`.

```
Sun Mar 31 20:57:39 2024 daemon.notice wpa_supplicant[1477]: ELOOP: remaining socket: sock=23 eloop_data=0x7777685c user_data=0x7777685c handler=0x55628089
```

yogo1212, 2024-04-01:
so it's the ucode script that freezes..

yogo1212, 2024-04-06:
Today, I noticed this:
```
Multiple BSSes specified in BSS-config
```

And just because it was a workaround [pre-ucode](https://forum.openwrt.org/t/multiple-bsses-specified-in-bss-config/62373/3), I tried `json_add_boolean keep 0` again and all BSSes were there and the `hostapd` ubus object was responsive.
It's not a real 'fix' but maybe it's a clue... 🐒

yogo1212, 2024-04-09:
Again, Github showing an old state of this issue..

It appears the freeze can happen in multiple ways.

One way for that to happen is through wpa_supplicant being flooded with `NL80211_CMD_NEW_PEER_CANDIDATE` events from netlink until the connection is finally completed:

```
Mon Apr  8 14:10:36 2024 daemon.notice wpa_supplicant[1482]: mesh24: MESH-GROUP-STARTED ssid="mesh-test" id=1
Mon Apr  8 14:10:36 2024 daemon.notice wpa_supplicant[1482]: mesh24: new peer notification for b2:4b:d6:26:ae:0c
# 295 more such lines cut
Mon Apr  8 14:10:37 2024 daemon.notice wpa_supplicant[1482]: mesh24: new peer notification for b2:4b:d6:26:ae:0c
Mon Apr  8 14:10:38 2024 daemon.notice wpa_supplicant[1482]: mesh24: new peer notification for b2:4b:d6:26:ae:0c
Mon Apr  8 14:10:39 2024 daemon.err wpa_supplicant[1482]: Mesh RSN: frame verification failed!
Mon Apr  8 14:10:39 2024 daemon.notice wpa_supplicant[1482]: mesh24: new peer notification for b2:4b:d6:26:ae:0c
Mon Apr  8 14:10:39 2024 daemon.notice wpa_supplicant[1482]: mesh24: mesh plink with b2:4b:d6:26:ae:0c established
Mon Apr  8 14:10:39 2024 daemon.notice wpa_supplicant[1482]: mesh24: MESH-PEER-CONNECTED b2:4b:d6:26:ae:0c
```
 
These events are also passed to `wpas_ucode_event`.
It appears that they starve other activities:

```
Mon Apr  8 14:08:00 2024 daemon.notice wpa_supplicant[1482]: mesh24: new peer notification for b2:4b:d6:26:ae:0c
Mon Apr  8 14:08:02 2024 daemon.notice netifd: radio0 (6489): Command failed: ubus call hostapd config_set { "phy": "phy0", "config": "", "prev_config": "/var/run/hostapd-phy0.conf" } (Request timed out)
```

The 'unclogging' coincides with a WARN_ON_ONCE on dmesg:

```
void ieee80211_link_info_change_notify(struct ieee80211_sub_if_data *sdata,
               struct ieee80211_link_data *link,
               u64 changed)
{
  struct ieee80211_local *local = sdata->local;

  WARN_ON_ONCE(changed & BSS_CHANGED_VIF_CFG_FLAGS);
```

```
Mon Apr  8 14:10:39 2024 kern.warn kernel: [  259.278487] [<831025e4>] ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
Mon Apr  8 14:10:39 2024 kern.warn kernel: [  259.286161] [<83176a3c>] ieee80211_mesh_work+0x198/0x300 [mac80211]
Mon Apr  8 14:10:39 2024 kern.warn kernel: [  259.292528] [<83080e28>] cfg80211_stop_iface+0x1c0/0x20c [cfg80211]
```

This particular behaviour appears to be specific to the mediatek target - at least I haven't seen it elsewhere (likely because other drivers aren't emitting out events like crazy).
These logs are from a `mt7615e`.

yogo1212, 2024-04-14:
My current approach is tracing how far [the ubus call](https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/wireless/mac80211.sh#L23) gets.

The `config_set` in [hostapd.uc](https://github.com/openwrt/openwrt/blob/main/package/network/services/hostapd/files/hostapd.uc#L817) certainly isn't reached.
I've added `printf`s to `uc_ubus_object_call_cb` but it looks like it isn't hit either.

Other ubus objects are responsive and, in fact, even other objects from the very same hostapd daemon respond normally.
Those other objects are registered via a different ubus context (connected from hostapd's C code) but using the same eloop/uloop translation mechanism.

So, it appears only ubus objects using the context created by the ucode library are affected - but not always and they work for a while.
I could imagine one of the following causes:
- global state in libubus 
- something about the eloop<->uloop translation

The C code in `uloop-mod-ubus` works in general. I suspect something happens early during hostapd's execution that causes `eloop_uloop_fd_cb` to loose events for one of its uloop users.

yogo1212, 2024-04-17:
i've added hacky debug functionality to ubus:

```
void log_if_hostapd(const char *fmt, ...)
{
  static int log_fd = -1;
  if (log_fd == -1) {
    int comm_fd = open("/proc/self/comm", O_RDONLY);
    char buf[20];
    read(comm_fd, buf, sizeof(buf));
    close(comm_fd);
    if (strncmp(buf, "hostapd", 7) != 0)
      return;

    log_fd = open("/tmp/hostapd-ubus.log", O_CREAT | O_APPEND | O_WRONLY);
  }

  if (log_fd == -1)
    return;

  va_list argp;
  va_start(argp, fmt);

  char buf[512];
  ssize_t len = vsnprintf(buf, sizeof(buf), fmt, argp);
  write(log_fd, buf, len);
}
```

```
ubus_add_object "hostapd" on fd 11
ubus_add_object "(null)" on fd 11
ubus_add_object "(null)" on fd 11
ubus_add_object "hostapd.test2-5_0" on fd 23
ubus_add_object "hostapd.test2-24_0" on fd 23
ubus_add_object "hostapd.test1-24_0" on fd 23
# cut a lot of ubus_handle_data before manual tests:

# ubus call hostapd.test1-24_0 rrm_nr_get_own
ubus_handle_data on fd 23, recv_fd -1
# works

# ubus call hostapd reload '{"phy":"dings"}'
ubus_handle_data on fd 11, recv_fd -1
# Command failed: Request timed out

# ubus call hostapd reload '{"phy":"dings"}'
ubus_handle_data on fd 11, recv_fd -1
# Command failed: Request timed out
```

so ubus sees the packets - continuing from there.

yogo1212, 2024-04-17:
the moment i add this to ubus_process_msg: `log_if_hostapd("ubus_process_msg hdr.type %d\n", (int) buf->hdr.type);` everything works normally.
i take it away again: bug reappears.

actually, make that a `log_if_hostapd("ubus_process_msg hdr.type\n");` - bug also appears.

let me fetch the old objdump from the tool shed..

yogo1212, 2024-04-17:
```
Wed Apr 17 17:08:56 2024 daemon.notice hostapd: Exception in ubus function: { "type": "Reference error", "message": "left-hand side expression is null", "stacktrace": [ { "filename": "/usr/share/hostap/hostapd.uc", "line": 739, "byte": 40, "function": "[anonymous function]", "context": "In [anonymous function](), file /usr/share/hostap/hostapd.uc, line 739, byte 40:\n  called from anonymous function (/usr/share/hostap/hostapd.uc:721:22)\n\n `                let config = iface_load_config(phy.orig_file);`\n  Near here -----------------------------------------^\n" }, { "filename": "/usr/share/hostap/hostapd.uc", "line": 721, "byte": 22 } ] } In [anonymous function](), file /usr/share/hostap/hostapd.uc, line 739, byte 40:   called from anonymous function (/usr/share/hostap/hostapd.uc:721:22)   `                let config = iface_load_config(phy.orig_file);`   Near here -----------------------------------------^
```

so `hostapd.data.config[phy_name]` can yield `null`.

yogo1212, 2024-04-17:
adding any kind of debug in `ubus_process_msg` that helps identifying the path taken in the switch statement appears to cause waveform collapse.
i've seen memory-corrupted core dumps (segfault at 0x0), the `null` config mentioned above, and everything working through an added `fprintf` (the output goes nowhere).

yogo1212, 2024-04-17:
i keep neglecting the probalistic effects..

`ubus_queue_msg` <- all the freezing ubus calls take this code path.

yogo1212, 2024-04-17:
so things go in but don't get out:

```
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
```

this happened:

```
ubus_complete_request stack_depth-- (now at 0)
ubus_complete_request stack_depth++ (now at 1)
ubus_handle_data ctx->msgbuf.hdr.type 1
ubus_handle_data ctx->msgbuf.hdr.type 2
ubus_handle_data ctx->msgbuf.hdr.type 1
ubus_complete_request stack_depth-- (now at 1) # huh?
ubus_complete_request stack_depth++ (now at 2)
```

BUT right now, the stack empty but not all interfaces are initialised.. so there must be something else going on.

yogo1212, 2024-04-17:
got another one of these stack_depths:

```
ubus_process_msg stack_depth++ (now at 1)
ubus_process_msg stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_process_msg stack_depth++ (now at 1)
ubus_process_msg stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_process_msg stack_depth++ (now at 1)
ubus_process_msg stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_process_msg stack_depth++ (now at 1)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_process_msg stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_complete_request stack_depth++ (now at 1)
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_complete_request stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 0
ubus_process_msg stack_depth++ (now at 1)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_process_msg stack_depth-- (now at 0)
ubus_process_pending_msg ctx->stack_depth 0
ubus_process_msg stack_depth++ (now at 1)
ubus_process_msg stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_process_pending_msg list_empty 1
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth-- (now at 0)
ubus_process_pending_msg list_empty 1
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth-- (now at 0)
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth++ (now at 2)
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_complete_request stack_depth-- (now at 1)
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth-- (now at 0)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_complete_request stack_depth++ (now at 1)
ubus_complete_request stack_depth-- (now at 0)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_complete_request stack_depth++ (now at 2)
ubus_complete_request stack_depth-- (now at 1)
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_queue_msg ctx->sock.registered 1
ubus_queue_msg ctx->sock.registered 1
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
```

yogo1212, 2024-04-17:
what else is fumbling with stack_depth?

```
ubus_complete_request stack_depth-- (now at 0)
ubus_complete_request stack_depth before 2 # huh?
ubus_complete_request stack_depth++ (now at 3)
```

yogo1212, 2024-04-17:
urgh. `ubus_poll_data`. right there under my nose :disappointed: 

```
# fine up until this point
ubus_complete_request (11) stack_depth++ (now at 1)
      ubus_poll_data (11) enter
      ubus_poll_data handle_data
ubus_handle_data ctx->msgbuf.hdr.type 2
ubus_handle_data ctx->msgbuf.hdr.type 1
ubus_complete_request (11) stack_depth++ (now at 2)
      ubus_poll_data (11) enter
      ubus_poll_data handle_data
ubus_handle_data ctx->msgbuf.hdr.type 5
ubus_queue_msg ctx->sock.registered 1
ubus_handle_data ctx->msgbuf.hdr.type 5
ubus_queue_msg ctx->sock.registered 1
ubus_handle_data ctx->msgbuf.hdr.type 1
ubus_handle_data ctx->msgbuf.hdr.type 2
ubus_handle_data ctx->msgbuf.hdr.type 1
      ubus_poll_data (11) leave
      ubus_poll_data (11) enter
```

so my take is: ubus waits for `status_msg` during `ubus_complete_request` to be set and calls `ubus_poll_data` with a timeout.
`ubus_poll_data` causes the next packet to be fetched, running into the very same code path (it looks to me like `_STATUS` does it).
now there are multiple `ubus_poll_data`s active. before all of them could run into a timeout, the next `ubus_complete_request` is called.

there's infrastructure in place (`if (ctx->stack_depth)` in `ubus_process_msg`) to prevent libubus from handling any new requests before previous ones have finished (added int [this commit](https://git.openwrt.org/?p=project/ubus.git;a=commit;h=a72457b61df045d3c499a6211362b751710590d7)).
but for `ubus_complete_request`, there is no such protection.

with my current level of knowledge of ubus and at this time of day, i couldn't tell the direction a possible fix lies in.
i'm leaving this here for today.

yogo1212, 2024-04-19:
looks like this is indeed a nested interaction with wpa_supplicant:

```
ubus_send_msg type 10
ubus_send_msg msg '{,"s.add":7094140481078322176,}'
ubus_send_msg type 4
ubus_send_msg msg '{"rvice":{}}'
ubus_complete_request stack_depth++ (now at 1)
      ubus_poll_data enter
ubus_process_msg buf->hdr.type 2
ubus_process_msg buf->hdr.type 1
ubus_send_msg type 4
ubus_send_msg msg '{"a_supplicant":{}}'
ubus_complete_request stack_depth++ (now at 2)
```

no idea why the messages are garbled.
i'm using this for debug:
```
str = blobmsg_format_json(msg, true);
log_if_hostapd(ctx->sock.fd, "%s msg '%s'\n", __func__, str);
free(str);
```

yogo1212, 2024-04-19:
:white_flag: 

First things first - misconception on my part:
It is not one uloop activity that is blocked - it is all the others!
The `poll` used in `ubus_poll_data` only works for that one ubus connection and, while inside `poll`, `stack_depth` is >0 causing invoke, unsubscribes, and notifies to be deferred (or in this case with the deadlock [2]: 'effectively disabled' unless you want to rely on oomkiller slamming in the door and hostapd being restarted by procd, which it doesn't do when ujail needs a sigkill).

I've come up with this list of causes for my lockup [2], at least some of which need to be addressed:

- `_complete_request` can (and does) nest which in turn can result in deadlocks/break timeouts [1].
  It is in an inherent problem that can never be solved entirely but can be alleviated a little by calling callbacks outside of the 'critical section'.
  The user can switch to async invokes but, ultimately, there are still lots of internal blocking invokes.
  The nesting could be made less annoying if the callback handling was deferred until after the 'critical section'. Again with a 'but': Other code paths can still call `ubus_complete_request`.
- `ubus_poll_data` calls `poll` only for the ubus socket fd. this starves the other ubus connections and events/timeouts registered in eloop/uloop.
  One solution could be to add an optional `event_loop` callback to the ubus context.
  It would at least allow for more interdependent interaction between ubus by limiting the effect to that one ubus connection but does not solve the general problem.
- invoke, unsubscribe, and notify aren't handled during a nested `ubus_process_msg`. if their handling (even concerning foreign objects) lies on the path for the completion of the current operation, this means deadlock.
  In principle, this should be possible to (re)enable after all problems internal to libubus are addressed.

`stack_depth > 0` looks to me like a semaphore for calling callbacks for invokes and notifies. The commit history states the prevention of list corruption being the motivation for raising the semaphore while handling invoke, unsubscribe, and notifiy in `ubus_process_message`.
The principal mechanism for deferring their handling is from before that ([October 11 2011](https://github.com/openwrt/ubus/commit/8e5a6af1db52a908b53e3e8fd44a6a0af496d257)) and protects only `ubus_complete_request`.

Aside from the work necessary inside libubus, there might be users (of libubus) who accidentally or intentionally 'rely' on this 'feature' to protect their critical sections.

[1]
Assume one `_complete_request` running with timeout 15 (registered at t=0). while ubus messages are being handled Another `_complete_request` runs with a timeout of 40 seconds (registered at t=12).
The status message from the first call only arrives at t=20 and is handled during the poll from the inner call but the outer call won't notice it was too late.
(There are calls to `_complete_request` from within libubus but the inner call could be made from the data outer callback.)
If the completion of the inner requests involves a reaction to an invoke, unsubscribe, or notify this scenario becomes a a deadlock.

[2]
```
ubus_complete_request stack_depth-- (now at 0)
ubus_invoke_async_fd obj 4187302718 method phy_status
ubus_send_msg type 5
ubus_send_msg msg '{,,}' (4)
ubus_process_pending_msg list_empty 1
__ubus_notify_async obj hostapd type bss.add
ubus_send_msg type 10
ubus_send_msg msg '{,"s.add":7094140481078322176,}' (31)
ubus_send_msg type 4
ubus_send_msg msg '{"rvice":{}}' (12)
ubus_complete_request stack_depth++ (now at 1)
      ubus_poll_data enter
ubus_process_msg buf->hdr.type 2
ubus_process_msg buf->hdr.type 1
ubus_send_msg type 4
ubus_send_msg msg '{"a_supplicant":{}}' (19)
ubus_complete_request stack_depth++ (now at 2)

# never goes back below 1 again

      ubus_poll_data enter
ubus_process_msg buf->hdr.type 5
ubus_queue_msg ctx->sock.registered 1
ubus_process_msg buf->hdr.type 5
ubus_queue_msg ctx->sock.registered 1
ubus_process_msg buf->hdr.type 1
ubus_process_msg buf->hdr.type 2
ubus_process_msg buf->hdr.type 1
      ubus_poll_data leave
      ubus_poll_data enter
ubus_process_msg buf->hdr.type 2
ubus_process_msg buf->hdr.type 1
      ubus_poll_data leave
ubus_complete_request stack_depth-- (now at 1)
ubus_invoke_async_fd obj 4187302718 method phy_set_state
ubus_send_msg type 5
ubus_send_msg msg '{,,}' (4)
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_process_msg buf->hdr.type 5
ubus_queue_msg ctx->sock.registered 1
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_process_msg buf->hdr.type 5
ubus_queue_msg ctx->sock.registered 1
ubus_process_pending_msg list_empty 0
ubus_process_pending_msg ctx->stack_depth 1
ubus_process_msg buf->hdr.type 1
ubus_send_msg type 4
ubus_send_msg msg '{"a_supplicant":{}}' (19)
ubus_complete_request stack_depth++ (now at 2)
      ubus_poll_data enter
ubus_process_msg buf->hdr.type 2
ubus_process_msg buf->hdr.type 1
      ubus_poll_data leave
ubus_complete_request stack_depth-- (now at 1)
ubus_invoke_async_fd obj 4187302718 method phy_set_state
ubus_send_msg type 5
ubus_send_msg msg '{,,}' (4)
ubus_process_msg buf->hdr.type 1
__ubus_notify_async obj hostapd type bss.add
ubus_send_msg type 10
ubus_send_msg msg '{,"s.add":7094140481078322176,}' (31)
ubus_send_msg type 4
ubus_send_msg msg '{"rvice":{}}' (12)
ubus_complete_request stack_depth++ (now at 2)
```

yogo1212, 2024-04-22:
I keep having ideas on how to fix this but none of them turn out to be very good or feasible to implement.
It just can't be solved as long as there is recursion of blocking completions (provided there are no cyclical dependencies in the code using ubus).

I've created a patch parsing packages one-by-one from `ubus_complete_request` (rather than using `ubus_handle_data` which has a loop itself and when nested doesn't reliably end the outer loop).
With the patch, I always get this under freeze conditions:

```
Exception in ubus function:
{ 
  "type": "Reference error",
  "message": "left-hand side expression is null",
  "stacktrace": [ 
    { 
      "filename": "/usr/share/hostap/hostapd.uc",
      "line": 739, "byte": 40,
      "function": "[anonymous function]",               
      "context": "In [anonymous function](), file /usr/share/hostap/hostapd.uc, line 739, byte 40:\n  called from anonymous function (/usr/share/hostap/hostapd.uc:721:22)\n\n `                let config = iface_load_config(phy.orig_file);`\n  Near here -----------------------------------------^\n"
    },
    { "filename": "/usr/share/hostap/hostapd.uc", "line": 721, "byte": 22 }
  ]
}
```

I'll start following that and hopefully find a fix.

Another bigger opportunity I see is making the ucode library do only async calls, pause the vm (atm, does so without returning, calling `uloop_run`), return to the event loop (atm, it only does uloop afaics), and resume once the result is ready.
Naturally, that only works if the code running ucode can handle the result asynchronously as well (users of wpa_ucode_call).
Much work. Much complexity. Probably not worth doing. Not that I would know - I've only just started looking into ubus.

yogo1212, 2024-04-28:
The lockup is still present after an update to the main branch (ff to 6e5edec159e2dc468607a5b9179f722857a38421).
 My patch for ubus only works on first glance.
 
<details><summary>Here's the config that reproduces the bug very well for me (Cudy M1800):</summary> 

```

config wifi-device 'radio24'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel '6'
	option band '2g'
	option htmode 'HE40'
	option country 'DE'

config wifi-device 'radio5'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option channel '48'
	option band '5g'
	option htmode 'HE80'
	option country 'DE'
	option cell_density '1'

config wifi-iface 'test1-24'
	option mode 'ap'
	option ifname 'test1-24'
	option device 'radio24'
	option network 'lan'
	option encryption 'psk2'
	option isolate '1'
	option hidden '0'
	option key 'manywoes'
	option ieee80211r '1'
	option reassociation_deadline '20000'
	option ft_over_ds '0'
	option ieee80211w '1'
	option ssid 'Woelan'

config wifi-iface 'test1-5'
	option mode 'ap'
	option ifname 'test1-5'
	option device 'radio5'
	option network 'lan'
	option encryption 'psk2'
	option isolate '1'
	option hidden '0'
	option key 'manywoes'
	option ieee80211r '1'
	option reassociation_deadline '20000'
	option ft_over_ds '0'
	option ieee80211w '1'
	option ssid 'Woelan'

config wifi-iface 'test2-24'
	option mode 'ap'
	option ifname 'test2-24'
	option device 'radio24'
	option network 'lan'
	option encryption 'psk2'
	option isolate '0'
	option hidden '0'
	option key 'donttrythis'
	option ieee80211r '0'
	option ieee80211w '1'
	option ssid 'no wifi'

config wifi-iface 'test2-5'
	option mode 'ap'
	option ifname 'test2-5'
	option device 'radio5'
	option network 'lan'
	option encryption 'psk2'
	option isolate '0'
	option hidden '0'
	option key 'donttrythis'
	option ieee80211r '0'
	option ieee80211w '1'
	option ssid 'no wifi'

config wifi-iface 'mesh24'
	option ifname 'mesh24'
	option network 'mesh24'
	option encryption 'sae'
	option mesh_id 'mesh'
	option bssid '04:21:f0:12:12:12'
	option key 'testmeshkey'
	option mode 'mesh'
	option mesh_fwding '0'
	option mesh_ttl '1'
	option device 'radio24'

```

</details>

The lockup happens with the snapshot build as well (see earlier comments).
Building a firmware with some debugging capabilities is probably required to work on this.

On restart from the locked up state, hostapd segfaults in a memory region that libc is loaded into but gdb can't find the beginning of the function and the stacktrace is garbage. I've looked at the config options, added `-g3`, `STRIP=/bin/true`, and `NO_STRIP=1` but gdb never does anything useful with the core file.

Because of that, I'm `schlogging`.
<details><summary>`schlog`:</summary>

```
#include <fcntl.h>
#include <unistd.h>

static void schlog(const char *fmt, ...)
{
	int fd = open("/proc/self/comm", O_RDONLY);
	char buf[512];
	read(fd, buf, sizeof(buf));
	close(fd);
	if (strncmp(buf, "hostapd", 7) != 0)
		return;

	fd = open("/tmp/schlog", O_CREAT | O_APPEND | O_WRONLY);
	if (fd == -1)
		return;

	va_list argp;
	va_start(argp, fmt);

	ssize_t len = vsnprintf(buf, sizeof(buf), fmt, argp);
	write(fd, buf, len);
	close(fd);
}
```

</details>

With the `schlog`, it's possible to trace hostapd's execution across libraries (add `schlog("%s\n", __func__);` anywhere you want). Very practical but I still get lost in the eloop/uloop/ucode/ubus engineering marvel that is hostapd.

Despite having followed many misinterpreted signs into dead ends, I'm fairly confident that a `ubus_lookup_id` for `"service"` called by ucode using an object loaded from [hostapd.uc](https://github.com/openwrt/openwrt/blob/main/package/network/services/hostapd/files/hostapd.uc#L882) never returns. At least, `stack_depth` never goes back down to zero - I can't confirm the call is actually still on the stack because I'm unable to obtain a proper stack trace.
There's some more meandering in the comments above that I'm not as sure of.

I have been throwing myself at this for weeks now and would I love to keep looking for the cause and, ultimately, create a fix but I'm leaving this be in order to stay sane and be able to tackle more important things in my life.

nbd168, 2024-05-01:
@yogo1212 thanks a lot for your detailed analysis. I just pushed fixes for two issues that I found (triggered with your config). Could you please test if this resolves your issue as well?

yogo1212, 2024-05-06:
@nbd168 indeed - that fixes it! :partying_face: 

now it can 't haunt me anymore... thank you! :smile:

yogo1212, 2024-05-06:
fixed

closed 2024-05-06 by yogo1212