openwrt/openwrt #14574: wifi-scripts: Script "wifi-detect.uc" fails if the object "wlan" is missing in "/etc/board.json"

Sybil127, 2024-02-07:
### Describe the bug

If the object "wlan" is missing in the file "/etc/board.json".
The ucode script "/usr/share/hostap/wifi-detect.uc" fails both when it's manually run, and when it's called from the program "/sbin/wifi", after running the command "wifi config". 

### OpenWrt version

r25029-07ea729761

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-B1300

### Image kind

Official downloaded image

### Steps to reproduce

1. Check that there is no object with the name "wlan" in the file "/etc/board.json", if there is, remove it.
2. Run the command 'wifi config' or run the script manually with 'ucode /usr/share/hostap/wifi-detect.uc'.

### Actual behaviour

```
root@OpenWrt:~# wifi config
Type error: attempt to set property on null value
In wiphy_get_entry(), file /usr/share/hostap/wifi-detect.uc, line 50, byte 2:
  called from function wiphy_detect (/usr/share/hostap/wifi-detect.uc:138:41)
  called from anonymous function (/usr/share/hostap/wifi-detect.uc:144:14)

 `    };`
      ^-- Near here

WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
```

### Additional info

One simple fix that I've tried, that seems to solve the issue.
Is to change line 42 in the script "/usr/share/hostap/wifi-detect.uc",
from `let wlan = board_data.wlan;` to `let wlan = board_data.wlan ??= {};`.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kimocoder, 2024-02-12:
`root@OpenWrt:/etc/modules.d# wifi config
Type error: attempt to set property on null value
In wiphy_get_entry(), file /usr/share/hostap/wifi-detect.uc, line 50, byte 2:
  called from function wiphy_detect (/usr/share/hostap/wifi-detect.uc:138:41)
  called from anonymous function (/usr/share/hostap/wifi-detect.uc:144:14)

 `    };`
      ^-- Near here

WARNING: Variable 'wlan' does not exist or is not an array/object`

edrikk, 2024-02-14:
Looks like this can be closed:  https://github.com/openwrt/openwrt/commit/2a2abed0be6078368b7dc6b7730794b481d32a0e

closed 2024-02-14 by nbd168