openwrt/openwrt #14575: Wireless 5GHz network crash when mesh network start

McGiverGim, 2024-02-07:
### Describe the bug

Follow up of: https://github.com/openwrt/openwrt/issues/14477

The original issue, shown a bootloop problem caused by the 5GHz network crashing when starting the mesh network and having a led attached to the mesh device. The issue was closed fixing the bootloop (the led crashing), but it seems, at least in my tests, that the problem with the wireless networks remain.

The problem is, that having this wireless configuration (one 5Ghz wireless network, one mesh network in the same radio, and one 2.4Ghz wireless network):
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/20000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '1'
        option cell_density '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'
        option country 'ES'

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi+1'
        option band '2g'
        option cell_density '0'
        option htmode 'HE40'
        option channel '6'
        option txpower '20'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'PIMINET'
        option ifname 'wlan5g'
        option key '*****'
        option network 'lan'
        option encryption 'psk2'

config wifi-iface 'wifinet2'
        option device 'radio1'
        option mode 'mesh'
        option mesh_fwding '1'
        option mesh_rssi_threshold '0'
        option network 'lan'
        option encryption 'sae'
        option mesh_id 'PIMIMESH'
        option key '*****'
        option ifname 'mesh'

config wifi-iface 'wifinet3'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'PIMINET_N'
        option ifname 'wlan24g'
        option key '*****'
        option encryption 'psk2'
```

When the router boots up, or after issuing a `wifi` command, the wireless 5Ghz (wlan5g) is down, in an inconsistent state:
[image]

After pressing `disable` and `enable` the wireless network is up again, and it works without problem.
The `wifi` command log is like this:
[... 129 lines cut ...]
qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Self-built image

### Steps to reproduce

Simply use the wireless config posted above, and issue the `wifi` command.

### Actual behaviour

The 5GHz network is down.

### Expected behaviour

The 5GHz network must be up.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-07:
Can you make wpad  stop and start 1 minute apart?

Ansuel, 2024-02-08:
@nbd168 Wonder if there is some kind of race with interface rename and wifi going up? The ubox changes made thing some ms slow due to kernel module parsing. Any clue?

McGiverGim, 2024-02-08:
> Can you make wpad stop and start 1 minute apart?

I tested and the same problem. The 5GHz network does not start. I can start it from LuCI without problem later (I don't know what is the command to start it from the command line).

brada4, 2024-02-08:
You can try to sabotage ujail in /etc/init.d/wpad. ie if problem is in hostapd+supplicant or ujail sandbox.
I wanted slow re-start and split start and stop logs out of it.

McGiverGim, 2024-02-08:
> You can try to sabotage ujail in /etc/init.d/wpad. ie if problem is in hostapd+supplicant or ujail sandbox.

I don't understand exactly what you are saying with this, my knowledge in OpenWrt is not too big.

Repeating the "slow" test, if I wait a little more than before, the crash is bigger and nothing starts:
`service wpad stop`
```
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: Remove interface 'phy2'
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan24g: interface state ENABLED->DISABLED
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan24g: AP-DISABLED
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan24g: CTRL-EVENT-TERMINATING
Thu Feb  8 08:11:04 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: MESH-GROUP-REMOVED mesh
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: leaving mesh
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan24g disabled_11b_rates=0
Thu Feb  8 08:11:04 2024 daemon.notice netifd: Network device 'wlan24g' link is down
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.011982] br-lan: port 5(wlan24g) entered disabled state
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.103856] br-lan: port 5(wlan24g) entered disabled state
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.104299] device wlan24g left promiscuous mode
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.108363] br-lan: port 5(wlan24g) entered disabled state
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: Remove interface 'phy1'
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan5g: interface state ENABLED->DISABLED
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan5g: AP-STA-DISCONNECTED ee:c3:e5:2d:d6:d4
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: interface state ENABLED->DISABLED
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: AP-DISABLED
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan5g: AP-DISABLED
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: wlan5g: CTRL-EVENT-TERMINATING
Thu Feb  8 08:11:04 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb  8 08:11:04 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan5g disabled_11b_rates=0
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.363130] device wlan5g left promiscuous mode
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.363296] br-lan: port 6(wlan5g) entered disabled state
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: CTRL-EVENT-DSCP-POLICY clear_all
Thu Feb  8 08:11:04 2024 daemon.notice netifd: Network device 'wlan5g' link is down
Thu Feb  8 08:11:04 2024 daemon.notice netifd: Network device 'mesh' link is down
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.455816] br-lan: port 7(mesh) entered disabled state
Thu Feb  8 08:11:04 2024 daemon.notice netifd: Interface 'lan6' has lost the connection
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: CTRL-EVENT-DSCP-POLICY clear_all
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: nl80211: deinit ifname=mesh disabled_11b_rates=0
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: mesh: CTRL-EVENT-TERMINATING
Thu Feb  8 08:11:04 2024 daemon.err wpa_supplicant[13234]: rmdir[ctrl_interface=/var/run/wpa_supplicant]: Permission denied
Thu Feb  8 08:11:04 2024 daemon.notice wpa_supplicant[13234]: Remove interface mesh
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.505951] br-lan: port 7(mesh) entered blocking state
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.506006] br-lan: port 7(mesh) entered forwarding state
Thu Feb  8 08:11:04 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb  8 08:11:04 2024 kern.info kernel: [147631.603951] br-lan: port 7(mesh) entered disabled state
Thu Feb  8 08:11:05 2024 kern.info kernel: [147631.643374] device mesh left promiscuous mode
Thu Feb  8 08:11:05 2024 kern.info kernel: [147631.643413] br-lan: port 7(mesh) entered disabled state
Thu Feb  8 08:11:05 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Thu Feb  8 08:11:05 2024 daemon.info dnsmasq[1]: using nameserver 192.168.100.1#53 for domain piminet
Thu Feb  8 08:11:05 2024 daemon.info dnsmasq[1]: using nameserver 192.168.100.1#53
Thu Feb  8 08:11:05 2024 daemon.info dnsmasq[1]: using only locally-known addresses for piminet
Thu Feb  8 08:11:05 2024 daemon.notice wpa_supplicant[13234]: ELOOP: remaining socket: sock=23 eloop_data=0x7f9ae76510 user_data=0x7f9ae76510 handler=0x5587bd3388
Thu Feb  8 08:11:05 2024 daemon.warn netifd: Wireless device 'radio1' setup failed, retry=3
Thu Feb  8 08:11:05 2024 daemon.notice netifd: radio1 (15526): Command failed: Not found
Thu Feb  8 08:11:05 2024 daemon.notice netifd: radio1 (15526): Command failed: Not found
Thu Feb  8 08:11:05 2024 daemon.notice netifd: Wireless device 'radio1' is now down
Thu Feb  8 08:11:05 2024 daemon.notice netifd: radio1 (15538): WARNING: Variable 'data' does not exist or is not an array/object
Thu Feb  8 08:11:05 2024 daemon.notice netifd: radio1 (15538): Command failed: Not found
Thu Feb  8 08:11:05 2024 daemon.notice netifd: radio1 (15538): Command failed: Not found
Thu Feb  8 08:11:05 2024 daemon.warn netifd: Wireless device 'radio2' setup failed, retry=3
Thu Feb  8 08:11:05 2024 daemon.notice netifd: radio2 (15587): Command failed: Not found
[... cut ...]
Thu Feb  8 08:16:06 2024 daemon.notice wpa_supplicant[16656]: Successfully initialized wpa_supplicant
Thu Feb  8 08:16:30 2024 kern.info kernel: [147957.620879] device br-lan entered promiscuous mode
Thu Feb  8 08:16:31 2024 daemon.warn odhcpd[2110]: No default route present, overriding ra_lifetime!
Thu Feb  8 08:16:32 2024 kern.info kernel: [147959.521762] device br-lan left promiscuous mode
```
but all the wireless network are down. Only a new `wifi` command starts them (except the 5GHz network, as expected)

nbd168, 2024-02-08:
I think ath11k (and other ath drivers) might be getting misdetected as fullmac. Could you please test this patch? https://nbd.name/p/6701a49d

McGiverGim, 2024-02-08:
> I think ath11k (and other ath drivers) might be getting misdetected as fullmac. Could you please test this patch? https://nbd.name/p/6701a49d

I applied directly the patch into the router manually, and after a reboot of the router, the 5GHz network is up. I issued various `wifi` commands and all the wireless networks are up.
If I do a `service wpad stop` the crash posted before continues, but I think this is another war. And the wireless networks don't start with a `service wpad start`, I don't know if it's expected, only after a `wifi` command the wireless networks are up again.

So the patch seems to work, great job and thanks! :-)

McGiverGim, 2024-02-09:
@nbd168 do you need some test more for my part?

closed 2024-02-09 by openwrt-bot

nbd168, 2024-02-09:
Applied, thanks for testing.

McGiverGim, 2024-02-09:
The crash when stopping the service wpad is "expected"? To open a new issue if there is interest.

brada4, 2024-02-09:
You mean ENOPERM?

McGiverGim, 2024-02-09:
I talk about all the ugly logs that appear after the wpad stop in the log. Maybe all of them are expected after stopping this service.