openwrt/openwrt #14576: openwrt.org dokuwiki error / down

stefanfoulis, 2024-02-07:
### Describe the bug

https://openwrt.org/toh/start and many other locations are showing an error.

```
[Wiki is down: Error: Call to undefined method syntax_plugin_data_table::_buildSQL()](https://forum.openwrt.org/t/wiki-is-down-error-call-to-undefined-method-syntax-plugin-data-table-buildsql/187185)
```

I followed the link for reporting infrastructure errors at https://openwrt.org/wiki/start#contact and ended up on this bug-tracker. I'm not sure this really is the right place to report it, but it since the contact page linked me here :man_shrugging: .
I couldn't find an issue reported here yet, so I created a new one.

There are already some forum posts reporting the issue:

- https://forum.openwrt.org/t/wiki-is-down-error-call-to-undefined-method-syntax-plugin-data-table-buildsql/187185
- https://forum.openwrt.org/t/issues-on-the-main-site-today/187215

### OpenWrt version

-

### OpenWrt release

-

### OpenWrt target/subtarget

-

### Device

-

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Snowiiii, 2024-02-07:
i get the same error

kazin-kharizma, 2024-02-08:
I am getting this error as well and its coming at a really bad time.

udanfacy22, 2024-02-08:
> I am getting this error as well and its coming at a really bad time.

For the time being, until this gets fixed, I suggest using the Internet Wayback Machine to access recent copies of the information on the website. https://web.archive.org/web/20240207025448/https://openwrt.org/

stintel, 2024-02-08:
Please don't report site issues here. Either report it to the openwrt-adm list or #openwrt-adm on IRC.

closed 2024-02-08 by stintel

stefanfoulis, 2024-02-08:
thanks for the hint @stintel. It would make sense to update the information at https://openwrt.org/wiki/start#contact then, because there it states that infrastructure outages should be reported here (it links to bugs.openwrt.org, which redirects here).
[image]