openwrt/openwrt #14589: Zyxel EX5601-T0 Randomically Unable to handle kernel paging request at virtual address

SuperKali, 2024-02-10:
### Describe the bug

Hello everyone,

I have been experiencing random router crashes on the mentioned router for some time, as detected by my syslog server. I'm unsure of the root cause. I attempted to compile the latest version of OpenWRT, version 23.05.2, but I encounter these crashes more frequently compared to the currently installed version.

Honestly, I am uncertain how to obtain the entire backlog since it is printed on the serial console. The syslog server doesn't provide me with sufficient details. If anyone knows how to obtain these advanced logs without using the serial console, please let me know, as these issues are not consistently encountered.

I am attaching what I managed to retrieve from my syslog server.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0

### Image kind

Self-built image

### Steps to reproduce

There are no steps to reproduce this crash as it is random and unpredictable.

### Actual behaviour

The router automatically crashes due to kernel errors.

### Expected behaviour

The correct operation of the router.

### Additional info

``` bash
10-Feb-2024 08:13:19.538 192.168.178.1 alert  [1194297.198858] Unable to handle kernel paging request at virtual address 001436ac663a8d70
10-Feb-2024 08:13:19.538 192.168.178.1 alert  [1194297.206952] Mem abort info:
10-Feb-2024 08:13:19.538 192.168.178.1 alert  [1194297.209911]   ESR = 0x0000000096000004
10-Feb-2024 08:13:19.548 192.168.178.1 alert  [1194297.213822]   EC = 0x25: DABT (current EL), IL = 32 bits
10-Feb-2024 08:13:19.548 192.168.178.1 alert  [1194297.219298]   SET = 0, FnV = 0
10-Feb-2024 08:13:19.559 192.168.178.1 alert  [1194297.222514]   EA = 0, S1PTW = 0
10-Feb-2024 08:13:19.559 192.168.178.1 alert  [1194297.225812]   FSC = 0x04: level 0 translation fault
10-Feb-2024 08:13:19.559 192.168.178.1 alert  [1194297.230848] Data abort info:
10-Feb-2024 08:13:19.569 192.168.178.1 alert  [1194297.233888]   ISV = 0, ISS = 0x00000004
10-Feb-2024 08:13:19.569 192.168.178.1 alert  [1194297.237885]   CM = 0, WnR = 0
10-Feb-2024 08:13:19.569 192.168.178.1 alert  [1194297.241016] [001436ac663a8d70] address between user and kernel address ranges
10-Feb-2024 08:14:16.380 192.168.178.1 info  exiting on receipt of SIGTERM
```

### Diffconfig

```text
i don't have it, i'm using the version of @pameruoso
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-10:
Since you are up/downgrading in own build tree try to install official binaries that do not depend on stale old package versions:
```
auc -n -f -B 23.05.0
```
(either sysupgrade with download link shown or remove -n parameter)
Also same way you can upgrade to 23.05.2 with clean binaries (assuming you have your "less crashing" image at hand.

Once on "official image" either establish remote syslog forwarding or run logread -f in console until crash. (or capture serial console for extended period, whatever seems viable)

SuperKali, 2024-02-10:
@brada4 Perhaps you misunderstood. Version 23.05.0 was initially compiled by another person, a friend of mine, starting from version 23.05.0. Subsequently, I compiled version 23.05.2 on my own, but both versions exhibit the same issue. Interestingly, the version compiled by my friend, 23.05.0, appears to be more stable than my 23.05.2, which seems to crash more frequently.

brada4, 2024-02-10:
Thats the reason I ask you to install pure produce of openwrt buildbots.

SuperKali, 2024-02-10:
I understand your concept, but I have always compiled OpenWRT without encountering any issues. I don't understand what changes between my compilation and the OpenWRT buildroot. I cannot install the buildroot as there are no necessary packages available, and some essential packages for establishing the PPPOE connection are missing. I would still need to use the OpenWRT firmware sector and manually write package by package, which doesn't seem like the correct approach. Otherwise, there would be no point in having people compile the images directly.

That being said, I started with the config.buildinfo from 23.05.2 on the OpenWRT website. Then, I added the necessary packages to enable the internet connection. However, it seems that 23.05.2 has more issues than 23.05.0, where I at least encountered a crash after only 14 days.

brada4, 2024-02-10:
Please convert to OpenWRT built packages using auc. Yes, your diffconfig lacks any pppoe package like kmods and pppd pppoe plugin.

SuperKali, 2024-02-10:
[image]
The command is not working

brada4, 2024-02-10:
@aparcar should know better wrt auc.

In the meantime you can try `opkg list-installed | cut -d\  -f1` and paste that over at firmware-selector.

SuperKali, 2024-02-10:
@brada4 I've do it now, I've upload the 23.05.2 version and the first problem I have noticed, the led 2.5Gbit is not working and the lan leds too

SuperKali, 2024-02-10:
it's like mdio is not correct configured, for reference https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=1c05388ab04c934ec240e8362321908f91381a90:

Leds description
------------
All the leds are working out of the box but the leds managed by the 2 maxlinear phy (phy 5 lan, phy6 wan).
To activate the phy5 led (rj45 ethernet port led on the back of the router) you have to use mdio-tools.
To activate the phy6 led (led on the front of the router for 2.5gbit link) you have to use mdio-tools.
Example:
Set lan5 led to fast blink on 2500/1000, slow blink on 10/100:
mdio mdio-bus mmd 5:30 raw 0x0001 0x33FC

Set wan 2.5gbit led to constant on when wan is 2.5gbit:
mdio mdio-bus mmd 6:30 raw 0x0001 0x0080

SuperKali, 2024-02-10:
In additional I tried also with 23.05.0 compiled with firmware selector and the same problem with leds

brada4, 2024-02-10:
Problem is memory protection error, certainly not caused by mdio programming or not programming,..
If you feel strong with dts editing you can add led control via PR there, but is is not related to memory error reported.

brada4, 2024-02-10:
What I suggest is to generate sysupgrade image via firmware selector or imagebuilder to have exact openwrt binaries in place to distinguish if your buildroot cached something (which is openwrt bug anyway) or there is broken memory management in place. Next step if it crashes with nonspecific fault address again will be to build kernel with debug symbols, courtesy of your buildroot....

SuperKali, 2024-02-10:
@brada4 So far there I knew it too, about the leds, yes they are definitely missing in the DTS, in fact my friend had put them on his version, about the memory problem I don't know how to take more logs apart from my syslog server. because as I said it happens randomly

brada4, 2024-02-10:
[image]

You have to reproduce problem with OpenWRT, customisations you describe clearly make your installation a fork.

SuperKali, 2024-03-08:
FYI, I'm testing a build by the firmware selector i need time

closed 2025-04-29 by SuperKali