openwrt/openwrt #14593: collectd service does not generate rrdtool config, resulting in flooding logs with errors

carbolymer, 2024-02-10:
### Describe the bug

`/etc/init.d/collectd` does not generate correct config for rrdtool which results in logs getting flooded by error messages.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Official downloaded image

### Steps to reproduce

1. I have following collectd plugins installed:
    ```
    # opkg list-installed | grep collectd
    collectd - 5.12.0-49
    collectd-mod-conntrack - 5.12.0-49
    collectd-mod-cpu - 5.12.0-49
    collectd-mod-cpufreq - 5.12.0-49
    collectd-mod-dns - 5.12.0-49
    collectd-mod-ethstat - 5.12.0-49
    collectd-mod-exec - 5.12.0-49
    collectd-mod-interface - 5.12.0-49
    collectd-mod-ipstatistics - 5.12.0-49
    collectd-mod-iptables - 5.12.0-49
    collectd-mod-irq - 5.12.0-49
    collectd-mod-iwinfo - 5.12.0-49
    collectd-mod-load - 5.12.0-49
    collectd-mod-memory - 5.12.0-49
    collectd-mod-network - 5.12.0-49
    collectd-mod-ping - 5.12.0-49
    collectd-mod-processes - 5.12.0-49
    collectd-mod-rrdtool - 5.12.0-49
    collectd-mod-sqm - 5.12.0-49
    collectd-mod-thermal - 5.12.0-49
    collectd-mod-uptime - 5.12.0-49
    collectd-mod-wireless - 5.12.0-49
    ```
2. I have the following configuration:
    ```
    # uci show collectd
    collectd.globals=globals
    collectd.globals.alt_config_file='/etc/collectd.conf'
    collectd.globals.BaseDir='/data/collectd'
    collectd.globals.ReadThreads='2'
    collectd.globals.WriteThreads='2'
    collectd.conntrack=plugin
    collectd.conntrack.enable='1'
    collectd.rrdtool=plugin
    collectd.rrdtool.enable='1'
    collectd.rrdtool.DataDir='/data/rrd'
    collectd.rrdtool.RRARows='100'
    collectd.rrdtool.RRASingle='1'
    collectd.rrdtool.RRATimespan='3600' '86400' '604800' '2678400' '31622400'
    collectd.thermal=plugin
    collectd.thermal.enable='0'
    collectd.thermal.IgnoreSelected='0'
    collectd.thermal.Device=''
    ```
3. After restarting `collectd` using: `service collectd restart` the following configuration file is generated:
    ```
    # cat /tmp/collectd.conf 
    BaseDir "/data/collectd"
    Include "/etc/collectd/conf.d/*.conf"
    PIDFile "/var/run/collectd.pid"
    PluginDir "/usr/lib/collectd"
    TypesDB "/usr/share/collectd/types.db"
    Interval 30
    ReadThreads 2
    
    LoadPlugin conntrack
    
    LoadPlugin cpu
    <Plugin cpu>
	    ValuesPercentage true
	    ReportByCpu true
	    ReportByState true
    </Plugin>
    
    LoadPlugin cpufreq
    
    LoadPlugin dns
    <Plugin dns>
	    Interface "br-lan"
	    IgnoreSource "127.0.0.1"
    </Plugin>
    
    LoadPlugin interface
    <Plugin interface>
	    IgnoreSelected false
	    Interface "br-lan"
    </Plugin>
    
    LoadPlugin ipstatistics
    ```

### Actual behaviour

The configuration file does not contain rrdtool configuration, which results in system logs getting flooded by multiple entries (45 new lines) every 30 seconds:
```
Sat Feb 10 15:44:47 2024 daemon.err collectd[29717]: Available write targets:: [none]
Sat Feb 10 15:44:47 2024 daemon.err collectd[29717]: Available write targets:: [none]
Sat Feb 10 15:44:47 2024 daemon.err collectd[29717]: Available write targets:: [none]
Sat Feb 10 15:44:47 2024 daemon.err collectd[29717]: Available write targets:: [none]
Sat Feb 10 15:44:47 2024 daemon.err collectd[29717]: Available write targets:: [none]
Sat Feb 10 15:44:47 2024 daemon.err collectd[29717]: Available write targets:: [none]
``` 

### Expected behaviour

The configuration file contains correct rrdtool configuration. `collectd` does not flood logs with errors.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2024-02-11:
sounds like this is possibly related to LuCI statistics (in luci feed repo), not into collectd itself (in the packages repo). Neither of which is in the main OpenWrt repo that this bug tracker concerns...

If you want "normal" OpenWrt statistics, please install the luci-app-statistics which then takes care of creating the config for the underlying collectd.

carbolymer, 2024-02-11:
@hnyman Thanks for the reply. I'm using `luci-app-statistics` too , and still `rrd` does not get configured if I enable it through web UI. How is that a different case? The configuration file is created by the [service file](https://github.com/openwrt/packages/blob/e73ac3534aafe688da54673cf345859744eafa37/utils/collectd/files/collectd.init#L268), so I understand that the configuration path is the same? 

Should I move this issue to openwrt/packages repo?

egc112, 2024-02-14:
I had the same spamming of the log after I (wrongly) disabled the rrdtool output plugin in luci-app-statistics