openwrt/openwrt #14596: EAP235-Wall: lan0 (PoE-PD uplink) dead and detached from eth0 on kernel 6.1

Borromini, 2024-02-10:
### Describe the bug

On kernel 6.1, the lan0 interface (PoE-PD port and uplink) fails to register and is not linked to eth0, rendering the device inaccessible. Error seen over serial:
```
[    8.974464] mtk_soc_eth 1e100000.ethernet lan0: validation of  with support 00000000,00000000,00006000 and advertisement 00000000,00000000,00000000 failed: -EINVAL
[    9.003873] mtk_soc_eth 1e100000.ethernet lan0: mtk_open: could not attach PHY: -22
ip: SIOCSIFFLAGS: Invalid argument
```

Full boot log linked below.

Ip will show lan0 as detached from eth0:
```
# ip a s
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::a62b:b0ff:fe2e:e988/64 scope link 
       valid_lft forever preferred_lft forever
3: lan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop master br-lan state DOWN qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
4: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
5: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
6: lan1@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
7: br-lan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
8: br-lan.1@br-lan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN qlen 1000
    link/ether a4:2b:b0:2e:e9:88 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.3/24 brd 10.0.0.255 scope global br-lan.1
       valid_lft forever preferred_lft forever
```

### OpenWrt version

r25115+29-7181eb9f81

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link EAP235-Wall v1

### Image kind

Self-built image

### Steps to reproduce

Check out main tree, compile for EAP235-Wall with testing kernel, flash to device (or boot from TFTP).

### Actual behaviour

Lan0 is not registered correctly, making the PoE-PD uplink port inaccessibile.

### Expected behaviour

[... 747 lines cut ...]
CONFIG_PACKAGE_umdns=y
CONFIG_PACKAGE_unbound-anchor=m
CONFIG_PACKAGE_unbound-control=m
CONFIG_PACKAGE_unbound-daemon=y
CONFIG_PACKAGE_usteer=y
CONFIG_PACKAGE_watchcat=m
CONFIG_PACKAGE_wifi-scripts=y
CONFIG_PACKAGE_wifischedule=y
CONFIG_PACKAGE_wireguard-tools=m
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_wpa-supplicant-openssl=m
CONFIG_PACKAGE_wpad-basic-mbedtls=m
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PREINITOPT=y
CONFIG_TARGET_PREINIT_BROADCAST="10.0.0.255"
CONFIG_TARGET_PREINIT_IP="10.0.0.254"
CONFIG_TESTING_KERNEL=y
# CONFIG_LIBCURL_FTP is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-10:
> not linked to eth0

it has separate SoC SerDes.
Can you explain 29 local patches?

Borromini, 2024-02-11:
> > not linked to eth0
> 
> it has separate SoC SerDes.

Are you referring to the `&gmac1` entry in the DTS? You might be right about that, I don't have a functional boot log handy ATM. Either way, it's clear that `lan0` is broken.

> Can you explain 29 local patches?

Sure. They're unrelated, see list below. The same tree with 5.15 runs just fine on this device.

```
$ git log --oneline|head -30
0e69b1d04a mvebu: RB5009: use port 8 as WAN port
543ddfad2c kernel: 5.15: refresh RB5009UG patches
48ff8c936b kernel: replace QCA8081 2.5Gb patch with upstream one
6f9fe7dab4 generic: replace at803x null pointer fix with upstream one
a060271cdc generic: backport 5.18 AT803x fiber support patch
9dd0e6467c mvebu: rb5009: add QCA8081 PHY support for 5.15
2fcf9cf6b7 kernel: 5.15: backport QCA8081 PHY support
ab52956508 mvebu: add Amethyst specific SMI setup
090f8bd874 mvebu: cortexa72: schedutil as default scheduler
3209758959 mvebu: cortexa72: enable SBSA Watchdog, Mikrotik RB Sysfs and MTD_SPLIT_FIT_FW
ce601e2fee mvebu: cortexa72: add missing symbol
6567d412d5 mvebu: rb5009: use QCA8081 label for 2.5 GbE port in DTS
157af752dc mvebu: rb5009: add SFP GPIO pins
c907663b97 mvebu: RB5009: remove sfp port from wan bridge
f1ee7b69fe mvebu: RB5009: update aux-loader.elf
05c86c0300 mvebu: RB5009: start to use UBI
720f752cbc mvebu: mvebu: add support for RB5009UG+S+IN.
7cd066f9bb mvebu: add aux-loader2 support
1b6f4c1c9a generic: fix mikrotik/rb_softconfig work(crc mismatch) with 64K block size
6dbe2e8133 octeon: disable CONFIG_CAVIUM_CN63XXP1 and EdgeRouter image
af411f8c57 octeon: rename erlite to EdgeRouter Lite
a007c737c6 realtek-poe: update to 2022-08-28 codebase
853b03431e realtek-poe: update to 2022-07-19 codebase
faea584641 realtek-poe: add GitHub based package
8b5cea080f Kernel: expose SERIAL_8250 and disable by default on SMALL_FLASH targets.
502e95bd18 Kernel: disable EARLY_PRINTK by default on mvebu
b3156e7834 Kernel: make EARLY_PRINTK available on MIPS as well.
c9040abf0e base-files: profile: personal enhancements
6f22772bf1 gitignore: ignore .precheck status file as well
7181eb9f81 ramips: add support for 6.1 kernel
```

brada4, 2024-02-11:
run ifconfig lan0 up and ethtool lan0 , boot message needsvto say like hsgmii wher€ 2 spaces are.
maybe mii-tool can find correct mii addres

and revert unrelated patches or rebuild in a clean tree.

Borromini, 2024-02-11:
From a clean tree, it boots just fine over TFTP, however, once flashed, exactly the same behaviour as my patched tree.

Any instructions on how to work with mii-tool and what I should be looking at?

<details>
<summary>Boot log r25115-7181eb9f81 over TFTP</summary>

```
MT7621 # bootm
## Booting image at 84f00000 ...
text base: 80001000
entry point: 80001000
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 6.1.77 (5800X@crunchbot) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24624+30-957a0921e2) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Feb 10 12:06:05 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is TP-Link EAP235-Wall v1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14944 r8192 d21920 u45056
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00028103
[    0.000000] Readback ErrCtl register=00028103
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 111432K/131072K available (7434K kernel code, 633K rwdata, 916K rodata, 8924K init, 220K bss, 19640K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.015996] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.088302] pid_max: default: 32768 minimum: 301
[    0.098329] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.112738] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.131297] cblist_init_generic: Setting adjustable number of callback queues.
[    0.145605] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.158153] rcu: Hierarchical SRCU implementation.
[    0.167524] rcu: 	Max phase no-delay instances is 1000.
[    0.178683] smp: Bringing up secondary CPUs ...
[    0.188458] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.188484] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.188499] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.188546] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.248175] Synchronize counters for CPU 1: done.
[    0.309839] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.309860] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.309873] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.309903] CPU2 revision is: 0001992f (MIPS 1004Kc)
[... 542 lines cut ...]
BusyBox v1.36.1 (2024-02-10 14:00:51 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r25115-7181eb9f81
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# dmesg -l err
[    2.265958] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    2.280548] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    2.340871] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    2.525497] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    9.022749] mtk_soc_eth 1e100000.ethernet lan0: mtk_open: could not attach PHY: -22
[   25.345900] mtk_soc_eth 1e100000.ethernet lan0: mtk_open: could not attach PHY: -22
```

</details>

brada4, 2024-02-11:
Do you netboot using that port or other?

Borromini, 2024-02-11:
I do, but TFTP boot involves U-boot, so it's not OpenWrt.

brada4, 2024-02-11:
It may perform some magic PHY initialisation routine vs cold boot from flash.
Try -r or -R to mii-tool or ethtool -r. one may repeat those.

Borromini, 2024-02-11:
While booted from RAM or booted from flash?

brada4, 2024-02-11:
When it does not work, i.e from tftp

Borromini, 2024-02-11:
```
root@OpenWrt:/# mii-tool -r lan0
SIOCGMIIPHY on 'lan0' failed: Not supported
root@OpenWrt:/# mii-tool -R lan0
SIOCGMIIPHY on 'lan0' failed: Not supported
root@OpenWrt:/# ethtool -r lan0
```

Turning off autonegotiation doesn't seem to do anything either:
```
root@OpenWrt:/# ethtool -s lan0 autoneg off speed 100
root@OpenWrt:/# ip link set lan0 up
[ 1978.879851] mtk_soc_eth 1e100000.ethernet lan0: validation of  with support 00000000,00000000,00006000 and advertisement 00000000,00000000L
[ 1978.909372] mtk_soc_eth 1e100000.ethernet lan0: mtk_open: could not attach PHY: -22
ip: SIOCSIFFLAGS: Invalid argument
```

I don't think this is a negotiation problem, 'could not attach PHY' sounds like something lower level. I understand your point about u-boot maybe initialising the PHY, what's weird is this issue did not pop up with older releases (hardware has been supported since 21.02), and I did not touch the bootloader.

What I do see in the EAP615-Wall DTS (which is very similar) is this, which is not in the EAP235-Wall DTS:

```
&ethernet {
        reg = <0x1e100000 0xe000>;
};
```

I have added that to the EAP235-Wall DTS, but doesn't make a difference, same -EINVAL and 'could not attach PHY' error.

brada4, 2024-02-12:
Can you try undoing this:
https://github.com/openwrt/openwrt/commit/f1c9afd801380a05a91d979b475c76cc0a67caae#diff-daa6157145752ee194c26809d2b56f174206f91da46d698223237847b94eb392
Your device is not the one wan-lan that would benefit from lan-wan split a lot.

Borromini, 2024-02-12:
That commit is part of 23.05 as well, and 23.05 works correctly on this device.

KliXeR, 2024-02-18:
Identical issue on TP-Link Archer C6U on kernel 6.1 where wan port is dead.

> mtk_soc_eth 1e100000.ethernet wan: validation of  with support 00000000,00000000,00006000 and advertisement 00000000,00000000,00000000 failed: -EINVAL
mtk_soc_eth 1e100000.ethernet wan: mtk_open: could not attach PHY: -22

Borromini, 2024-02-19:
@KliXeR Does reverting https://github.com/openwrt/openwrt/commit/f1c9afd801380a05a91d979b475c76cc0a67caae for your device fix it? Mind you, you're not able to revert it cleanly, so you have to edit your DTS manually.

brada4, 2024-02-19:
@Borromini semi-dup https://github.com/openwrt/openwrt/issues/14673 says it used to work with linux-5.15 so hardware is well wired (often)

Borromini, 2024-02-19:
> @Borromini semi-dup #14673 says it used to work with linux-5.15 so hardware is well wired (often)

Yes, as I stated in my bug report above:

>**Expected behaviour**
>
>Lan0 correctly registered and linked to eth0 as on kernel 5.15.

Thanks a lot for your help in pinpointing this btw. Hoping stintel can chime in concerning the EAP615, he ported it so pretty sure he's got serial.

Borromini, 2024-03-24:
@brada4 Would you mind PM'ing me your name & e-mail so I can add your suggested-by when I send this in as a patch?

Thanks

brada4, 2024-03-24:
Seems problem is more widespread, likely applies to all mtk eth to mii direct connections and introduced in kernel 6.1. You can find me at fw4, but fine uncredited too.

arinc9, 2024-03-24:
@Borromini send what in as a patch?

brada4, 2024-03-24:
I think it is not worth reverting it, verified on RT-AX54 that kernel 6.1 as of today cannot init the split-out extra mii while 5.15 works fine on same hw.

Borromini, 2024-03-24:
> I think it is not worth reverting it, verified on RT-AX54 that kernel 6.1 as of today cannot init the split-out extra mii while 5.15 works fine on same hw.

Why do you think it's not worth reverting? It breaks multiple devices. Unless there's a way to _fix_ it of course...

> @Borromini send what in as a patch?

Reverting https://github.com/openwrt/openwrt/commit/f1c9afd801380a05a91d979b475c76cc0a67caae for the EAP235-Wall and probably the EAP615-Wall as well. I meant to ask you if you had found the time to look into it, sorry.

brada4, 2024-03-24:
v23 introduced both SoC side MII-s used which doubles (permitting full gigabit NAT) the capacity of those routers. It is like downgrade just to have latest version number shown.

Borromini, 2024-03-24:
I'm fully aware. But a dead AP is of no use to me, so if this cannot fixed, then reverting that commit is the next best thing.

arinc9, 2024-03-24:
@Borromini do you still have this issue on the latest main branch? I couldn't reproduce it. I've used the exact device tree of EAP235-Wall v1 on my Unielec U7621-06 to reduce variables.

<details>
  <summary>Boot log</summary>

```
[    0.000000] Linux version 6.1.82 (arinc9@arinc9-PC) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.2.0 r23773+1864-6b63fcf8df) 13.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Mar 23 22:47:29 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is TP-Link EAP235-Wall v1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s18016 r8192 d22944 u49152
[    0.000000] pcpu-alloc: s18016 r8192 d22944 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00022630
[    0.000000] Readback ErrCtl register=00022630
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 245844K/262144K available (7367K kernel code, 627K rwdata, 1596K rodata, 3708K init, 222K bss, 16300K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008008] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066205] pid_max: default: 32768 minimum: 301
[    0.071652] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078862] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.089635] cblist_init_generic: Setting adjustable number of callback queues.
[    0.096856] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.103311] rcu: Hierarchical SRCU implementation.
[    0.108035] rcu: 	Max phase no-delay instances is 1000.
[    0.113879] smp: Bringing up secondary CPUs ...
[    0.119166] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.119192] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.119206] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.119255] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.178518] Synchronize counters for CPU 1: done.
[    0.210373] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.210393] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.210405] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.210434] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.269320] Synchronize counters for CPU 2: done.
[    0.299954] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.299975] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.299986] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.300019] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.354507] Synchronize counters for CPU 3: done.
[    0.384363] smp: Brought up 1 node, 4 CPUs
[    0.393739] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.403550] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.410607] pinctrl core: initialized pinctrl subsystem
[    0.417289] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... 167 lines cut ...]
[   34.488494] mtk_soc_eth 1e100000.ethernet lan0: configuring for phy/rgmii link mode
[   34.506301] br-lan: port 1(lan0) entered blocking state
[   34.516894] br-lan: port 1(lan0) entered disabled state
[   34.528093] device lan0 entered promiscuous mode
[   34.577609] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   34.598518] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.613652] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
[   34.631149] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   34.644688] br-lan: port 2(lan1) entered blocking state
[   34.655362] br-lan: port 2(lan1) entered disabled state
[... cut ...]
[   34.764914] mt7530-mdio mdio-bus:1f lan3: configuring for phy/gmii link mode
[   34.781174] br-lan: port 4(lan3) entered blocking state
[   34.791869] br-lan: port 4(lan3) entered disabled state
[   34.804654] device lan3 entered promiscuous mode
[   37.631562] mtk_soc_eth 1e100000.ethernet lan0: Link is Up - 1Gbps/Full - flow control off
[   37.648155] br-lan: port 1(lan0) entered blocking state
[   37.658598] br-lan: port 1(lan0) entered forwarding state
[   37.670676] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
``` 
</details>

brada4, 2024-03-24:
@arinc9 mine works now with 6.1 self built and dl snapshot.

Borromini, 2024-03-25:
A clean build works again, custom settings break. So I'll have to do some digging. Thanks @arinc9.

DragonBluep, 2024-03-25:
If only initramfs works, I think there are some conflicts or dependence issues with some kernel modules. Any output by command `ls /sys/class/devlink`?

DragonBluep, 2024-03-25:
If initramfs works, I think there are some conflicts or dependence issues with some kernel modules. Any output by command `ls /sys/class/devlink`?

Borromini, 2024-03-25:
That looks identical at first sight.

Initramfs:
```
ls /sys/class/devlink
phy:phy-1e149000.pcie-phy.0--platform:1e140000.pcie
phy:phy-1e14a000.pcie-phy.1--platform:1e140000.pcie
platform:1e000000.syscon--mdio_bus:mdio-bus
platform:1e000000.syscon--pci:0000:00:00.0
platform:1e000000.syscon--pci:0000:00:01.0
platform:1e100000.ethernet--mdio_bus:mdio-bus:1f
platform:1e149000.pcie-phy--pci:0000:00:00.0
platform:1e149000.pcie-phy--pci:0000:00:01.0
```

Flashed image:
```
# ls /sys/class/devlink
phy:phy-1e149000.pcie-phy.0--platform:1e140000.pcie
phy:phy-1e14a000.pcie-phy.1--platform:1e140000.pcie
platform:1e000000.syscon--mdio_bus:mdio-bus
platform:1e000000.syscon--pci:0000:00:00.0
platform:1e000000.syscon--pci:0000:00:01.0
platform:1e100000.ethernet--mdio_bus:mdio-bus:1f
platform:1e149000.pcie-phy--pci:0000:00:00.0
platform:1e149000.pcie-phy--pci:0000:00:01.0

```

arinc9, 2024-03-26:
It looks like my patch is not the root cause of the problem. It only made the problem apparent. Since booting the initramfs image from memory produces healthy behaviour, it's either something wrong with your hardware or there's a flaw with the squashfs image creation that results in the kernel, kernel modules, or maybe even the device tree being written to the flash in an improper manner. Not mediatek-ge (the MT753X phy driver) nor mt7530 (DSA subdriver) nor mt7530-mdio (MDIO interface driver) are compiled as modules on mt7621 subtarget of OpenWrt so I don't think there're any issues with module dependencies.

I've just flashed the squashfs image on my Unielec U7621-06 16m, and booted from the flash. I don't see the issue reported here. But there's a lot of moving parts with OpenWrt (namely tp-link safeloader) so I can't speak for certain that, for your device model, there's no issue with the squashfs image creation. I also had to tweak the filesystem description on the device tree to fit the custom partitions I maintain on the device with mainline U-Boot.

Did anybody else with EAP235-Wall report this same issue? Are we sure this issue is not specific to your hardware?

The last thing that I can think of is that maybe your flash chip cannot operate stably at the frequency defined on the device tree. I see it's `spi-max-frequency = <20000000>;`. Try `spi-max-frequency = <14000000>;` and see if that works.

Borromini, 2024-03-26:
I asked a few people that own an EAP615-Wall, and are savvy enough to test (and have serial or are able to solder it). The 615 has a very similar layout, but they're reluctant to test since they use it in production (as I do with mine, the 235 is a testing device). So it's basically wait and see till someone tries a snapshot and starts complaining. I can ask on the forums, but I'm afraid few people have serial set up on these.

The Archer C6U is affected too, see https://github.com/openwrt/openwrt/issues/14673. That makes me think this is not just a quirk my 235 is running into.

I'll get back to you when I tested the lower SPI frequency.

arinc9, 2024-03-26:
Ok, the device tree of all devices that were being reported with the issue present have the SPI max frequency higher than 14MHz. So this is still a loose end for someone tie up.

Borromini, 2024-03-27:
@arinc9 I have tested with the lower flash speed, the issue still persists unfortunately.

arinc9, 2024-03-28:
Ok. At this point, I think you're justified to revert the phy muxing patch. As far as I can tell, this issue only exists on the main branch, not 23.05. The snapshot is supposed to be unstable, that's what stable releases are for. So I would suggest you to wait until a branch for the next stable version is made then test on that stable branch. If the issue still exists, only then revert the phy muxing patch for the device on the stable and main branches.

Borromini, 2024-04-05:
FWIW, regression persists on 6.6, just tested.

Borromini, 2024-04-27:
I'm going to test the main tree again, this in-tree revert looks like it might 'fix' it for the time being:

https://github.com/openwrt/openwrt/commit/a8dde7e5bd6d289db6485cf57d3512ea62eaa827

arinc9, 2024-04-27:
No, that's irrelevant.

arinc9, 2024-04-28:
@Borromini please test the image for your board at this link:

https://files.arinc9.com/OpenWrt/

Diff:

```
diff --git a/target/linux/generic/hack-6.1/940-net-dsa-mt7530-use-switch-MDIO-bus-for-PHY-muxing.patch b/target/linux/generic/hack-6.1/940-net-dsa-mt7530-use-switch-MDIO-bus-for-PHY-muxing.patch
new file mode 100644
index 0000000000..3acb21ab25
--- /dev/null
+++ b/target/linux/generic/hack-6.1/940-net-dsa-mt7530-use-switch-MDIO-bus-for-PHY-muxing.patch
@@ -0,0 +1,29 @@
+From 5f94d7358cceda721983b29245010a9d4ab0c808 Mon Sep 17 00:00:00 2001
+From: =?UTF-8?q?Ar=C4=B1n=C3=A7=20=C3=9CNAL?= <arinc.unal@arinc9.com>
+Date: Sun, 28 Apr 2024 13:38:46 +0300
+Subject: [PATCH] net: dsa: mt7530: use switch MDIO bus for PHY muxing
+MIME-Version: 1.0
+Content-Type: text/plain; charset=UTF-8
+Content-Transfer-Encoding: 8bit
+
+Signed-off-by: Arınç ÜNAL <arinc.unal@arinc9.com>
+---
+ drivers/net/dsa/mt7530.c | 2 +-
+ 1 file changed, 1 insertion(+), 1 deletion(-)
+
+diff --git a/drivers/net/dsa/mt7530.c b/drivers/net/dsa/mt7530.c
+index 7dc2f0ac8962..fa7a1ab292d2 100644
+--- a/drivers/net/dsa/mt7530.c
++++ b/drivers/net/dsa/mt7530.c
+@@ -2585,7 +2585,7 @@ mt7530_setup(struct dsa_switch *ds)
+ 			if (!phy_node)
+ 				continue;
+ 
+-			if (phy_node->parent == priv->dev->of_node->parent) {
++			if (phy_node->parent->parent == priv->dev->of_node) {
+ 				ret = of_get_phy_mode(mac_np, &interface);
+ 				if (ret && ret != -ENODEV) {
+ 					of_node_put(mac_np);
+-- 
+2.40.1
+
diff --git a/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts b/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts
index fb14bd7829..2e573fff27 100644
--- a/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts
+++ b/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts
@@ -145,27 +145,48 @@
 	nvmem-cell-names = "mac-address";
 };
 
-&mdio {
-	ethphy0: ethernet-phy@0 {
-		reg = <0>;
+&switch0 {
+	mdio {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		ethphy0: ethernet-phy@0 {
+			reg = <0>;
+		};
+
+		ethphy1: ethernet-phy@1 {
+			reg = <1>;
+			interrupts = <1>;
+		};
+
+		ethphy2: ethernet-phy@2 {
+			reg = <2>;
+			interrupts = <2>;
+		};
+
+		ethphy3: ethernet-phy@3 {
+			reg = <3>;
+			interrupts = <3>;
+		};
 	};
-};
 
-&switch0 {
 	ports {
 		port@1 {
 			status = "okay";
 			label = "lan1";
+			phy-handle = <&ethphy1>;
 		};
 
 		port@2 {
 			status = "okay";
 			label = "lan2";
+			phy-handle = <&ethphy2>;
 		};
 
 		port@3 {
 			status = "okay";
 			label = "lan3";
+			phy-handle = <&ethphy3>;
 		};
 	};
 };
diff --git a/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts b/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts
index d814cba261..057f0b11a7 100644
--- a/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts
+++ b/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts
@@ -189,27 +189,48 @@
 	nvmem-cell-names = "mac-address";
 };
 
-&mdio {
-	ethphy0: ethernet-phy@0 {
-		reg = <0>;
+&switch0 {
+	mdio {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		ethphy0: ethernet-phy@0 {
+			reg = <0>;
+		};
+
+		ethphy1: ethernet-phy@1 {
+			reg = <1>;
+			interrupts = <1>;
+		};
+
+		ethphy2: ethernet-phy@2 {
+			reg = <2>;
+			interrupts = <2>;
+		};
+
+		ethphy3: ethernet-phy@3 {
+			reg = <3>;
+			interrupts = <3>;
+		};
 	};
-};
 
-&switch0 {
 	ports {
 		port@1 {
 			status = "okay";
 			label = "lan3";
+			phy-handle = <&ethphy1>;
 		};
 
 		port@2 {
 			status = "okay";
 			label = "lan2";
+			phy-handle = <&ethphy2>;
 		};
 
 		port@3 {
 			status = "okay";
 			label = "lan1";
+			phy-handle = <&ethphy3>;
 		};
 	};
 };
```

Borromini, 2024-04-28:
Thanks, that works on 6.1. Can the MT7530 patch be ported to 6.6? I suppose so since it's limited in scope?

I was wondering if something like the below would work for the Archer AX23? I broke mine trying to revert to a vanilla OpenWrt build so can't test yet :smile: 

```
diff --git a/target/linux/ramips/dts/mt7621_tplink_archer-ax23-v1.dts b/target/linux/ramips/dts/mt7621_tplink_archer-ax23-v1.dts
index 71ef4bc6b6..48fea6deb4 100644
--- a/target/linux/ramips/dts/mt7621_tplink_archer-ax23-v1.dts
+++ b/target/linux/ramips/dts/mt7621_tplink_archer-ax23-v1.dts
@@ -181,32 +181,59 @@
 	nvmem-cell-names = "mac-address";
 };
 
-&mdio {
-	ethphy4: ethernet-phy@4 {
-		reg = <4>;
+&switch0 {
+	mdio0 {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		ethphy0: ethernet-phy@0 {
+			reg = <0>;
+			interrupts = <0>;
+		}
+
+		ethphy1: ethernet-phy@1 {
+			reg = <1>;
+			interrupts = <1>;
+		}
+
+		ethphy2: ethernet-phy@2 {
+			reg = <2>;
+			interrupts = <2>;
+		}
+
+		ethphy3: ethernet-phy@3 {
+			reg = <3>;
+			interrupts = <3>;
+		}
+
+		ethphy4: ethernet-phy@4 {
+			reg = <4>;
+		}
 	};
-};
 
-&switch0 {
 	ports {
 		port@0 {
 			status = "okay";
 			label = "lan1";
+			phy-handle = <&ethphy0>;
 		};
 
 		port@1 {
 			status = "okay";
 			label = "lan2";
+			phy-handle = <&ethphy1>;
 		};
 
 		port@2 {
 			status = "okay";
 			label = "lan3";
+			phy-handle = <&ethphy2>;
 		};
 
 		port@3 {
 			status = "okay";
 			label = "lan4";
+			phy-handle = <&ethphy3>;
 		};
 	};
 };
```

arinc9, 2024-04-28:
I'll submit a proper fix for all devices, this was just for you to confirm whether or not it fixed the issue.

Borromini, 2024-04-28:
Thanks, appreciate it. It does indeed on the EAP235-Wall v1.

arinc9, 2024-04-29:
Could you boot the image on the link below and provide here the kernel log that says "look here: TRAP=???". Either flash and boot the squashfs image or boot the initramfs image, whichever is quicker for you.

https://files.arinc9.com/OpenWrt/

I am trying to find out why this issue exists on a select number of devices.

calin2k, 2024-04-29:
@arinc9: can you provide an image for tp-link archer c6u and/or tpl-link ax23?

arinc9, 2024-04-29:
I've put the images for these boards on the same link. Please provide me the boot log as well.

calin2k, 2024-04-29:
here is output of dmesg, ip a after sysupgrade on archer c6u
https://pastebin.com/jfUp1G5f

arinc9, 2024-04-29:
do these devices have the issue that was reported here?

calin2k, 2024-04-29:
maybe not, I got here with the dead wan port and PHY error following a link in the discussion.

Borromini, 2024-04-29:
@arinc9 Exact same error km the C6U, see:

https://github.com/openwrt/openwrt/issues/14673

arinc9, 2024-04-29:
calin2k's test result was enough for me. You don't need to test anything anymore @Borromini. I will coordinate with Daniel to resolve this issue on all devices soon.

arinc9, 2024-05-01:
Daniel commited my patches that is supposed to address this issue. Please try the latest snapshot.

Borromini, 2024-05-02:
Thanks @arinc9 for your work on this. I can confirm my EAP235-Wall v1 is fully functional again. Will test the AX23, but I suppose it won't be any different.

closed 2024-05-02 by Borromini