openwrt/openwrt #14626: The 40 MHz mode does not work at 2.4 GHz (GL-INET MT6000)

Mummify1867, 2024-02-13:
### Describe the bug

Devices are not connecting at the 40 MHz frequency in WiFi 2.4 GHz

### OpenWrt version

r25142-a42075435a

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Configure the 40 MHz mode on the Operating Frequency option on WiFi 2.4 GHz

### Actual behaviour

Devices will connect at a frequency of 20 MHz

### Expected behaviour

Devices will connect at a frequency of 40 MHz

### Additional info

This occurs in all connected devices, whether smartphones or PCs

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-02-13:
Did you force 40MHz operation?

[image]

Mummify1867, 2024-02-13:
> Did you force 40MHz operation? [...]

Use forcing 40 MHz worked. Thanks for the support and sorry, I did not observe this option in OpenWRT settings.

closed 2024-02-13 by Mummify1867