openwrt/openwrt #14628: qualcommax/ipq807x/QNAP 301w: LEDs not properly ordered

sppmasterspp, 2024-02-14:
### Describe the bug

Updating to the latest snapshot (compared to the 2 weeks older build) with sysupgrade keeping the settings and the LEDs no longer work.
These can be seen in kernel log
`[    4.246494] leds-gpio leds: Led green: renamed to green:_1 due to name collision
[    4.248978] leds-gpio leds: Led green: renamed to green:_2 due to name collision
[    4.256345] leds-gpio leds: Led green: renamed to green:_3 due to name collision
[    4.263770] leds-gpio leds: Led green: renamed to green:_4 due to name collision
[    4.271105] leds-gpio leds: Led amber: renamed to amber:_1 due to name collision
[    4.278487] leds-gpio leds: Led green: renamed to green:_5 due to name collision
[    4.285856] leds-gpio leds: Led amber: renamed to amber:_2 due to name collision
[    4.293257] leds-gpio leds: Led green: renamed to green:_6 due to name collision
[    4.300616] leds-gpio leds: Led amber: renamed to amber:_3 due to name collision
[    4.308002] leds-gpio leds: Led green: renamed to green:_7 due to name collision
[    4.315384] leds-gpio leds: Led amber: renamed to amber:_4 due to name collision
[    4.322760] leds-gpio leds: Led green: renamed to green:_8 due to name collision
[    4.330135] leds-gpio leds: Led amber: renamed to amber:_5 due to name collision`

I played a bit with the LEDs configuration and it seems that LED names were changed (as stated in kernel log due to name collision) and that was the reason they didn't work. I didn't try to reset all settings to deafault to see if they would work without playing with their settings.
Now they are a bit stirred. But switching between the numbers I've found this order of all 9 LEDs.

[image]

Previously the device names matched the LED names.

### OpenWrt version

r25275+3-1128290dbf

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

QNAP 301w

### Image kind

Self-built image

### Steps to reproduce

Usual sysupgrade from older snapshot to the latest one keeping the settings.

### Expected behaviour

All the LEDs should work and have names matching device names. This was the case before the latest LED changes for qualcommax.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-02-14:
It seems to be due to the lack of ```function``` property in the LED nodes in the dts.

2e659930d3f3e131b28879447d61faf58cf8add7 removes ```label``` property, but b7a7a793b140175cd02a863412b5464cad2269fc didn't add ```function``` property for 301W. Then, there are multiple ```green:```/```red:```/```amber:``` LEDs.

CC: @Ansuel

l33tlinuxh4x0r, 2024-02-22:
Can confirm that this is a bug on the clean snapshot builds form firmware-selector.openwrt.org

termomaks, 2024-02-22:
> Can confirm that this is a bug on the clean snapshot builds form firmware-selector.openwrt.org

Let's hope @Ansuel will be able to fix it soon.

sppmasterspp, 2024-02-22:
> Can confirm that this is a bug on the clean snapshot builds form firmware-selector.openwrt.org

Let's hope @Ansuel will be able to fix it soon.

termomaks, 2024-03-14:
@Ansuel or @robimarko 
If you have time, can you look at this bug, please.
Thanks.

musashino205, 2024-03-16:
Can someone test this attached firmware that contains fixes for this issue?
Note: If you use initramfs image, you can test without flashing to NAND.

[openwrt-qualcommax-ipq807x-qnap_301w_ledfix.zip](https://github.com/openwrt/openwrt/files/14623341/openwrt-qualcommax-ipq807x-qnap_301w_ledfix.zip)

termomaks, 2024-03-16:
I'll try it once the router is idle.
Maybe @l33tlinuxh4x0r can do it earlier than me.
Thanks a lot.

sppmasterspp, 2024-03-16:
> Can someone test this attached firmware that contains fixes for this issue?
> Note: If you use initramfs image, you can test without flashing to NAND.

I'll try it once the router is idle.
Maybe @l33tlinuxh4x0r can do it earlier than me.
Thanks a lot.

sppmasterspp, 2024-03-17:
> Can someone test this attached firmware that contains fixes for this issue?
> Note: If you use initramfs image, you can test without flashing to NAND.

I've tested your image and it works OK.
[image]

Thanks a lot.

robimarko, 2024-03-17:
Should be fixed via:
https://github.com/openwrt/openwrt/pull/14910

closed 2024-03-18 by robimarko