openwrt/openwrt #14631: Dnsmasq out of date (2.89 vs 2.90)

ascendbeing, 2024-02-14:
### Describe the bug

dnsmasq 2.90 stable is out as of 13/02/2024. Owrt is using 2.89.

### OpenWrt version

r25141-6f83a708c8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Additional info

See https://dnsmasq.org/ 2.90 stable is available.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-02-14:
This is not a bug, please post to the forum or the ML.
(or you can also send a patch to bump the version)

oskarirauta, 2024-02-14:
I took a quick look at it and I got to admit, that making the ubus patch to work with 2.90, requires some expertise on dnsmasq, so it might take some time until dnsmasq is updated to 2.90.

jow-, 2024-02-14:
Packages not being the latest upstream version is not a bug.

closed 2024-02-14 by jow-

ascendbeing, 2024-02-19:
> Packages not being the latest upstream version is not a bug.

I thought  this was the appropriate forum to give notice to this. I don't know where else one would note it. It's not that I viewed it as a serious show stopping bug, I was just noting it for anyone who may be interested in taking this up.