openwrt/openwrt #14633: after upgrading the packages luci login page no longer accessible

mrusli70, 2024-02-14:
### Describe the bug

I am using raspberry pi 4 with version 23.05.0 and after the upgrading the packages yesterday and now that my luci login page of http://192.168.1.1 does not seems to work at all. I did try http:.//openwrt.lan, that too cannot seems to work at all. 

### OpenWrt version

23.05.0

### OpenWrt release

nil

### OpenWrt target/subtarget

nil

### Device

raspberry pi 4

### Image kind

Official downloaded image

### Steps to reproduce

after upgrading the packages the luci login page is no longer accesible

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrusli70, 2024-02-14:
Yes! It is raspberry pi 4.  after upgrading the packages from the software update. it cannot seems to go to luci login web page of http://192.168.1.1 or even http://openwrt.lan

brada4, 2024-02-14:
Set static IP in 192.168.1/24 network and try connecting the router and reboot it. Some package installs restart other services and kill them unwillingly. Unless you say which package hard to tell what could go wrong.
Failing that take SD card and do clean install.
auc/attended sysupgrade has better success than upgrading packages one by one.