openwrt/openwrt #14638: WI-FI light bulbs blink in time with WAN/LAN

YoricII, 2024-02-15:
### Describe the bug

Hello, sorry fo my bad English.
WI-FI light bulbs blink in time with WAN/LAN when there is no WI-FI traffic.
21.XX and later are also susceptible to this. Version 19.X does not have this defect.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

OpenWrt 23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WDR3600 v1

### Image kind

Official downloaded image

### Steps to reproduce

Exclude WI-FI traffic and organize WAN/LAN traffic

### Actual behaviour

WI-FI light bulbs blink in time with WAN/LAN

### Expected behaviour

WI-FI light bulbs should not blink

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

YoricII, 2024-02-15:
23.05.2

brada4, 2024-02-15:
How that is OpenWRT issue? It is physical/workplace safety issue that has to be addressed by bulb vendor.

yichya, 2024-02-15:
> How that is OpenWRT issue? It is physical/workplace safety issue that has to be addressed by bulb vendor.

Maybe the 'light bulbs' here means LEDs on the device

brada4, 2024-02-15:
https://openwrt.org/docs/guide-user/base-system/led_configuration#network_activity
You can change led to wifi-assoc or netdev-up trigger in place of radio environment activity disco.

YoricII, 2024-02-15:
As far as I can see in
https://archive.openwrt.org/releases/packages-23.05/mips_24kc/[base|packages]
the kmod-ledtrig-netdev package is missing.

brada4, 2024-02-15:
Your platform is not DSA, could be has only switch ledtrig and you are stuck with just wifi values (and on/off)

YoricII, 2024-02-15:
Please name the appropriate luci module.
```
$ grep -i luci .config | grep trig
# CONFIG_PACKAGE_luci-app-ledtrig-rssi is not set
# CONFIG_PACKAGE_luci-app-ledtrig-switch is not set
# CONFIG_PACKAGE_luci-app-ledtrig-usbport is not set
```
Was there a DSA before 19.XX, but not after? That is, the LEDs worked as they should

brada4, 2024-02-15:
Short google search should reveal the history.
luci is always installed as a package be it snapshots or releases.

brada4, 2024-02-15:
Luci/System/Leds. Set green wlan2g greenwlan2g to your preference. Default wiring is from vendor bootloader or similar.

YoricII, 2024-02-16:
That is, does LUCI itself include LED functionality, do you need other packages?

Please tell me, then what is the luci-app-ledtrig-switch package for?

brada4, 2024-02-16:
luci is add-on package, you need extra add-on to change led behaviour via web, but /etc/config/system (doc mentions legacy nonexistent ig/leds) configs can be parsed by init script with prebuilt kernel modules by default, kmods IF customized, and once set up you dont need luci add-on anymore, nor unused kernel modules. Optimisation for small flash space ie routers like yours.

brada4, 2024-02-16:
netdev module can be attached to any adapter seen by ifconfig including DSA switch ports, while switch is to monitor swconfig port(s) that are not adapters in system.

YoricII, 2024-02-17:
Thanks for the help, everything worked out, sorry to bother you, the bug can be deleted))

brada4, 2024-02-17:
Close it under comment box

closed 2024-02-17 by YoricII