openwrt/openwrt #14648: regdom escape via `option channels`

brada4, 2024-02-16:
### Describe the bug

```
        option channel 'auto'
        option band '5g'
        option htmode 'HE80'
        option country 'BE'
        option channels '36 52'
```
happily selects channel 132 for eventual beaconing at 26dBm
```
                        * 5660 MHz [132] (26.0 dBm) (radar detection)
                        * 5680 MHz [136] (26.0 dBm) (radar detection)
                        * 5700 MHz [140] (26.0 dBm) (radar detection)
                        * 5720 MHz [144] (13.0 dBm) (radar detection)
```
Quite obvious selected channel is way off intended channel choice too.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Just set channels option in config/wireless anywhere in EU/ETSI and restart 5ghz radio few times.
freqlist same.

### Actual behaviour

Firstly the channels outside scope are selected more often than not
Secondly radio parameters outside regulatory spec is chosen at times.

### Expected behaviour

Users' choices and local traditions is obeyed.

### Additional info

Same on ath97, presumably problem is hostapd not scripts.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-16:
Verified that parameters are properly conveyed to hostapd conf in /var/tmp
v22 or snapshot - same behaviour.

Shine-, 2024-02-16:
This:
```
        option channels '36 52'
```
is an invalid channel selection for:
```
        option htmode 'HE80'
```
thus ignored.

Use `option channels '36-64'` or `option channels '36-48 52-64'` instead.

brada4, 2024-02-16:
If I dare to select channel 36 then script generates invalid-your-way config of chanlist 36.
I insist my report is 100% accurate

brada4, 2024-02-16:
fyi: https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n205

brada4, 2024-02-17:
When I explicitly set channel 132 it comes up at unpleasant 13dbm, when I select any `channels` option and auto channel (0, acs_survey also tried) random channel is selected including off-regdom 132@80@26
android , linux, windows connect 40wide while apple refuses to connect at all.

brada4, 2024-11-17:
Works correctly with v23.05, i am astonished by amount of BS thrown at me to disqualify genuine regulatory fokup report.

closed 2024-11-17 by brada4