openwrt/openwrt #14655: WED hang on latest 6.1 kernel on mt7981

ptpt52, 2024-02-17:
### Describe the bug

```
insmod mt7915e wed_enable=Y
[   74.734159] mt798x-wmac 18000000.wifi: attaching wed device 0 version 2
[   74.787920] platform 15010000.wed: MTK WED WO Firmware Version: DEV_000000, Build Time: 20221208202138
[   74.797280] platform 15010000.wed: MTK WED WO Chip ID 00 Region 3
(hang...and...reboot)
```

### OpenWrt version

xx

### OpenWrt release

xx

### OpenWrt target/subtarget

xx

### Device

xx

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chbgdn, 2024-02-17:
My Xiaomi AX3000T was bricked after upgrade to 6.1 kernel with enabled WED, so I think that same issue. Disabling WED from recovery mode fixed it.

ptpt52, 2024-02-17:
ping @nbd168

ptpt52, 2024-02-17:
also ping @dangowrt

brada4, 2024-02-17:
@ptpt52 wed is device to device DMA, some platforms are not properly wired for that, you need to indicate failing hw.

dangowrt, 2024-02-17:
Is this is newly introduced problem after https://github.com/openwrt/openwrt/commit/c55c56b3fd742d069941953357d56097fe692b4f or could it be that WED on MT7981 was already broken on Linux 6.1 also before that?
The described behavior sounds like a clock not setup correctly, so comparing clk driver in 5.15 (downstream files) with 6.1 (upstream) would be the first place to look (I did and it looks innocent for me). Also comparing `/sys/kernel/debug/clk/clk_summary` on 5.15 vs. 6.1 could give insights.

ptpt52, 2024-02-17:
the diff in `/sys/kernel/debug/clk/clk_summary` @dangowrt 
```diff
diff --git a/clk_summary.k-5.15.txt b/clk_summary.k-6.1.txt
index 4860236..4241d91 100644
--- a/clk_summary.k-5.15.txt
+++ b/clk_summary.k-6.1.txt
@@ -2,9 +2,9 @@
    clock                          count    count    count        rate   accuracy phase  cycle    enable
 -------------------------------------------------------------------------------------------------------
  clkxtal                              7        7        0    40000000          0     0  50000         Y
-    cb_cksq_40m                       5        5        0    40000000          0     0  50000         Y
+    cb_cksq_40m                       2        2        0    40000000          0     0  50000         Y
        infra_ipcie_pipe               0        0        0    40000000          0     0  50000         N
-       ap2cnn_host_sel                0        0        0    40000000          0     0  50000         N
+       ap2cnn_host_sel                1        1        0    40000000          0     0  50000         Y
           ap2cnn_host                 0        0        0    40000000          0     0  50000         Y
        uart_sel                       1        1        0    40000000          0     0  50000         Y
           infra_uart2_sel             0        0        0    40000000          0     0  50000         Y
@@ -21,18 +21,10 @@
        spinfi_sel                     0        0        0    40000000          0     0  50000         N
           spinfi_bck                  0        0        0    40000000          0     0  50000         Y
              infra_spinfi1            0        0        0    40000000          0     0  50000         N
-       usb_ln0                        2        2        0    40000000          0     0  50000         Y
-          sgm0_ck0_en                 1        1        0    40000000          0     0  50000         Y
-          sgm1_ck1_en                 1        1        0    40000000          0     0  50000         Y
-       usb_cdr                        2        2        0    40000000          0     0  50000         Y
-          sgm0_cdr_ck0_en             1        1        0    40000000          0     0  50000         Y
-          sgm1_cdr_ck1_en             1        1        0    40000000          0     0  50000         Y
-       usb_eq_rx250m                  2        2        0    40000000          0     0  50000         Y
-          sgm0_rx_en                  1        1        0    40000000          0     0  50000         Y
-          sgm1_rx_en                  1        1        0    40000000          0     0  50000         Y
-       usb_tx250m                     2        2        0    40000000          0     0  50000         Y
-          sgm0_tx_en                  1        1        0    40000000          0     0  50000         Y
-          sgm1_tx_en                  1        1        0    40000000          0     0  50000         Y
+       usb_ln0                        0        0        0    40000000          0     0  50000         Y
+       usb_cdr                        0        0        0    40000000          0     0  50000         Y
+       usb_eq_rx250m                  0        0        0    40000000          0     0  50000         Y
+       usb_tx250m                     0        0        0    40000000          0     0  50000         Y
        cb_rtc_32p7k                   0        0        0       32786          0     0  50000         Y
        cb_rtc_32k                     0        0        0       32000          0     0  50000         Y
           infra_frtc                  0        0        0       32000          0     0  50000         N
@@ -140,9 +132,9 @@
              i2c_bck                  0        0        0   125000000          0     0  50000         Y
                 infra_i2c0            0        0        0   125000000          0     0  50000         N
        net1_d5_d2                     0        0        0   250000000          0     0  50000         Y
-       cb_net1_d5                     1        1        0   500000000          0     0  50000         Y
-          eip97b_sel                  0        0        0   500000000          0     0  50000         N
-             eip97b                   0        0        0   500000000          0     0  50000         Y
+       cb_net1_d5                     2        2        0   500000000          0     0  50000         Y
+          eip97b_sel                  1        1        0   500000000          0     0  50000         Y
+             eip97b                   1        1        0   500000000          0     0  50000         Y
           netsys_500m_sel             1        1        0   500000000          0     0  50000         Y
              netsys_500m              0        0        0   500000000          0     0  50000         Y
        cb_net1_d4                     0        0        0   625000000          0     0  50000         Y
@@ -151,10 +143,10 @@
        cb_wedmcu_208m                 0        0        0   208000000          0     0  50000         Y
     sgmpll                            1        1        0   325000000          0     0  50000         Y
        cb_sgm_325m                    1        1        0   325000000          0     0  50000         Y
-          sgm_325m_sel                1        1        0   325000000          0     0  50000         Y
[... cut ...]
+ sgmii1_tx250m_en                     1        1        0           0          0     0  50000         Y
+ sgmii0_cdr_fb                        1        1        0           0          0     0  50000         Y
+ sgmii0_cdr_ref                       1        1        0           0          0     0  50000         Y
+ sgmii0_rx250m_en                     1        1        0           0          0     0  50000         Y
+ sgmii0_tx250m_en                     1        1        0           0          0     0  50000         Y
```

ptpt52, 2024-02-17:
[clk_summary.k-5.15.txt](https://github.com/openwrt/openwrt/files/14318096/clk_summary.k-5.15.txt)
[clk_summary.k-6.1.txt](https://github.com/openwrt/openwrt/files/14318098/clk_summary.k-6.1.txt)

dangowrt, 2024-02-17:
So this looks like the wrong ethsys clock driver gets bound (from MT7986 which got wocpu0 and wocpu1, that's how I recognize).
Please confirm that in `/sys/bus/platform/drivers/clk-mt7981-eth/` there isn't any bound hardware while in `/sys/bus/platform/drivers/clk-mt7986-eth/` there is.

If so, try this please:
```
diff --git a/target/linux/mediatek/files-6.1/arch/arm64/boot/dts/mediatek/mt7981.dtsi b/target/linux/mediatek/files-6.1/arch/arm64/boot/dts/mediatek/mt7981.dtsi
index cb161df81e..c763b4ebb3 100644
--- a/target/linux/mediatek/files-6.1/arch/arm64/boot/dts/mediatek/mt7981.dtsi
+++ b/target/linux/mediatek/files-6.1/arch/arm64/boot/dts/mediatek/mt7981.dtsi
@@ -450,7 +450,6 @@
                #address-cells = <1>;
                #size-cells = <1>;
                compatible = "mediatek,mt7981-ethsys",
-                            "mediatek,mt7986-ethsys",
                             "syscon";
                reg = <0 0x15000000 0 0x1000>;
                #clock-cells = <1>;
```

dangowrt, 2024-02-17:
I've investigated it myself and pushed a fix https://github.com/openwrt/openwrt/commit/8955cadfb582b8f5fd005432a0d565abb1c1d370

closed 2024-02-17 by dangowrt

ptpt52, 2024-02-18:
tested and it works fine.