openwrt/openwrt #14656: dnsmasq 200-ubus_dns.patch no longer useful?

nwf, 2024-02-17:
### Describe the bug

https://github.com/openwrt/openwrt/blob/24d3eb7629d3e20a9c0d3dd47c84e75003df1199/package/network/services/dnsmasq/patches/200-ubus_dns.patch adds `ubus` interposition on all `dnsmasq` queries.  It does so by making blocking calls to the `ubus` and awaiting responses to its notification, which is of type `"dns_result"`.

However, the only listener for this notification was https://git.openwrt.org/?p=project/qosify.git which, as of https://git.openwrt.org/?p=project/qosify.git;a=commitdiff;h=6ff06d66c36c52e321f5d2ead5b15fa3215914e0;hp=558eabc13c64bdbd2d380e9520d56107d60a13cd prefers to snoop on DNS traffic.  (It still listens "as a backup", I admit, but probably it should switch to using nftsets that dnsmasq and other resolvers can maintain?)

In any case, I would like 200-ubus_dns.patch removed because it...
- Requires work to adapt to `dnsmasq` 2.90, which has a DNSSEC CVE fix in it.
- Allows any `ubus` consumer to mangle DNS responses, which is frankly kind of terrifying.

Does that seem sensible?

### OpenWrt version

main

### OpenWrt release

n/a

### OpenWrt target/subtarget

n/a

### Device

Not device specific

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nwf, 2024-02-17:
OK, sorry, how _should_ I have spelled "not device specific, affects HEAD"?

brada4, 2024-02-17:
> Ok, sorry

Fill the form with real device to distinguish OpenWRT user apart from robot or a potato?

nwf, 2024-02-17:
> > Ok, sorry
> 
> Fill the form with real device to distinguish OpenWRT user apart from robot or a potato?

But the issue isn't about just one device.  But sure, fine, in the future I'll do that to avoid ticking off the bot.

> The patch is a CPU hog uselessly reporting queries to ubus which discards them in all practical scenarios,

That's harsher commentary on the nature of the patch than I would have given, but, indeed, the CPU load is probably significantly more than zero.

> not related to dnssec issues applying to dnssec validation memory leaks fixed in later releases, those apply to (almost) all DNS software validating DNSSEC.

I didn't say that the patch was related to DNSSEC issues.  I said that the patch was _less than trivial to adapt to dnsmasq 2.90_, which contains fixes for those DNSSEC issues.

nbd168, 2024-02-17:
I would like to keep this patch. There is other code that I'm working on (which is not in OpenWrt yet), which relies on the same API. The CPU load concern does not make any sense to me. As long as nothing is listening on the ubus object, the CPU overhead is so close to zero, it's probably not even measurable.

nwf, 2024-02-17:
> I would like to keep this patch. There is other code that I'm working on (which is not in OpenWrt yet), which relies on the same API.

I'm curious about your use cases, if you wouldn't mind spilling a little ink here.  Do you have something in mind that can't work with resolver-maintained nftsets?  Would you be OK with a modified, notify-only version that didn't allow the listeners to change DNS results?

> The CPU load concern does not make any sense to me. As long as nothing is listening on the ubus object, the CPU overhead is so close to zero, it's probably not even measurable.

It's still a context switch from `dnsmasq` to `ubusd` and back just to discover that nobody's listening.  A notify-only version could do that asynchronously rather than blocking on the return, which would also help with the low CPU load argument.

brada4, 2024-02-17:
@nbd168 would be nice to make it a command line parameter as to nor make ubus a sink for unnecessary log.

ynezz, 2024-02-18:
> But the issue isn't about just one device. But sure, fine, in the future I'll do that to avoid ticking off the bot.

Well, we could argue, that this is not an issue/bug, but basically a multiple discussion topics better suited for a mailing list thread.

>  I said that the patch was less than trivial to adapt to dnsmasq 2.90, which contains fixes for those DNSSEC issues.

Great, BTW do you plan to contribute that not so trivial fix of yours?

> It's still a context switch from dnsmasq to ubusd and back just to discover that nobody's listening.

BTW where do you see this context switch exactly, do you've link/line number handy? I've just tried to find it, maybe I'm blind, but all I just see is a static struct member check `!ubus_dns_object.has_subscribers`  which already looks like the  cheapest solution.

> Allows any ubus consumer to mangle DNS responses, which is frankly kind of terrifying. 

We are all aware, that due to the lack of any ACLs etc. any ubus consumer can do much more terrifying stuff anyway, so not a threat model.

Nonetheless we all agree, that maintaining any downstream patch is a PITA, moreover in such core network exposed service, so the best course of actions would be upstreaming this (dont know if it was already tried and the outcome of that).

closed 2024-02-18 by ynezz

nbd168, 2024-02-18:
Quick note regarding ACLs. ubus does have built-in ACL support, and the default for non-root users is that they can't access anything unless explicitly whitelisted.

brada4, 2024-02-18:
Something like --log-queries=ubus would ease ubus tracing for average case.

nwf, 2024-02-18:
> > But the issue isn't about just one device. But sure, fine, in the future I'll do that to avoid ticking off the bot.
> 
> Well, we could argue, that this is not an issue/bug, but basically a multiple discussion topics better suited for a mailing list thread.

Fair enough.

> > I said that the patch was less than trivial to adapt to dnsmasq 2.90, which contains fixes for those DNSSEC issues.
> 
> Great, BTW do you plan to contribute that not so trivial fix of yours?

Without this patch, the jump to `dnsmasq` 2.90 is just

```
--- a/package/network/services/dnsmasq/Makefile
+++ b/package/network/services/dnsmasq/Makefile
@@ -8,13 +8,13 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=dnsmasq
-PKG_UPSTREAM_VERSION:=2.89
+PKG_UPSTREAM_VERSION:=2.90
 PKG_VERSION:=$(subst test,~~test,$(subst rc,~rc,$(PKG_UPSTREAM_VERSION)))
 PKG_RELEASE:=7
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_UPSTREAM_VERSION).tar.xz
 PKG_SOURCE_URL:=https://thekelleys.org.uk/dnsmasq/
-PKG_HASH:=02bd230346cf0b9d5909f5e151df168b2707103785eb616b56685855adebb609
+PKG_HASH:=8e50309bd837bfec9649a812e066c09b6988b73d749b7d293c06c57d46a109e4
 
 PKG_LICENSE:=GPL-2.0
 PKG_LICENSE_FILES:=COPYING
```

I have not adapted this patch to 2.90, and that's why I was asking about it.  As there's desire to keep it, I can try, but I suspect I'll be beaten to the punch.

> BTW where do you see this context switch exactly, do you've link/line number handy? I've just tried to find it, maybe I'm blind, but all I just see is a static struct member check [...]

I missed that in my read of the patch; I stand corrected.

> > Allows any ubus consumer to mangle DNS responses, which is frankly kind of terrifying.
> 
> We are all aware, that due to the lack of any ACLs etc. any ubus consumer can do much more terrifying stuff anyway, so not a threat model.

Fine.