openwrt/openwrt #14661: scripts/feeds update very slow.

Vonger, 2024-02-17:
### Describe the bug

download openwrt 23.05.2 then call sh ./scripts/feeds update, it will download full packages and luci, that is over 200K files, very very slow...
I checked the script, at line 157, seems like init_commit line do not have **--depth 1**, after add that, feed update speed increase over 10 times.
I do not know if that is intend to or just a bug, but really important for me :)

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt76x8

### Device

VoCore2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-02-17:
They are intentionally NOT shallow cloned, if the default `git.openwrt.org` is slow you can switch to GH mirrors

ynezz, 2024-02-18:
> I checked the script, at line 157, seems like init_commit line do not have --depth 1, after add that, feed update speed increase over 10 times.

I'm quite puzzled now, I've thought, that this was already the case since [ Revert "feeds: use git-src-full to allow Git versioning" ](https://github.com/openwrt/openwrt/commit/478a3e609fec5034ab5acef54deb2f7907eebfbc), basically previously there was `src-git-full` full clone performed and now it should be `src-git` shallow clone.

If its still slow, then ` sed -i -E 's;git.openwrt.org/(feed|project);github.com/openwrt;' feeds.conf.default` and use GitHub mirrors.

Indeed, its sometimes slow probably due to some crazy CI builds happening out there.