openwrt/openwrt #14671: Bridge port isolation doesn't working for wireless devices

notgood, 2024-02-18:
### Describe the bug

Trying to enable bridge port isolation for wireless devices

/etc/config/network
```
config device
	option name 'phy0-guest'
	option isolate '1'

config device
	option name 'phy0-iot'
	option isolate '1'

config device
	option name 'phy1-guest'
	option isolate '1'

config device
	option name 'phy1-iot'
	option isolate '1'
```

After bringing up relevant guest and iot networks, bridge ports remain unisolated.

grep -H '' /sys/class/net/*/brport/isolated

```
/sys/class/net/lan1/brport/isolated:0
/sys/class/net/lan2/brport/isolated:0
/sys/class/net/lan3/brport/isolated:0
/sys/class/net/phy0-guest/brport/isolated:0
/sys/class/net/phy0-iot/brport/isolated:0
/sys/class/net/phy0-lan/brport/isolated:0
/sys/class/net/phy1-guest/brport/isolated:0
/sys/class/net/phy1-iot/brport/isolated:0
/sys/class/net/phy1-lan/brport/isolated:0
/sys/class/net/wan/brport/isolated:0
```

Setting isolate flag on bridge port for ethernet devices works just fine.

Anyway, here is my workaround hotplug script for now
```
cat << 'EOF' > /etc/hotplug.d/net/99-wifi-isolation
#!/bin/sh
if [ "$ACTION" = "add" ]; then
  if [ "$DEVICENAME" = "phy0-guest" ]; then
    sleep 1
    echo 1 > /sys/class/net/phy0-guest/brport/isolated
  fi
  if [ "$DEVICENAME" = "phy1-guest" ]; then
    sleep 1
    echo 1 > /sys/class/net/phy1-guest/brport/isolated
  fi
  if [ "$DEVICENAME" = "phy0-iot" ]; then
    sleep 1
    echo 1 > /sys/class/net/phy0-iot/brport/isolated
  fi
  if [ "$DEVICENAME" = "phy1-iot" ]; then
    sleep 1
    echo 1 > /sys/class/net/phy1-iot/brport/isolated
  fi
fi
EOF
```

My uneducated guess: there could some race issue trying to set brport/isolated before wireless device is actually up and added to bridge.

### OpenWrt version

latest master

### OpenWrt release

latest master

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000 v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-18:
Wifi interfaces are configured in wireless file. You provide too little context to help.

notgood, 2024-02-19:
> Wifi interfaces are configured in wireless file

Bridge port isolation is configured in /etc/network, including wifi devices.
[image]

> You provide too little context to help

There is plenty.

brada4, 2024-02-19:
Can you help yourself with one vlan and one wifi configu$ation sections?
netifd does not configure wifi.

notgood, 2024-02-20:
Right, a few months ago "bridge_isolate" option was added to netifd, which overrides generic "isolate" option for network devices included in bridge.
https://github.com/openwrt/netifd/commit/f3e06e81b347bbdec1c6c71603328b6e442728d4

`dev->settings.isolate = vif->bridge_isolate;`

This option is undocumented, and there is no GUI checkboxes for it in LUCI.... but it works.
So it solves this issue for me, but I guess not for others, until LUCI and documentation is adjusted to reflect netifd changes.

> netifd does not configure wifi.

@brada4, please at least try and do research things before commenting.
netifd source code could be a good start, wireless.c in particular

https://github.com/openwrt/netifd/blob/master/wireless.c

brada4, 2024-02-20:
You need to configure station isolation in /etc/config/wireless to make sure all traffic betwen stations reaches bridges like for ebtables.
netifd does not create phy-ap devices, it can configure only open network client, functionality which is not used in general.

brada4, 2024-02-20:
https://openwrt.org/docs/guide-user/network/wifi/basic
search for `isolate` network parameter.

notgood, 2024-02-20:
> https://openwrt.org/docs/guide-user/network/wifi/basic
> search for `isolate` network parameter.

Can't believe I have to spell it, the very first three words of this bugports title are "Bridge port isolation",
This bugreport has absolutely nothing to do with "isolate" option in /etc/config/wireless, and which in turn sets ap_isolate=1 in hostapd. Of course I have it set, and it certainly works for intra-band wifi client isolation, but it's useless for isolation traffic between bridge ports.

So again, this bugreport is about "Bridge port isolation".
1) It is configured by "isolate" option in /etc/config/network. Toggle is available in LUCI, but not documented in https://openwrt.org/docs/guide-user/network/network_configuration
2) It is also configured by "bridge_isolate" option in /etc/config/wireless, overriding setting nr.1. Not available in LUCI, not documented in https://openwrt.org/docs/guide-user/network/wifi/basic

brada4, 2024-02-20:
You have to go with option 2.
Same effect for mtu btw. It works via apply in luci just that one time.