openwrt/openwrt #14672: Proper support for gssapi

pmehax0r, 2024-02-18:
### Describe the bug

Hi guys, 

Thank you for the great distro!

I have been playing around openwrt for a while and always had some gssapi problems. I finally had time to dig in properly and found out that while I get all necessary KRB5 libraries with "krb5-libs" there isn't a proper support for gssapi, so "pip install gssapi" always fails. 

My workaround that was to first copy all files from /usr/include/mit-krb5 on a debian distro and transfer them to my openwrt distro. After that I had to create symlinks for all header files and finally had to add and make a link to com_err.h header file.
That exercise took quite some time to figure out (considering my noob skills). 

Would it be possible to somehow include those header files or create a separate package that ensures their installation. On debian all krb5 files are part of libkrb5-dev. Only the com_err header is part of comerr-dev package.

Feel free to let me know if I have been doing something very dumb and there was way easier setup :)

Thank you!

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi Compute Module 4 Rev 1.1

### Image kind

Official downloaded image

### Steps to reproduce

opkg install krb5-lib
pip install gssapi

### Actual behaviour

It throws an error that the pip installation failed

### Expected behaviour

Successfully installed gssapi via pip

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-02-18:
OpenWrt does not support on-device compilation and thus does not provide any development or header packages either. Consider requesting a `python-gssapi` package in the [packages feed](https://github.com/openwrt/packages).

closed 2024-02-18 by jow-

pmehax0r, 2024-02-18:
Cool, thanks for you quick reply @jow-  cheers