openwrt/openwrt #14673: Dead wan port on testing kernel - TP-Link Archer C6U

KliXeR, 2024-02-18:
### Describe the bug

On testing kernel 6.1 on mt7621 TP-Link Archer C6U v1 wan port is completely dead. On 5.15 works fine.

> mtk_soc_eth 1e100000.ethernet wan: validation of with support 00000000,00000000,00006000 and advertisement 00000000,00000000,00000000 failed: -EINVAL
mtk_soc_eth 1e100000.ethernet wan: mtk_open: could not attach PHY: -22

### OpenWrt version

r24132-359a6e36e9

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6U v1

### Image kind

Self-built image

### Steps to reproduce

Compile snapshot for TP-Link Archer C6U with testing kernel, flash to device.

### Actual behaviour

Wan port is dead.

### Expected behaviour

Wan port is working.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

KliXeR, 2024-02-19:
On 5.15 -p 0 is correct phy.

On 6.1 i got this on both phy (0, 4):
> SIOCGMIIPHY on 'wan' failed: Not supported

brada4, 2024-02-19:
As a temporary workaround you may find unused "LAN" port, remove it from br-lan and transfer WAN-related configurations to it. But otherwise indeed it is broken.

NorthP8, 2024-02-23:
I did a default reset again and reinstalled https-dns-proxy. The service is enabled in startup. Problem still exists.

NorthP8, 2024-02-23:
Sorry for the offtopic. We have the same devices TP-Link Archer C6U v1. Could you try to reproduce this problem? (If you have free time and desire, of course) https://github.com/openwrt/openwrt/issues/14696

KosmiK2001, 2024-02-24:
Tp-link AX23 v1.x
Same problem. Dead wan port. Snapshot!

calin2k, 2024-03-27:
bug also present on xiaomi_redmi-router-ax6s mediatek 7622 target

KosmiK2001, 2024-05-01:
Check snapshot.
WAN in AX23 v1 now works fine.

Borromini, 2024-05-02:
@KliXeR Can you test a recent snapshot build? This should be fixed now (see my EAP235-Wall bug report).

KliXeR, 2024-05-02:
Yeah, thanks. Everything is working now!

closed 2024-05-02 by KliXeR