openwrt/openwrt #14680: mt76: WED does not work with WDS

ttc0419, 2024-02-19:
### Describe the bug

I have a XDR6088 as the main router and AX6S as the secondary router wireless bridging to the main router.  If I turned on WED on the main router, any devices connected to the secondary router will lose internet connection.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

TP-Link TL-XDR6088

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-19:
WDS uses packet with 4 MAC addresses, WED or flow offload handles only two.

rany2, 2024-02-19:
This is expected, you could use my tree if you want mtk's patches working on snapshot: https://github.com/rany2/openwrt/commits/main_w_mtk/

Alternatively, you could try to apply the relevant patches yourself from https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild_mac80211_release/package/kernel/mt76/patches/

ttc0419, 2024-02-19:
Opened https://github.com/openwrt/mt76/issues/861 as well.

brada4, 2024-02-19:
I think it can be tracked over there. In the meantime do not enable WED as long as you use WDS ;-)

taylorkline, 2024-02-27:
Funny to stumble across this. I can confirm this on my Redmi AX6000.

The same problem does **not** happen on my RT3200, oddly enough.

brada4, 2024-02-27:
@taylorkline is wed and nat offloads enabled (i.e check module parameter and if offloaded device is wifi with wed)
2nd address pair is constructed from zeroes in wed+wds case, no chance it could forward properly.

taylorkline, 2024-02-27:
Well, I've switched over to 802.11s now, but `cat /sys/kernel/debug/ppe0/bind` is empty, so it seems likely that I have absolutely no idea what I am talking about.

taylorkline, 2024-03-12:
I tracked down that [there was some activity in late 2022](https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=207859&state=*&q=wds&archive=&delegate=) to support WED + WDS. It seems to have fizzled out, perhaps due to Mediatek losing interest. 

@rany2 you may already have a very good reason for having not done so already, or there may be a lot of context that I am not aware of, but since there has not been any activity on this since 2022 and you have it working on your fork, would it make sense to submit your changes to the linux-wireless mailing list?

lePaulo, 2024-03-20:
Hi, 
Thanks for this thread, actually the `2001-wifi-mt76-mt7915-wed-add-wds-support-when-wed-is-ena.patch` patch fails. Maybe it's due to the fact that my GL-MT6000 device is now dealing with Linux 6.1 kernel. @rany2, any clue on how to update it?
Many thanks!

```
Applying /home/ubuntu/environment/openwrt/package/kernel/mt76/patches/2001-wifi-mt76-mt7915-wed-add-wds-support-when-wed-is-ena.patch using plaintext: 
patching file mt76.h
Hunk #1 succeeded at 70 (offset -8 lines).
patching file mt7915/main.c
Hunk #1 FAILED at 804.
Hunk #2 succeeded at 1203 (offset -93 lines).
Hunk #3 FAILED at 1742.
2 out of 3 hunks FAILED -- saving rejects to file mt7915/main.c.rej
patching file mt7915/mcu.c
Hunk #1 succeeded at 2351 (offset -226 lines).
patching file mt7915/mcu.h
Hunk #1 FAILED at 392.
1 out of 1 hunk FAILED -- saving rejects to file mt7915/mcu.h.rej
patching file util.c
patching file util.h
Patch failed!  Please fix /home/ubuntu/environment/openwrt/package/kernel/mt76/patches/2001-wifi-mt76-mt7915-wed-add-wds-support-when-wed-is-ena.patch!
```

rany2, 2024-03-23:
@lePaulo You could use my OpenWRT tree https://github.com/openwrt/openwrt. It has the patches updated. Just note that you also need mac80211 patches.

lePaulo, 2024-03-23:
Thanks @rany2. Actually the file `2001-wifi-mt76-mt7915-wed-add-wds-support-when-wed-is-ena.patch` in your repository is not that different from the patch I used. I thought the failure messages I had were because of the use of Linux kernel 6.1... but maybe I'm doing something wrong elsewhere.

Anyway, I've tried also with your repository, but it ended up with the following messages:
```
Collected errors:
 * check_data_file_clashes: Package mt7986-wo-firmware wants to install file /home/ubuntu/environment/openwrt-gl-mt6000/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/lib/firmware/mediatek/mt7986_wo_0.bin
        But that file is already provided by package  * kmod-mt7986-firmware
 * check_data_file_clashes: Package mt7986-wo-firmware wants to install file /home/ubuntu/environment/openwrt-gl-mt6000/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/lib/firmware/mediatek/mt7986_wo_1.bin
        But that file is already provided by package  * kmod-mt7986-firmware
 * opkg_install_cmd: Cannot install package mt7986-wo-firmware.
package/Makefile:67: recipe for target 'package/install' failed
```
Is it due to the lack of mac80211 patches (if so, where should I get them?), or should I simply change some Makefiles to avoid `mt7986_wo_0.bin` file getting included twice?

Thanks!

rany2, 2024-03-23:
> or should I simply change some Makefiles to avoid `mt7986_wo_0.bin` file getting included twice?

You should just deselect `mt7986-wo-firmware` as it is already provided by `mt7986-firmware` from the Makefile in my repo. The offload firmware in `mt7986-firmware` is more up-to-date from MTK's OpenWRT feed whereas `mt7986-wo-firmware` is from linux-firmware.

lePaulo, 2024-03-23:
Nice. Make sense. Thanks @rany2. I'll test it next week.

Headcrabed, 2024-07-08:
I'm sending the patch to fix this issue to mainline kernel: https://patchwork.kernel.org/project/linux-wireless/patch/OS3P286MB259748F6FA6BE628C3295D7498D92@OS3P286MB2597.JPNP286.PROD.OUTLOOK.COM/

pcslide, 2024-07-15:
@Headcrabed 
Can you have a look at this? 
[issue #815](https://github.com/openwrt/mt76/issues/815)
Any suggestion?

Headcrabed, 2025-07-01:
Could anyone test the patch here? https://lore.kernel.org/linux-wireless/3903ee21-c87d-4a2c-b688-d64a5796b6c0@nbd.name/

raenye, 2025-10-17:
I tested this patch on Xiaomi AX3000T (MT7981) and can verify that it solved the problem :)
Without the patch I had the same symptoms discussed above (small packets such as ping and ssh work, but downloads are iperf3 grind to a halt) and now all work fine.

Thank you @Headcrabed, you're welcome to add my Tested-By to the patch.

closed 2025-10-23 by ttc0419

Headcrabed, 2025-10-24:
> I just tested on [3faa3a0](https://github.com/openwrt/openwrt/commit/3faa3a04bb44c3a265e95cb5f8ec31b37ba37520) with xdr6088. Clients connected to the main router (xdr6088) are fine, but the ones connected to the wds client router still has the same symptoms.

@nbd168 ping.

repaxan, 2026-11-04:
@ttc0419 @Headcrabed https://github.com/openwrt/openwrt/pull/26464 may fix that issue