openwrt/openwrt #14681: busybox: segmentation fault

oskarirauta, 2024-02-19:
### Describe the bug

I've now built openwrt twice and with both times, same issue; boot is impossible, also chroot to new root is impossible, actually executing even busybox binary files.

### OpenWrt version

r25226-8955cadfb5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Cerberus Interceptor

### Image kind

Self-built image

### Steps to reproduce

build openwrt with almost every package

### Actual behaviour

segmentation fault, I copied busybox from previous build (from few months back) - and system works with that without any problems.. Something has changed..

### Expected behaviour

no segmentation fault

### Additional info

gdb results:
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f9d568 in strlen (s=s@entry=0x555555543f0b <error: Cannot access memory at address 0x555555543f0b>) at src/string/strlen.c:16
16      src/string/strlen.c: No such file or directory.

valgrind results:
==1662== Memcheck, a memory error detector
==1662== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==1662== Using Valgrind-3.21.0 and LibVEX; rerun with -h for copyright info
==1662== Command: ./busybox_unstripped
==1662== 
==1662== Invalid read of size 1
==1662==    at 0x4017568: strlen (strlen.c:16)
==1662==    by 0x48A437B: get_cie_encoding (unwind-dw2-fde.c:347)
==1662==    by 0x48A44A9: classify_object_over_fdes (unwind-dw2-fde.c:742)
==1662==    by 0x48A5110: __register_frame_info_bases (unwind-dw2-fde.c:129)
==1662==    by 0x48A5110: __register_frame_info_bases (unwind-dw2-fde.c:109)
==1662==    by 0x405A661: do_init_fini (dynlink.c:1606)
==1662==    by 0x405C3C6: __libc_start_init (dynlink.c:1619)
==1662==    by 0x401FE8E: libc_start_main_stage2 (__libc_start_main.c:92)
==1662==    by 0x13027D: ??? (in /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox_unstripped)
==1662==  Address 0xf7f0b is not stack'd, malloc'd or (recently) free'd
==1662== 
==1662== 
==1662== Process terminating with default action of signal 11 (SIGSEGV)
==1662==  Access not within mapped region at address 0xF7F0B
==1662==    at 0x4017568: strlen (strlen.c:16)
==1662==    by 0x48A437B: get_cie_encoding (unwind-dw2-fde.c:347)
==1662==    by 0x48A44A9: classify_object_over_fdes (unwind-dw2-fde.c:742)
==1662==    by 0x48A5110: __register_frame_info_bases (unwind-dw2-fde.c:129)
==1662==    by 0x48A5110: __register_frame_info_bases (unwind-dw2-fde.c:109)
==1662==    by 0x405A661: do_init_fini (dynlink.c:1606)
[... 331 lines cut ...]
CONFIG_KERNEL_SOFTLOCKUP_DETECTOR=y
CONFIG_KERNEL_SQUASHFS_XATTR=y
CONFIG_KEXEC_LZMA=y
CONFIG_KEXEC_ZLIB=y

CONFIG_LLDPD_WITH_CDP=y
CONFIG_LLDPD_WITH_CUSTOM=y
CONFIG_LLDPD_WITH_DOT1=y
CONFIG_LLDPD_WITH_DOT3=y
CONFIG_LLDPD_WITH_EDP=y
CONFIG_LLDPD_WITH_FDP=y
CONFIG_LLDPD_WITH_JSON=y
CONFIG_LLDPD_WITH_LLDPMED=y
CONFIG_LLDPD_WITH_PRIVSEP=y
CONFIG_LLDPD_WITH_SONMP=y

CONFIG_MOLD=y
CONFIG_USE_MOLD=y

Since my list is really long, I removed all unnecessary packages (about 99% selected/compiled) and kmods (only few were opted out)
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

oskarirauta, 2024-02-19:
```
-rwxr-xr-x    1 root     root        573467 Feb 19 19:34 busybox
-rwxr-xr-x    1 root     root        573104 Feb 17 20:54 busybox.orig
```
size is almost same.. busybox.orig is new segfaulting version..

```
root@gateway:/bin# ldd busybox
	/lib/ld-musl-x86_64.so.1 (0x7fc193638000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fc193590000)
	libc.so => /lib/ld-musl-x86_64.so.1 (0x7fc193638000)

root@gateway:/bin# ldd busybox.orig
	/lib/ld-musl-x86_64.so.1 (0x7fd0ad122000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fd0ad078000)
	libc.so => /lib/ld-musl-x86_64.so.1 (0x7fd0ad122000)
```

oskarirauta, 2024-02-19:
musl strlen.c is here: https://github.com/bminor/musl/blob/master/src/string/strlen.c

```
#include <string.h>
#include <stdint.h>
#include <limits.h>

#define ALIGN (sizeof(size_t))
#define ONES ((size_t)-1/UCHAR_MAX)
#define HIGHS (ONES * (UCHAR_MAX/2+1))
#define HASZERO(x) ((x)-ONES & ~(x) & HIGHS)

size_t strlen(const char *s)
{
	const char *a = s;
#ifdef __GNUC__
	typedef size_t __attribute__((__may_alias__)) word;
	const word *w;
	for (; (uintptr_t)s % ALIGN; s++) if (!*s) return s-a;
	for (w = (const void *)s; !HASZERO(*w); w++);
	s = (const void *)w;
#endif
	for (; *s; s++);
	return s-a;
}
```

where line 16 is:
`for (; (uintptr_t)s % ALIGN; s++) if (!*s) return s-a;`

It seems ok to me...
So, busybox then must be calling strlen with a pointer that has not been initialised. I used grep -rnw . -e strlen in busybox's code, but that gave me way too many results to narrow down the problem..

brada4, 2024-02-19:
it is invalid *s that is hit by strlen, somewhere before in `gdb> bt` chain.
you can expand source or copy file by file into `pwd` ....

oskarirauta, 2024-02-19:
@brada4

said what? Sorry, got no idea how to perform any of that..

brada4, 2024-02-19:
gdb busybox
> r grep ..... stren .... whatever
CRASH
> bt
> info proc mapping 
You can repeat crashes as muc as you want....
> run grep... ......
CRASH
> q

oskarirauta, 2024-02-19:
Thanks..

```
(gdb) r pwd
Starting program: /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox_unstripped pwd

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f9d568 in strlen (s=s@entry=0x555555543f0b <error: Cannot access memory at address 0x555555543f0b>) at src/string/strlen.c:16
16	src/string/strlen.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7f9d568 in strlen (s=s@entry=0x555555543f0b <error: Cannot access memory at address 0x555555543f0b>)
    at src/string/strlen.c:16
#1  0x00007ffff7f7937c in get_cie_encoding (cie=cie@entry=0x555555543f02)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:347
#2  0x00007ffff7f794aa in classify_object_over_fdes (ob=0x5555555e2ee0 <object>, this_fde=0x555555554000, 
    range=0x7fffffffe910)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:742
#3  0x00007ffff7f7a111 in __register_frame_info_bases (dbase=<optimized out>, tbase=<optimized out>, 
    ob=0x5555555e2ee0 <object>, begin=<optimized out>)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:129
#4  __register_frame_info_bases (begin=<optimized out>, ob=0x5555555e2ee0 <object>, tbase=<optimized out>, 
    dbase=<optimized out>)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:109
#5  0x00007ffff7fe0662 in do_init_fini (queue=<optimized out>) at ldso/dynlink.c:1606
#6  0x00007ffff7fe23c7 in __libc_start_init () at ldso/dynlink.c:1619
#7  0x00007ffff7fa5e8f in libc_start_main_stage2 (main=0x55555559d9c0 <main>, argc=2, argv=0x7fffffffeae8)
    at src/env/__libc_start_main.c:92
#8  0x000055555557c27e in _start ()
(gdb) info proc mapping
process 19155
Mapped address spaces:

          Start Addr           End Addr       Size     Offset  Perms  objfile
      0x555555554000     0x55555557c000    0x28000        0x0  r--p   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox_unstripped
      0x55555557c000     0x5555555df000    0x63000    0x27000  r-xp   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox_unstripped
      0x5555555df000     0x5555555e2000     0x3000    0x89000  r--p   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox_unstripped
      0x5555555e2000     0x5555555e3000     0x1000    0x8b000  rw-p   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox_unstripped
      0x5555555e3000     0x5555555e4000     0x1000        0x0  rw-p   [heap]
      0x5555555e4000     0x5555555e5000     0x1000        0x0  ---p   [heap]
      0x5555555e5000     0x5555555e6000     0x1000        0x0  rw-p   [heap]
      0x7ffff7f67000     0x7ffff7f6b000     0x4000        0x0  r--p   /lib/libgcc_s.so.1
      0x7ffff7f6b000     0x7ffff7f7c000    0x11000     0x4000  r-xp   /lib/libgcc_s.so.1
      0x7ffff7f7c000     0x7ffff7f7f000     0x3000    0x15000  r--p   /lib/libgcc_s.so.1
      0x7ffff7f7f000     0x7ffff7f80000     0x1000    0x17000  r--p   /lib/libgcc_s.so.1
      0x7ffff7f80000     0x7ffff7f81000     0x1000    0x18000  rw-p   /lib/libgcc_s.so.1
      0x7ffff7f81000     0x7ffff7f85000     0x4000        0x0  r--p   [vvar]
      0x7ffff7f85000     0x7ffff7f86000     0x1000        0x0  r-xp   [vdso]
      0x7ffff7f86000     0x7ffff7f9a000    0x14000        0x0  r--p   /lib/libc.so
      0x7ffff7f9a000     0x7ffff7fe5000    0x4b000    0x14000  r-xp   /lib/libc.so
      0x7ffff7fe5000     0x7ffff7ffa000    0x15000    0x5f000  r--p   /lib/libc.so
      0x7ffff7ffa000     0x7ffff7ffb000     0x1000    0x73000  r--p   /lib/libc.so
      0x7ffff7ffb000     0x7ffff7ffc000     0x1000    0x74000  rw-p   /lib/libc.so
      0x7ffff7ffc000     0x7ffff7fff000     0x3000        0x0  rw-p   
      0x7ffffffde000     0x7ffffffff000    0x21000        0x0  rw-p   [stack]
(gdb) 

```

brada4, 2024-02-19:
now make src directory and add mentioned source files there to get debug symbols (tm)
The *s address looks taken from before any mapped memory - could be unintialized memory piece taken as pointer, before was somehow left after by ?ld.so? to map objects.

oskarirauta, 2024-02-19:
results:

```
(gdb) r
Starting program: /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox 

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f9d568 in strlen (s=s@entry=0x555555543f0b <error: Cannot access memory at address 0x555555543f0b>) at src/string/strlen.c:16
warning: Source file is more recent than executable.
16		for (; (uintptr_t)s % ALIGN; s++) if (!*s) return s-a;
(gdb) r grep
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox grep

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f9d568 in strlen (s=s@entry=0x555555543f0b <error: Cannot access memory at address 0x555555543f0b>) at src/string/strlen.c:16
16		for (; (uintptr_t)s % ALIGN; s++) if (!*s) return s-a;
(gdb) bt
#0  0x00007ffff7f9d568 in strlen (s=s@entry=0x555555543f0b <error: Cannot access memory at address 0x555555543f0b>) at src/string/strlen.c:16
#1  0x00007ffff7f7937c in get_cie_encoding (cie=cie@entry=0x555555543f02)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:347
#2  0x00007ffff7f794aa in classify_object_over_fdes (ob=0x5555555e2ee0 <object>, this_fde=0x555555554000, range=0x7fffffffe930)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:742
#3  0x00007ffff7f7a111 in __register_frame_info_bases (dbase=<optimized out>, tbase=<optimized out>, ob=0x5555555e2ee0 <object>, begin=<optimized out>)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:129
#4  __register_frame_info_bases (begin=<optimized out>, ob=0x5555555e2ee0 <object>, tbase=<optimized out>, dbase=<optimized out>)
    at /usr/src/openwrt/build_dir/toolchain-x86_64_gcc-13.2.0_musl/gcc-13.2.0/libgcc/unwind-dw2-fde.c:109
#5  0x00007ffff7fe0662 in do_init_fini (queue=<optimized out>) at ldso/dynlink.c:1606
#6  0x00007ffff7fe23c7 in __libc_start_init () at ldso/dynlink.c:1619
#7  0x00007ffff7fa5e8f in libc_start_main_stage2 (main=0x55555559d9c0 <main>, argc=2, argv=0x7fffffffeb08) at src/env/__libc_start_main.c:92
#8  0x000055555557c27e in _start ()
(gdb) info proc mapping
process 19256
Mapped address spaces:

          Start Addr           End Addr       Size     Offset  Perms  objfile
      0x555555554000     0x55555557c000    0x28000        0x0  r--p   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox
      0x55555557c000     0x5555555df000    0x63000    0x27000  r-xp   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox
      0x5555555df000     0x5555555e2000     0x3000    0x89000  r--p   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox
      0x5555555e2000     0x5555555e3000     0x1000    0x8b000  rw-p   /usr/src/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.1/busybox
      0x5555555e3000     0x5555555e4000     0x1000        0x0  rw-p   [heap]
      0x5555555e4000     0x5555555e5000     0x1000        0x0  ---p   [heap]
      0x5555555e5000     0x5555555e6000     0x1000        0x0  rw-p   [heap]
      0x7ffff7f67000     0x7ffff7f6b000     0x4000        0x0  r--p   /lib/libgcc_s.so.1
      0x7ffff7f6b000     0x7ffff7f7c000    0x11000     0x4000  r-xp   /lib/libgcc_s.so.1
      0x7ffff7f7c000     0x7ffff7f7f000     0x3000    0x15000  r--p   /lib/libgcc_s.so.1
      0x7ffff7f7f000     0x7ffff7f80000     0x1000    0x17000  r--p   /lib/libgcc_s.so.1
      0x7ffff7f80000     0x7ffff7f81000     0x1000    0x18000  rw-p   /lib/libgcc_s.so.1
      0x7ffff7f81000     0x7ffff7f85000     0x4000        0x0  r--p   [vvar]
      0x7ffff7f85000     0x7ffff7f86000     0x1000        0x0  r-xp   [vdso]
      0x7ffff7f86000     0x7ffff7f9a000    0x14000        0x0  r--p   /lib/libc.so
      0x7ffff7f9a000     0x7ffff7fe5000    0x4b000    0x14000  r-xp   /lib/libc.so
      0x7ffff7fe5000     0x7ffff7ffa000    0x15000    0x5f000  r--p   /lib/libc.so
      0x7ffff7ffa000     0x7ffff7ffb000     0x1000    0x73000  r--p   /lib/libc.so
      0x7ffff7ffb000     0x7ffff7ffc000     0x1000    0x74000  rw-p   /lib/libc.so
      0x7ffff7ffc000     0x7ffff7fff000     0x3000        0x0  rw-p   
      0x7ffffffde000     0x7ffffffff000    0x21000        0x0  rw-p   [stack]
(gdb)

```

brada4, 2024-02-19:
It is libgcc exception handler??? I am not smart enough to debug early program init before main().

oskarirauta, 2024-02-19:
you and me both.. what the f*ck..

jow-, 2024-02-19:
It's also entirely unexpected, especially on x86... @oskarirauta - please test without `CONFIG_MOLD` and without `CONFIG_GCC_DEFAULT_PIE`

oskarirauta, 2024-02-19:
I'll keep gcc_default_pie for now.. I've had it for ages.. Biggest thing that has changed, is mold..
Just erased my whole build.. Takes 2-3 days now to get it finished..

brada4, 2024-02-19:
it may suffice to have musl+busybox to verify issue is gone.

oskarirauta, 2024-02-19:
Maybe. Playing for safe bet here..
I already earlier changed busybox's makefile to opt-out for mold, but that didn't do the trick..

oskarirauta, 2024-02-21:
@jow- @brada4

Build has completed. I haven't yet installed the image, but I went to busybox's build dir and executed busybox_unstripped directly from there with ls and sh arguments, both succeeded, no segmentation fault, so I am pretty confident that issue is fixed by disabling CONFIG_MOLD and CONFIG_USE_MOLD.

I previously added manually tag to opt-opt busybox from mold and re-built it without mold, with same result; therefore I suspect that problem is not with busybox (or busybox alone), instead somewhere else, for example musl--
Anyway, without finding the source of the problem, I can verify that use of MOLD links something in a way that stuff gets broken; I'd like to recommend for removal of MOLD, but on the other hand, I am huge fan of improvement and development, so maybe then not, it's work in progress; not so sure if it comes up only when experimental is enabled, but it should be- and STILL, it should at least have mention in it's description that describes better what MOLD even is (had to google it as I had never heard of it before), with a huge warning sign that very most likely this will brake your build- and should be only used for if agenda is to compile packages but still with caution, and not for building a image, at least until problem is found and fixed.

Currently known software that had this issue, is only busybox; but I didn't install every package and try them- who knows how many packages suffer this issue when built with mold- so it should be mentioned that use should be experimental and only for development purposes and comes with risk.

I am not yet closing this issue, until this _has been seen_..

brada4, 2024-02-21:
If you could try dumpelf/readelf to get differences in resulting executables you could report that "upstream"

oskarirauta, 2024-02-21:
@brada4 

Unfortunately I don't have that broken version anymore; I made sure everything is re-built, erased it all and built all from the beginning..

closed 2024-02-21 by oskarirauta