openwrt/openwrt #14685: brcmfmac makes CPU stalls

Mile-Lile, 2024-02-20:
### Describe the bug

brcmfmac makes CPU stalls on latest SNAPSHOT

### OpenWrt version

25252

### OpenWrt release

SNAPSHOT 

### OpenWrt target/subtarget

arm/bcm53xx

### Device

r8000

### Image kind

Official downloaded image

### Steps to reproduce

```
 Feb 19 17:40:48 R8000 kernel [102516.015596] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:40:48 R8000 kernel [102516.017386] ieee80211 phy0: brcmf_msgbuf_query_dcmd: Timeout on response for query command
 Feb 19 17:40:48 R8000 kernel [102516.023350] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: C"
 Feb 19 17:40:48 R8000 kernel [102516.031597] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-5
 Feb 19 17:40:48 R8000 kernel [102516.039059] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:40:48 R8000 kernel [102516.051217] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
 Feb 19 17:40:48 R8000 kernel [102516.055023] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: Ԏ%
 Feb 19 17:40:48 R8000 kernel [102516.063108] ieee80211 phy1: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-12
 Feb 19 17:40:48 R8000 kernel [102516.070272] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:40:48 R8000 kernel [102516.082240] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
 Feb 19 17:40:48 R8000 kernel [102516.086979] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:40:48 R8000 kernel [102516.094810] ieee80211 phy1: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-12
 Feb 19 17:40:48 R8000 kernel [102516.101785] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:40:48 R8000 kernel [102516.113993] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
 Feb 19 17:40:48 R8000 kernel [102516.117745] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:40:48 R8000 kernel [102516.125528] ieee80211 phy1: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-12
 Feb 19 17:40:48 R8000 kernel [102516.133230] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: C"
 Feb 19 17:40:48 R8000 kernel [102516.145267] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
 Feb 19 17:40:48 R8000 kernel [102516.149340] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:40:48 R8000 kernel [102516.157135] ieee80211 phy1: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-12
 Feb 19 17:41:26 R8000 kernel [102552.415271] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:26 R8000 kernel [102552.416873] net_ratelimit: 4 callbacks suppressed
 Feb 19 17:41:26 R8000 kernel [102552.416888] ieee80211 phy2: brcmf_msgbuf_query_dcmd: Timeout on response for query command
 Feb 19 17:41:26 R8000 kernel [102552.423026] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: C"
 Feb 19 17:41:26 R8000 kernel [102552.427713] ieee80211 phy2: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-5
 Feb 19 17:41:26 R8000 kernel [102552.436481] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:26 R8000 kernel [102552.459504] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: Ԏ%
 Feb 19 17:41:26 R8000 kernel [102552.466699] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:46 R8000 kernel [102572.405842] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:46 R8000 kernel [102572.413874] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:41:46 R8000 kernel [102572.420898] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:41:46 R8000 kernel [102572.428237] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:46 R8000 kernel [102572.435975] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: C"
 Feb 19 17:41:46 R8000 kernel [102572.443045] rcu: INFO: rcu_sched self-detected stall on CPU
 Feb 19 17:41:47 R8000 kernel [102572.443052] rcu: 	0-....: (131866 ticks this GP) idle=5f1/1/0x40000002 softirq=1406219/1406219 fqs=63770
 Feb 19 17:41:47 R8000 kernel [102572.443068] 	(t=132021 jiffies g=2346757 q=146212)
 Feb 19 17:41:47 R8000 kernel [102572.443077] NMI backtrace for cpu 0
 Feb 19 17:41:47 R8000 kernel [102572.443083] CPU: 0 PID: 20762 Comm: kworker/0:2 Not tainted 5.15.148 #0
 Feb 19 17:41:47 R8000 kernel [102572.443092] Hardware name: BCM5301X
 Feb 19 17:41:47 R8000 kernel [102572.443097] Workqueue: events brcmf_core_bus_reset [brcmfmac]
 Feb 19 17:41:47 R8000 kernel [102572.443204] [<c01096e0>] (unwind_backtrace) from [<c0105270>] (show_stack+0x10/0x20)
 Feb 19 17:41:47 R8000 kernel [102572.443234] [<c0105270>] (show_stack) from [<c041e720>] (dump_stack_lvl+0x40/0x4c)
 Feb 19 17:41:47 R8000 kernel [102572.443269] [<c041e720>] (dump_stack_lvl) from [<c0426b1c>] (nmi_cpu_backtrace+0xdc/0x120)
[... 27 lines cut ...]
 Feb 19 17:41:47 R8000 kernel [102572.444112] [<c013724c>] (worker_thread) from [<c013e5e8>] (kthread+0x148/0x180)
 Feb 19 17:41:47 R8000 kernel [102572.444133] [<c013e5e8>] (kthread) from [<c0100114>] (ret_from_fork+0x14/0x20)
 Feb 19 17:41:47 R8000 kernel [102572.444145] Exception stack(0xc388ffb0 to 0xc388fff8)
 Feb 19 17:41:47 R8000 kernel [102572.444152] ffa0:                                     00000000 00000000 00000000 00000000
 Feb 19 17:41:47 R8000 kernel [102572.444159] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 Feb 19 17:41:47 R8000 kernel [102572.444166] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
 Feb 19 17:41:47 R8000 kernel [102572.781920] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:47 R8000 kernel [102572.789671] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: Ԏ%
 Feb 19 17:41:47 R8000 kernel [102572.796892] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:47 R8000 kernel [102572.804913] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:41:47 R8000 kernel [102572.811922] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:41:47 R8000 kernel [102572.819261] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:47 R8000 kernel [102572.827016] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: C"
 Feb 19 17:41:47 R8000 kernel [102572.834506] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:47 R8000 kernel [102572.842252] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: Ԏ%
 Feb 19 17:41:47 R8000 kernel [102572.849472] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|
 Feb 19 17:41:47 R8000 kernel [102572.857499] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:41:47 R8000 kernel [102572.864507] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE:
 Feb 19 17:41:47 R8000 kernel [102572.871844] brcmfmac 0001:03:00.0: brcmf_pcie_bus_console_read: CONSOLE: g� ��|

```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Mile-Lile, 2024-02-20:
@rmilecki
[2024-02-19-logs-stall.zip](https://github.com/openwrt/openwrt/files/14348223/2024-02-19-logs-stall.zip)

Mile-Lile, 2024-02-20:
Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269

Mile-Lile, 2024-02-20:
including maintainer of brcmfmac:
@arend.vanspriel@broadcom.com

LuckyDima, 2024-02-29:
I have the same bug.

Netgear R8000 (BCM4709)
ARMv7 Processor rev 0 (v7l)
bcm53xx/generic
OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256
5.15.137

`[80077.494399] ieee80211 phy1: brcmf_cfg80211_get_station: GET STA INFO failed, -12
[80135.547535] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
[80135.555711] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
[80135.563850] ieee80211 phy1: brcmf_cfg80211_get_station: GET STA INFO failed, -12
[80161.591473] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
[80161.599647] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
[80161.607896] ieee80211 phy1: brcmf_cfg80211_get_station: GET STA INFO failed, -12
[80181.173566] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
[80181.181734] ieee80211 phy1: brcmf_msgbuf_tx_ioctl: Failed to reserve space in commonring
[80181.189959] ieee80211 phy1: brcmf_cfg80211_get_station: GET STA INFO failed, -12`

Mile-Lile, 2024-03-06:
let me add this one as well:

```
 Mar 06 05:20:56 178.220.177.32 kernel [41739.178467] 0002:03:00.0: CONSOLE: 041746.172 phy0: PSM microcode watchdog fired at 41733 (seconds) 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.187768] 0002:03:00.0: CONSOLE: 041746.172 phy0: psmdebug 0x00ec8111, phydebug 0x00000048, psm_brc 0x0501 psm_brc_1 0x0000maccontrol 0x84160403 ifsstat 0x00af m_cts_duration 0x0000 m_ucode_dbgst 0x0002 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.206038] 0002:03:00.0: CONSOLE: 041746.172 wepctl 0x0050 wepstat 0x0000 wep_hdrloc 0x001e wep_psdulen 281 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.216025] 0002:03:00.0: CONSOLE: 041746.172      RXE_RXCNT: 0x011f   DAGG: ctl: 0x0000, bleft: 0x0096 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.225523] 0002:03:00.0: CONSOLE:      offs: 0x000c, stat: 0x1056, len: 0x0000 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.232995] 0002:03:00.0: CONSOLE: 041746.172 M_RXFRM_BLK scr(0:63) btc_params(0:63) phydebug PC ampdu clk off 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.243076] 0002:03:00.0: CONSOLE: 041746.172 wlc_dump_aggfifo: 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.249089] 0002:03:00.0: CONSOLE: 041746.172 framerdy 0x0 bmccmd 7 framecnt 1024  
 Mar 06 05:20:56 178.220.177.32 kernel [41739.256822] 0002:03:00.0: CONSOLE: 041746.172 AQM agg params 0xfc0 maxlen hi/lo 0x0 0xffff minlen 0x0 adjlen 0x0 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.267153] 0002:03:00.0: CONSOLE: 041746.172 AQM agg results 0x8001 len hi/lo: 0x0 0x22 BAbitmap(0-3) 0 0 0 0 
 Mar 06 05:20:56 178.220.177.32 kernel [41739.277244] ieee80211 phy1: brcmf_psm_watchdog_notify: PSM's watchdog has fired! ```

```