openwrt/openwrt #14686: FRITZ!box 7530 LAN MAC not set according to TFFS data

mrkiko, 2024-02-20:
I am experiencing the MAC address not set for the LAN interface of this device.
MAC addresses for the VDSL interface is correct.

In the TFFS area I see those MAC related entries:
maca=3C:37:12:XX:XX:XX
macb=3C:37:12:XX:XX:XX
macwlan=3C:37:12:XX:XX:XX
macwlan2=3C:37:12:XX:XX:XX
macdsl=3C:37:12:XX:XX:XX

I am running
Linux ???? 6.1.77 #0 SMP Fri Feb  9 00:44:19 2024 armv7l GNU/Linux
OpenWrt SNAPSHOT r25082-c6ed34940d

Thanks a lot for your great work!

francisuk1989, 2024-02-22:
Same issue, Same router however im getting the following MAC Address to 3C:37:12 @mrkiko

OpenWrt SNAPSHOT, r25277-44168fda78
uname -a
`Linux DSL 6.1.78 #0 SMP Wed Feb 21 20:42:23 2024 armv7l GNU/Linux`

eth026:74:23:46:AC:A8
lan1 26:74:23
lan2 26:74:23
lan3 26:74:23
lan4 26:74:23

https://maclookup.app/macaddress/dc396f
theses are correct for me
MAC: DC:39:6F:XX:XX > phy0-ap0
MAC: DC:39:6F:XX:XX >phy1-ap0
MAC: DC:39:6F:XX:XX > dsl0.101

janh, 2024-12-23:
I can confirm this issue using OpenWrt 24.10. The LAN interfaces use a random MAC address instead of the proper one.

This is a regression, it worked fine in 23.05.

mrkiko, 2025-01-07:
This issue has been fixed via commit d09dd75fbd1314ec5f3571511195da4012d7a51a
... thanks!

closed 2025-01-07 by mrkiko