openwrt/openwrt #14687: firewall4: Missing a nft define for uplink IPv6 prefix

Shine-, 2024-02-20:
### Describe the bug

fw4 does, by default, create a number of `define`s that can e.g. be used for dynamic rules in custom `.nft` includes,

Imagine my ISP delegates an IPv6 prefix (IA-PD) `2001:db8:7654:3200::/56` and assigns an IPv6 address (IA-NA) `2001:db8:7654::16e9/128`.

fw4 will, for example. create the following `define`s:
```
root@OpenWrt:~# fw4 print
[...]
        #
        # Defines
        #

        define lan_devices = { "br-lan" }
        define lan_subnets = { 192.168.1.0/24, 2001:db8:7654:3200::/60, fdcd:b72c:3960::/60 }

        define wan_devices = { "eth0.2" }
        define wan_subnets = { 198.51.100.0/24, 2001:db8:7654::16e9 }
[...]
```
As you can see, fw4 fills the "wan_subnets" define with the IA-NA, and it also fills the "lan_subnets" define with the sub-prefix that OpenWrt assigned to the LAN interface (`2001:db8:7654:3200::/60`).

What I'm missing here is a `define` for the uplink IPv6 prefix that was assigned to me by my ISP, something like this, for example:

`define wan_prefixes = { 2001:db8:7654:3200::/56 }`

This may not count as a bug report but a feature request instead. But somehow I was kindof expecting fw4 to also expose my uplink prefix, so I can work with it in custom `.nft` includes.

If the current behavior is on purpose and the uplink IA-PD isn't supposed to be available as a `define`, please feel free to close this as "wontfix".

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WDR4300 v1

### Image kind

Official downloaded image

### Steps to reproduce

`fw4 print`

### Actual behaviour

The only IPv6 `define`s that are available are WAN IPv6 address (IA-NA) and the downstream sub-prefix that OpenWrt assigned to the LAN interface.

### Expected behaviour

fw4 should also expose the upstream prefix (IA-PD) as a `define`.

### Additional info

n/a

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Shine-, 2024-02-20:
Yes, **finding** the prefix is not the issue, can also be done easily via `ubus`. But that still won't help for using it in `*.nft` includes.

Shine-, 2024-02-21:
So I thought I could help myself with a `script` include, instead.

And I failed as well, because apparently, `fw4` will call the `script` include _before_ populating the `fw4` rulebase (opposite to `fw3`, which called `firewall.user` _after_ populating the rulebase.)

So I can't even manipulate the default chains using a `script` include, because they don't exist yet at this point. Or rather, I _could_ pre-fill the default chains, but they would then be moved to the very beginning, instead of where `fw4` would place them, potentially causing undesired effects.

So if I may, I'd like to extend this suggestion:

- either add a `define` for the upstream IA-PD(s)
- or add an option to choose whether to call `script` includes before or after populating the `fw4` rule base
- or perhaps even both?

systemcrash, 2024-04-05:
If you could pull one together, I can make a commit. 

What's the use case to have this WAN defined PD included? Just 'everything in the right place'? 

If the IA-PD is `2001:db8:7654:3200::/56` - this is the PD for assignment 'internally' (or, wherever you like, actually), which the router picks from and does `2001:db8:7654:3200::/60` on the LAN. Your link-net is just `2001:db8:7654::16e9/128`. 

Where the prefix is assigned and split is implementation detail. You can chop it up how you wish. So yeah, it could also go to the WAN.

The problem (I think) is that you have a /56 on the WAN, and the 60 on the LAN, so the firewall might now straddle a zone across LAN and WAN. Conceptually, no problem, but for clean configurations, maybe not desirable.

But in terms of LAN WAN range delegation/separation, it's OK.

Shine-, 2024-04-05:
Having the ISP-assigned PD available as a named variable will greatly help with building dynamic firewall rules, which are self-updating whenever fw4 rebuilds the rule base (e.g. on WAN I/F status change). You can think of allow/block rules between public prefix and cascaded sub-routers, or adding IPv6 privacy using source NAT over the whole IA-PD, instead of `masq6` (which will always use only the IA-NA as source address). And countless other possibilities.

Here's a `script` include I put together as an example for a more privacy-aware replacement of `masq6`. Only problem is, as described above, it will be called **prior** to filling the fw4 ruleset, instead of **after** filling the ruleset (like fw3 did it).
```
P=$(ubus call network.interface.wan6 status | jsonfilter -e "@['ipv6-prefix'][0].address")
L=$(ubus call network.interface.wan6 status | jsonfilter -e "@['ipv6-prefix'][0].mask")
#echo $P / $L
[ -n "$P" -a -n "$L" ] && {
  nft add rule inet fw4 srcnat_wan snat ip6 prefix to ip6 saddr map { 2001:db8::/32 : $P/$L }
  nft add rule inet fw4 srcnat_wan snat ip6 prefix to ip6 saddr map { fc00::/7 : $P/$L }
}
```
And yes, it's also for completeness' / correctness' sake. As you can see from the fw4 dump above, it's for example mixing the IPv4 external subnet (which doesn't even belong to me) with my IPv6 IA-NA in the same variable. Likely this stems from a PPPoE view of things, where external IPv4 subnet == external IPv4 address (/32). Nevertheless, for all other intents and purposes, it's simply wrong to mix these up.

In addition, I do see problems to actually use these mixed (IPv4/IPv6) nft variables in fw4 rules, like e.g. a `.nft` version of the `masq6` replacement example above. Nftables usually refuses to add any rules that contain an address family that it doesn't expect.

Yes, it should, in fact, be rather easy to attend to both requests I'm describing here (`position` option for `script` include, and a named NFT variable for IA-PD). I could probably do it myself if I spend some time getting to know ucode. Which would, on the other hand, mean that I will end up with even more personal patches that I have to maintain on my own forever.

I was hoping to gain @jow-'s attention with this issue, since he's the master of fw4, and nothing he doesn't approve or deem useful will land in fw4 anyway. So either he hasn't noticed it (unlikely - he's regularly participating in GH issues) or he's simply not interested at this time. Which is bad luck for me, end of story.

systemcrash, 2024-04-05:
Yeah, he's likely the right person to tag. Summarise the need into two lines which is easy to understand, and concise to decide whether it's worth pursuing (unless you have a patch available).