openwrt/openwrt #14688: XDR6088 ports order mismatch with the physical ports

pupboss, 2024-02-20:
### Describe the bug

XDR6088 ports order mismatch with the physical ports.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

TP-Link TL-XDR6088

### Image kind

Official downloaded image

### Steps to reproduce

Login to LuCI, and check status overview

### Actual behaviour

Actual order in openwrt is:

eth1, lan1, lan2, lan3, lan4, lan5

### Expected behaviour

The physical ports order is:
eth1, lan5, lan1, lan2, lan3, lan4

while eth1 and lan5 are 2.5G ports.

### Additional info

For the reason why lan5 is a 2.5G port and located at the second physical port, its a long story.

The CPU has only two HSGMII bus eth0 and eth1, and lan1-lan5 ports are extended by a switch chip which takes eth0 communicating with CPU.

```
eth0 -> ../../devices/platform/soc/15100000.ethernet/net/eth0
eth1 -> ../../devices/platform/soc/15100000.ethernet/net/eth1
lan1 -> ../../devices/platform/soc/15100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan1
lan2 -> ../../devices/platform/soc/15100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan2
lan3 -> ../../devices/platform/soc/15100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan3
lan4 -> ../../devices/platform/soc/15100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan4
lan5 -> ../../devices/platform/soc/15100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan5
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pupboss, 2024-02-20:
Hi @brada4 , it has labels, let me share a more detailed picture. All 6 ports can be either WAN or LAN

[image]
Source: https://www.acwifi.net/20864.html

Thanks for the reply...though I am not familiar with the `pause` dts file 😆

brada4, 2024-02-20:
What I am saying ports in DTS file are configured as 2.5Gbps (like optical) fixed rate, not the hsgmii which can negotiate speed.
PAUSE means ethernet flow control, which is 99% wrong whenever enabled, since both end devices are perfectly capable of wire rate.

brada4, 2024-02-20:
(i was not posting permalink to DTS, you can browse in repo to latest revicsion, port description is in the beginning of file)

pupboss, 2024-02-20:
@brada4 About the reason, I encountered some issue upon the first time flashing openwrt with WAN cable connected to lan4.

By default lan4 was included in br-lan, therefore my PC cannot obtain an internal IP address, making it impossible for me to access LuCI to manage the router. I was scared, thinking the flashing process had failed haha.

Afterwards I disconnected the WAN cable and was able to login to LuCI.

pupboss, 2024-02-20:
> 1/ 2.5GBPS port speed in kernel DTS is wrong 2/ the port mapping script does not help either As per https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=21be2c26d555b33e9faf85be3e980876810b29d3 @dangowrt should be able to adjust.

Got it. Thanks a lot. @brada4 

@dangowrt TLDR;

https://github.com/openwrt/luci/blob/c8cddc35302380882bab910679f717d65f715029/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/29_ports.js#L349

As per the above file, the sorting function in LuCI status page, it compares names such as eth0 lan1 lan2.

Can you help to see how to adjust label naming to perfectly match the order of physical network ports?

dangowrt, 2024-02-20:
> 1/ 2.5GBPS port speed in kernel DTS is wrong

Please elaborate more where you believe something is wrong. I got XDR-6068 which got the same PHYs and shares most of the DTS, and I can confirm the the 2.5G ports work fine in all speeds.

dangowrt, 2024-02-20:
> it should be hsgmii,

The SoC does not support HSGMII which is a proprietary RealTek thing and not really a standard. As an interface mode this isn't even defined in Linux or OpenWrt (outside of the `realtek` target used on RealTek switches).

Therefore both PHY and MAC are switching between "classic" 10M/100M/1000M SGMII and 2500Base-X depending on the established link speed. The PHY mode states in device tree merely serves as a fall-back and starting point, it is subsequently modified by the PHY driver.

> the 2500base-x kind of dictates advertised speed is 2.5gbit only?

Also that is not true. Aquantia/Marvell AQRate AQR112 PHY for example always connects to the MAC in 2500Base-X mode and uses pause-frames/flow-control to perform rate-adaption to lower speeds inside the PHY. This is not ideal, of course, but very common.

brada4, 2024-02-20:
Cannot guess without actual device. SGMII would be repeating symbol from the NIC side to match wire rate.

dangowrt, 2024-02-20:
> SGMII would be repeating symbol from the NIC side to match wire rate.

Yes, and this is how it is done for 10M/100M and 1000M mode. Once a link comes up with 2500M, the interface mode used between MAC and PHY is switched from Cisco's SGMII to 2500Base-X by the PHY driver.

dangowrt, 2024-02-21:
@pupboss I guess I'm the wrong person for that task, I'm not into UI stuff at all (as in: I highly appreciate that other people are taking care of that).

pupboss, 2024-02-21:
@dangowrt Hi Daniel

https://github.com/openwrt/openwrt/blob/c42b915af0ea53bc1b65bc72afc9875da08b14d2/target/linux/mediatek/dts/mt7986a-tplink-tl-xdr6088.dts#L38

The "adjusting label" we mentioned earlier refers to this line of code, which needs some adjustments to ensure it can be sorted according to the sequence of physical interfaces. Actually its not a UI part.

Now the naming is `eth1 lan1 lan2 lan3 lan4 lan5`, however the physical interfaces are `eth1 lan5 lan1 lan2 lan3 lan4`

What we suggested was, renaming the `lan5` to `eth2`, to ensure it can be sorted correctly.

Or if you want to make it looks better, could consider to rename as the below mapping:

| Current name | Renaming name |
|--------------|---------------|
| eth1         | eth1          |
| lan1         | eth3          |
| lan2         | eth4          |
| lan3         | eth5          |
| lan4         | eth6          |
| lan5         | eth2          |

Anyhow, our ultimate goal is simply to ensure that the naming sequence aligns with the order of physical interfaces. What do you suggest

dangowrt, 2024-02-21:
Renaming as such is ok as long as the new name doesn't start by `eth`. Stuff like renaming `eth2` to `eth0` and `eth0` to `eth1` and `eth1` to `eth2` looks easy, but it's a mess in practice and worth avoiding, esp. as then we also will not be able to tell if renaming has happened or not.

I'm not opposed to change the order or `lan` ports, or even rename `eth1` to `lan1`, given that this would then more closely reflect how the device is labeled.

pupboss, 2024-02-21:
I am fine with the `dedicated wan` too. Thanks for your reply and looking forward to the change 😄 

Let me check what I can contribute.

pupboss, 2024-02-21:
I just got a workaround to fix this issue on XDR6088, just incase anyone needed.

[image]
In `/etc/board.json`, adjust the order as what you want.

[image]
in `/www/luci-static/resources/view/status/include/29_ports.js` L69

original code
```
known_ports.sort(function(a,b){return L.naturalCompare(a.device,b.device);});
```

change with
```
known_ports.sort(function(a,b){return L.naturalCompare(b.role,a.role);});
```

[image]

ALL DONE 🎉 

----

For the permanent solution, seems we will have to refactor `borad.json` and add additional information to support customized index, then sort with the index in `29_ports.js`

aiamadeus, 2024-02-22:
I don't think this is a bug as the port order is not marked on the shell at all.
I can change it to this: 'usb | eth1 lan5 lan4 lan3 lan2 lan1 |power'.
But I don't know if it makes sense to do this.

pupboss, 2024-02-22:
Hi @aiamadeus

The sequence of ports is only crucial for those first time installing OpenWrt, as they need to know which port is named what during port configuration. Otherwise, there's a risk of misconfiguring the WAN port as the LAN port, resulting in devices directly obtaining IP addresses from the modem, thus making it unable to obtain lan ip to manage the router.

For those who have already installed OpenWrt and are aware of the port mapping, the order of ports is not as important.

Personally I vote for changing to "usb | eth1 lan5 lan4 lan3 lan2 lan1 | power" 👍

aiamadeus, 2024-02-23:
@pupboss You can use this: https://github.com/aiamadeus/openwrt/commit/70d5191e123be8fb9ab4ad239599f2d44fb6dcf3

pupboss, 2024-02-23:
@aiamadeus thanks a lot for your timely action!

One more thing I would like to mention, this line of code will sort the display regardless of the original port order. Therefore, it may be necessary to make change to the LuCI code accordingly. (However I don't have a spare device to test with, and I can't disrupt the internet at home because we're all still working from home). Sorry for being a freeloader

https://github.com/openwrt/luci/blob/c8cddc35302380882bab910679f717d65f715029/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/29_ports.js#L349

aiamadeus, 2024-02-23:
I don't think luci needs to be changed, if you really mind you can rename eth1 to wan in user space.

pupboss, 2024-02-23:
@aiamadeus  Understood, with the renaming of `lan1`, users who want to set `lan1` as WAN port wont getting confused, that is enough 👍