openwrt/openwrt #14693: wifi crash

mhench, 2024-02-21:
### Describe the bug

wifi controller crashes on disconnect due to weak signal. does NOT crash I just turn off wifi on the tablet. (Samsung SM-T837T)
If I drive away with tablet it will crash

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq807x/generic

### Device

Dynalink DL-WRX36

### Image kind

Official downloaded image

### Steps to reproduce

seems to do it fairly often If I drive away with tablet. (connected via 5G)

### Actual behaviour

wifi stops working until reboot.
[logread.txt](https://github.com/openwrt/openwrt/files/14363262/logread.txt)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-22:
/etc/config/wireless /etc/config/firewall /etc/config/network
wipe out security keys and addresses before posting files.

`logread -e 88:9f:6f:5f:99:e6`
it shows some AP changing activity happened
Can you set AP-s with same names on both radio devices as to trigger roaming with other devices walking off?

mhench, 2024-02-22:
sorry should have included that stuff.
there ARE 2 wifi networks with different SSID's one IOT on 2.4 and one
regular on 5Ghz
I have an IOT device that gets weird if it sees a 5Ghz network on the SSID
it wants.
the tablet should not have tried to connect to the IOT network.
I left something out because it seems the tablet seems to be triggering the
problem
I also always carry a phone. THAT might have tried to connect to 2.4ghz.

oh and while I can't see how this matters both devices then proceed to
connect to the
motorhome wifi (totally different SSID, 5GHZ only)
(motorhome is what I am driving off)
also motorhome is the only time I drive off with the tablet. it is either
in the house or the motorhome.

I don't understand your last sentence.

Thanks.

brada4, 2024-02-22:
Ok, leave 2.4 iot network name and make 2-band normal network.
If you filter hostapd you see your tablet tried iot network once radio kind of crashed.
For better observations rise log size in Luci/System.

mhench, 2024-02-22:
log size issue was due to my wifidog script (before I added the sleep 60)
#!/bin/sh

sleep 60
while true
do
        sleep 30
        logread | grep "c000000.wifi: failed"
        if [ $? -eq 0 ]; then
                sleep 60
                logread > /root/wifilog
                reboot
        fi
done

included a log for 60 seconds after failure detected. removed stuff before
station disconnect. it looked generic and normal.
need reboot above wifi down/up does not do it.

Thu Feb 22 15:47:37 2024 daemon.notice hostapd: phy0-ap1: AP-STA-DISCONNECTED SM-T837EA
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707]
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] wal_peer_control.c:2870 Assertion is_graceful_to_handle failedparam0 :zero, param1 :zero, param2 :zero.
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] Thread ID      : 0x00000069  Thread name    : WLAN RT0  Process ID     : 0
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] Register:
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] SP : 0x4bfb9230
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] FP : 0x4bfb9238
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] PC : 0x4b1080c4
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] SSR : 0x00000008
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] BADVA : 0x00020000
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] LR : 0x4b107860
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707]
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] Stack Dump
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] from : 0x4bfb9230
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707] to   : 0x4bfb9aa8
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.702707]
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.751681] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.773930] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
Thu Feb 22 15:47:37 2024 kern.err kernel: [  281.782158] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
Thu Feb 22 15:47:37 2024 kern.info kernel: [  281.814769] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.092194] ath11k c000000.wifi: failed to find peer SM-T837EA on vdev 1 after creation
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.092251] ath11k c000000.wifi: failed to find peer vdev_id 1 addr SM-T837EA in delete
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.099714] ath11k c000000.wifi: failed peer SM-T837EA delete vdev_id 1 fallback ret -22
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.108415] ath11k c000000.wifi: Failed to add peer: SM-T837EA for VDEV: 1
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.117432] ath11k c000000.wifi: Failed to add station: SM-T837EA for VDEV: 1
Thu Feb 22 15:47:38 2024 daemon.notice hostapd: phy0-ap0: STA SM-T837EA IEEE 802.11: Could not add STA to kernel driver
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.152573] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.152611] ath11k c000000.wifi: failed to send pdev bss chan info request
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.159220] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.165910] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 0
Thu Feb 22 15:47:38 2024 kern.warn kernel: [  282.172443] ath11k c000000.wifi: failed to send WMI_BCN_TMPL_CMDID
Thu Feb 22 15:47:38 2024 kern.info kernel: [  282.179176] phy0-ap1: HW problem - can not stop rx aggregation for SM-T837EA tid 0
Thu Feb 22 15:47:43 2024 kern.err kernel: [  287.132305] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: start timed out
Thu Feb 22 15:47:43 2024 kern.err kernel: [  287.132355] remoteproc remoteproc0: can't start rproc cd00000.q6v5_wcss: -110
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.292249] ath11k_warn: 18 callbacks suppressed
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.292257] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 2
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.296007] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.303688] ath11k c000000.wifi: failed to delete peer vdev_id 2 addr SM-T837EA ret -108
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.309657] ath11k c000000.wifi: Failed to delete peer: SM-T837EA for VDEV: 2
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.318694] ath11k c000000.wifi: Found peer entry 4e:ab:f8:47:d3:9b n vdev 2 after it was supposedly removed
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.326463] ------------[ cut here ]------------
Thu Feb 22 15:47:43 2024 kern.warn kernel: [  287.336297] WARNING: CPU: 1 PID: 1657 at 0xffffffc000a8aa70 ***@***.***+0x85000]
[... cut ...]
Thu Feb 22 15:48:37 2024 kern.warn kernel: [  341.827168] ---[ end trace 4353ceb0ee531a45 ]---
Thu Feb 22 15:48:38 2024 kern.warn kernel: [  342.171161] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
Thu Feb 22 15:48:38 2024 kern.warn kernel: [  342.171195] ath11k c000000.wifi: failed to send pdev bss chan info request
Thu Feb 22 15:48:38 2024 kern.warn kernel: [  342.177821] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
Thu Feb 22 15:48:38 2024 kern.warn kernel: [  342.184492] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 1

brada4, 2024-02-23:
Enclose log between 3 backticks.
The necessary piece is a minute or two before firmware crashes, it is closed source blob running on wifi adapter, once it crashes more than once eithout recovering no point watching further.
If you have permanent device like raspberry or nas on lan  it could ideally gather logs continuously.

brada4, 2024-02-23:
Please format your posts and set proper country code, attaching respective config files.

mhench, 2024-02-23:
boot to 1 minute past failure.
sorry about the formatting. was not aware what was happening.

[firewall.txt](https://github.com/openwrt/openwrt/files/14389973/firewall.txt)
[network.txt](https://github.com/openwrt/openwrt/files/14389975/network.txt)
[wireless.txt](https://github.com/openwrt/openwrt/files/14389976/wireless.txt)

```

Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Fri Feb 23 05:41:40 2024 kern.notice kernel: [    0.000000] Linux version 5.15.137 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Nov 14 13:38:11 2023
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Machine model: Dynalink DL-WRX36
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Zone ranges:
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   DMA32    empty
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   Normal   empty
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Early memory node ranges
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000051100000-0x000000007fffffff]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] percpu: Embedded 17 pages/cpu s30808 r8192 d30632 u69632
Fri Feb 23 05:41:40 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s30808 r8192 d30632 u69632 alloc=17*4096
Fri Feb 23 05:41:40 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
Fri Feb 23 05:41:40 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Memory: 888452K/1048576K available (7808K kernel code, 860K rwdata, 1400K rodata, 1280K init, 289K bss, 160124K reserved, 0K cma-reserved)
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] Root IRQ handler: 0xffffffc00837cdb0
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000113] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000130] pid_max: default: 32768 minimum: 301
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000601] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.000619] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Fri Feb 23 05:41:40 2024 kern.info kernel: [    0.002587] rcu: Hierarchical SRCU implementation.
[... cut ...]
Fri Feb 23 08:44:55 2024 kern.warn kernel: [  879.167605] ath11k c000000.wifi: failed to submit beacon template command: -108
Fri Feb 23 08:44:55 2024 kern.warn kernel: [  879.173776] ath11k c000000.wifi: failed to update bcn template: -108
Fri Feb 23 08:44:55 2024 kern.warn kernel: [  879.180961] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
Fri Feb 23 08:44:55 2024 kern.warn kernel: [  879.187569] ath11k c000000.wifi: failed to set BA BUFFER SIZE 256 for vdev: 0
Fri Feb 23 08:44:55 2024 kern.warn kernel: [  879.194261] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID

```

brada4, 2024-02-23:
Set country code via luci on both radios, that should change code paths around firmware. 1% chance it stops crashing.

brada4, 2024-02-23:
(firewall and network are sane)

xsokolikx, 2024-03-04:
Hi!

I'm getting this firmware crash on a Xiaomi AX3600. Making some bisect work, I found that reverting this 2 commits solves the firmware crash:

- https://patchwork.kernel.org/project/linux-wireless/patch/20220603164559.27769-1-ansuelsmth@gmail.com/
- https://patchwork.kernel.org/project/linux-wireless/patch/20230209222622.1751-1-ansuelsmth@gmail.com/

The side effect is that removing these patches from mac80211 makes roaming unusable when trying to roam between different radios on the same AP (one on 2.4GHz, another one on 5Ghz).  I made these tests using different ath11k firmware versions, like WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1 and WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1, crashing on all of them.

I already tried to contact to commiter of this patches, unsuccessfully.

brada4, 2024-03-04:
@Ansuel ?

bcarlson-dev, 2024-10-11:
Are there any updates on the status of this issue? I'm experiencing something similar. Has anyone found a fix?

Neustradamus, 2024-10-12:
To follow this ticket.

brozikcz, 2024-10-30:
the similar crash on my AX3600

```
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.589878] ath11k_warn: 49 callbacks suppressed
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.589902] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 3
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.593618] ath11k c000000.wifi: failed to send WMI_VDEV_INSTALL_KEY cmd
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.601347] ath11k c000000.wifi: failed to remove peer key 0: -108
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.608001] ath11k c000000.wifi: failed to clear all peer keys for vdev 0: -108
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.614006] ath11k c000000.wifi: Failed to disassociate station: xxx
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.621259] ------------[ cut here ]------------
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.628830] WARNING: CPU: 1 PID: 1679 at 0xffffffc000a88b3c [mac80211@000000000570aaf3+0x86000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.633611] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath11k_ahb ath11k ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 slhc qrtr_smd qrtr qmi_helpers nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat sha512_generic seqiv jitterentropy_rng drbg michael_mic hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug ext4 mbcache jbd2 aquantia hwmon crc_ccitt crc32c_generic
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.691666] CPU: 1 PID: 1679 Comm: hostapd Tainted: G        W         5.15.167 #0
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.713903] Hardware name: Xiaomi AX3600 (DT)
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.721538] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.725968] pc : 0xffffffc000a88b3c [mac80211@000000000570aaf3+0x86000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.732739] lr : 0xffffffc000a88a8c [mac80211@000000000570aaf3+0x86000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.739335] sp : ffffffc00bccb840
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.745927] x29: ffffffc00bccb840 x28: ffffff8002982340 x27: ffffffc00bccbdd0
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.749406] x26: ffffff8001c73880 x25: ffffffc008b760c0 x24: ffffffc000a79000
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.756525] x23: ffffffc00bccba38 x22: ffffff800ec9c900 x21: ffffff801267cab0
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.763642] x20: ffffff8005c691a0 x19: ffffff801267c000 x18: ffffffc008a77b60
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.770760] x17: 63613a63333a3133 x16: 3a3034203a6e6f69 x15: 0000000000000c00
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.777879] x14: 0000000000000400 x13: ffffffc00bccb358 x12: ffffffc008acfb60
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.784996] x11: 0000000000000004 x10: ffffffc008acfb60 x9 : 0000000000000000
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.792114] x8 : ffffffc008a77b10 x7 : ffffffc008a77b60 x6 : 0000000000000001
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.799233] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffffff8002982340
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.806351] x2 : 0000000000000000 x1 : ffffff8002982340 x0 : 00000000ffffff94
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.813469] Call trace:
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.820578]  0xffffffc000a88b3c [mac80211@000000000570aaf3+0x86000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.822841]  0xffffffc000a88e34 [mac80211@000000000570aaf3+0x86000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.829091]  0xffffffc000aa14cc [mac80211@000000000570aaf3+0x86000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.835342]  0xffffffc000a49908 [cfg80211@000000003c7e763b+0x45000]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.841591]  0xffffffc00864288c
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.847836]  0xffffffc008642d38
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.850960]  0xffffffc00864191c
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.854086]  0xffffffc008642198
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.857210]  0xffffffc0086410ac
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.860335]  0xffffffc008641350
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.863461]  0xffffffc00859eae8
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.866586]  0xffffffc0085a14c8
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.869710]  0xffffffc0085a1618
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.872835]  0xffffffc0085a16a4
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.875960]  0xffffffc008022f1c
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.879086]  0xffffffc00802303c
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.882210]  0xffffffc008795d88
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.885335]  0xffffffc008796ba8
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62868.888461]  0xffffffc0080115f4
[... cut ...]
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62869.191672]  0xffffffc008795d88
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62869.194796]  0xffffffc008796ba8
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62869.197921]  0xffffffc0080115f4
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62869.201047] ---[ end trace 301109501b568f0d ]---
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62869.204217] ath11k c000000.wifi: failed to clear all peer keys for vdev 0: -2
Wed Oct 30 07:53:22 2024 kern.warn kernel: [62869.209043] ------------[ cut here ]------------
```