openwrt/openwrt #14696: The service "https-dns-proxy" does not start after rebooting the router

NorthP8, 2024-02-22:
### Describe the bug

I have problems with "https-dns-proxy"
After rebooting the router "https-dns-proxy" does not start. Starts working only after a forced reboot of the "https-dns-proxy" service in the control panel.

In the previous version of OpenWRT 22.03 everything worked without problems.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6U v1

### Image kind

Official downloaded image

### Steps to reproduce

Reboot the router. Check (if Cloudflare) https://1.1.1.1/help and https://browserleaks.com/dns

### Actual behaviour

"https-dns-proxy" does not start after rebooting the router. Check status: "Using DNS over HTTPS (DoH) - No"

### Expected behaviour

"https-dns-proxy" automatically starts after rebooting the router. When checking the status: "Using DNS over HTTPS (DoH) - Yes" 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-22:
You have to report optional package issues in ../packages
If you upgraded keeping configs  from v22 then it is worth trying "clean install" at least of particular package and/or resetting default dnsmasq config from /rom.

NorthP8, 2024-02-22:
> You have to report optional package issues in ../packages If you upgraded keeping configs from v22 then it is worth trying "clean install" at least of particular package and/or resetting default dnsmasq config from /rom.

There are also problems with "luci-app-wifischedule". Works, but incorrectly.
After update I did a full reset.

brada4, 2024-02-22:
Report bugs in respective trackers https://openwrt.org/bugs
Or ask for general advice in forums.
Just a note on credibility - it is your burden to help to reproduce instead of saying nobody was able to reproduce.

NorthP8, 2024-02-23:
I did a reset router to default and reinstalled https-dns-proxy. The service is enabled in startup. Problem still exists.

brada4, 2024-02-23:
It is a package, you have to report bug in other tracker. stubby and dnscrypt are viable alternatives.

closed 2024-03-14 by NorthP8