openwrt/openwrt #14701: TP-Link Archer C7 v5 OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions

howl, 2024-02-23:
### Describe the bug

System log shows:
```
...
Thu Feb 22 23:37:30 2024 kern.notice kernel: [    0.361463] 8 fixed-partitions partitions found on MTD device spi0.0
Thu Feb 22 23:37:30 2024 kern.err kernel: [    0.368103] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Feb 22 23:37:30 2024 kern.err kernel: [    0.374969] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Feb 22 23:37:30 2024 kern.err kernel: [    0.382348] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Feb 22 23:37:30 2024 kern.err kernel: [    0.389269] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
Thu Feb 22 23:37:30 2024 kern.notice kernel: [    0.396309] Creating 8 MTD partitions on "spi0.0":
...
```

Searching for it seems that this was fixed months ago.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Additional info

I have select Official image but it has been built with ASU, just some packages added.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-23:
You do not describe any problem?
I see same message at boot and never paid attention, everything works.

howl, 2024-02-23:
Only the err messages in the log, according previous bugs reports these errors were resolved by not registering some ofs also to save memory, doesn't know is it's a regression or this is as it should be.

howl, 2024-02-23:
Only the err messages in the log, according previous bugs reports these errors were resolved by not registering some ofs also to save memory, doesn't know if it's a regression or this is as it should be.

brada4, 2024-02-23:
A quick forum search could have helped you:
https://forum.openwrt.org/t/what-does-the-presence-of-spi-nor-bad-cell-count-mean-for-a-firmware/157470

namiltd, 2024-02-23:
Add this to the qca9563_tplink_archer-c7-v5.dts file:
'''
&spi0 {
	flash@0 {
		#address-cells = <1>;
		#size-cells = <1>;
	};
};

howl, 2024-02-23:
> A quick forum search could have helped you: https://forum.openwrt.org/t/what-does-the-presence-of-spi-nor-bad-cell-count-mean-for-a-firmware/157470

Thanks for pointing me to the same post that I read before that point to a bug that is marked as solved and is what I mention in this issue.

> Add this to the qca9563_tplink_archer-c7-v5.dts file:
> 
> ```
> &spi0 {
> 	flash@0 {
> 		#address-cells = <1>;
> 		#size-cells = <1>;
> 	};
> };
> ```

Is needed to recompile or is possible to alter dts files in running systems? I don't have physical access to the router most of the time and I could do tests but with some v4 given they have the same issue.

howl, 2024-02-23:
Ok I see that is possible to read only dts from sysfs but no alter. I will check this with an v4 because I can TFTP and so on in case of eventual disaster.

namiltd, 2024-02-23:
This information in the log does not affect the operation of the router, so it is not worth recompiling just for this reason.

howl, 2024-02-23:
I have two spare v4, just taking dust can check it. Even could serve for me to try ImageBuilder as years ago I only did full compilations.

closed 2025-11-27 by openwrt-bot